openwrt/mt76 #285: mt7610u USB AP mode - no traffic

tluciomiranda, 2019-06-10:
Hi,
Using a TP-LINK T2U as AP on a RPi 3B+ and despite initially everything seems working, after some time traffic stops passing via the AP. I can still communicate with RPi, open Luci, SSH via ethernet but not via wireless AP. Also when traffic stops, new stations can’t see wifi network listed as available, or can’t connect if it appears. 

 No logs, no errors, at least that I can see. 

If you need some debug info I am happy to provide. 

Thanks!

yesnomaybeyes, 2019-06-11:
Getting the same here, also noteworthy is that if I saturate the link on one laptop, traffic flow just stops.

```
root@OpenWrt:~# modinfo mt76
module:         /lib/modules/4.14.123/mt76.ko
license:        Dual BSD/GPL
depends:        mac80211
retpoline:      Y
```

UCI config:

```
config wifi-device 'radio0'
        option type 'mac80211'
        option path 'pci0000:00/0000:00:1d.7/usb1/1-1'
        option country 'GB'
        option legacy_rates '0'
        option txpower '17'
        option hwmode '11g'
        option channel '4'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option mode 'ap'
        option ssid 'h'
        option encryption 'psk2'
        option key 'stuff!!!'
        option network 'BGL'

```

It seems the issue doesn't happen when using the "Legacy" selection, N and AC seem to experience this issue. Tried many different settings (seems `auto` just prevents the AP coming up, so I select specifically the channels and such)

I am using a TP-Link T2UH

tluciomiranda, 2019-06-11:
> Getting the same here, also noteworthy is that if I saturate the link on one laptop, traffic flow just stops.
>
> ```
> root@OpenWrt:~# modinfo mt76
> module:         /lib/modules/4.14.123/mt76.ko [...]

@yesnomaybeyes That's right, precisely what I am getting here too. Interestingly, you realized that "Legacy" mode does not cause the problem, could not figure that one... Have you noticed something more ? Debug info, logging, ... ?

Forgot to say that I am using a TP-Link T2U.

Thank you!

LorenzoBianconi, 2019-06-12:
@tluciomiranda could you please provide the hostpad configuration file? /var/run/hostapd-phy0.conf

tluciomiranda, 2019-06-12:
> @tluciomiranda could you please provide the hostpad configuration file? /var/run/hostapd-phy0.conf

@LorenzoBianconi Please find below the contents of /var/run/hostapd-phy1.conf, as I am using phy1 as AP:

`root@gw_movel:/tmp/run# cat hostapd-phy1.conf
driver=nl80211
logger_syslog=127
logger_syslog_level=2
logger_stdout=127
logger_stdout_level=2
country_code=JP
ieee80211d=1
ieee80211h=1
hw_mode=a
beacon_int=100
channel=36

ieee80211n=1
ht_coex=0
ht_capab=[HT40+][SHORT-GI-20][SHORT-GI-40][RX-STBC1]
vht_oper_chwidth=1
vht_oper_centr_freq_seg0_idx=42
ieee80211ac=1
vht_capab=[SHORT-GI-80][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN][RX-STBC-1][MAX-A-MPDU-LEN-EXP3]

interface=wlan1
ctrl_interface=/var/run/hostapd
ap_isolate=1
bss_load_update_period=60
chan_util_avg_period=600
disassoc_low_ack=1
preamble=1
wmm_enabled=1
ignore_broadcast_ssid=0
uapsd_advertisement_enabled=1
utf8_ssid=1
multi_ap=0
auth_algs=1
wpa=0
ssid=redecasa_test
bridge=br-lan
macaddr_acl=1
accept_mac_file=/var/run/hostapd-wlan1.maclist
bssid=50:3e:aa:89:e0:63`

yesnomaybeyes, 2019-06-12:
Hi all,

Sorry, I was wrong, I was misremembering possibly, I had the issue twice today on Legacy mode:

[image]

```
root@OpenWrt:~# cat /var/run/hostapd-phy0.conf 
driver=nl80211
logger_syslog=127
logger_syslog_level=2
logger_stdout=127
logger_stdout_level=2
country_code=GB
ieee80211d=1
hw_mode=g
supported_rates=60 90 120 180 240 360 480 540
basic_rates=60 120 240
beacon_int=100
channel=4

interface=wlan0
ctrl_interface=/var/run/hostapd
ap_isolate=1
bss_load_update_period=60
chan_util_avg_period=600
disassoc_low_ack=1
preamble=1
wmm_enabled=1
ignore_broadcast_ssid=0
uapsd_advertisement_enabled=1
utf8_ssid=1
multi_ap=0
wpa_passphrase=stuff!!!
auth_algs=1
wpa=2
wpa_pairwise=CCMP
ssid=h
bridge=br-BGL
wpa_disable_eapol_key_retries=0
wpa_key_mgmt=WPA-PSK
okc=0
disable_pmksa_caching=1
bssid=50:3e:aa:e6:d5:50

```

tluciomiranda, 2019-06-26:
Updated driver with latest commits from yesterday, just to test if something changed. Everything appears the same.

Also noticed that in LuCI, RX Rate of connected stations (when they can connect) won't pass 6 Mbit/s, even with devices 50 cm away from AP. Please see attachment as an example.

[image]

LorenzoBianconi, 2019-07-01:
Could you please post the syslog output when the dongle hangs?
Moreover could you please test latest [commit](https://github.com/openwrt/mt76/commit/8919516006e44567bd0dd65d5af368ce8eeb1235) since running this driver version I have not not been able to trigger the issue

tluciomiranda, 2019-07-02:
Hi @LorenzoBianconi , thank you for your efforts.

Just to clarify, my setup is the following:
  - phy0 (RPi 3B+ integrated WIFI) connects to a network as a client (wwan)
  - phy1 (T2U) hosts wifi network 'redecasa_test' (wlan1)

eth0 and wlan1 are bridged to form an internal network called br-lan.

So, traffic going from stations connected by ethernet to wwan goes perfectly (~80Mbit/s); traffic from stations connected to wlan1 will pass with extreme difficult, even a ping has a million dropped packets. Sometimes clients can't even connect.

Also, running iperf3 between hosts in the bridge network sometimes is impossible, sometimes works, but traffic seems to be going in "chunks", like sending host connected to wifi sends 128KB at a time and then 0 for some seconds, and then 128KB again, and so on; receiving station connected via ethernet seems to be receiving the traffic steadily at a very low bitrate.

Please find below an updated version of hostapd-phy1.conf and syslog:

[syslog.txt](https://github.com/openwrt/mt76/files/3349869/syslog.txt)
[hostapd-phy1.conf.txt](https://github.com/openwrt/mt76/files/3349870/hostapd-phy1.conf.txt)

If you need more info feel free to ask.

Tiago

tluciomiranda, 2019-08-26:
Hi @LorenzoBianconi 

Just tested the latest snapshot and driver presents the same problem.
Do you have any methods so I can collect some debug info that could be useful to you ? Like logs, captures, ...

Regards

tluciomiranda, 2019-11-22:
Hi @LorenzoBianconi 

Regarding this problem, I discovered via tcpdump on monitor interface that wifi beacons are being sent every ~300ms instead of the 100ms that are set in the file above. Can't tell if this can be related or not with the problem, but thought I should share.

[beacon.txt](https://github.com/openwrt/mt76/files/3881751/beacon.txt)

Again, I would be glad to help you debug this. If you would like me to do some tests just ask.

Regards

tluciomiranda, 2020-03-11:
Seems like this problem may have been solved in the last few commits. At least, traffic seems to be flowing without issues for 3 days now with T2U in master mode. 

Tried to trace which was the commit that changed the behavior and it seems that [fd892bc0](https://github.com/openwrt/mt76/commit/fd892bc033fbfad7b1a5bcba1432709a74327a1e) changed something that made the symptoms disappear. @yesnomaybeyes can you please also confirm ?

Closing this. Thanks for your work.

closed 2020-03-11 by tluciomiranda