openwrt/mt76 #288: Mt7603e driver use PA, wifi can not find, receive normal

listenerln, 2019-06-27:
The device uses PA, mt7603e starts normally, the phone can't find wifi, the test shows that the sending power is very low. But the MTK official driver PA can work normally.
Modify e2p After the PA is turned off, the wifi signal is very poor.

How can I modify the PA to work properly?
Thanks!

wlan0     ESSID: "A1111111"
          Access Point: 00:CD:88:00:00:81
          Mode: Master  Channel: 9 (2.452 GHz)
          Tx-Power: 12 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: unknown
          Bit Rate: unknown
          Encryption: none
          Type: nl80211  HW Mode(s): 802.11bgn
          Hardware: 14C3:7603 14C3:7603 [Generic MAC80211]
          TX power offset: unknown
          Frequency offset: unknown
          Supports VAPs: yes  PHY name: phy0

nbd168, 2019-06-27:
Which openwrt/mt76 version were you using in your test?

listenerln, 2019-06-28:
> Which openwrt/mt76 version were you using in your test?

mt76-2019-03-27-a11b67348aeb2d3175c5995c747e7fc050c7df60

openwrt version:
     commit 3209f5ae3d54e6f028e2e54314180a1be5b4896e

nbd168, 2019-06-28:
What's the displayed tx power on the 2019-06-25 version?

listenerln, 2019-06-28:
> What's the displayed tx power on the 2019-06-25 version?

wlan0     ESSID: "A1111111"
          Access Point: 00:CD:88:00:00:81
          Mode: Master  Channel: 11 (2.462 GHz)
          Tx-Power: 20 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: unknown
          Bit Rate: unknown
          Encryption: none
          Type: nl80211  HW Mode(s): 802.11bgn
          Hardware: 14C3:7603 14C3:7603 [MediaTek MT7603E]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy0

paullai58, 2019-08-21:
hello everyone,
Is there any one could help to explain below.
In mt76x02_eeprom.h
enum mt76x02_eeprom_modes {
MT_EE_READ,
MT_EE_PHYSICAL_READ,
};
what different is between MT_EE_READ and MT_EE_PHYSICAL_READ?
Thanks.

libiao0120, 2020-03-16:
This issue is a bug in the mt7603 driver. I have solved the problem you encountered. You try to modify the config1 field of eeprom to using external PA is invalid, because this driver uses the default value instead of selecting the external PA based on the eeprom configuration . his may be a problem with your hardware circuit. The mt7603 has several ANTSEL pins. It will open the enable pin of the external PA during data transmission. Your problem may be that the pins are not connected properly. In addition, mt7603 is a 2T2R device. The driver is only configured with 1T2R. Only one channel can use an external PA, and the other channel does not work at all. That is, if you connect two antennas, whether the other antenna is connected or not has no effect on the transmitted signal strength.

libiao0120, 2020-03-16:
> hello everyone,
> Is there any one could help to explain below.
> In mt76x02_eeprom.h
> enum mt76x02_eeprom_modes {
> MT_EE_READ,
> MT_EE_PHYSICAL_READ,
> };
> what different is between MT_EE_READ and MT_EE_PHYSICAL_READ?
> Thanks.

MT_EE_READ

> hello everyone,
> Is there any one could help to explain below.
> In mt76x02_eeprom.h
> enum mt76x02_eeprom_modes {
> MT_EE_READ,
> MT_EE_PHYSICAL_READ,
> };
> what different is between MT_EE_READ and MT_EE_PHYSICAL_READ?
> Thanks.

You need to check the manual of mt7603 or mt7612 to understand the introduction of the registers. I guess it should be a different mode of internal efuse reading, or setting the source of reading. Refer to the source code to write this parameter before reading. Also if you have the user manual of this mt7603 let me see.

static int
mt76x02_efuse_read(struct mt76x02_dev *dev, u16 addr, u8 *data,
		   enum mt76x02_eeprom_modes mode)
{
	u32 val;
	int i;

	val = mt76_rr(dev, MT_EFUSE_CTRL);
	val &= ~(MT_EFUSE_CTRL_AIN |
		 MT_EFUSE_CTRL_MODE);
	val |= FIELD_PREP(MT_EFUSE_CTRL_AIN, addr & ~0xf);
	val |= FIELD_PREP(MT_EFUSE_CTRL_MODE, mode);
	val |= MT_EFUSE_CTRL_KICK;
	mt76_wr(dev, MT_EFUSE_CTRL, val);