openwrt/mt76 #289: mt7610u USB mesh mode - accept peerings always NO

chennakeshavaSoC, 2019-06-27:
I am trying to setup mesh with TP Link wifi usb dongle (T2U) with Rpi3B+ (openwrt snapshot builds).

I am seeing that the mesh is not getting formed with even unsecured mode as well. I could find the issue by looking at the beacon frames transmitted, the following are details:
Tag: Mesh Configuration
0000   71 07 01 01 00 01 00 00 **_00_**                       q........
The last byte (7th one) is always set to zero, due to this the neighbor node is not progressing further (attempting to send request probe, etc).

I did captured the data over USB as well to see whats going on between dongle and host, I am clear on the last 3 bytes exchanged.

ffffffc03715fb00 398833123 S Co:1:004:0 s 40 06 0000 c05c 0004 4 = 73687107
ffffffc03715fb00 398833241 C Co:1:004:0 0 4 >
ffffffc03715fb00 398833259 S Co:1:004:0 s 40 06 0000 c060 0004 4 = 01010001
ffffffc03715fb00 398833386 C Co:1:004:0 0 4 >
ffffffc03715fb00 398833405 S Co:1:004:0 s 40 06 0000 108c 0004 4 = **feff**ffff
ffffffc03715fb00 398833519 C Co:1:004:0 0 4 >
ffffffc03715fb00 398833537 S Ci:1:004:0 s c0 07 0000 1014 0004 4 <
ffffffc03715fb00 398833651 C Ci:1:004:0 0 4 = 00001c00
ffffffc03715fb00 398833668 S Co:1:004:0 s 40 06 0000 1014 0004 4 = 00000000
ffffffc03715fb00 398833782 C Co:1:004:0 0 4 >

Please let me know is it a know issue or not, and I could provide more logs if required.

chennakeshavaSoC, 2019-06-27:
I could find root cause for the issue, its due to alignment.
In my case the mesh beacon frame length is coming as 83 bytes, now I just adjusted the mesh_id to set the frame length as multiple of 4. Now I am getting the expected results.
Tag: Mesh Configuration
0000 71 07 01 01 00 01 00 00 **09** q........

sgruszka, 2019-06-29:
This looks to be a bug in mac80211 mesh code that require proper fix.

chennakeshavaSoC, 2019-07-01:
Yes, I could be able to find it out, its here 
https://github.com/openwrt/mt76/blob/6cafaca7225d3676e0dce232aa8036637309f261/usb.c#L167

sgruszka, 2019-07-01:
I think we really can/should assume that len in mt76_wr_copy() is multiple of 4,  i.e mmio version of this is:

static void mt76_mmio_copy(struct mt76_dev *dev, u32 offset, const void *data,
                           int len)
{
        __iowrite32_copy(dev->mmio.regs + offset, data, len >> 2);
}

Not sure about that, but seems that mesh mac80211 code should create beacon that size is 4 bytes multitude, i.e. add 2 bytes zero pad at the end if needed.  If not, we should fix mt76_wr_copy() on in mmio and usb version, or add pad by ourselves in mt76x02_beacon.c code.

Though I think in the pointed code we have a bug , we should not change endian by using put_unaligned_le32() , just ordinary put_unaligned() . Looks that is was never triggered as nobody use mt76x{02}u on big endian machines.

nbd168, 2019-07-01:
I don't think mac80211 should pad beacon frames to work around this quirk. I will send out a patch that rounds up length in mmio/usb copy()

sgruszka, 2019-07-01:
Yeah, I get similar impression after looking at mac80211 beacon generation code. For usb we should be able to just use single request for copy like this:
[mt76u_copy-patch.txt](https://github.com/openwrt/mt76/files/3345297/mt76u_copy-patch.txt)

nbd168, 2019-07-01:
I sent out the simple fix for now, which simply adds the round-up. We can add that change on top later (after careful testing to avoid regressions)

sgruszka, 2019-07-01:
Agree.

nbd168, 2019-07-01:
Pushed my fix in 9e2e0b8577104806f40e469c8948843c67bddcb6 and verified that it fixes the issue (before that fix, it the WMM VO TXOP value was corrupted in my test).

closed 2019-07-01 by nbd168

sgruszka, 2019-07-01:
For the record , change to single usb request as proposed in  https://github.com/openwrt/mt76/issues/289#issuecomment-507226987 doest not work. There are 'usb 1-1: vendor request req:01 off:b420 failed:-11' errors, so I'm not going to sent that patch,  happy to have @nbd168 DIV_ROUND_UP fix.