openwrt/mt76 #290: Waking up from suspend freezes if MediaTek MT7610U wifi is plugged in

Yanpas, 2019-07-07:
[Kernel.org issue](https://bugzilla.kernel.org/show_bug.cgi?id=204091)

I have the following device plugged in:
Bus 001 Device 011: ID 0586:3425 ZyXEL Communications Corp. NWD6505 802.11a/b/g/n/ac Wireless Adapter [MediaTek MT7610U]

When I wake up from suspend I see only underscore in the top left corner of the screen. Only SysRq+B works, so I have nothing do but reboot.

This device is often not detected on boot and I have to replug it in, otherwise lsusb does not show it.

lsmod gives
```
mt76x0u                20480  0
mt76x0_common          49152  1 mt76x0u                                                                                                                     
mt76x02_usb            16384  1 mt76x0u                                                                                                                     
mt76_usb               36864  2 mt76x02_usb,mt76x0u                                                                                                         
mt76x02_lib            77824  3 mt76x02_usb,mt76x0_common,mt76x0u                                                                                           
mt76                   57344  5 mt76_usb,mt76x02_lib,mt76x02_usb,mt76x0_common,mt76x0u                                                                      
mac80211              958464  5 mt76,mt76_usb,mt76x02_lib,mt76x0_common,mt76x0u                                                                             
cfg80211              815104  4 mt76,mt76x02_lib,mac80211,mt76x02_usb
```

On 4.19 I did manage to get input devices working after waking up in a minute or so. Here is a dmesg.

Graphical subsystem died, I had to manually stop display manager and kill all my user processes.

The most notable lines:
```
[  475.480062] mt76x0 1-8:1.0: Error: MCU response pre-completed!
[  475.990066] mt76x0 1-8:1.0: Error: send MCU cmd failed:-110
[  475.990075] mt76x0 1-8:1.0: Error: MCU response pre-completed!
[  476.500003] mt76x0 1-8:1.0: Error: send MCU cmd failed:-110
[  476.500012] mt76x0 1-8:1.0: Error: MCU response pre-completed!
[  477.010046] mt76x0 1-8:1.0: Error: send MCU cmd failed:-110
[  477.010055] mt76x0 1-8:1.0: Error: MCU response pre-completed!
[  477.529997] mt76x0 1-8:1.0: Error: send MCU cmd failed:-110
[  477.530006] mt76x0 1-8:1.0: Error: MCU response pre-completed!
[  477.824907] mt76x0 1-8:1.0: Error: send MCU cmd failed:-71
[  477.824916] mt76x0 1-8:1.0: Error: MCU response pre-completed!
[  477.825029] usb 1-8: USB disconnect, device number 6
[  477.825639] usb 1-8:1.0: Error: send MCU cmd failed:-71
[  477.825647] usb 1-8:1.0: Error: MCU response pre-completed!
[  477.825653] usb 1-8:1.0: Error: send MCU cmd failed:-19
[  477.825658] usb 1-8:1.0: Error: MCU response pre-completed!
[  477.825662] usb 1-8:1.0: Error: send MCU cmd failed:-19
[  477.825667] usb 1-8:1.0: Error: MCU response pre-completed!
[  477.825671] usb 1-8:1.0: Error: send MCU cmd failed:-19
```

On 5.1 I see following errors on 
```
[   23.405299] usb 1-8: device descriptor read/64, error -110
[   23.635280] usb 1-8: new high-speed USB device number 7 using xhci_hcd
[   27.116384] audit: type=1131 audit(1562519777.616:38): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=user@995 comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[   27.124458] audit: type=1131 audit(1562519777.623:39): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=user-runtime-dir@995 comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[   28.948680] usb 1-8: device descriptor read/64, error -110
[   36.804418] audit: type=1130 audit(1562519787.303:40): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=packagekit comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[   37.836974] audit: type=1131 audit(1562519788.336:41): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=systemd-hostnamed comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[   44.525331] usb 1-8: device descriptor read/64, error -110
[   44.632035] usb usb1-port8: attempt power cycle
[   45.278620] usb 1-8: new high-speed USB device number 8 using xhci_hcd
[   50.495338] usb 1-8: device descriptor read/64, error -110
[   66.071964] usb 1-8: device descriptor read/64, error -110
[   66.308653] usb 1-8: new high-speed USB device number 9 using xhci_hcd
[   71.615233] usb 1-8: device descriptor read/64, error -110
[   87.188660] usb 1-8: device descriptor read/64, error -110
[   87.295375] usb usb1-port8: unable to enumerate USB device
```

Yanpas, 2019-07-07:
Installing `mt76-dkms-git` didn't help

sgruszka, 2019-07-08:
We should handle usb errors better to do not hung the system. But the root of the issue most likely is in usb hcd driver. There is pending know problem with xhci_hcd:

https://bugzilla.kernel.org/show_bug.cgi?id=202541

You can try to revert: 

commit f8f80be501aa2f10669585c3e328fad079d8cb3a
Author: Mathias Nyman <mathias.nyman@linux.intel.com>
Date:   Thu Sep 20 19:13:37 2018 +0300
xhci: Use soft retry to recover faster from transaction errors

and see if that will help. 

To clarify , suspend/resume doest not work for you on 4.19 bit it didn't hung the system, on 4.20 and upwards system hung? Or suspend/resume fully works on 4.19 ?

Yanpas, 2019-07-08:
Well, on 4.19 after some period of time with a great probability I am able to get to tty2-6. X server becomes unusable and I have to restart the whole session.

On 5.1 I didn't succeed to get access to PC. I even plugged out wifi adapter during this freeze and got kernel panic ( caps lock and scroll lock were blinking).

I also noticed that the bug with connect on boot exists on Windows too. After start of windows 10 I had to replug adapter to connect to network. Today I'll try to return this adapter to store.

sgruszka, 2019-07-18:
I was able to reproduce the problem on one of my machines (with ehci-hcd usb host controler, on machines with xhci-hcd suspend/resume works for me).
It turned out we can avoid the hung if we do not reset radio on resume like in below:

diff --git a/drivers/net/wireless/mediatek/mt76/mt76x0/usb.c b/drivers/net/wireless/mediatek/mt76/mt76x0/usb.c
index f878ad8b4b76..645f4d15fb61 100644
--- a/drivers/net/wireless/mediatek/mt76/mt76x0/usb.c
+++ b/drivers/net/wireless/mediatek/mt76/mt76x0/usb.c
@@ -136,11 +136,11 @@ static int mt76x0u_start(struct ieee80211_hw *hw)
        .release_buffered_frames = mt76_release_buffered_frames,
 };
 
-static int mt76x0u_init_hardware(struct mt76x02_dev *dev)
+static int mt76x0u_init_hardware(struct mt76x02_dev *dev, bool reset)
 {
        int err;
 
-       mt76x0_chip_onoff(dev, true, false);
+       mt76x0_chip_onoff(dev, true, reset);
 
        if (!mt76x02_wait_for_mac(&dev->mt76))
                return -ETIMEDOUT;
@@ -173,7 +173,7 @@ static int mt76x0u_register_device(struct mt76x02_dev *dev)
        if (err < 0)
                goto out_err;
 
-       err = mt76x0u_init_hardware(dev);
+       err = mt76x0u_init_hardware(dev, true);
        if (err < 0)
                goto out_err;
 
@@ -309,7 +309,7 @@ static int __maybe_unused mt76x0_resume(struct usb_interface *usb_intf)
        if (ret < 0)
                goto err;
 
-       ret = mt76x0u_init_hardware(dev);
+       ret = mt76x0u_init_hardware(dev, false);
        if (ret)
                goto err;

sgruszka, 2019-07-18:
I was able to reproduce the problem on one of my machines (with ehci-hcd , other machines with xhci-hcd work fine). It turned out we can avoid hung by do not reset rf on resume like in the patch below:
[mt76x0u_do_not_rf_reset_on_resume.patch.txt](https://github.com/openwrt/mt76/files/3406008/mt76x0u_do_not_rf_reset_on_resume.patch.txt)

sgruszka, 2019-07-18:
Fix posted:
https://lore.kernel.org/linux-wireless/1563446290-2813-1-git-send-email-sgruszka@redhat.com/T/#u
This can be closed once merged in mt76.

Yanpas, 2019-07-19:
Thanks

The issue that device is not always properly configured after boot will be always solved in this patch?
`usb 1-8: device descriptor read/64, error -110`

sgruszka, 2019-07-19:
If you see also  'xhci_hcd 0000:15:00.0: WARN Set TR Deq Ptr cmd failed due to incorrect slot or ep state' in dmesg , it is xhci driver issue: 
https://bugzilla.kernel.org/show_bug.cgi?id=202541
Not solved yet. Looks xhci maintainer requested debug information, which he doesn't get:
https://lore.kernel.org/linux-usb/520462de-9a7f-d3f3-2cce-e4be045cf8b2@linux.intel.com/ 

If there is no xhci_hcd warning, this could be similar problem like on resume . Patch does not address it, but you can easily make similar change on init: change reset to false in mt76x0u_register_device(),  and check if that helps (I'm not able to reproduce boot problem by myself).