openwrt/mt76 #299: kernel panic with kernel 4.14.132 on version 3d7f7384bef0950ceea3a8c96ce79333d208e07d

ybbid, 2019-07-21:
router:Newifi-D2(mt7621+mt7612+mt7603 rom:32m ram:512m)
arch:MediaTek MT7621 ver:1 eco:3
openwrt-git-version: db8e08a5a4469f98ed5d9b0ff3189e356f53d924
mt76-git-version: 3d7f7384bef0950ceea3a8c96ce79333d208e07d
kernel version:4.14.132

blow is the kernel crash log:
DRAM: 512MB
Platform: MediaTek MT7621A ver 1, eco 3
Board: Newifi D2
Clocks: CPU: 880MHz, DDR: 1066MHz, Bus: 293MHz, Ref: 40MHz
Flash: Winbond W25Q256 (32MB) on mt7621-spi.0
rt2880-eth: MAC address from EEPROM is invalid, using default settings.
rt2880-eth: Using MAC address 00:0c:43:00:00:01
eth0: MediaTek MT7530 Gigabit switch

Network started on eth0, inet addr 192.168.1.1, netmask 255.255.255.0

Press any key to interrupt autoboot ... 0   

Trying to boot firmware from 0x00050000 in flash bank 0 ...
Reading data into memory ...
U-Boot firmware image header detected.
    Image Name:   MIPS OpenWrt Linux-4.14.132
    Data Size:    19385695 Bytes
    Load Address: 80001000
    Entry Point:  80001000
Uncompressing data (LZMA) ... done.
Flushing cache ... done.

Starting kernel at 0x80001000...

[    0.000000] Linux version 4.14.132 (ybbid@ybbid-Virtual-Machine) (gcc version 7.4.0 (OpenWrt GCC 7.4.0 r10561-ff7909f69e)) #0 SMP Fri Jul 19 21:04:23 2019
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
[    0.000000] bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
[    0.000000] MIPS: machine is Newifi-D2
[    0.000000] Determined physical RAM map:
[    0.000000]  memory: 1c000000 @ 00000000 (usable)
[    0.000000]  memory: 04000000 @ 20000000 (usable)
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] VPE topology {2,2} total 4
[    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000]   HighMem  [mem 0x0000000010000000-0x0000000023ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000001bffffff]
[    0.000000]   node   0: [mem 0x0000000020000000-0x0000000023ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000023ffffff]
[    0.000000] random: get_random_bytes called from start_kernel+0x9c/0x4d8 with crng_init=0
[    0.000000] percpu: Embedded 14 pages/cpu s26064 r8192 d23088 u57344
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 130496
[    0.000000] Kernel command line: console=ttyS0,115200 rootfstype=squashfs,jffs2
[    0.000000] PID hash table entries: 1024 (order: 0, 4096 bytes)
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes)
[    0.000000] Writing ErrCtl register=00019041
[    0.000000] Readback ErrCtl register=00019041
[    0.000000] Memory: 494256K/524288K available (4790K kernel code, 246K rwdata, 1036K rodata, 17980K init, 254K bss, 30032K reserved, 0K cma-reserved, 262144K highmem)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000] NR_IRQS: 256
[    0.000000] CPU Clock: 880MHz
[    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4343773742 ns
[    0.000010] sched_clock: 32 bits at 440MHz, resolution 2ns, wraps every 4880645118ns
[... 215 lines cut ...]
[   30.933860] epc   : 80f6b5c8 mt76_eeprom_override+0x38/0xdc [mt76]
[   30.940013] ra    : 80f6b5bc mt76_eeprom_override+0x2c/0xdc [mt76]
[   30.946158] Status: 11008403 KERNEL EXL IE 
[   30.950328] Cause : 40800008 (ExcCode 02)
[   30.954314] BadVA : fffffff0
[   30.957177] PrId  : 0001992f (MIPS 1004Kc)
[   30.961248] Modules linked in: mt7603e(+) mt76 mac80211 lz4 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_FLOWOFFLOAD ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda uinput snd_usbmidi_lib slhc rfcomm nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack lz4_decompress lz4_compress iptable_mangle iptable_filter ip_tables hidp hci_uart crc_ccitt compat cdc_wdm btusb btintel bnep bluetooth hid evdev input_core ledtrig_usbport nf_log_ipv6 nf_log_common ip6table_mangle
[   31.031569]  ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 snd_rawmidi snd_seq_device snd_pcm_oss snd_pcm snd_timer snd_mixer_oss snd_hwdep snd_compress snd soundcore nls_utf8 zram zsmalloc ecdh_generic kpp ecb cmac uas usb_storage leds_gpio xhci_plat_hcd xhci_pci xhci_mtk xhci_hcd uhci_hcd ohci_platform ohci_hcd ehci_platform sd_mod scsi_mod ehci_hcd gpio_button_hotplug usbcore nls_base usb_common
[   31.068263] Process kmodloader (pid: 588, threadinfo=80e0e000, task=8fdad480, tls=77fbbeb8)
[   31.076569] Stack : 00000008 00000400 8fc35980 80f4d5c0 000f0040 8fc35980 80f4d5c0 80fc7664
[   31.084902]         80fc8a38 802e3900 00000000 80fc3214 000003e8 80f4d5c0 80f4d5c0 80fc8894
[   31.093234]         80f4d5c0 8fcef000 80f4d5c0 80fc8a38 80f4cbe0 80f4c1a0 00000008 00000028
[   31.101566]         805c0000 80fc1d00 00000018 8f6cc580 00000000 80f4d5c0 00000018 80080cfc
[   31.109897]         8fe58100 80e0fba8 80fc8a38 802dbb74 80fc85e8 80f4d5c0 00000000 8fcef000
[   31.118227]         ...
[   31.120664] Call Trace:
[   31.123102] [<80f6b5c8>] mt76_eeprom_override+0x38/0xdc [mt76]
[   31.128948] [<80fc7664>] mt7603_eeprom_init+0x2e4/0x504 [mt7603e]
[   31.135032] [<80fc1d00>] mt7603_register_device+0x130/0xb8c [mt7603e]
[   31.141462] [<80fc0188>] init_module+0x8e188/0x8f240 [mt7603e]
[   31.147275] Code: 14400002  00000000  2402ffed <88430003> 26110a9c  98430000  aa230003  ba230000  90430004 
[   31.156999] 
[   31.158634] ---[ end trace 00a2537da2f0a4a3 ]---
[   31.165037] Kernel panic - not syncing: Fatal exception
[   31.171651] Rebooting in 1 seconds..

airend, 2019-07-21:
This might be related to [mac80211: Update to version 5.2-rc7](https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=0b2c42ced21a7bc053e0d729f85041f1e3b54fbc). I also get panics on the Xiaomi MiWiFi Mini (`MT7620A`).

ynezz, 2019-07-24:
Should be fixed in 75656a4590a3288503d42194adf014f4a9eeaa84

closed 2019-07-24 by ynezz