openwrt/mt76 #302: Error - Dropping frame due to full tx queue 2

532910, 2019-07-25:
OpenWrt 18.06.4, r7808-ef686b7292
system type : MediaTek MT7620A ver:2 eco:6
machine : Xiaomi MiWiFi Mini

All dmesg is spammed wth:

```
[33427.871649] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[33427.881093] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[33427.890532] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[33427.899972] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[33427.909409] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[33427.918849] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[33427.928268] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[33427.937705] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[33427.947151] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[33427.956589] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[33427.966025] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[33427.975462] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[33427.984901] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[33427.994341] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[33428.003781] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[33428.013221] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[33428.022657] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[33428.032108] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[33428.041556] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[33428.051002] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[33428.060440] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[33428.069880] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[33428.079317] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
[33428.088758] ieee80211 phy1: rt2x00queue_write_tx_frame: Error - Dropping frame due to full tx queue 2
```

LGA1150, 2019-07-25:
Have you tried the latest snapshot or 19.07?
btw MT7620 uses rt2800-soc driver, not mt76

dangowrt, 2019-07-25:
Yes, wrong place to discuss this, however, this should be fixed in 19.07 and current master, please retry using the preview-version of OpenWrt 19.07 available on https://downloads.openwrt.org/releases/19.07-SNAPSHOT/targets/ramips/mt7620/openwrt-19.07-snapshot-r10269-5100629e32-ramips-mt7620-miwifi-mini-squashfs-sysupgrade.bin

closed 2019-07-25 by dangowrt

dangowrt, 2019-07-25:
> Sorry!
No worries, the repository naming is a bit misleading and it's still better to report a bug in the wrong place than not reporting it at all. So thanks :)

532910, 2019-07-26:
19.07 works fine, thank you!

masterchop, 2020-05-02:
not sure this has been resolved, most of my devices wont connect when i installed this.
for example the raspberry pi 3b cant see this network at all.

532910, 2020-05-02:
pretty sure it's due to CRDA policy and out of the scope of this issue

masterchop, 2020-05-02:
> CRDA policy

could you please point me out where should i go? i have like 50 wifi devices only 3 connect :(

532910, 2020-05-02:
just play with `channel`/`freq`, `country` and also `ieee80211w` and `encryption` options

masterchop, 2020-05-02:
nothing worked, rolling back to pandora :(,  i open a thread :

https://github.com/openwrt/mt76/issues/400

so the question is regit but thanks for replying