openwrt/mt76 #303: kernel panic on Newifi D2

pataha, 2019-07-29:
The device is Newifi D2 with latest snapshot, it usually crash while watching youtube videos
```
<1>[170366.384783] CPU 2 Unable to handle kernel paging request at virtual address 07406014, epc == 8f7ea37c, ra == 8f7ea35c
<4>[170366.395475] Oops[#1]:
<4>[170366.397828] CPU: 2 PID: 0 Comm: swapper/2 Not tainted 4.14.132 #0
<4>[170366.403978] task: 8fc3c780 task.stack: 8fc5e000
<4>[170366.408569] $ 0   : 00000000 00000001 00009c40 00000001
<4>[170366.413877] $ 4   : 8e5844a8 8e98c838 ffffffac 00000018
<4>[170366.419182] $ 8   : 00000068 00000040 0000ffff 0000ffff
<4>[170366.424485] $12   : 00000000 3fff0000 ffffffff 8f100a80
<4>[170366.429788] $16   : 07406000 8e584488 00000040 8e98c820
<4>[170366.435084] $20   : 8ee30000 00000050 8e5844a0 8e98c838
<4>[170366.440385] $24   : 0e995000 00000d50
<4>[170366.445687] $28   : 8fc5e000 8fc0dc50 8e5844a8 8f7ea35c
<4>[170366.450987] Hi    : 000000d6
<4>[170366.453934] Lo    : 00000000
<4>[170366.456977] epc   : 8f7ea37c cfg80211_rx_mgmt+0xb4/0x180 [cfg80211]
<4>[170366.463314] ra    : 8f7ea35c cfg80211_rx_mgmt+0x94/0x180 [cfg80211]
<4>[170366.469636] Status: 11007c03	KERNEL EXL IE
<4>[170366.473895] Cause : 40800008 (ExcCode 02)
<4>[170366.477972] BadVA : 07406014
<4>[170366.480925] PrId  : 0001992f (MIPS 1004Kc)
<4>[170366.485084] Modules linked in: pppoe ppp_async pppox ppp_generic nf_conntrack_ipv6 mt76x2e mt76x2_common mt76x02_lib mt7603e mt76 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_FLOWOFFLOAD xt_CT slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_mangle iptable_filter ip_tables crc_ccitt compat ledtrig_usbport nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 leds_gpio xhci_plat_hcd xhci_pci xhci_mtk xhci_hcd gpio_button_hotplug usbcore nls_base usb_common
<4>[170366.555849] Process swapper/2 (pid: 0, threadinfo=8fc5e000, task=8fc3c780, tls=00000000)
<4>[170366.563986] Stack : 00000000 8f5ac000 8fdff800 804ab918 8fcd53c0 8fcf3480 8fcf3480 8fcf3480
<4>[170366.572406]         8fcd5300 8fcf3480 8fcf3480 0000143c 01080020 ffffffac 8e584480 8fc0ddc4
<4>[170366.580833]         8fc0dd58 00000000 8e98c820 8f100a80 8ec6564c 8ec65680 80600000 8ec28dc0
<4>[170366.589259]         00000000 00000000 8f5ac480 8f1006c0 00000068 00000000 8ec6ad70 80600000
<4>[170366.597682]         8ee30be0 00000000 00000000 8e6c5a00 8fc24380 8006284c 00000000 00000000
<4>[170366.606106]         ...
<4>[170366.608637] Call Trace:
<4>[170366.611200] [<8f7ea37c>] cfg80211_rx_mgmt+0xb4/0x180 [cfg80211]
<4>[170366.617351] [<8ec28dc0>] ieee80211_sta_uapsd_trigger+0x35c8/0x3a8c [mac80211]
<4>[170366.624607] Code: 2694fe60  10000020  afa20030 <96020014> 10520003  02e02825  1000001b  8e100000  8e020010
<4>[170366.634427]
<4>[170366.636111] ---[ end trace 4403ef0fb54eb9ab ]---
```

rufengsuixing, 2019-09-09:
d1同样不定时炸掉
炸的时候无线没了网卡灯常亮
日志只能记到前两行
才知道原来是驱动炸了

redchenjs, 2019-09-09:
Very stable.
[image]

rufengsuixing, 2019-09-13:
重新编译了之后关掉了hwnet，现稳定

siwind, 2019-09-18:
2019-09-16编译的newifi-d2固件。 内核版本4.14.143。
在LUCI-web界面那里， 网络-->防火墙处，不开启任何硬/软件 "Routing/NAT 分载(offload )"。
安装transmission并且外挂USB移动硬盘下载bt。

2-5分钟内不定时重启，无法正常使用。实在没办法，切换回半年以前编译的内核为： 4.14.114内核的固件，一切正常。

尽管驱动有更新，但是往往引入更多bug~~

siwind, 2019-10-12:
Latest mt76 driver version

`
root@OpenWrt:~# opkg list-installed | grep mt76
kmod-mt76-core - 4.14.148+2019-10-10-d3a58958-1
kmod-mt7603 - 4.14.148+2019-10-10-d3a58958-1
kmod-mt76x02-common - 4.14.148+2019-10-10-d3a58958-1
kmod-mt76x2 - 4.14.148+2019-10-10-d3a58958-1
kmod-mt76x2-common - 4.14.148+2019-10-10-d3a58958-1
root@OpenWrt:~# 
`

**And kernel Oops log:**

`
root@OpenWrt:~# cat /sys/kernel/debug/crashlog 
Time: 1570871129.402026
Modules:        pppoe@8ebf4000+2290     ppp_async@8ea98000+1bc0 pppox@8eac3000+58a      ppp_generic@8f1b8000+5b62     mt76x2e@8ec64000+2910    mt76x2_common@8eb64000+2d88     mt76x02_lib@8f120000+a421       mt7603e@8ebd0000+96ac   mt76@8eb78000+7f1d     mac80211@8f000000+76a81 iptable_nat@8ef11000+310        ipt_REJECT@8f6e2000+410 ipt_MASQUERADE@8ee50000+2f0    cfg80211@8eb80000+3e2e9 xt_time@8eb40000+730    xt_tcpudp@8eb3d000+750  xt_tcpmss@8ea46000+470  xt_statistic@8eb3a000+390      xt_state@8eb36000+310   xt_socket@8eb1f000+a20  xt_recent@8eb30000+1c40 xt_nat@8eb18000+650     xt_multiport@8eb15000+550      xt_mark@8eb12000+2f0    xt_mac@8eb01000+2b0     xt_limit@8eafe000+4f0   xt_length@8eafa000+310xt_hl@8eaf7000+370       xt_helper@8eae7000+3b0  xt_ecn@8eae4000+590     xt_dscp@8eae2000+430    xt_conntrack@8eade000+910      xt_connmark@8eadb000+610        xt_connlimit@8ead8000+15cd      xt_connbytes@8eabd000+690       xt_comment@8eab9000+230        xt_TPROXY@8eaca000+1210 xt_TCPMSS@8eab3000+b30  xt_REDIRECT@8eaa8000+310        xt_LOG@8ea85000+350   xt_HL@8ea82000+590       xt_FLOWOFFLOAD@8ea0d000+c00     xt_DSCP@8ea91000+630    xt_CT@8ea8e000+ab0      xt_CLASSIFY@8ea8a000+290       ums_usbat@8eaa4000+20b0 ums_sddr55@8ef7e000+1750        ums_sddr09@8eaa0000+23f0        ums_karma@8edbf000+6f0 ums_jumpshot@8ee3e000+1010      ums_isd200@8ea96000+1760        ums_freecom@8ea0f000+8d0        ums_datafab@8ea88000+1370      ums_cypress@8ef0a000+a70        ums_alauda@8ea74000+22f0        slhc@8ea6a000+14bb      nf_socket_ipv6@8ea49000+516    nf_socket_ipv4@8edcf000+616     nf_reject_ipv4@8e9ff000+883     nf_nat_redirect@8ed9f000+63b    nf_nat_masquerade_ipv4@8ea7f000+67c    nf_conntrack_ipv4@8ea5e000+1450 nf_nat_ipv4@8ee5b000+dd1        nf_log_ipv4@8edbb000+db0       nf_flow_table_hw@8e924000+920   nf_flow_table@8ea38000+365f     nf_defrag_ipv4@8ea1d000+476     nf_conntrack_rtcache@8e978000+ad0      nf_conntrack_netlink@8ea78000+5940      macvlan@8ea6c000+3462   iptable_raw@8ea53000+2b0      iptable_mangle@8ef3c000+3d0      iptable_filter@8eda7000+2d0     ipt_ECN@8edd7000+5b0    ip_tables@8edcc000+291d crc_ccitt@8edff000+42b compat@8ea64000+1de4    act_connmark@8ea4b000+880       sch_tbf@8ea5a000+1730   sch_ingress@8ea45000+510       sch_htb@8ea40000+3720   sch_hfsc@8ea3c000+3a70  em_u32@8ea2a000+270     cls_u32@8ea2c000+2320   cls_tcindex@8eca6000+14f0      cls_route@8ed82000+14c0 cls_matchall@8ea27000+a30       cls_fw@8ea24000+f70     cls_flow@8f7d6000+16d0cls_basic@8ea1f000+c90   act_skbedit@8ea1c000+a20        act_mirred@8f7e2000+fc0 ledtrig_usbport@8f7fd000+b00    xt_set@8ea0a000+1b70   ip_set_list_set@8ea04000+1cd0   ip_set_hash_netportnet@8edf8000+6cd0    ip_set_hash_netport@8edf0000+6290      ip_set_hash_netnet@8ede8000+67f0        ip_set_hash_netiface@8edd8000+62f0      ip_set_hash_net@8edd0000+5c90 ip_set_hash_mac@8edbc000+2970    ip_set_hash_ipportnet@8edc0000+6810     ip_set_hash_ipportip@8edb0000+5590      ip_set_hash_ipport@8eda0000+51f0       ip_set_hash_ipmark@8ed98000+4ef0        ip_set_hash_ip@8ed90000+4dd0    ip_set_bitmap_port@8ee02000+1830       ip_set_bitmap_ipmac@8e92c000+1bb0       ip_set_bitmap_ip@8eeb2000+1b10  ip_set@8ed88000+5a4a  nfnetlink@8ec82000+1277  ip6table_nat@8e87d000+310       nf_conntrack_ipv6@8ec8a000+15c0 nf_defrag_ipv6@8ef72000+137e  nf_nat_ipv6@8ed84000+1033        ip6t_NPT@8e91c000+650   ip6t_MASQUERADE@8e9fb000+270    nf_nat_masquerade_ipv6@8ef16000+71c    nf_nat@8ed30000+387a    nf_conntrack@8ed60000+1151e     nf_log_ipv6@8ee4a000+1070       nf_log_common@8ecf8000+acf     ip6table_mangle@8ecfb000+4d0    ip6table_filter@8e917000+2d0    ip6_tables@8ef6c000+2821        ip6t_REJECT@8ef4b000+450       x_tables@8e9f4000+37cf  nf_reject_ipv6@8ecbf000+a08     nfs@8ed40000+1db18      msdos@8ef06000+1990   ifb@8eec6000+d30 vfat@8ee24000+2250      fat@8ed20000+d14e       ntfs@8ed00000+180f3     lockd@8ecb0000+d43f     sunrpc@8ecc0000+29341  grace@8eedb000+621      cifs@8ec00000+5b7e2     autofs4@8eea8000+57bb   nls_utf8@8e934000+390   nls_iso8859_1@8e907000+b70     nls_cp936@8ee80000+1d770        nls_cp437@8ee7c000+1170 wp512@8ee70000+62b0     twofish_generic@8ef20000+1f30  twofish_common@8ee60000+3351    tgr192@8ee54000+3530    tea@8f74f000+9b0        serpent_generic@8ee40000+4d34  khazad@8ee28000+4a30    cast6_generic@8ee20000+38ac     cast5_generic@8ee18000+2f8d     cast_common@8ee14000+11d1      camellia_generic@8ee08000+5530  blowfish_generic@8ee00000+10b0  blowfish_common@8ef7c000+1a91   anubis@8ef74000+26b0   xts@8ef70000+13ca       sha512_generic@8ef68000+25f9    sha256_generic@8ef48000+29f9    md5@8ef4c000+11b0     md4@8ef27000+cb0 hmac@8ef19000+a10       gf128mul@8e8b8000+1a9e  ecb@8fdc2000+5b0        des_generic@8ef30000+48eb     arc4@8ef47000+6b0        usb_storage@8f750000+9d2b       leds_gpio@8f643000+bb0  xhci_plat_hcd@8f64a000+15f0     xhci_pci@8f5a3000+c30  xhci_mtk@8f744000+12f0  xhci_hcd@8f720000+16de9 sd_mod@8f670000+75e0    scsi_mod@8f700000+16d72 gpio_button_hotplug@8f644000+1ad0      ext4@8f680000+5fb2e     mbcac [... cut ...]
<4>[26210.905596] [<8f003448>] ieee80211_tx_status_ext+0x15c/0x1d8 [mac80211]
<4>[26210.912246] [<8ebd614c>] mt7603_mac_add_txs+0x1e0/0x228 [mt7603e]
<4>[26210.918328] [<8ebd3860>] mt7603_queue_rx_skb+0x198/0x21c [mt7603e]
<4>[26210.924511] [<8eb79474>] mt76_dma_cleanup+0x760/0xdac [mt76]
<4>[26210.930154] Code: 00000000  8ca504dc  24a5fac0 <8ca202fc> 30430020  14600017  00000000  8ca702f4  14e00002 
<4>[26210.939899] 
<4>[26210.941618] ---[ end trace fd7ee539f927d452 ]---

===================================
Time: 1570871129.4root@OpenWrt:~# 
`

siwind, 2019-10-14:
And another kernel panic log:

`
root@OpenWrt:~# cat /sys/kernel/debug/crashlog 
Time: 1571028230.110638
Modules:        pppoe@8ee20000+2290     ppp_async@8eaee000+1bc0 pppox@8ee66000+58a      ppp_generic@8f1e8000+5b62     mt76x2e@8ed2c000+2910    mt76x2_common@8ed30000+2d88     mt76x02_lib@8f140000+a421       mt7603e@8eba0000+96ac   mt76@8eb80000+7f1d     mac80211@8f000000+76a81 iptable_nat@8ef72000+310        ipt_REJECT@8ed3d000+410 ipt_MASQUERADE@8f7f5000+2f0    cfg80211@8ebc0000+3e2e9 xt_time@8eb52000+730    xt_tcpudp@8eb50000+750  xt_tcpmss@8eb4c000+470  xt_statistic@8eb49000+390      xt_state@8eb45000+310   xt_socket@8eb36000+a20  xt_recent@8eb3e000+1c40 xt_nat@8eb30000+650     xt_multiport@8eb2c000+550      xt_mark@8eb21000+2f0    xt_mac@8eb1e000+2b0     xt_limit@8eb1b000+4f0   xt_length@8eb18000+310xt_hl@8eb14000+370       xt_helper@8eb0b000+3b0  xt_ecn@8eb08000+590     xt_dscp@8eb05000+430    xt_conntrack@8ece0000+910      xt_connmark@8ee6b000+610        xt_connlimit@8eb10000+15cd      xt_connbytes@8eaff000+690       xt_comment@8eaec000+230        xt_TPROXY@8eaf6000+1210 xt_TCPMSS@8eae7000+b30  xt_REDIRECT@8eae3000+310        xt_LOG@8eae0000+350   xt_HL@8ead2000+590       xt_FLOWOFFLOAD@8edc7000+c00     xt_DSCP@8ea85000+630    xt_CT@8ea8e000+ab0      xt_CLASSIFY@8ee80000+290       ums_usbat@8ea80000+20b0 ums_sddr55@8ef7c000+1750        ums_sddr09@8ee68000+23f0        ums_karma@8edf6000+6f0 ums_jumpshot@8ed74000+1010      ums_isd200@8e8f0000+1760        ums_freecom@8eacf000+8d0        ums_datafab@8eab0000+1370      ums_cypress@8eab9000+a70        ums_alauda@8eacc000+22f0        slhc@8e8f2000+14bb      nf_socket_ipv6@8ea5f000+516    nf_socket_ipv4@8ece9000+616     nf_reject_ipv4@8ed5a000+883     nf_nat_redirect@8ecdd000+63b    nf_nat_masquerade_ipv4@8ee2f000+67c    nf_conntrack_ipv4@8ef84000+1450 nf_nat_ipv4@8eabb000+dd1        nf_log_ipv4@8eab8000+db0       nf_flow_table_hw@8eab4000+920   nf_flow_table@8eac4000+365f     nf_defrag_ipv4@8eee0000+476     nf_conntrack_rtcache@8ec04000+ad0      nf_conntrack_netlink@8eaa8000+5940      macvlan@8fd58000+3462   iptable_raw@8ea3f000+2b0      iptable_mangle@8edbc000+3d0      iptable_filter@8edb4000+2d0     ipt_ECN@8ea17000+5b0    ip_tables@8ea00000+291d crc_ccitt@8ea47000+42b compat@8ea9e000+1de4    act_connmark@8ef6e000+880       sch_tbf@8ea96000+1730   sch_ingress@8e8ec000+510       sch_htb@8ea90000+3720   sch_hfsc@8ea88000+3a70  em_u32@8ee05000+270     cls_u32@8edd0000+2320   cls_tcindex@8ef82000+14f0      cls_route@8ea86000+14c0 cls_matchall@8e98d000+a30       cls_fw@8ecdc000+f70     cls_flow@8ea7e000+16d0cls_basic@8ec65000+c90   act_skbedit@8ec16000+a20        act_mirred@8ece1000+fc0 ledtrig_usbport@8ece4000+b00    xt_set@8ec06000+1b70   ip_set_list_set@8e8ba000+1cd0   ip_set_hash_netportnet@8ea68000+6cd0    ip_set_hash_netport@8ea60000+6290      ip_set_hash_netnet@8ea58000+67f0        ip_set_hash_netiface@8ea50000+62f0      ip_set_hash_net@8ea40000+5c90 ip_set_hash_mac@8ea2c000+2970    ip_set_hash_ipportnet@8ea20000+6810     ip_set_hash_ipportip@8ea18000+5590      ip_set_hash_ipport@8ea10000+51f0       ip_set_hash_ipmark@8ea08000+4ef0        ip_set_hash_ip@8edf8000+4dd0    ip_set_bitmap_port@8edf4000+1830       ip_set_bitmap_ipmac@8edf0000+1bb0       ip_set_bitmap_ip@8ee42000+1b10  ip_set@8ede8000+5a4a  nfnetlink@8eef4000+1277  ip6table_nat@8ef64000+310       nf_conntrack_ipv6@8ee7c000+15c0 nf_defrag_ipv6@8ef5a000+137e  nf_nat_ipv6@8edd4000+1033        ip6t_NPT@8ee54000+650   ip6t_MASQUERADE@8ec3f000+270    nf_nat_masquerade_ipv6@8ec62000+71c    nf_nat@8edc0000+387a    nf_conntrack@8eda0000+1151e     nf_log_ipv6@8ee64000+1070       nf_log_common@8ecec000+acf     ip6table_mangle@8ecf0000+4d0    ip6table_filter@8ecf4000+2d0    ip6_tables@8ee98000+2821        ip6t_REJECT@8ecf9000+450       x_tables@8ef5c000+37cf  nf_reject_ipv6@8ee9f000+a08     nfs@8ed80000+1db18      msdos@8ee60000+1990   ifb@8ed5b000+d30 vfat@8eef0000+2250      fat@8ed60000+d14e       ntfs@8ed40000+180f3     lockd@8ec70000+d43f     sunrpc@8ed00000+29341  grace@8ee8f000+621      cifs@8ec80000+5b7e2     autofs4@8ec08000+57bb   nls_utf8@8ee19000+390   nls_iso8859_1@8ee0a000+b70     nls_cp936@8ec20000+1d770        nls_cp437@8ec02000+1170 wp512@8eef8000+62b0     twofish_generic@8eee2000+1f30  twofish_common@8eedc000+3351    tgr192@8eed0000+3530    tea@8eec5000+9b0        serpent_generic@8eec0000+4d34  khazad@8eeb0000+4a30    cast6_generic@8ee3c000+38ac     cast5_generic@8ee00000+2f8d     cast_common@8e888000+11d1      camellia_generic@8eea8000+5530  blowfish_generic@8eea6000+10b0  blowfish_common@8eea2000+1a91   anubis@8ee9c000+26b0   xts@8ee96000+13ca       sha512_generic@8ee8c000+25f9    sha256_generic@8ee84000+29f9    md5@8ee6c000+11b0     md4@8e904000+cb0 hmac@8e854000+a10       gf128mul@8ee62000+1a9e  ecb@8e9e6000+5b0        des_generic@8ee58000+48eb     arc4@8ee52000+6b0        usb_storage@8f7a0000+9d2b       leds_gpio@8f6b6000+bb0  xhci_plat_hcd@8f614000+15f0     xhci_pci@8f5f5000+c30  xhci_mtk@8f6f0000+12f0  xhci_hcd@8f780000+16de9 sd_mod@8f6e0000+75e0    scsi_mod@8f6c0000+16d72 gpio_button_hotplug@8f686000+1ad0      ext4@8f700000+5fb2e     mbcache@8f67b000+fce    jbd2@8f620000+e481      usbcore@8f640000+2127d nls_base@8f606000+1570  usb_common@8fe7e000+a37 crc32c_generic@8fedf000+5b0
<6>[   17.810646] nf_conntrack version 0.5.0 (8192 buckets, 32768 max)
<6>[   17.864434] Netfilter messages via NETLINK v0.30.
<6>[   17.872778] ip_set: protocol 6
[... cut ...]
<4>[12589.847022]         ...
<4>[12589.849459] Call Trace:
<4>[12589.851914] [<80331944>] skb_put+0x14/0x54
<4>[12589.856009] [<8024ad38>] __nla_reserve+0x34/0x78
<4>[12589.860610] [<8024af40>] __nla_put+0x1c/0x3c
<4>[12589.864865] [<8024afa8>] nla_put+0x48/0x60
<4>[12589.868949] Code: 8c820098  afbf0014  0006302b <00060336> 8c870054  8c86009c  00454021  00e53821  00c8302b 
<4>[12589.878678] 
<4>[12589.880411] ---[ end trace ad49d02a9e222775 ]---

===================================
Time: 1571028230.119root@OpenWrt:~# 
`

siwind, 2019-10-20:
[newifi-d2-crashlog.txt](https://github.com/openwrt/mt76/files/3747423/newifi-d2-crashlog.txt)

siwind, 2019-11-23:
root@OpenWrt:~# cat /sys/kernel/debug/crashlog 
Time: 1574477766.803429
Modules:        pppoe@8eb34000+2290     ppp_async@8ea50000+1bc0 pppox@8eaf0000+58a      ppp_generic@8f1d0000+5b62     mt76x2e@8eb0c000+28b0    mt76x2_common@8eac4000+2d88     mt76x02_lib@8f0b0000+a421       mt7603e@8ebd0000+970c   mt76@8eb60000+81e4     mac80211@8f000000+77c29 iptable_nat@8eee2000+310        ipt_REJECT@8ecee000+410 ipt_MASQUERADE@8f7c1000+2f0    cfg80211@8eb80000+3f219 xt_time@8eb4c000+730    xt_tcpudp@8e9ab000+750  xt_tcpmss@8eb49000+470  xt_statistic@8eb46000+390      xt_state@8eb33000+310   xt_socket@8eb2f000+a20  xt_recent@8eb3c000+1c40 xt_nat@8eb28000+650     xt_multiport@8eb25000+550      xt_mark@8eb18000+2f0    xt_mac@8eb15000+2b0     xt_limit@8eb11000+4f0   xt_length@8ed56000+310xt_hl@8ed54000+370       xt_helper@8eb08000+3b0  xt_ecn@8eb05000+590     xt_dscp@8eb01000+430    xt_conntrack@8eafe000+910      xt_connmark@8eceb000+610        xt_connlimit@8ee78000+15cd      xt_connbytes@8edd3000+690       xt_comment@8ea32000+230        xt_TPROXY@8ea64000+1210 xt_TCPMSS@8ea49000+b30  xt_REDIRECT@8eaf5000+310        xt_LOG@8ea7e000+350   xt_HL@8eaf3000+590       xt_FLOWOFFLOAD@8eaed000+c00     xt_DSCP@8eae9000+630    xt_CT@8eae5000+ab0      xt_CLASSIFY@8eae3000+290       ums_usbat@8ea94000+20b0 ums_sddr55@8eaba000+1750        ums_sddr09@8ead8000+23f0        ums_karma@8e9aa000+6f0 ums_jumpshot@8eacc000+1010      ums_isd200@8ed44000+1760        ums_freecom@8ee0c000+8d0        ums_datafab@8ec8e000+1370      ums_cypress@8ea35000+a70        ums_alauda@8edd0000+22f0        slhc@8edf6000+14bb      nf_socket_ipv6@8eda6000+516    nf_socket_ipv4@8ee03000+616     nf_reject_ipv4@8ecea000+883     nf_nat_redirect@8e9b3000+63b    nf_nat_masquerade_ipv4@8ecf8000+67c    nf_conntrack_ipv4@8ea2c000+1450 nf_nat_ipv4@8eab1000+dd1        nf_log_ipv4@8eaae000+db0       nf_flow_table_hw@8eaaa000+920   nf_flow_table@8eabc000+365f     nf_defrag_ipv4@8ea8d000+476     nf_conntrack_rtcache@8ea8a000+ad0      nf_conntrack_netlink@8eaa0000+5940      macvlan@8ea98000+3462   iptable_raw@8ea83000+2b0      iptable_mangle@8ec8d000+3d0      iptable_filter@8e937000+2d0     ipt_ECN@8f784000+5b0    ip_tables@8ea74000+291d crc_ccitt@8e95d000+42b compat@8ee2a000+1de4    act_connmark@8ea5c000+880       sch_tbf@8e9ae000+1730   sch_ingress@8ec6e000+510       sch_htb@8e87c000+3720   sch_hfsc@8ed80000+3a70  em_u32@8ed76000+270     cls_u32@8ed90000+2320   cls_tcindex@8ed94000+14f0      cls_route@8ee08000+14c0 cls_matchall@8ea52000+a30       cls_fw@8ea40000+f70     cls_flow@8ed42000+16d0cls_basic@8ea3a000+c90   act_skbedit@8ea36000+a20        act_mirred@8ee9b000+fc0 ledtrig_usbport@8ee97000+b00    xt_set@8ea2e000+1b70   ip_set_list_set@8ea2a000+1cd0   ip_set_hash_netportnet@8ea20000+6cd0    ip_set_hash_netport@8ea18000+6290      ip_set_hash_netnet@8ea10000+67f0        ip_set_hash_netiface@8ea00000+62f0      ip_set_hash_net@8edf8000+5c90 ip_set_hash_mac@8edf0000+2970    ip_set_hash_ipportnet@8ede8000+6810     ip_set_hash_ipportip@8ede0000+5590      ip_set_hash_ipport@8edd8000+51f0       ip_set_hash_ipmark@8edc8000+4ef0        ip_set_hash_ip@8edc0000+4dd0    ip_set_bitmap_port@8edb8000+1830       ip_set_bitmap_ipmac@8edb4000+1bb0       ip_set_bitmap_ip@8edb0000+1b10  ip_set@8eda8000+5a2a  nfnetlink@8ee9e000+1277  ip6table_nat@8e9b9000+310       nf_conntrack_ipv6@8eeda000+15c0 nf_defrag_ipv6@8eed4000+137e  nf_nat_ipv6@8eda2000+1033        ip6t_NPT@8ed77000+650   ip6t_MASQUERADE@8ed7a000+270    nf_nat_masquerade_ipv6@8ec9a000+71c    nf_nat@8ed84000+387a    nf_conntrack@8ed60000+1151e     nf_log_ipv6@8ed48000+1070       nf_log_common@8ec71000+acf     ip6table_mangle@8ec76000+4d0    ip6table_filter@8ec79000+2d0    ip6_tables@8e994000+2821        ip6t_REJECT@8ec7e000+450       x_tables@8e9f0000+37cf  nf_reject_ipv6@8e9cd000+a08     nfs@8ed20000+1dad8      msdos@8e9e0000+1990   ifb@8ed1b000+d30 vfat@8ee58000+2250      fat@8ecb0000+d14e       ntfs@8ed00000+180f3     lockd@8eca0000+d43f     sunrpc@8ecc0000+29341  grace@8e90a000+621      cifs@8ec00000+5b8c2     autofs4@8eec8000+57bb   nls_utf8@8ee93000+390   nls_iso8859_1@8ee8f000+b70     nls_cp936@8eea0000+1d770        nls_cp437@8ee8c000+1170 wp512@8ee80000+62b0     twofish_generic@8ee6c000+1f30  twofish_common@8ee68000+3351    tgr192@8ee5c000+3530    tea@8ee4d000+9b0        serpent_generic@8ee48000+4d34  khazad@8ee38000+4a30    cast6_generic@8ee2c000+38ac     cast5_generic@8ee24000+2f8d     cast_common@8ee0e000+11d1      camellia_generic@8ee10000+5530  blowfish_generic@8ee04000+10b0  blowfish_common@8ee00000+1a91   anubis@8e9f8000+26b0   xts@8e9f4000+13ca       sha512_generic@8e9e4000+25f9    sha256_generic@8e980000+29f9    md5@8e9a0000+11b0     md4@8e9cc000+cb0 hmac@8e9e2000+a10       gf128mul@8e8a6000+1a9e  ecb@8e9ee000+5b0        des_generic@8e9e8000+48eb     arc4@8e9d3000+6b0        usb_storage@8f760000+9cab       leds_gpio@8f5ab000+bb0  xhci_plat_hcd@8f658000+15f0     xhci_pci@8f5b1000+c30  xhci_mtk@8f5ca000+12f0  xhci_hcd@8f740000+16de9 sd_mod@8f728000+7600    scsi_mod@8f700000+16dd2 gpio_button_hotplug@8f648000+1ad0      ext4@8f680000+5fb2e     mbcache@8f639000+fce    jbd2@8f5f0000+e481      usbcore@8f600000+2125d nls_base@8f5cc000+1570  usb_common@8ff8a000+a37 crc32c_generic@8ff88000+5b0
<6>[    8.050505] hub 2-0:1.0: USB hub found
<6>[    8.054332] hub 2-0:1.0: 1 port detected
<6>[    8.067413] usbcore: registered new interface driver usb-storage
[... cut ...]
<4>[55353.572861]         8fc0fdb8 00000084 00000000 ac07f583 8fc0fdb8 8ebde600 8eb429a8 8dbf5064
<4>[55353.581194]         8d8a0000 8fc0fdb8 00000084 00000000 835e3f60 8f01c568 8e904684 8fc0fd64
<4>[55353.589527]         ...
<4>[55353.591967] Call Trace:
<4>[55353.594475] [<8f01b148>] ieee80211_rate_control_unregister+0xdc/0x950 [mac80211]
<4>[55353.601877] Code: 10000002  00405825  240b0020 <8cad0010> 00005025  00001025  1000000b  240e0001  8ca90004 
<4>[55353.611606] 
<4>[55353.613323] ---[ end trace c5c4ae5fb7ca81fc ]---

===========================root@OpenWrt:~#

siwind, 2019-11-25:
root@OpenWrt:~# cat /sys/kernel/debug/crashlog 
Time: 1574640813.932966
Modules:        pppoe@8ea80000+2290     ppp_async@8ed66000+1bc0 pppox@8ea1e000+58a      ppp_generic@8ea18000+5b62     mt76x2e@8edf4000+28b0    mt76x2_common@8ec50000+2d88     mt76x02_lib@8f310000+a421       mt7603e@8ede0000+970c   mt76@8ed70000+81e4     mac80211@8f200000+77c69 iptable_nat@8e8dd000+310        ipt_REJECT@8ef0f000+410 ipt_MASQUERADE@8e85c000+2f0    cfg80211@8ed80000+3f299 xt_time@8ed59000+730    xt_tcpudp@8ed56000+750  xt_tcpmss@8ed45000+470  xt_statistic@8ed42000+390      xt_state@8ed3e000+310   xt_socket@8ed3b000+a20  xt_recent@8ed34000+1c40 xt_nat@8ed23000+650     xt_multiport@8e84b000+550      xt_mark@8eefa000+2f0    xt_mac@8ec64000+2b0     xt_limit@8ed0d000+4f0   xt_length@8ed09000+310xt_hl@8ed07000+370       xt_helper@8ed04000+3b0  xt_ecn@8fe34000+590     xt_dscp@8f0ea000+430    xt_conntrack@8ec91000+910      xt_connmark@8ecaf000+610        xt_connlimit@8ed12000+15cd      xt_connbytes@8eced000+690       xt_comment@8ece9000+230        xt_TPROXY@8ecf8000+1210 xt_TCPMSS@8ecdb000+b30  xt_REDIRECT@8f0c4000+310        xt_LOG@8ecd4000+350   xt_HL@8ecd0000+590       xt_FLOWOFFLOAD@8eccd000+c00     xt_DSCP@8ecc9000+630    xt_CT@8ec85000+ab0      xt_CLASSIFY@8ec2f000+290       ums_usbat@8ece0000+20b0 ums_sddr55@8eeda000+1750        ums_sddr09@8ecdc000+23f0        ums_karma@8ecb5000+6f0 ums_jumpshot@8ecba000+1010      ums_isd200@8f094000+1760        ums_freecom@8f1e6000+8d0        ums_datafab@8f0c6000+1370      ums_cypress@8ec90000+a70        ums_alauda@8ec18000+22f0        slhc@8ec76000+14bb      nf_socket_ipv6@8f0e8000+516    nf_socket_ipv4@8f0b2000+616     nf_reject_ipv4@8ee5b000+883     nf_nat_redirect@8f041000+63b    nf_nat_masquerade_ipv4@8f0d0000+71c    nf_conntrack_ipv4@8eca0000+1450 nf_nat_ipv4@8f1e1000+dd1        nf_log_ipv4@8eef7000+db0       nf_flow_table_hw@8eebb000+920   nf_flow_table@8ef68000+365f     nf_defrag_ipv4@8f09d000+476     nf_conntrack_rtcache@8ec27000+af0      nf_conntrack_netlink@8ec28000+5940      macvlan@8ec78000+3462   iptable_raw@8e9e5000+2b0      iptable_mangle@8ec6a000+3d0      iptable_filter@8ec86000+2d0     ipt_ECN@8ecab000+5b0    ip_tables@8f084000+291d crc_ccitt@8f1ee000+42b compat@8ef14000+1de4    act_connmark@8ef3f000+880       sch_tbf@8f0b8000+1730   sch_ingress@8f081000+510       sch_htb@8ef44000+3720   sch_hfsc@8eed0000+3a70  em_u32@8f1f0000+270     cls_u32@8eca8000+2320   cls_tcindex@8eca2000+14f0      cls_route@8ec9e000+14c0 cls_matchall@8ec84000+a30       cls_fw@8ec72000+f70     cls_flow@8ec7e000+16d0cls_basic@8ec6d000+c90   act_skbedit@8ec69000+a20        act_mirred@8ec66000+fc0 ledtrig_usbport@8e9e8000+b00    xt_set@8ec62000+1b70   ip_set_list_set@8ec54000+1cd0   ip_set_hash_netportnet@8ec58000+6cd0    ip_set_hash_netport@8ec48000+6290      ip_set_hash_netnet@8ec40000+67f0        ip_set_hash_netiface@8ec38000+62f0      ip_set_hash_net@8ec30000+5c90 ip_set_hash_mac@8ec1c000+2970    ip_set_hash_ipportnet@8ec20000+6810     ip_set_hash_ipportip@8ec10000+5590      ip_set_hash_ipport@8ec08000+51f0       ip_set_hash_ipmark@8ec00000+4ef0        ip_set_hash_ip@8eef0000+4dd0    ip_set_bitmap_port@8f0fa000+1830       ip_set_bitmap_ipmac@8f0ce000+1bb0       ip_set_bitmap_ip@8f0a0000+1b10  ip_set@8eee8000+5a2a  nfnetlink@8e974000+1277  ip6table_nat@8f1eb000+310       nf_conntrack_ipv6@8ef98000+15c0 nf_defrag_ipv6@8ef2c000+137e  nf_nat_ipv6@8eed8000+1033        ip6t_NPT@8f1f1000+650   ip6t_MASQUERADE@8eeb4000+270    nf_nat_masquerade_ipv6@8eeb8000+7fc    nf_nat@8ee74000+387a    nf_conntrack@8eea0000+1151e     nf_log_ipv6@8f0c2000+1070       nf_log_common@8f1f7000+acf     ip6table_mangle@8f1fc000+4d0    ip6table_filter@8ef37000+2d0    ip6_tables@8ef3c000+2821        ip6t_REJECT@8ef24000+450       x_tables@8ef5c000+37cf  nf_reject_ipv6@8f0df000+a08     nfs@8ee80000+1dad8      msdos@8f15e000+1990   ifb@8ee5c000+d30 vfat@8f0e4000+2250      fat@8ee60000+d14e       ntfs@8ee40000+180f3     lockd@8f170000+d43f     sunrpc@8ee00000+29341  grace@8f048000+621      cifs@8f180000+5b8c2     autofs4@8f0f0000+57bb   nls_utf8@8f03d000+390   nls_iso8859_1@8e90f000+b70     nls_cp936@8f100000+1d770        nls_cp437@8f0e2000+1170 wp512@8f0d8000+62b0     twofish_generic@8f0cc000+1f30  twofish_common@8f0c8000+3351    tgr192@8f0bc000+3530    tea@8f09e000+9b0        serpent_generic@8f0a8000+4d34  khazad@8f098000+4a30    cast6_generic@8f090000+38ac     cast5_generic@8ef40000+2f8d     cast_common@8ef58000+11d1      camellia_generic@8f088000+5530  blowfish_generic@8ef7c000+10b0  blowfish_common@8ef78000+1a91   anubis@8ef70000+26b0   xts@8ef6c000+13ca       sha512_generic@8ef64000+25f9    sha256_generic@8ef00000+29f9    md5@8ef28000+11b0     md4@8e830000+cb0 hmac@8e9f3000+a10       gf128mul@8ef5a000+1a9e  ecb@8e9fd000+5b0        des_generic@8e9f8000+48eb     arc4@8f02e000+6b0        usb_storage@8f6f0000+9cab       leds_gpio@8f691000+bb0  xhci_plat_hcd@8f6ee000+15f0     xhci_pci@8f6b4000+c30  xhci_mtk@8f608000+12f0  xhci_hcd@8f780000+16e09 sd_mod@8f6e0000+7600    scsi_mod@8f6c0000+16df2 gpio_button_hotplug@8f684000+1ad0      ext4@8f700000+5fb2e     mbcache@8f67c000+fce    jbd2@8f630000+e481      usbcore@8f640000+2125d nls_base@8f606000+1570  usb_common@8f5b3000+a37 crc32c_generic@8f5b1000+5b0
<6>[    8.484168] scsi host0: usb-storage 2-1:1.0
<5>[    8.609905] random: procd: uninitialized urandom read (4 bytes read)
<6>[    9.582469] mtk_soc_eth 1e100000.ethernet: PPE started
[... cut ...]
<4>[40837.929451] [<8ede3528>] mt7603_reg_map+0x90/0x280 [mt7603e]
<4>[40837.935086] [<803455d8>] net_rps_send_ipi+0x5c/0x78
<4>[40837.939979] [<8034d498>] net_rx_action+0x150/0x30c
<4>[40837.944834] [<80475a80>] __do_softirq+0x128/0x2e8
<4>[40837.949539] [<80033f90>] irq_exit+0xa8/0xc4
<4>[40837.953703] [<8024eca4>] plat_irq_dispatch+0xf0/0x13c
<4>[40837.958767] [<800074c8>] except_vec_vi_end+0xb8/0xc4
<4>[40837.963718] 
<4>[40837.965198] Code: (Bad address in epc)
<4>[40837.968948] 
<4>[40837.970634] ---[ end trace dda22eacc5040cd6 ]---

root@OpenWrt:~#

araujorm, 2019-11-25:
The latest commit mentions something about softirq: https://github.com/openwrt/mt76/commit/44825e88a42dbd570b142271ea1b4cca9ad5b3c3

May be a longshot but if I were you I'd try compiling with the latest master branch of the driver and see if it solves your issue...

siwind, 2019-12-19:
compiled with the latest master and the latest branch of "openwrt-19.07",  kernel crash  still happen.

siwind, 2019-12-28:
Now, after 2.4G/5Gwifi  is disabled,  the router is very stable.
The best way is to use it as a wired router.

xavier83, 2019-12-28:
> Now, after 2.4G/5Gwifi is disabled, the router is very stable.
> The best way is to use it as a wired router.

OpenW(ired)R(ou)T(er)

nbd168, 2020-02-20:
Crashes fixed in 8f8e9161b3550c9b53a70dec6ceeac0d194a221d (and some other commits)

closed 2020-02-20 by nbd168