openwrt/mt76 #304: Kernel Panic Xiaomi MIR3G on boot

sylvainduchesne, 2019-08-02:
Hi,
Flashed Xiaomi MIR3G using snapshot from https://downloads.openwrt.org/snapshots/targets/ramips/mt7621/ ; downloaded maybe 2 weeks ago. 
After flashing (followed instructions on https://www.reddit.com/r/Xiaomi/comments/abkqyu/mi_3g_router_openwrt_instructions_r3g_the_one/ ) , the router now goes into boot loop.

Here is the output from serial:
```
===================================================================

     		MT7621   stage1 code 10:33:11 (ASIC)

     		CPU=50000000 HZ BUS=12500000 HZ

==================================================================

Change MPLL source from XTAL to CR...

do MEMPLL setting..

MEMPLL Config : 0x11100000

3PLL mode + External loopback

=== XTAL-40Mhz === DDR-1200Mhz ===

PLL4 FB_DL: 0x0, 1/0 = 894/130 01000000

PLL3 FB_DL: 0x8, 1/0 = 622/402 21000000

PLL2 FB_DL: 0xc, 1/0 = 693/331 31000000

do DDR setting..[00320381]

Apply DDR3 Setting...(use customer AC)

          0    8   16   24   32   40   48   56   64   72   80   88   96  104  112  120

      --------------------------------------------------------------------------------

0000:|    0    0    0    0    0    0    0    0    0    0    0    0    0    0    0    0

0001:|    0    0    0    0    0    0    0    0    0    0    0    0    0    0    0    0

0002:|    0    0    0    0    0    0    0    0    0    0    0    0    0    0    0    0

0003:|    0    0    0    0    0    0    0    0    0    0    0    0    0    0    0    0

0004:|    0    0    0    0    0    0    0    0    0    0    0    0    0    0    0    0

0005:|    0    0    0    0    0    0    0    0    0    0    0    0    0    0    0    0

0006:|    0    0    0    0    0    0    0    0    0    0    0    0    0    0    0    0

0007:|    0    0    0    0    0    0    0    0    0    0    0    0    0    0    0    0

0008:|    0    0    0    0    0    0    0    0    0    0    0    0    0    0    0    0

0009:|    0    0    0    0    0    0    0    0    0    0    0    0    0    0    0    0

000A:|    0    0    0    0    0    0    0    0    0    0    0    0    0    0    0    0

000B:|    0    0    0    0    0    0    0    0    0    0    0    0    0    0    0    0

000C:|    0    0    0    0    0    0    0    0    0    0    0    0    0    0    0    0

000D:|    0    0    0    0    0    0    0    0    0    0    0    0    0    0    0    1

000E:|    0    0    0    0    0    0    0    0    0    1    1    1    1    1    1    1

000F:|    0    0    0    0    1    1    1    1    1    1    1    1    1    1    0    0

[... 536 lines cut ...]
[   12.794336] Status: 11007c03	KERNEL EXL IE 
[   12.798508] Cause : 40800008 (ExcCode 02)
[   12.802495] BadVA : fffffff0
[   12.805359] PrId  : 0001992f (MIPS 1004Kc)
[   12.809432] Modules linked in: mt7603e(+) mt76 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_FLOWOFFLOAD xt_CT slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_mangle iptable_filter ip_tables crc_ccitt compat ledtrig_usbport nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 leds_gpio xhci_plat_hcd xhci_pci xhci_mtk xhci_hcd gpio_button_hotplug usbcore nls_base usb_common
[   12.872906] Process kmodloader (pid: 633, threadinfo=8e9ee000, task=8fcf6180, tls=77ffaeb8)
[   12.881214] Stack : 00000008 00000400 8f7ed180 8f1fd5c0 000f0040 8f7ed180 8f1fd5c0 8f257664
[   12.889552]         8f258a48 802fa7e0 00000000 8f253214 000003e8 8f1fd5c0 8f1fd5c0 8f258898
[   12.897887]         8f1fd5c0 8fe9cc00 8f1fd5c0 8f258a48 8f1fcbe0 8f1fc1a0 00000008 00000028
[   12.906220]         80600000 8f251d00 00000017 8f7b9280 00000000 8f1fd5c0 00000017 80080ebc
[   12.914556]         8fe71c00 8e9efba8 8f258a48 802f2a54 8f2585e8 8f1fd5c0 00000000 8fe9cc00
[   12.922894]         ...
[   12.925333] Call Trace:
[   12.927770] [<8f2835c8>] mt76_eeprom_override+0x38/0xdc [mt76]
[   12.933619] [<8f257664>] mt7603_eeprom_init+0x2e4/0x504 [mt7603e]
[   12.939707] [<8f251d00>] mt7603_register_device+0x130/0xb8c [mt7603e]
[   12.946135] [<8f250188>] init_module+0x8b4188/0x8b5240 [mt7603e]
[   12.952125] Code: 14400002  00000000  2402ffed <88430003> 26110a9c  98430000  aa230003  ba230000  90430004 
[   12.961854] 
[   12.963516] ---[ end trace 2cae12dc8e0d9023 ]---
[   12.969816] Kernel panic - not syncing: Fatal exception
[   12.976412] Rebooting in 1 seconds..
```

Is there something wrong with the firmware I used or did I do something wrong?

araujorm, 2019-08-02:
Please try latest version, it should be fixed by a commit made on 22 of July: https://github.com/openwrt/mt76/commit/75656a4590a3288503d42194adf014f4a9eeaa84

sylvainduchesne, 2019-08-02:
Thanks I will give it a try. 
I typically don't reinstall FW through serial, so I am not sure how to fully proceed. Do I need to reflash kernel0.bin, reload original firmware, and then reflash openwrt, or is there a simpler way to do so?

araujorm, 2019-08-02:
I don't have that router, but judging from the instructions on the thread you mentioned, I'd say you'll have to recover as if it was bricked, because currently you can't SSH to the router (as it kernel panics during the boot).

Maybe something like this:
https://forum.openwrt.org/t/xiaomi-wifi-router-3g/5377/104

If it doesn't work, I'd say to try asking in that forum for help on how to recover from a brick on Xiaomi 3G router. And when you do, please report back here if the latest firmware worked for you or if there still is an issue with the mt76 driver.

sylvainduchesne, 2019-08-02:
yea ok, thats also what I figured. Will report back when I test the latest. 
Thanks for the help.

ynezz, 2019-08-02:
Try latest snapshot, it should be fixed.

sylvainduchesne, 2019-08-05:
Reporting back. Indeed latest snapshot fixed the issue. Thanks!

closed 2019-08-05 by sylvainduchesne