openwrt/mt76 #305: mt7615: slow AC wifi speed on Linksys EA7500 v2

jackcolentern, 2019-08-03:
Hi, I am trying to bring support to Linksys EA7500 V2 with a MT7615
The wifi works but the AC speed is about 50~150Mbps, which is not normal because the original firmware can do ～350Mbps, 

```
[    0.000000] Linux version 4.14.132 (jksl@jksl-ZenBook-Pro-15-UX550GDX-UX580GD) (gcc version 7.4.0 (OpenWrt GCC 7.4.0 r10588-7b73cad7be)) #0 SMP Tue Jul 30 13:00:52 2019
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
[    0.000000] bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
[    0.000000] MIPS: machine is Linksys EA7500 V2
[    0.000000] Determined physical RAM map:
[    0.000000]  memory: 10000000 @ 00000000 (usable)
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] VPE topology {2,2} total 4
[    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000]   HighMem  empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000] On node 0 totalpages: 65536
[    0.000000] free_area_init_node: node 0, pgdat 805ee4c0, node_mem_map 81003000
[    0.000000]   Normal zone: 576 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 65536 pages, LIFO batch:15
[    0.000000] random: get_random_bytes called from start_kernel+0x9c/0x4d8 with crng_init=0
[    0.000000] percpu: Embedded 14 pages/cpu s26064 r8192 d23088 u57344
[    0.000000] pcpu-alloc: s26064 r8192 d23088 u57344 alloc=14*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 64960
[    0.000000] Kernel command line: console=ttyS0,115200 rootfstype=squashfs,jffs2
[    0.000000] PID hash table entries: 1024 (order: 0, 4096 bytes)
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes)
[    0.000000] Writing ErrCtl register=00012400
[    0.000000] Readback ErrCtl register=00012400
[    0.000000] Memory: 251692K/262144K available (4784K kernel code, 251K rwdata, 1036K rodata, 1276K init, 254K bss, 10452K reserved, 0K cma-reserved, 0K highmem)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000] NR_IRQS: 256
[    0.000000] CPU Clock: 880MHz
[    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4343773742 ns
[    0.000009] sched_clock: 32 bits at 440MHz, resolution 2ns, wraps every 4880645118ns
[    0.007806] Calibrating delay loop... 586.13 BogoMIPS (lpj=2930688)
[    0.073978] pid_max: default: 32768 minimum: 301
[    0.078799] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.085308] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.094469] Hierarchical SRCU implementation.
[    0.099691] smp: Bringing up secondary CPUs ...
[    0.104925] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.104935] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.104946] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.105083] CPU1 revision is: 0001992f (MIPS 1004Kc)
[    0.164447] Synchronize counters for CPU 1: done.
[    0.198797] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.198806] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.198815] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.198892] CPU2 revision is: 0001992f (MIPS 1004Kc)
[    0.255670] Synchronize counters for CPU 2: done.
[    0.286155] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.286164] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.286172] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.286250] CPU3 revision is: 0001992f (MIPS 1004Kc)
[    0.340855] Synchronize counters for CPU 3: done.
[    0.370721] smp: Brought up 1 node, 4 CPUs
[    0.379006] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[... 207 lines cut ...]
[   20.223050] mt7615e 0000:02:00.0: HW/SW Version: 0x8a108a10, Build Time: 20180518100604a
[   20.223050] 
[   20.248386] mt7615e 0000:02:00.0: N9 Firmware Version: _reserved_, Build Time: 20190103180756
[   20.264395] mt7615e 0000:02:00.0: CR4 Firmware Version: _reserved_, Build Time: 20181207140436
[   27.910461] ieee80211 phy1: Selected rate control algorithm 'minstrel_ht'
[   27.918468] PPP generic driver version 2.4.2
[   27.924323] NET: Registered protocol family 24
[   27.931813] kmodloader: done loading kernel modules from /etc/modules.d/*
[   34.245615] mtk_soc_eth 1e100000.ethernet: PPE started
[   34.259745] device eth0 entered promiscuous mode
[   34.265531] br-lan: port 1(eth0.2) entered blocking state
[   34.271009] br-lan: port 1(eth0.2) entered disabled state
[   34.277269] device eth0.2 entered promiscuous mode
[   34.287508] br-lan: port 1(eth0.2) entered blocking state
[   34.292965] br-lan: port 1(eth0.2) entered forwarding state
[   34.299140] IPv6: ADDRCONF(NETDEV_UP): br-lan: link is not ready
[   35.283365] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   36.585296] IPv6: ADDRCONF(NETDEV_UP): wlan1: link is not ready
[   36.595642] br-lan: port 2(wlan1) entered blocking state
[   36.600994] br-lan: port 2(wlan1) entered disabled state
[   36.606979] device wlan1 entered promiscuous mode
[   39.546267] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
[   39.552951] br-lan: port 2(wlan1) entered blocking state
[   39.558252] br-lan: port 2(wlan1) entered forwarding state
```

dissent1, 2019-08-03:
In my case the tx speed is ~350 but rx is around 50 and drops very quickly with distance. At around 5 meters the wifi is unusable.
TP-Link RE650 with eLNA/ePA.

sanitariu, 2019-08-04:
Same is with Mediatek 7628 for example TP-Link 840 version 6.
2 meters away and signal is 50% low.
Also from latest reisions router start to stuck each 3-4 days.
I am unable to print crash log because it needs power cycle.
Main problem is txpower. Seems like it is working at 15db or 15Miliwatt
Maybe this is the default.
All mediatek chips have that txpower problems.

iscilyas, 2019-08-04:
same with xiaomi mir3p (mt7621 with dual mt7615)... getting rates of 200mbit (on AC) while on closed-source driver could achieve 400mbit... N is also about half the speed of closed-source driver.

otherwise, congratulations and good job! very nice that it's working!

iscilyas, 2019-08-09:
sorry just wanted to update/fix my previous post. i've done more testing, and what i see is more like 336+mbps on 5ghz and 115mbps on 2.4ghz (40mhz channel). this matches the speed of the closed-source mt7615e driver.

i'm also not seeing any significant quality drop with reasonable distance (1 very thick wall, 10-15 metres away from router)... my impression is that it's fine at even greater distances (but the network here is rather complicated so i can't be sure without testing more)....

so: fantastic work!

ashbenz, 2019-08-14:
Hey @dissent1 !
I was eyeing the same unit you have (RE650), how do you like it? I was under the impression the low txpower issue was fixed on recent mt76 releases, but maybe not yet for the 7615?

Generally speaking are you happy with the device? If it was possible to power it with USB, or at least DC in I would've liked it much better than the wall-plug form factor, but generally it seems to fit a particular need.. Did you test it with a QCA chipset? Eg using it as wireless bridge and seeing how fast it can push when linked to another 4x4 unit..

pjgowtham, 2019-08-15:
I only tested 5Ghz since 2.4Ghz is mostly being used by IOT anyways.
I tested it using iperf on an Asus TUF FX505DT, which probably has a 2x2 antenna. The power mode was on high performance while testing. I did a 60 seconds run each. Surprisingly, the speeds were better than that of the Archer C7 v4.

Mi R3P Tx @20dbm channel 36, Band width 80Mhz. 160Mhz wouldn't work anyways.
Run 1 - 287Mbit/sec
Run 2 - 271Mbit/sec

whereas my archer c7 1950 v4 with latest snapshot

Run 1 - 244Mbit/sec
Run 2 - 258Mbit/sec

nbd168, 2019-08-28:
Please try the latest version, it should work better now

jackcolentern, 2020-01-22:
I retested using a EA8100-ah which also use MT7615N, same problem occur

I noticed that in LUCI, Network -> Wireless -> Associated Stations is only showing

RX Rate / TX Rate
6.0 Mbit/s, 20MHz
585.1 Mbit/s, 80MHz, VHT-MCS 6, VHT-NSS 2, Short GI

it this normal?

closed 2020-01-23 by nbd168