openwrt/mt76 #307: xhci_hcd 0000:00:14.0: WARN Wrong bounce buffer write length

meeuw, 2019-08-13:
Since I'm using mt76x0u from this repository (instead of the version supplied by 5.1.20) for my ASUS AC51 my connection is much better. Thanks!

I only get the following warnings in dmesg:
```
[  528.344778] xhci_hcd 0000:00:14.0: WARN Wrong bounce buffer write length: 208 != 0
[  535.784660] xhci_hcd 0000:00:14.0: WARN Wrong bounce buffer write length: 424 != 208
[  536.058058] xhci_hcd 0000:00:14.0: WARN Wrong bounce buffer write length: 284 != 0
[  536.123252] xhci_hcd 0000:00:14.0: WARN Wrong bounce buffer write length: 158 != 424
[  536.533650] xhci_hcd 0000:00:14.0: WARN Wrong bounce buffer write length: 208 != 158
[  542.161184] xhci_hcd 0000:00:14.0: WARN Wrong bounce buffer write length: 208 != 284
[  543.672329] xhci_hcd 0000:00:14.0: WARN Wrong bounce buffer write length: 134 != 208
[  569.489808] xhci_hcd 0000:00:14.0: WARN Wrong bounce buffer write length: 208 != 134
```
(and this goes on)

Is this something I should/can fix? Do you know what has been fixed in this repository that isn't in linux stable yet? I'd like to switch back when that fix is merged.

sgruszka, 2019-08-13:
There is patch for this warning
https://marc.info/?l=linux-usb&m=156152916509643&w=2

Why on 5.1.20 connection is worse over this repo? Do we need to backport some mt76x0u fix to linux 5.1 ?

meeuw, 2019-08-13:
Thanks, if I look at the patch it's just a false-positive warning which I ignore for now.

I will have a look for what the differences are for mt76x0u in this repository and linux stable.

closed 2019-08-13 by meeuw