openwrt/mt76 #308: openwrt-19.07 status?

jsantala, 2019-08-13:
I switched to openwrt-19.07 branch to get ready for the eventual 19.07 release which I could then keep using for some time, but a quick test resulted in abysmal wifi performance on a mt7628 device. What is the openwrt-19.07 branch status regarding the mt76 driver and what will it be going forward? Thanks for your great work on the driver.

julianocss, 2019-08-13:
Hello @jsantala.

The driver in 19.07 branch it´s not the latest version, the latest version had a lot of improvements and is much more stable.

You should always use the Openwrt snapshot builds to get the latest drivers, which is based on master openwrt branch.
What is your device?

Juliano

jsantala, 2019-08-13:
I tested on VoCore2, but the one I'll really be using is Omega2S+. Any chance of getting the updated driver into the openwrt-19.07 branch? I'd feel a lot more comfortable using a release branch (and eventually a release) in the long run, since I'm sure master is not going to be 100% stable at all times.

julianocss, 2019-08-13:
Yes, it is possible. Are you familiar with Openwrt build system?

jsantala, 2019-08-14:
I've been building my own firmware images for some time, but that's as deep as I've gone so far. Let me know if there's anything I can do to help in this matter.

julianocss, 2019-08-14:
Hello @jsantala.

You will have to work with mt76 Makefile in the build_dir.

Take a look in this thread, i explained how to do this.

https://forum.openwrt.org/t/tl-wr841n-v13-wifi-not-stable/16018/124

Regards.
Juliano

jsantala, 2019-08-14:
Thanks! I think I got it. Building with commit https://github.com/openwrt/mt76/tree/3245ca8b8aeb74c404ce28e0de41ce15772022f9 now.

closed 2019-08-14 by jsantala

jsantala, 2019-08-16:
I suppose I can just do this in my build script to always use the latest mt76 driver:

```
DATE=`date +%y-%m-%d`
sed -i s/PKG_SOURCE_VERSION:=.*/PKG_SOURCE_VERSION:=master/ package/kernel/mt76/Makefile
sed -i s/PKG_SOURCE_DATE:=.*/PKG_SOURCE_DATE:=$DATE/ package/kernel/mt76/Makefile
```

julianocss, 2019-08-16:
Hello @jsantala.

To get the latest changes, you have to use mt76 repo. To get the latest official commits to Openwrt master, you can use your script.

Tell us about the latest changes.
It is more stable to you?

Juliano

jsantala, 2019-08-16:
Resulting list of variables would be, for example:

```
PKG_SOURCE_URL:=https://github.com/openwrt/mt76
PKG_SOURCE_DATE:=19-08-16
PKG_SOURCE_VERSION:=master
```

I think this should result in the latest drivers directly from mt76?

Current situation seems usable. Interestingly, when one connects to the AP, the connection seems pretty bad at first, but when you put some traffic on it, it seems to get better.

julianocss, 2019-08-16:
@jsantala.

No.
The version is the "name" of the commit.

I.E.: https://github.com/openwrt/mt76/tree/3245ca8b8aeb74c404ce28e0de41ce15772022f9

VERSION:=3245ca8b8aeb74c404ce28e0de41ce15772022f9

I really recommend you to try the latest snapshot. It is stable and have fixes that affects mt76 users.

Juliano

jsantala, 2019-08-16:
Ok. Will do.

jsantala, 2019-08-16:
Seems really good with 3245ca8b8aeb74c404ce28e0de41ce15772022f9. Thanks!