openwrt/mt76 #311: rcu_sched self-detected stall on CPU after calling mt76_dma_attach

araujorm, 2019-08-30:
Hello.

Not sure if it's a mt76 issue, but the fact is that I get freezes that only seem to happen (not always) after mt76_dma_attach has been invoked, with the error ```rcu_sched self-detected stall on CPU``` (see full log below).

The device is a Xiaomi Mi Router 4A Gigabit Edition (SoC mt7621). It's not yet on the official Openwrt tree, but support is being worked on and discussed on a thread in OpenWrt forum.

I updated OpenWrt to the latest master (before applying the device's WIP support patch), so mt76 version is updated and is the following:
```
PKG_SOURCE_DATE:=2019-08-22
PKG_SOURCE_VERSION:=2a0edbb4473b71dfaa4756f7189abf6cb7f7be8b
```
Device is working in AP mode, one of the clients is an IP camera that is constantly recording and sending traffic, and that camera is monitorized (ping) by a icinga2 instance so I usually get warned when this happens after a few minutes at most.

It happened today again and device lost WiFi for about 15 minutes. All devices connected to this router lost connection during that time. After that period, the connections started working on their own again, however much slower than usual (until I rebooted the router). This usually happens again 1 to 2 days after rebooting (so if I reboot the router every 2 days via crontab I don't have much issues, but of course that's not a good solution).

During the "freeze", several messages like the following appeared on the kernel ring buffer:
```
[134641.605038] INFO: rcu_sched self-detected stall on CPU
[134641.610304] 	2-...: (330089 ticks this GP) idle=2b2/140000000000001/0 softirq=7415304/7415304 fqs=164677 
[134641.619916] 	 (t=330091 jiffies g=3011804 c=3011803 q=22895)
[134641.625028] INFO: rcu_sched detected stalls on CPUs/tasks:
[134641.625670] NMI backtrace for cpu 2
[134641.634772] CPU: 2 PID: 20863 Comm: kworker/u8:2 Not tainted 4.14.140 #0
[134641.641574] Workqueue: phy0 mt7603_mac_work [mt7603e]
[134641.646700] Stack : 804d8a1c 87c0dd24 862a5900 80550000 804d8a34 80570000 80550000 8054a0cc
[134641.655160]         00000000 8006875c 877e378c 00000001 862a5900 00000001 87c0dce8 ac07f595
[134641.663614]         00000000 00000000 00000007 00000000 000006e7 00000000 00000018 20307968
[134641.672066]         3637746d 8629bdc8 00000000 6d5f3330 00000000 ffffffff 00000000 80570000
[134641.680520]         80550000 8054a0cc 00000000 80550000 00000008 8027cad0 00000008 806b0008
[134641.688974]         ...
[134641.691511] Call Trace:
[134641.694077] [<8000bff4>] show_stack+0x74/0x104
[134641.698637] [<8040e1c4>] dump_stack+0x9c/0xd8
[134641.703092] [<80414a1c>] nmi_cpu_backtrace+0xc8/0xd8
[134641.708139] [<80414ae8>] nmi_trigger_cpumask_backtrace+0xbc/0x134
[134641.714324] [<8007b48c>] rcu_dump_cpu_stacks+0x110/0x11c
[134641.719716] [<8007a9a0>] rcu_check_callbacks+0x334/0x7c0
[134641.725122] [<8007db88>] update_process_times+0x34/0x7c
[134641.730431] [<8008cea0>] tick_handle_periodic+0x34/0xd0
[134641.735767] [<802f30bc>] gic_compare_interrupt+0x94/0xb4
[134641.741166] [<8006f270>] handle_percpu_devid_irq+0x78/0x140
[134641.746840] [<80069ebc>] generic_handle_irq+0x24/0x3c
[134641.751987] [<80237dac>] gic_handle_local_int+0xbc/0xe4
[134641.757294] [<80237f30>] gic_irq_dispatch+0x10/0x20
[134641.762253] [<80069ebc>] generic_handle_irq+0x24/0x3c
[134641.767410] [<80428154>] do_IRQ+0x1c/0x2c
[134641.771505] [<80236f10>] plat_irq_dispatch+0x108/0x124
[134641.776726] [<80006c08>] except_vec_vi_end+0xb8/0xc4
[134641.781792] [<869a9688>] mt76_dma_attach+0x108c/0x1228 [mt76]
[134641.787651] 	2-...: (330089 ticks this GP) idle=2b2/140000000000001/0 softirq=7415304/7415304 fqs=164679 
[134641.797262] 	(detected by 1, t=330094 jiffies, g=3011804, c=3011803, q=22895)
[134641.804473] Sending NMI from CPU 1 to CPUs 2:
[134641.808911] NMI backtrace for cpu 2
[134641.812490] CPU: 2 PID: 20863 Comm: kworker/u8:2 Not tainted 4.14.140 #0
[134641.819275] Workqueue: phy0 mt7603_mac_work [mt7603e]
[134641.824404] task: 877e3400 task.stack: 87e96000
[134641.829003] $ 0   : 00000000 00000001 8042ba10 00000000
[134641.834321] $ 4   : 87e99068 06ab1c6e 000005fe 00000001
[134641.839641] $ 8   : 00000010 80423f20 00000000 00000000
[134641.844959] $12   : 7fc78610 8629bdc8 00000000 77f9d2a0
[134641.850280] $16   : 87e99068 869fdde0 86e68360 ffffffff
[134641.855602] $20   : 00000003 00000001 a6c0a2e0 f9e22be8
[134641.860920] $24   : 00000000 869a802c                  
[134641.866238] $28   : 87e96000 87e97da0 802045fe 869a883c
[134641.871560] Hi    : 0000002e
[134641.874516] Lo    : 00000000
[134641.877491] epc   : 869a883c mt76_dma_attach+0x240/0x1228 [mt76]
[134641.883573] ra    : 869a883c mt76_dma_attach+0x240/0x1228 [mt76]
[134641.889643] Status: 11007c03	KERNEL EXL IE 
[134641.893918] Cause : 50800400 (ExcCode 00)
[134641.898001] PrId  : 0001992f (MIPS 1004Kc)
[134641.902177] CPU: 2 PID: 20863 Comm: kworker/u8:2 Not tainted 4.14.140 #0
[134641.908949] Workqueue: phy0 mt7603_mac_work [mt7603e]
[134641.914072] Stack : 804d8a1c 87c0dd6c 862a5900 80550000 804d8a34 80570000 00000000 804ddf3c
[134641.922527]         87c0ded4 8006875c 877e378c 00000001 862a5900 00000001 87c0dd30 ac07f595
[134641.930983]         00000000 00000000 00000007 00000000 00000718 00000000 00000018 20307968
[134641.939435]         3637746d 44000000 00000000 6d5f3330 00000000 ffffffff 00000000 80570000
[134641.947891]         00000000 804ddf3c 87c0ded4 80550000 00000008 8027cad0 00000008 806b0008
[134641.956345]         ...
[134641.958881] Call Trace:
[134641.961428] [<8000bff4>] show_stack+0x74/0x104
[134641.965975] [<8040e1c4>] dump_stack+0x9c/0xd8
[134641.970418] [<80414a0c>] nmi_cpu_backtrace+0xb8/0xd8
[134641.975464] [<80008800>] handle_backtrace+0x10/0x28
[134641.980444] [<8009366c>] flush_smp_call_function_queue+0xe0/0x1b0
[134641.986627] [<80010e64>] ipi_call_interrupt+0x10/0x20
[134641.991767] [<8006a998>] __handle_irq_event_percpu+0x98/0x168
[134641.997593] [<8006aa88>] handle_irq_event_percpu+0x20/0x64
[134642.003157] [<8006ab2c>] handle_irq_event+0x60/0xa4
[134642.008120] [<8006f120>] handle_edge_irq+0x118/0x144
[134642.013165] [<80069ebc>] generic_handle_irq+0x24/0x3c
[134642.018307] [<80237ef8>] gic_handle_shared_int+0x124/0x14c
[134642.023871] [<80069ebc>] generic_handle_irq+0x24/0x3c
[134642.029019] [<80428154>] do_IRQ+0x1c/0x2c
[134642.033114] [<80236f10>] plat_irq_dispatch+0x108/0x124
[134642.038332] [<80006c08>] except_vec_vi_end+0xb8/0xc4
[134642.043387] [<869a883c>] mt76_dma_attach+0x240/0x1228 [mt76]
```
For the time being, I've disabled the reboot via crontab to see if anything new appears.
Meanwhile, with this information, do you have any advice? If it's not related to mt76, then sorry and please let me know.
Thanks in advance.

araujorm, 2019-09-07:
Hello.

Meanwhile a different support patch was tried to the device (Xiaomi Mi Router 4A Gigabit Edition) and I've had no more errors after that, for about a week. At this point I can't relate the errors I had with the mt76 driver, so I'll close this issue. Should anything else happens that's more likely to be related, I'll open another issue if appropriate.

Thanks to @nbd168, @LorenzoBianconi, @sgruszka and all the others that have been doing this extremely great work on this driver.

closed 2019-09-07 by araujorm

dhimant77, 2024-09-12:
@araujorm 
Can you share which patch you applied for resolving this issue ?

I am also facing same issue on mt76 device.
Any help is appreciated!

araujorm, 2024-11-13:
> @araujorm Can you share which patch you applied for resolving this issue ?
> 
> I am also facing same issue on mt76 device. Any help is appreciated!

@dhimant77 I'm sorry, I no longer have the device (it eventually failed, the hardware was pretty weak) and whatever patch I may have applied back then is now gone. But I have the impression that everything was eventually merged in the mt76 repository, so it should already be present in recent kernels and OpenWRT releases. Sorry I can't be of much more help.