openwrt/mt76 #312: mt76: unusually high load

dmke, 2019-09-16:
We've recently switched to this OSS driver on our router fleet and are now monitoring unusual load behaviour.

After boot, the load stays at an expected 0.05-0.1, before jumping to a load of 1.0. It stays there and does not recover back to "normal" load (unless we reboot the device; turning wifi off/on again does not help). Some routers even jump to a load of 2.0:

[image]

Here's a version [without annotations](https://user-images.githubusercontent.com/327411/64959178-df044500-d890-11e9-94fe-360425a46f87.png) and [for single router](https://user-images.githubusercontent.com/327411/64959408-576b0600-d891-11e9-9bde-96299a7ac8d3.png) (both graphs show a 2h load average window over the last 32 days). If we zoom in to the last 24 hours, we observe a sharp border at load level 2 (the *minumum* load is exactly 2.0):

[image]

The frustrating part is that the logs (`dmesg`/`logread`) don't show anything unusual: Wifi clients connect, request a DHCP lease and, after some time disconnect (where available, clients also successfully perform the RADIUS/WPA2-Enterprise handshake).

<details><summary><code>/etc/config/wireless</code></summary>

```
config wifi-device 'radio0'
	option type 'mac80211'
	option channel '36'
	option hwmode '11a'
	option path 'pci0000:00/0000:00:01.0/0000:02:00.0'
	option htmode 'VHT80'
	option disabled '0'
	option country 'DE'

config wifi-device 'radio1'
	option type 'mac80211'
	option channel '11'
	option hwmode '11g'
	option path 'pci0000:00/0000:00:00.0/0000:01:00.0'
	option htmode 'HT20'
	option basic_rate '6000 9000 12000 24000 54000'
	option supported_rates '6000 9000 12000 18000 24000 36000 48000 54000'
	option disabled '0'

config wifi-iface 'guest_radio0'
	option device 'radio0'
	option mode 'ap'
	option network 'guest'
	option ssid '<omitted>'
	option isolate '1'
	option disabled '0'

config wifi-iface 'guest_radio1'
	option device 'radio1'
	option network 'guest'
	option mode 'ap'
	option ssid '<omitted>'
	option disabled '0'
	option isolate '1'
```

</details>

We're currently running [OpenWRT `4c46bbbd`](https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=4c46bbbd930406449f4dbb1a07204ef9187de05c), which includes 3d7f7384 (I realise this is a few commits behind `master`), on ZBT-WG3526-23M hardware (MT7612E (ac) and MT7603E (b/g/n) chips). We do compile with `glibc` (for GCCGO support), though.

Is there a way to debug this? Are there increased log levels (maybe behind compile-time flags)?

/cc @corny, @0x5a17ed

ynezz, 2019-09-16:
> Is there a way to debug this?

[perf](http://www.brendangregg.com/perf.html) & co.

dmke, 2019-09-16:
I will look into perf, thanks for the pointer.

corny, 2019-09-25:
When the load increases the 2.4 GHz wifi (`mt7603e`) stops working.

nbd168, 2019-10-11:
Please try the latest version

closed 2019-12-13 by nbd168