openwrt/mt76 #313: Frequent Xiaomi Mi WiFi Router 3G reboots (crashes) under certain circumstances

lumixen, 2019-09-21:
Hello everyone.

Last few months I observe quite frequent reboots with my Xiaomi Mi WiFi Router 3G + openwrt. I'm not 100% confident that the root cause of these reboots is `mt76` driver although I found some certain correlation between connected devices in the house and reboots.

It's pretty hard for me to do traffic sniffing\debugging because I don't know really when reboot might happen again but what I actually can do is to capture crashlogs after each reboot.
So far I can confirm that it happens for me on `18.06.4` and latest openwrt master builds (e.g. `5ef9e4`)

Example of the crashlog:
```
<1>[ 1880.325436] CPU 3 Unable to handle kernel paging request at virtual address 00000804, epc == 8047d2c4, ra == 8e7e1a08
<4>[ 1880.336052] Oops[#1]:
<4>[ 1880.338320] CPU: 3 PID: 0 Comm: swapper/3 Not tainted 4.14.143 #0
<4>[ 1880.344385] task: 8fc3cb00 task.stack: 8fc68000
<4>[ 1880.348891] $ 0   : 00000000 00000001 00000fe5 00000001
<4>[ 1880.354102] $ 4   : 00000804 00000000 00000040 00000001
<4>[ 1880.359311] $ 8   : 00000001 00000002 00000023 00000000
<4>[ 1880.364521] $12   : 00000000 3fff0000 ffffffff 40000000
<4>[ 1880.369731] $16   : 8dc413c0 00000800 00000804 00000000
<4>[ 1880.374941] $20   : 0d27f000 000000a4 8e825dc0 8e826288
<4>[ 1880.380151] $24   : 0d236800 8e7e002c
<4>[ 1880.385361] $28   : 8fc68000 8fc0fdb8 00000088 8e7e1a08
<4>[ 1880.390571] Hi    : 00000000
<4>[ 1880.393433] Lo    : 00000080
<4>[ 1880.396333] epc   : 8047d2c4 _raw_spin_lock_bh+0x18/0x58
<4>[ 1880.401644] ra    : 8e7e1a08 mt76_dma_cleanup+0x5c4/0x63c [mt76]
<4>[ 1880.407617] Status: 11007c03      KERNEL EXL IE
<4>[ 1880.411790] Cause : 40800008 (ExcCode 02)
<4>[ 1880.415777] BadVA : 00000804
<4>[ 1880.418639] PrId  : 0001992f (MIPS 1004Kc)
<4>[ 1880.422712] Modules linked in: pppoe ppp_async l2tp_ppp pptp pppox ppp_mppe ppp_generic nf_nat_pptp nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet nf_conntrack_pptp nf_conntrack_ipv6 mt76x2e mt76x2_common mt76x02_lib mt7603e mt76 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_tcpmss xt_string xt_statistic xt_state xt_recent xt_quota xt_pkttype xt_owner xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_iprange xt_hl xt_helper xt_esp xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_bpf xt_addrtype xt_TCPMSS xt_REDIRECT xt_NETMAP xt_LOG xt_IPMARK xt_HL xt_FLOWOFFLOAD xt_DSCP xt_CT xt_CLASSIFY usblp ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda ts_fsm ts_bm
<4>[ 1880.493046]  slhc nft_set_rbtree nft_set_hash nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir_ipv4 nft_redir nft_quota nft_numgen nft_nat nft_meta nft_masq_ipv4 nft_masq nft_log nft_limit nft_flow_offload nft_exthdr nft_ct nft_counter nft_chain_route_ipv6 nft_chain_route_ipv4 nft_chain_nat_ipv4 nf_tables_ipv6 nf_tables_ipv4 nf_tables_inet nf_tables_arp nf_tables nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_redirect nf_nat_proto_gre nf_nat_masquerade_ipv4 nf_nat_irc nf_conntrack_ipv4 nf_nat_ipv4 nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_nat nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_rtcache nf_conntrack_proto_gre nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast
<4>[ 1880.565151]  ts_kmp nf_conntrack_amanda nf_conntrack iptable_raw iptable_mangle iptable_filter ipt_ah ipt_ECN ip_tables crc_ccitt compat_xtables compat fuse ledtrig_usbport xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6t_rt ip6t_mh ip6t_ipv6header ip6t_hbh ip6t_frag ip6t_eui64 ip6t_ah nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 nfsv4 nfsv3 nfs msdos ip_gre gre l2tp_netlink l2tp_core udp_tunnel ip6_udp_tunnel ip_tunnel tun vfat fat ntfs lockd sunrpc grace dns_resolver nls_utf8 nls_iso8859_1
<4>[ 1880.636622]  nls_cp437 sha1_generic ecb arc4 uas usb_storage leds_gpio xhci_plat_hcd xhci_pci xhci_mtk xhci_hcd uhci_hcd ohci_platform ohci_hcd sd_mod scsi_mod gpio_button_hotplug ext4 mbcache jbd2 usbcore nls_base usb_common crc32c_generic
<4>[ 1880.657918] Process swapper/3 (pid: 0, threadinfo=8fc68000, task=8fc3cb00, tls=00000000)
<4>[ 1880.665969] Stack : 00000088 8e7e19c4 8e824bc0 8e826890 8e825dc0 00000001 00000001 00000000
<4>[ 1880.674306]         0d27f000 00000fe5 8fc0fe28 8e825dc0 8e825dc0 8d3ed63c 8e825dc0 8ee50464
<4>[ 1880.682642]         00000001 8d3ed63c 00000000 8e825dc0 8d3ed63c 8d3ed63c 8e826ca8 8ee51868
<4>[ 1880.690978]         8c09e000 00000740 00000000 8e825dfc 8fc0fe28 8fc0fe28 00000000 8e7e0a7c
<4>[ 1880.699312]         8ea21180 8d3ed73c 8e826368 8e7e3228 8d236800 00000000 8fd1a2b0 80590000
<4>[ 1880.707645]         ...
<4>[ 1880.710083] Call Trace:
<4>[ 1880.712558] [<8047d2c4>] _raw_spin_lock_bh+0x18/0x58
<4>[ 1880.717536] [<8e7e1a08>] mt76_dma_cleanup+0x5c4/0x63c [mt76]
<4>[ 1880.723177] Code: af830010  0000000f  24030001 <c0850000> 14a00005  00000000  00600825  e0810000  1020fffa
<4>[ 1880.732908]
<4>[ 1880.734640] ---[ end trace f0933f4a7a0dcfa9 ]---
```

I know that the description isn't really exhaustive and it might be hard to figure out the real cause of these so I'm going to iteratively update this issue with crashlogs after each reboot. Who knows, perhaps there exists some correlation between them?

[crash_1.log](https://github.com/openwrt/mt76/files/3638748/crash_1.log)
[crash_2.log](https://github.com/openwrt/mt76/files/3638750/crash_2.log)
[crash_3.log](https://github.com/openwrt/mt76/files/3638751/crash_3.log)

lumixen, 2019-10-05:
Hey @nbd168 , I don't know if it's related, but after series of these patches
[image]
I haven't seen reboots any longer. Weird though, I've tried once built image without flow offload modules though still got reboots🤔

nbd168, 2019-10-11:
I think the last instances of these crashes (even without flow offload) should be fixed in the latest version. Please test

lumixen, 2019-10-12:
Indeed, so far - no reboots. Going to close this ticket, thank you!

lumixen, 2019-12-22:
Unfortunately, after some additional testing I'm still observing occasional reboots even on the latest master.
```
<1>[55228.126458] CPU 2 Unable to handle kernel paging request at virtual address 0740607c, epc == 8e9dc578, ra == 8e9ddaa4
<4>[55228.137075] Oops[#1]:
<4>[55228.139345] CPU: 2 PID: 0 Comm: swapper/2 Not tainted 4.14.158 #0
<4>[55228.145414] task: 8fc3c620 task.stack: 8fc5e000
<4>[55228.149920] $ 0   : 00000000 00000001 0000001d 00000000
<4>[55228.155133] $ 4   : 00000003 07406000 8dcfed80 00000000
<4>[55228.160345] $ 8   : 00000100 00000019 00000002 00000003
<4>[55228.165564] $12   : 8e9e6344 00000380 00000000 00000002
<4>[55228.170780] $16   : 0759aaed 813495ed 00000000 000003ff
<4>[55228.175994] $20   : 00000000 00000000 00000084 00000000
<4>[55228.181209] $24   : 160d02b0 8d50a210
<4>[55228.186421] $28   : 8fc5e000 8fc0dd50 8dd9d6a2 8e9ddaa4
<4>[55228.191642] Hi    : 065abb70
<4>[55228.194509] Lo    : a9666e40
<4>[55228.197518] epc   : 8e9dc578 minstrel_calc_rate_stats+0xad4/0x3164 [mac80211]
<4>[55228.204655] ra    : 8e9ddaa4 minstrel_calc_rate_stats+0x2000/0x3164 [mac80211]
<4>[55228.211842] Status: 11007c03      KERNEL EXL IE
<4>[55228.216014] Cause : 40800008 (ExcCode 02)
<4>[55228.220001] BadVA : 0740607c
<4>[55228.222870] PrId  : 0001992f (MIPS 1004Kc)
<4>[55228.226943] Modules linked in: pppoe ppp_async l2tp_ppp pptp pppox ppp_mppe ppp_generic nf_nat_pptp nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet nf_conntrack_pptp nf_conntrack_ipv6 mt76x2e mt76x2_common mt76x02_lib mt7603e mt76 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_tcpmss xt_string xt_statistic xt_state xt_recent xt_quota xt_pkttype xt_owner xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_iprange xt_hl xt_helper xt_esp xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_bpf xt_addrtype xt_TCPMSS xt_REDIRECT xt_NETMAP xt_LOG xt_HL xt_FLOWOFFLOAD xt_DSCP xt_CT xt_CLASSIFY wireguard usblp ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda ts_fsm ts_bm
<4>[55228.297288]  slhc nft_set_rbtree nft_set_hash nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir_ipv4 nft_redir nft_quota nft_numgen nft_nat nft_meta nft_masq_ipv4 nft_masq nft_log nft_limit nft_flow_offload nft_exthdr nft_ct nft_counter nft_chain_route_ipv6 nft_chain_route_ipv4 nft_chain_nat_ipv4 nf_tables_ipv6 nf_tables_ipv4 nf_tables_inet nf_tables_arp nf_tables nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_redirect nf_nat_proto_gre nf_nat_masquerade_ipv4 nf_nat_irc nf_conntrack_ipv4 nf_nat_ipv4 nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_nat nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_rtcache nf_conntrack_proto_gre nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast
<4>[55228.369413]  ts_kmp nf_conntrack_amanda iptable_raw iptable_mangle iptable_filter ipt_ah ipt_ECN ip_tables crc_ccitt compat fuse sch_cake nf_conntrack sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred ledtrig_usbport xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6t_rt ip6t_mh ip6t_ipv6header ip6t_hbh ip6t_frag ip6t_eui64 ip6t_ah nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 nfsv4 nfsv3 nfs msdos
<4>[55228.439744]  ip_gre gre ifb l2tp_netlink l2tp_core udp_tunnel ip6_udp_tunnel ip_tunnel tun vfat fat ntfs lockd sunrpc grace dns_resolver nls_utf8 nls_iso8859_1 nls_cp437 sha1_generic ecb uas usb_storage leds_gpio xhci_plat_hcd xhci_pci xhci_mtk xhci_hcd uhci_hcd ohci_platform ohci_hcd sd_mod scsi_mod gpio_button_hotplug ext4 mbcache jbd2 usbcore nls_base usb_common crc32c_generic
<4>[55228.473364] Process swapper/2 (pid: 0, threadinfo=8fc5e000, task=8fc3c620, tls=00000000)
<4>[55228.481427] Stack : 8d508000 8dcfed80 8e7b4100 0000012b 8d508000 8fc0de34 00000001 00000000
<4>[55228.489766]         8fc0de34 8ef0ad80 0000012b 8dfed864 8d508000 8fc0de34 8ef0ad80 8e9f60d8
<4>[55228.498102]         8dfed864 8e99b70c ffffffff 0000012c 8fc0df18 80007588 81232d80 8005b6a0
<4>[55228.506439]         8d2c9cd8 8e9f4bc0 8dfede3c 8d2c9cc0 00000200 8e982c98 8e9f5d68 00000001
<4>[55228.514775]         00000001 00000003 00000000 00000000 8e9f60d8 00000084 00000000 8dfed800
<4>[55228.523116]         ...
<4>[55228.525570] Call Trace:
<4>[55228.528139] [<8e9dc578>] minstrel_calc_rate_stats+0xad4/0x3164 [mac80211]
<4>[55228.534952] [<80007588>] except_vec_vi_end+0xb8/0xc4
<4>[55228.539896] Code: 14e0000b  24040003  8ca50000 <8ca5007c> 14a4000d  2404001a  70442802  00ac1021  90420002
<4>[55228.549622]
<4>[55228.551305] ---[ end trace be84731cba6c5c14 ]---
```

Soberia, 2020-01-05:
Same here

<1>[  135.481602] CPU 2 Unable to handle kernel paging request at virtual address 63610319, epc == 8eaa8be0, ra == 8eaa8                                                                        be0
<4>[  135.492201] Oops[#1]:
<4>[  135.494468] CPU: 2 PID: 0 Comm: swapper/2 Not tainted 4.14.156 #0
<4>[  135.500535] task: 8fc3c4c0 task.stack: 8fc5e000
<4>[  135.505041] $ 0   : 00000000 00000001 0000001a 00000188
<4>[  135.510255] $ 4   : 00000188 00000088 00000018 00000018
<4>[  135.515465] $ 8   : ffffffc4 00000003 0000003f 00000000
<4>[  135.520675] $12   : 00000000 3fff0000 ffffffff 40000000
<4>[  135.525884] $16   : 8fc0ddc4 8f2d7f00 8b17a000 00000001
<4>[  135.531095] $20   : 8b050022 63610300 0000009b 70770d00
<4>[  135.536309] $24   : 0dd98800 000003e0
<4>[  135.541523] $28   : 8fc5e000 8fc0dd28 00000188 8eaa8be0
<4>[  135.546736] Hi    : 0000003f
<4>[  135.549600] Lo    : 00000000
<4>[  135.552469] epc   : 8eaa8be0 0x8eaa8be0 [mac80211@8ea80000+0x6c980]
<4>[  135.558708] ra    : 8eaa8be0 0x8eaa8be0 [mac80211@8ea80000+0x6c980]
<4>[  135.564941] Status: 11007c03      KERNEL EXL IE
<4>[  135.569120] Cause : 40800008 (ExcCode 02)
<4>[  135.573109] BadVA : 63610319
<4>[  135.575972] PrId  : 0001992f (MIPS 1004Kc)
<4>[  135.580043] Modules linked in: pppoe ppp_async pptp pppox ppp_mppe ppp_generic nf_nat_pptp nf_conntrack_pptp nf_co                                                                        nntrack_ipv6 mt76x2e mt76x2_common mt76x02_lib mt7603e mt76 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 xt_t                                                                        ime xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_h                                                                        elper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt                                                                        _FLOWOFFLOAD xt_DSCP xt_CT xt_CLASSIFY ts_fsm ts_bm slhc nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_                                                                        redirect nf_nat_proto_gre nf_nat_masquerade_ipv4 nf_nat_irc nf_conntrack_ipv4 nf_nat_ipv4 nf_nat_h323 nf_nat_amanda nf_n                                                                        at nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_tftp
<4>[  135.650455]  nf_conntrack_snmp nf_conntrack_sip nf_conntrack_rtcache nf_conntrack_proto_gre nf_conntrack_netlink n                                                                        f_conntrack_irc nf_conntrack_h323 nf_conntrack_broadcast ts_kmp nf_conntrack_amanda iptable_raw iptable_mangle iptable_f                                                                        ilter ipt_ECN ip_tables crc_ccitt compat fuse sch_cake nf_conntrack sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32                                                                         cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred ledtrig_usbport xt_set ip_set_list_s                                                                        et ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip                                                                        _set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_                                                                        bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter
<4>[  135.721781]  ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 ip_gre gre ifb ip_tunnel tun vfat fat nls_utf8 nls_iso                                                                        8859_1 nls_cp437 sha1_generic ecb usb_storage sd_mod scsi_mod leds_gpio xhci_plat_hcd xhci_pci xhci_mtk xhci_hcd gpio_bu                                                                        tton_hotplug usbcore nls_base usb_common
<4>[  135.746006] Process swapper/2 (pid: 0, threadinfo=8fc5e000, task=8fc3c4c0, tls=00000000)
<4>[  135.754057] Stack : 8f39f260 80550000 00000001 00000000 001ed73b 00000100 804b0000 8d780bc0
<4>[  135.762395]         8fc0ddfc 80540000 8d730401 00000000 00000000 00000000 00000000 0000049a
<4>[  135.765369] YAMon: IN= OUT=pppoe-wan SRC=100.66.26.51 DST=91.134.166.227 LEN=126 TOS=0x00 PREC=0x00 TTL=64 ID=34888 DF PROTO=UDP SPT=59671 DPT=800 LEN=106
<4>[  135.770730]         8f2d7f00 8d780bc0 8b05003a 00000008 8b050022 0000009b 8d781a88 8b17a000
<4>[  135.770765]         00000006 8eaaa860 00000000 0000af33 81232da0 80540000 ffffbfb8 8005d1ec
<4>[  135.801261]         00000000 8d6e90c0 00000000 00000000 00000000 00000003 ffffffc4 8d781a88
<4>[  135.809600]         ...
<4>[  135.812043] Call Trace:
<4>[  135.812179] [<8eaaa860>] 0x8eaaa860 [mac80211@8ea80000+0x6c980]
<4>[  135.820495] [<8005d1ec>] 0x8005d1ec
<4>[  135.824045] [<8f2e2efc>] 0x8f2e2efc [mt76@8f2e0000+0x6580]
<4>[  135.829545] [<8f2e31c8>] 0x8f2e31c8 [mt76@8f2e0000+0x6580]
<4>[  135.835043] [<8f2e0da8>] 0x8f2e0da8 [mt76@8f2e0000+0x6580]
<4>[  135.840541] [<80363074>] 0x80363074
<4>[  135.844031] [<804a1d78>] 0x804a1d78
<4>[  135.847513] [<80032f74>] 0x80032f74
<4>[  135.850987] [<80259af0>] 0x80259af0
<4>[  135.854459] [<80294b34>] 0x80294b34
<4>[  135.857933] [<80007488>] 0x80007488
<4>[  135.861401]
<4>[  135.862879] Code: 969e0000  0fcf1053  03c02025 <92a30019> afa20024  30620008  14400004  3c048055  26420130
<4>[  135.872605]
<4>[  135.874264] ---[ end trace 3070450821c955e6 ]---

ptpt52, 2020-01-10:
same problem here @nbd168 
```
<6>[160956.862291] mtk_soc_eth 1e100000.ethernet eth0: port 3 link down
<6>[160960.089299] mtk_soc_eth 1e100000.ethernet eth0: port 3 link up
<6>[181053.395284] mtk_soc_eth 1e100000.ethernet eth0: port 3 link down
<6>[181055.931435] mtk_soc_eth 1e100000.ethernet eth0: port 3 link up
<6>[181058.935467] mtk_soc_eth 1e100000.ethernet eth0: port 3 link down
<1>[191259.291186] CPU 3 Unable to handle kernel paging request at virtual address 07406000, epc == 8012f7b0, ra == 80050738
<4>[191259.302060] Oops[#1]:
<4>[191259.304414] CPU: 3 PID: 1863 Comm: timeout Not tainted 4.14.160 #0
<4>[191259.310653] task: 8fe99a80 task.stack: 8cf78000
<4>[191259.315245] $ 0   : 00000000 00000001 00000000 812841d8
<4>[191259.320544] $ 4   : 805d21fc 00000001 00000001 07406000
<4>[191259.325841] $ 8   : 0001b11f 0001b11e 00c4f000 00000000
<4>[191259.331139] $12   : 7f865d48 77f142a0 00000000 00000000
<4>[191259.336438] $16   : 8fc02200 014000c0 80770000 8c7d8000
<4>[191259.341735] $20   : 00000000 8fe99a80 00000000 00000000
<4>[191259.347032] $24   : 00000000 77ec89a4                  
<4>[191259.352330] $28   : 8cf78000 8cf79dd8 00000000 80050738
<4>[191259.357636] Hi    : 000002f4
<4>[191259.360589] Lo    : 0001cae9
<4>[191259.363553] epc   : 8012f7b0 kmem_cache_alloc+0x160/0x1dc
<4>[191259.369025] ra    : 80050738 prepare_creds+0x28/0xb4
<4>[191259.374048] Status: 11007c03	KERNEL EXL IE 
<4>[191259.378310] Cause : 40800008 (ExcCode 02)
<4>[191259.382383] BadVA : 07406000
<4>[191259.385331] PrId  : 0001992f (MIPS 1004Kc)
<4>[191259.389491] Modules linked in: qcserial pppoe ppp_async option l2tp_ppp cdc_mbim usb_wwan sierra_net sierra rndis_host qmi_wwan pptp pppox ppp_mppe ppp_generic nf_nat_pptp nf_conntrack_pptp mt76x2e mt76x2_common mt76x02_lib mt7603e mt76 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE huawei_cdc_ncm cfg80211 cdc_ncm cdc_ether xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_socket xt_recent xt_quota xt_pkttype xt_physdev xt_owner xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_iprange xt_ipp2p xt_iface xt_hl xt_helper xt_hashlimit xt_esp xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_addrtype xt_TPROXY xt_TCPMSS xt_REDIRECT xt_NETMAP xt_LOG xt_IPMARK xt_HL xt_DSCP xt_CT xt_CLASSIFY wireguard usbserial usbnet usblp ts_fsm ts_bm slhc r8152 nft_set_rbtree
<4>[191259.459941]  nft_set_hash nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject_bridge nft_reject nft_redir nft_quota nft_numgen nft_meta_bridge nft_meta nft_log nft_limit nft_fwd_netdev nft_exthdr nft_dup_netdev nft_ct nft_counter nft_chain_route_ipv6 nft_chain_route_ipv4 nf_tables_netdev nf_tables_ipv6 nf_tables_ipv4 nf_tables_inet nf_tables_bridge nf_tables nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_rtsp nf_nat_redirect nf_nat_proto_gre nf_nat_masquerade_ipv4 nf_nat_irc nf_conntrack_ipv4 nf_nat_ipv4 nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_ipv4 nf_dup_netdev nf_defrag_ipv4 nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_rtsp nf_conntrack_rtcache nf_conntrack_proto_gre nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323
<4>[191259.530735]  nf_conntrack_ftp nf_conntrack_broadcast ts_kmp nf_conntrack_amanda macvlan iptable_raw iptable_mangle iptable_filter ipt_ah ipt_ECN ipheth ip_tables crc_itu_t crc_ccitt compat_xtables compat cdc_wdm br_netfilter natflow natcap fuse sch_cake tcp_bbr sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred ledtrig_usbport xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_nat nf_conntrack_ipv6 nf_defrag_ipv6 nf_nat_ipv6 ip6t_NPT ip6t_MASQUERADE nf_nat_masquerade_ipv6
<4>[191259.602852]  nf_nat nf_conntrack ip6t_rt ip6t_mh ip6t_ipv6header ip6t_hbh ip6t_frag ip6t_eui64 ip6t_ah nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 msdos ip6_gre ip_gre gre ifb sit l2tp_netlink l2tp_core udp_tunnel ip6_udp_tunnel ip6_tunnel tunnel6 tunnel4 ip_tunnel tun vfat fat autofs4 nls_utf8 nls_iso8859_1 nls_cp437 sha256_generic sha1_generic seqiv jitterentropy_rng drbg hmac ghash_generic gf128mul gcm ecb ctr cmac arc4 uas usb_storage leds_gpio xhci_plat_hcd xhci_pci xhci_mtk xhci_hcd ohci_platform ohci_hcd softdog ehci_platform sd_mod scsi_mod ehci_hcd gpio_button_hotplug ext4 mbcache jbd2 exfat usbcore nls_base usb_common mii crc32c_generic
<4>[191259.665427] Process timeout (pid: 1863, threadinfo=8cf78000, task=8fe99a80, tls=77f15ec8)
<4>[191259.673651] Stack : 00000002 8011527c 805fd0c0 00000000 805fd0c0 8ff0f0a0 8fe99a80 8ff0f0a0
<4>[191259.682075]         00000000 80050738 8ce5b540 00000054 014200ca 00000063 7f865d40 8ff0f0a0
<4>[191259.690495]         00000012 80050d74 00000000 00000000 00000000 8ff0f0a0 8ff0f0a0 00000012
<4>[191259.698915]         8d0ee000 8002d5ac 77ec89c4 8ce5b540 8fe99a80 8cf79f28 00000054 77ec89c4
<4>[191259.707338]         80740000 77efa890 00000000 77efa890 8cf79f10 77efa880 00000010 8003c598
<4>[191259.715759]         ...
<4>[191259.718282] Call Trace:
<4>[191259.720810] [<8012f7b0>] kmem_cache_alloc+0x160/0x1dc
[... cut ...]
<4>[191259.730622] [<80050d74>] copy_creds+0x8c/0x148
<4>[191259.735138] [<8002d5ac>] copy_process.part.84+0x29c/0x1680
<4>[191259.740687] [<8002eb28>] _do_fork+0xe0/0x370
<4>[191259.745023] [<8002ee18>] sys_fork+0x24/0x30
<4>[191259.749290] [<80016578>] syscall_common+0x34/0x58
<4>[191259.754062] Code: 00000000  8e020014  00e23821 <8ce20000> 10000010  cc400000  10400005  00000000  8e060010 
<4>[191259.763878] 
<4>[191259.766625] ---[ end trace f02f4eabb0e92071 ]---

===================================

```

Gingernut1978, 2020-01-12:
Snapshot build from 11/01/2020 on Xiaomi R3Gv2. 
Very unstable if used as a WDS+AP Station on the MT76x2E radio.

root@Xiaomi:~# cat /sys/kernel/debug/crashlog
Time: 1578782729.862026
Modules:        pppoe@869dc000+22f0     ppp_async@86f4c000+1bc0 pppox@8774d000+58a      ppp_generic@86cd8000+5c82       nf_conntrack_ipv6@86f76000+15c0       mt76x2e@869d0000+28b0   mt76x2_common@86924000+2d88     mt76x02_lib@86cc0000+a2e1       mt7603e@869f0000+972c   mt76@86e10000+8909    mac80211@86c00000+77c89 iptable_nat@86f7e000+3b0        ipt_REJECT@868dd000+410 ipt_MASQUERADE@86800000+2f0     cfg80211@86e40000+3f4f9       xt_time@877ea000+730    xt_tcpudp@86fb6000+750  xt_state@8698c000+310   xt_nat@86989000+650     xt_multiport@86981000+550       xt_mark@86976000+2f0  xt_mac@8696f000+2b0     xt_limit@8696c000+4f0   xt_conntrack@86969000+910       xt_comment@86961000+230 xt_TCPMSS@8695f000+b30xt_REDIRECT@86952000+310        xt_LOG@86950000+350     xt_FLOWOFFLOAD@86949000+c30     xt_CT@86946000+ab0      slhc@86942000+14bb      nf_reject_ipv4@86931000+903   nf_nat_redirect@8692c000+63b    nf_nat_masquerade_ipv4@86929000+71c     nf_conntrack_ipv4@86936000+1450 nf_nat_ipv4@86922000+1031     nf_nat@8691c000+3a7c    nf_log_ipv4@86f42000+df0        nf_flow_table_hw@86f56000+920   nf_flow_table@86910000+367f     nf_defrag_ipv6@8690a000+147e  nf_defrag_ipv4@86f4f000+4b6     nf_conntrack_rtcache@86f52000+b10       nf_conntrack@86f60000+11bca     iptable_mangle@86f45000+470   iptable_filter@86f38000+3d0     ip_tables@86f48000+2bdd crc_ccitt@877de000+42b  compat@86f3c000+1de4    nf_log_ipv6@86f30000+10d0     nf_log_common@86f19000+baf      ip6table_mangle@86814000+570    ip6table_filter@86810000+3d0    ip6_tables@877d8000+2ae1        ip6t_REJECT@86811000+450      x_tables@877f4000+38af  nf_reject_ipv6@87770000+ac8     leds_gpio@8773f000+bb0  gpio_button_hotplug@87750000+1ad0
<6>[    1.940018] mt7621_gpio 1e000600.gpio: registering 32 gpios
<6>[    1.945802] mt7621_gpio 1e000600.gpio: registering 32 gpios
<6>[    1.951505] mt7621_gpio 1e000600.gpio: registering 32 gpios
<6>[    1.958536] PCI host bridge to bus 0000:00
<6>[    1.962567] pci_bus 0000:00: root bus resource [mem 0x60000000-0x6fffffff]
<6>[    1.969427] pci_bus 0000:00: root bus resource [io  0xffffffff]
<6>[    1.975253] pci_bus 0000:00: root bus resource [??? 0x00000000 flags 0x0]
<6>[    1.982013] pci_bus 0000:00: No busn resource found for root bus, will use [bus 00-ff]
<7>[    1.989901] pci 0000:00:00.0: [0e8d:0801] type 01 class 0x060400
<7>[    1.989940] pci 0000:00:00.0: reg 0x10: [mem 0x00000000-0x7fffffff]
<7>[    1.989952] pci 0000:00:00.0: reg 0x14: [mem 0x60300000-0x6030ffff]
<7>[    1.990015] pci 0000:00:00.0: supports D1
<7>[    1.990024] pci 0000:00:00.0: PME# supported from D0 D1 D3hot
<7>[    1.990281] pci 0000:00:01.0: [0e8d:0801] type 01 class 0x060400
<7>[    1.990309] pci 0000:00:01.0: reg 0x10: [mem 0x00000000-0x7fffffff]
<7>[    1.990323] pci 0000:00:01.0: reg 0x14: [mem 0x60310000-0x6031ffff]
<7>[    1.990376] pci 0000:00:01.0: supports D1
<7>[    1.990385] pci 0000:00:01.0: PME# supported from D0 D1 D3hot
<7>[    1.990804] pci 0000:01:00.0: [14c3:7662] type 00 class 0x028000
<7>[    1.990858] pci 0000:01:00.0: reg 0x10: [mem 0x00000000-0x000fffff 64bit]
<7>[    1.990902] pci 0000:01:00.0: reg 0x30: [mem 0x00000000-0x0000ffff pref]
<7>[    1.990994] pci 0000:01:00.0: PME# supported from D0 D3hot D3cold
<7>[    1.991175] pci_bus 0000:01: busn_res: [bus 01-ff] end is updated to 01
<7>[    1.991364] pci 0000:02:00.0: [14c3:7603] type 00 class 0x028000
<7>[    1.991411] pci 0000:02:00.0: reg 0x10: [mem 0x00000000-0x000fffff]
<7>[    1.991537] pci 0000:02:00.0: PME# supported from D0 D3hot D3cold
<7>[    1.991715] pci_bus 0000:02: busn_res: [bus 02-ff] end is updated to 02
<7>[    1.991731] pci_bus 0000:00: busn_res: [bus 00-ff] end is updated to 02
<6>[    1.991790] pci 0000:00:00.0: BAR 0: no space for [mem size 0x80000000]
<6>[    1.998344] pci 0000:00:00.0: BAR 0: failed to assign [mem size 0x80000000]
<6>[    2.005211] pci 0000:00:01.0: BAR 0: no space for [mem size 0x80000000]
<6>[    2.011804] pci 0000:00:01.0: BAR 0: failed to assign [mem size 0x80000000]
<6>[    2.018698] pci 0000:00:00.0: BAR 8: assigned [mem 0x60000000-0x600fffff]
<6>[    2.025450] pci 0000:00:00.0: BAR 9: assigned [mem 0x60100000-0x601fffff pref]
<6>[    2.032599] pci 0000:00:01.0: BAR 8: assigned [mem 0x60200000-0x602fffff]
<6>[    2.039359] pci 0000:00:00.0: BAR 1: assigned [mem 0x60300000-0x6030ffff]
<6>[    2.046089] pci 0000:00:01.0: BAR 1: assigned [mem 0x60310000-0x6031ffff]
<6>[    2.052853] pci 0000:01:00.0: BAR 0: assigned [mem 0x60000000-0x600fffff 64bit]
<6>[    2.060088] pci 0000:01:00.0: BAR 6: assigned [mem 0x60100000-0x6010ffff pref]
<6>[    2.067268] pci 0000:00:00.0: PCI bridge to [bus 01]
<6>[    2.072172] pci 0000:00:00.0:   bridge window [mem 0x60000000-0x600fffff]
<6>[    2.078930] pci 0000:00:00.0:   bridge window [mem 0x60100000-0x601fffff pref]
<6>[    2.086094] pci 0000:02:00.0: BAR 0: assigned [mem 0x60200000-0x602fffff]
<6>[    2.092852] pci 0000:00:01.0: PCI bridge to [bus 02]
<6>[    2.097747] pci 0000:00:01.0:   bridge window [mem 0x60200000-0x602fffff]
<6>[    2.105864] clocksource: Switched to clocksource GIC
<6>[    2.112448] NET: Registered protocol family 2
<6>[    2.117575] TCP established hash table entries: 1024 (order: 0, 4096 bytes)
<6>[    2.124448] TCP bind hash table entries: 1024 (order: 1, 8192 bytes)
[... cut ...]
<4>[ 3302.247649]         87c0fed0 86a72200 00000001 806061fc 806069d4 00000000 80600000 00000000
<4>[ 3302.255985]         850b00a4 850b00b0 869ede20 850b00b8 00000000 869ede20 869ede40 8778c584
<4>[ 3302.264318]         ...
<4>[ 3302.266759] Call Trace:
<4>[ 3302.269272] [<86c30970>] ieee80211_tx_dequeue+0x188/0x107c [mac80211]
<4>[ 3302.275735] [<86e14c24>] mt76_tx+0x248/0x28c [mt76]
<4>[ 3302.280595] Code: afb60024  8fa20024  8c530000 <8e650024> 1ca0000a  00000000  8ee200cc  8e640000  8e630004
<4>[ 3302.290319]
<4>[ 3302.291965] ---[ end trace c626dbb08b363d1a ]---

nbd168, 2020-02-20:
Fixed in 8f8e9161b3550c9b53a70dec6ceeac0d194a221d

closed 2020-02-20 by nbd168