openwrt/mt76 #315: MT7615: Add support for DBDC mode

VitaliySh, 2019-09-30:
The driver currently does not support DBDC mode. So this is a feature request to add support for this functionality. Can anyone explain how to do this? Do you have any plan?

Assader, 2019-10-02:
So what changes has to be made to get at least primal DBDC support?
As far as i see:
- mt7615_mcu_init_mac has to be called two times, for each band;
- mt7615_mcu_ctrl_pm_state -- PM state has to be turned off for each band;
- mt7615_mcu_set_channel -- channel_band has to be set in respect to a band;
- something like mt7615_mcu_dbdc_ctrl has to be called to send MCU_EXT_CMD_DBDC_CTRL (0x45) to MCU with proper filling.
Does someone have any other clues?

VitaliySh, 2019-10-03:
Should we have an additional "virtual" PCI device to let mac80211 do its work in the case of DBDC mode in 7615?

Assader, 2019-10-03:
The source code contains traces, related to DBDC.
Our team is currently working on it (well, trying to work), but we'd like to know, what was the plan?
ieee80211_hw seems to be constructed to work with one band only at a time.
Now i can imagine two ways of dealing with it.
The first one is, somehow, duplicate one physical PCI device to get a "virtual" one.
The second one is dirty: hardcode some actions (mt7615_mcu_set_channel, mt7615_mcu_dbdc_ctrl, mt7615_mcu_set_tx_power, mt7615_mcu_set_bcn, mt7615_mcu_set_dev_info) for the second call of mt7615_add_interface.
Any help would be appreciated.

Assader, 2019-10-28:
Hello.
We had some progress at the field of DBDC. Now we have:
- still working 2G at the band0;
- beacon frames are being sent simultaneous via band0 and band1 (2G and 5G);
- frames from both bands are being received and classified (pushed into wlan0 and wlan1);
But. All the frames are sent into band1 happens to fall into 2G.
Headers generated at the mac.c:mt7615_mac_write_txwi are seems to be correct (judjing by omac_idx, wlan_idx).
I think that the problem comes out of DBDC configuration, which is set into mt7615_mcu_dbdc_ctrl (MtCmdSetDbdcCtrl analogue). Does anyone sees it?
`    enum {
        DBDC_TYPE_WMM = 0,
        DBDC_TYPE_MGMT,
        DBDC_TYPE_BSS,
        DBDC_TYPE_MBSS,
        DBDC_TYPE_REPEATER,
        DBDC_TYPE_MU,
        DBDC_TYPE_BF,
        DBDC_TYPE_PTA,
    };
    struct {
        u8 dbdc_enable;
        u8 total_num;
        u8 rsv[2];
        struct {
            u8 type;
            u8 index;
            u8 band_index;
            u8 rsv;
        } entry[64];
    } req = {
        .dbdc_enable = 1,
    };
    int i = 0, j = 0;

    /* OMAC */
    req.entry[i].type = DBDC_TYPE_BSS;
    req.entry[i].index = 0;
    req.entry[i++].band_index = 0;
    req.entry[i].type = DBDC_TYPE_BSS;
    req.entry[i].index = 1;
    req.entry[i++].band_index = 0;
    req.entry[i].type = DBDC_TYPE_MBSS;
    req.entry[i].index = 0;
    req.entry[i++].band_index = 1;
    req.entry[i].type = DBDC_TYPE_BSS;
    req.entry[i].index = 2;
    req.entry[i++].band_index = 1;
    /* BFMU */
    /* MU is 5G only */
    req.entry[i].type = DBDC_TYPE_MU;
    req.entry[i].index = 0;
    req.entry[i++].band_index = 1;
    /* Couple BFs. */
    for (j = 0; j < 3; ++j) {
        req.entry[i].type = DBDC_TYPE_BF;
        req.entry[i].index = j;
        req.entry[i++].band_index = 1;
    }
    /* WMM */
    req.entry[i].type = DBDC_TYPE_WMM;
    req.entry[i].index = 0;
    req.entry[i++].band_index = 0;
    req.entry[i].type = DBDC_TYPE_WMM;
    req.entry[i].index = 1;
    req.entry[i++].band_index = 1;
    /* MGMT */
    req.entry[i].type = DBDC_TYPE_MGMT;
    req.entry[i].index = 0;
    req.entry[i++].band_index = 0;
    req.entry[i].type = DBDC_TYPE_MGMT;
    req.entry[i].index = 1;
    req.entry[i++].band_index = 1;

    req.total_num = i;`
Or maybe cr4 soft-mac blob is not capable of DBDC-ing?

LGA1150, 2019-11-09:
@Assader Have you tried the new firmware at #282 ?

Assader, 2019-11-09:
@LGA1150 it turned out that current firmware is ok. mt7615 branch of this repo contains DBDC-capable version of the driver. Some work there still has to be done, but soon the branch will be merged to master.

porentak, 2019-12-19:
@Assader did you make any progress on this topic?

Assader, 2019-12-19:
@porentak master branch of this repo now contains a code, required for DBDC support. You could turn it on via debugfs (/sys/kernel/debug/ieee80211/phy0/mt76/dbdc) and then tune with /etc/config/wireless. Some issues still has to be solved, that's why you manage the mode via debugfs, but overall it's quite usable.
This feature request, probably, could be closed.

porentak, 2020-01-17:
@Assader can you share with me some more details on "Some issues still has to be solved". What issues are known to you?

I have encountered one. There are no broadcast beacon frames on phy1 interface, unless addition configuration steps are made.

Steps to reproduce the issue:

1. factory reset of the board
2. after boot "wifi down"
3. echo 1 > /sys/kernel/debug/ieee80211/phy0/mt76/dbdc
4. change wifi configuration in /etc/config/wireless to:

config wifi-device 'radio0'
	   option type 'mac80211'
	   option path 'pci0000:00/0000:00:01.0/0000:02:00.0'
	   option hwmode '11g'
	   option country 'DE'
	   option htmode 'HT20'
	   option channel '6'
	   option log_level 1

config wifi-iface 'default_radio0'
	   option device 'radio0'
	   option network 'lan'
	   option mode 'ap'
	   option encryption 'none'
	   option ssid 'TestOpenWrt2G'

config wifi-device 'radio1'
	   option type 'mac80211'
	   option path 'pci0000:00/0000:00:01.0/0000:02:00.0+1'
	   option country 'DE'
	   option channel '40'
	   option hwmode '11a'
	   option htmode 'VHT20'
	   option log_level 1

config wifi-iface 'default_radio1'
	   option device 'radio1'
	   option network 'lan'
	   option mode 'ap'
	   option ssid 'TestOpenWrtMT5G'
	   option encryption 'none'

5. wifi up
In this case phy1 only responds to beacon probe requests, but does not broadcast beacon frames on 100ms interval. And client can not connect to AP.

To make phy1 interface to work we need to folow this steps:
1. factory reset of the board
2. after boot execute "wifi down"
3. echo 1 > /sys/kernel/debug/ieee80211/phy0/mt76/dbdc
4. change wifi configuration in /etc/config/wireless and disable both interfaces:

config wifi-device 'radio0'
+	   option disabled 1

config wifi-device 'radio1'
+	   option disabled 1

5. wifi up
6. wifi down
7. remove "option disabled 1" from configuration
8. wifi up
In this case phy1 works as it should.

Tests were made on master branch, commit 43db699b1ad87ad446808341b591b62b3d41948d (Jan 13, 2020)

If this issue is not on your list, can you share some ideas what to check.

Should I open new issue for this?

Thank you, Klemen

Assader, 2020-01-18:
@porentak the issue i was talking about is MT7615 death under high load in DBDC mode, caused by aggregated frames (AMSDU).
Unfortunately, i wasn't able to reproduce your case.
I was following your steps, including factory reset (and have even used your config), except setting wifi down (step 2), because there is a [guard](https://github.com/openwrt/mt76/blob/master/mt7615/init.c#L390), stopping DBDC mode from being turned on, when wifi is down, and got both bands sending beacons:
```
a@robot:~$ sudo iw dev wlan0 scan | grep OpenWrt
	SSID: TestOpenWrtMT5G
a@robot:~$ sudo iw dev wlan1 scan | grep OpenWrt
	SSID: TestOpenWrt2G
```
If you still suffer from this bug, maybe you should create an issue.

porentak, 2020-01-19:
@Assader thanks for feedback. Can you please let me know what OpenWRT version and host CPU you are using.

Can you also send hostapd debug output (after enabling dbdc and wifi up call)? As it looks to me, I see the issue if phy1 hostapd is ready before phy0 hostapd.

Assader, 2020-01-20:
OpenWrt = 19.07.
CPU = MT7621 SoC.
Logs:
```
root@OpenWrt:~# logread | grep hostapd
Mon Jan 20 05:38:10 2020 daemon.err hostapd: Configuration file: /var/run/hostapd-phy0.conf
Mon Jan 20 05:38:10 2020 daemon.notice hostapd: wlan0: interface state UNINITIALIZED->COUNTRY_UPDATE
Mon Jan 20 05:38:10 2020 daemon.err hostapd: Using interface wlan0 with hwaddr f8:f0:82:9b:c4:b0 and ssid "TestOpenWrt2G"
Mon Jan 20 05:38:11 2020 daemon.notice hostapd: wlan0: interface state COUNTRY_UPDATE->ENABLED
Mon Jan 20 05:38:11 2020 daemon.notice hostapd: wlan0: AP-ENABLED
Mon Jan 20 05:38:11 2020 daemon.err hostapd: Configuration file: /var/run/hostapd-phy1.conf
Mon Jan 20 05:38:11 2020 daemon.notice hostapd: wlan1: interface state UNINITIALIZED->COUNTRY_UPDATE
Mon Jan 20 05:38:11 2020 daemon.err hostapd: Using interface wlan1 with hwaddr f8:f0:82:9b:c4:b0 and ssid "TestOpenWrtMT5G"
Mon Jan 20 05:38:12 2020 daemon.notice hostapd: wlan1: interface state COUNTRY_UPDATE->ENABLED
Mon Jan 20 05:38:12 2020 daemon.notice hostapd: wlan1: AP-ENABLED
Mon Jan 20 05:38:21 2020 daemon.notice hostapd: wlan0: interface state ENABLED->DISABLED
Mon Jan 20 05:38:21 2020 daemon.notice hostapd: wlan0: AP-DISABLED
Mon Jan 20 05:38:21 2020 daemon.notice hostapd: wlan0: CTRL-EVENT-TERMINATING
Mon Jan 20 05:38:21 2020 daemon.notice hostapd: nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Mon Jan 20 05:38:21 2020 daemon.err hostapd: Could not read interface wlan0 flags: No such device
Mon Jan 20 05:38:21 2020 daemon.err hostapd: Configuration file: /var/run/hostapd-phy0.conf
Mon Jan 20 05:38:21 2020 daemon.notice hostapd: wlan0: interface state UNINITIALIZED->COUNTRY_UPDATE
Mon Jan 20 05:38:21 2020 daemon.err hostapd: Using interface wlan0 with hwaddr f8:f0:82:9b:c4:b0 and ssid "TestOpenWrt2G"
Mon Jan 20 05:38:21 2020 daemon.notice hostapd: wlan0: interface state COUNTRY_UPDATE->ENABLED
Mon Jan 20 05:38:21 2020 daemon.notice hostapd: wlan0: AP-ENABLED
Mon Jan 20 05:38:22 2020 daemon.notice hostapd: wlan1: interface state ENABLED->DISABLED
Mon Jan 20 05:38:22 2020 daemon.notice hostapd: wlan1: AP-DISABLED
Mon Jan 20 05:38:22 2020 daemon.notice hostapd: wlan1: CTRL-EVENT-TERMINATING
Mon Jan 20 05:38:22 2020 daemon.notice hostapd: nl80211: deinit ifname=wlan1 disabled_11b_rates=0
Mon Jan 20 05:38:23 2020 daemon.err hostapd: Configuration file: /var/run/hostapd-phy1.conf
Mon Jan 20 05:38:23 2020 daemon.notice hostapd: wlan1: interface state UNINITIALIZED->COUNTRY_UPDATE
Mon Jan 20 05:38:23 2020 daemon.err hostapd: Using interface wlan1 with hwaddr f8:f0:82:9b:c4:b0 and ssid "TestOpenWrtMT5G"
Mon Jan 20 05:38:23 2020 daemon.notice hostapd: wlan1: interface state COUNTRY_UPDATE->ENABLED
Mon Jan 20 05:38:23 2020 daemon.notice hostapd: wlan1: AP-ENABLED
```

closed 2020-02-06 by VitaliySh