openwrt/mt76 #317: Does the driver support hardware NAT offload？

MeIsReallyBa, 2019-10-03:
If not，is there any way to achieve it？

lumixen, 2019-10-05:
@MeIsReallyBa Hey, what do you mean by "driver support for NAT offload"? `mt76` driver handles only wireless module -> kernel network stack communication, while NAT is "kernel network stack" part. Or I'm missing something? 
Also, there already present support for hardware NAT offload (currently only mt7621 chipset) in openwrt since this commit: https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=424a9ae128bd2045cd4bfd6e3229f2529d150a25

MeIsReallyBa, 2019-10-05:
@lumixen It shows that it only support ethernet->ethernet connection, So I want to know whether it works in wifi. In MTK Stock firmware,hardware nat need some special setting to work in wifi or it would only work in wire connection.

lumixen, 2019-10-06:
@MeIsReallyBa indeed commit message explicitly mentions `ethernet->ethernet connections`. Did you make any tests with several wifi clients -> wan to see if
```
option flow_offloading '1'/'0'
option flow_offloading_hw '1'/'0'
```
makes any difference?

closed 2019-10-09 by MeIsReallyBa