openwrt/mt76 #318: Preemption bug with MT7612U (AVM AC860 USB)

thillux, 2019-10-11:
I used commit d3a589586d1b1 on top of Linux 5.4-rc2 with a AVM AC860 USB adapter.

Running iw dev wlan0 survey dump yields the following output in dmesg:

[  291.790866] BUG: scheduling while atomic: iw/2161/0x00000202
[  291.790873] Modules linked in: ccm fuse snd_hda_codec_hdmi snd_hda_codec_realtek snd_hda_codec_generic nft_chain_nat nf_nat intel_rapl_msr intel_rapl_common nft_reject_inet nf_reject_ipv4 nf_reject_ipv6 nft_reject x86_pkg_temp_thermal intel_powerclamp joydev coretemp mousedev kvm_intel nft_ct nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c kvm nf_tables_set irqbypass nf_tables nfnetlink crct10dif_pclmul crc32_pclmul mt76x2u mt76x02_usb mt76_usb ghash_clmulni_intel mt76x2_common iwldvm mt76x02_lib mt76 i915 iTCO_wdt mei_wdt iwlwifi iTCO_vendor_support mac80211 wmi_bmof uvcvideo aesni_intel i2c_algo_bit videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 btusb videobuf2_common crypto_simd snd_hda_intel btrtl btbcm drm_kms_helper snd_intel_nhlt cryptd btintel bluetooth glue_helper intel_cstate snd_hda_codec videodev drm intel_uncore cfg80211 snd_hda_core intel_rapl_perf psmouse pcspkr input_leds ecdh_generic intel_gtt snd_hwdep i2c_i801 ecc mc snd_pcm agpgart thinkpad_acpi syscopyarea
[  291.790941]  mei_me libarc4 nvram sysfillrect ledtrig_audio snd_timer sysimgblt e1000e rfkill lpc_ich mei fb_sys_fops wmi snd tpm_tis tpm_tis_core ac battery soundcore tpm rng_core evdev mac_hid usbip_host usbip_core pkcs8_key_parser sg crypto_user ip_tables x_tables ext4 crc32c_generic crc16 mbcache jbd2 sr_mod cdrom sd_mod serio_raw atkbd libps2 ahci libahci sdhci_pci libata crc32c_intel cqhci sdhci xhci_pci ehci_pci mmc_core scsi_mod ehci_hcd xhci_hcd i8042 serio
[  291.791002] Preemption disabled at:
[  291.791007] [<0000000000000000>] 0x0
[  291.791015] CPU: 0 PID: 2161 Comm: iw Tainted: G        W         5.4.0-rc2-3-ARCH-00104-g9e208aa06c21 #1
[  291.791017] Hardware name: LENOVO 2349QM6/2349QM6, BIOS G1ETC2WW (2.82 ) 08/07/2019
[  291.791019] Call Trace:
[  291.791042]  dump_stack+0x5c/0x80
[  291.791049]  __schedule_bug.cold+0x8e/0x9b
[  291.791055]  __schedule+0x5f8/0x770
[  291.791062]  schedule+0x43/0xd0
[  291.791068]  schedule_preempt_disabled+0x14/0x20
[  291.791074]  __mutex_lock.isra.0+0x18a/0x530
[  291.791099]  mt76u_rr+0x1f/0x40 [mt76_usb]
[  291.791113]  mt76x02_update_channel+0x22/0x40 [mt76x02_lib]
[  291.791122]  mt76_update_survey+0x42/0xe0 [mt76]
[  291.791129]  mt76_get_survey+0x2f/0x1b0 [mt76]
[  291.791170]  ieee80211_dump_survey+0x5e/0x140 [mac80211]
[  291.791217]  nl80211_dump_survey+0x13c/0x2f0 [cfg80211]
[  291.791222]  ? __kmalloc_reserve.isra.0+0x2d/0x70
[  291.791225]  ? __alloc_skb+0x96/0x1d0
[  291.791229]  netlink_dump+0x17b/0x370
[  291.791247]  __netlink_dump_start+0x16f/0x1e0
[  291.791253]  genl_family_rcv_msg+0x396/0x410
[  291.791290]  ? nl80211_prepare_wdev_dump+0x1b0/0x1b0 [cfg80211]
[  291.791297]  ? _raw_spin_unlock_irqrestore+0x20/0x40
[  291.791312]  ? __wake_up_common_lock+0x8a/0xc0
[  291.791316]  genl_rcv_msg+0x47/0x90
[  291.791320]  ? genl_family_rcv_msg+0x410/0x410
[  291.791323]  netlink_rcv_skb+0x49/0x110
[  291.791329]  genl_rcv+0x24/0x40
[  291.791333]  netlink_unicast+0x171/0x200
[  291.791340]  netlink_sendmsg+0x208/0x3d0
[  291.791358]  sock_sendmsg+0x5e/0x60
[  291.791361]  ___sys_sendmsg+0x2ae/0x330
[  291.791368]  ? filemap_map_pages+0x272/0x390
[  291.791374]  ? _raw_spin_unlock+0x16/0x30
[  291.791379]  ? __handle_mm_fault+0x112f/0x1390
[  291.791388]  __sys_sendmsg+0x59/0xa0
[  291.791396]  do_syscall_64+0x5b/0x1a0
[  291.791400]  entry_SYSCALL_64_after_hwframe+0x44/0xa9
[  291.791404] RIP: 0033:0x7f5d0c7f37b7
[  291.791418] Code: 64 89 02 48 c7 c0 ff ff ff ff eb bb 0f 1f 80 00 00 00 00 f3 0f 1e fa 64 8b 04 25 18 00 00 00 85 c0 75 10 b8 2e 00 00 00 0f 05 <48> 3d 00 f0 ff ff 77 51 c3 48 83 ec 28 89 54 24 1c 48 89 74 24 10
[  291.791421] RSP: 002b:00007ffe8b5d0538 EFLAGS: 00000246 ORIG_RAX: 000000000000002e
[  291.791426] RAX: ffffffffffffffda RBX: 000055a038e6c390 RCX: 00007f5d0c7f37b7
[  291.791430] RDX: 0000000000000000 RSI: 00007ffe8b5d0570 RDI: 0000000000000003
[  291.791434] RBP: 000055a038e718c0 R08: 000055a038e6c02a R09: 0000000000000002
[  291.791438] R10: 000055a03808cb00 R11: 0000000000000246 R12: 000055a038e71780
[  291.791440] R13: 00007ffe8b5d0570 R14: 000055a038e717d0 R15: 000055a038e718c0
[  291.791480] NOHZ: local_softirq_pending 202

thillux, 2019-10-11:
To me, this seems to be a rather fundamental issue with the current USB part in mt76, as long as the usb ctrl mutex is necessary because of slow and atomic register access.

thillux, 2019-10-11:
This dmesg output is also triggered through the periodical mac_work.

[  919.681641] BUG: scheduling while atomic: kworker/u16:19/2668/0x00000202
[  919.681645] Modules linked in: usb_storage ccm fuse snd_hda_codec_hdmi snd_hda_codec_realtek snd_hda_codec_generic nft_chain_nat nf_nat intel_rapl_msr intel_rapl_common nft_reject_inet nf_reject_ipv4 nf_reject_ipv6 nft_reject x86_pkg_temp_thermal intel_powerclamp joydev coretemp mousedev kvm_intel nft_ct nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c kvm nf_tables_set irqbypass nf_tables nfnetlink crct10dif_pclmul crc32_pclmul mt76x2u mt76x02_usb mt76_usb ghash_clmulni_intel mt76x2_common iwldvm mt76x02_lib mt76 i915 iTCO_wdt mei_wdt iwlwifi iTCO_vendor_support mac80211 wmi_bmof uvcvideo aesni_intel i2c_algo_bit videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 btusb videobuf2_common crypto_simd snd_hda_intel btrtl btbcm drm_kms_helper snd_intel_nhlt cryptd btintel bluetooth glue_helper intel_cstate snd_hda_codec videodev drm intel_uncore cfg80211 snd_hda_core intel_rapl_perf psmouse pcspkr input_leds ecdh_generic intel_gtt snd_hwdep i2c_i801 ecc mc snd_pcm agpgart thinkpad_acpi
[  919.681680]  syscopyarea mei_me libarc4 nvram sysfillrect ledtrig_audio snd_timer sysimgblt e1000e rfkill lpc_ich mei fb_sys_fops wmi snd tpm_tis tpm_tis_core ac battery soundcore tpm rng_core evdev mac_hid usbip_host usbip_core pkcs8_key_parser sg crypto_user ip_tables x_tables ext4 crc32c_generic crc16 mbcache jbd2 sr_mod cdrom sd_mod serio_raw atkbd libps2 ahci libahci sdhci_pci libata crc32c_intel cqhci sdhci xhci_pci ehci_pci mmc_core scsi_mod ehci_hcd xhci_hcd i8042 serio
[  919.681708] Preemption disabled at:
[  919.681712] [<0000000000000000>] 0x0
[  919.681717] CPU: 1 PID: 2668 Comm: kworker/u16:19 Tainted: G        W         5.4.0-rc2-3-ARCH-00104-g9e208aa06c21 #1
[  919.681718] Hardware name: LENOVO 2349QM6/2349QM6, BIOS G1ETC2WW (2.82 ) 08/07/2019
[  919.681726] Workqueue: phy2 mt76x02_mac_work [mt76x02_lib]
[  919.681728] Call Trace:
[  919.681736]  dump_stack+0x5c/0x80
[  919.681739]  __schedule_bug.cold+0x8e/0x9b
[  919.681743]  __schedule+0x5f8/0x770
[  919.681746]  schedule+0x43/0xd0
[  919.681749]  schedule_timeout+0x163/0x320
[  919.681753]  ? __next_timer_interrupt+0xd0/0xd0
[  919.681756]  wait_for_completion_timeout+0x9f/0x110
[  919.681760]  ? wake_up_q+0x60/0x60
[  919.681764]  usb_start_wait_urb+0x83/0x160
[  919.681768]  usb_control_msg+0xe1/0x140
[  919.681770]  ? __switch_to_asm+0x34/0x70
[  919.681774]  __mt76u_vendor_request+0xbf/0x100 [mt76_usb]
[  919.681778]  __mt76u_rr+0x54/0xd0 [mt76_usb]
[  919.681781]  mt76u_rr+0x2a/0x40 [mt76_usb]
[  919.681785]  mt76x02_update_channel+0x22/0x40 [mt76x02_lib]
[  919.681790]  mt76_update_survey+0x42/0xe0 [mt76]
[  919.681795]  mt76x02_mac_work+0x2e/0x260 [mt76x02_lib]
[  919.681799]  process_one_work+0x1e2/0x3b0
[  919.681801]  worker_thread+0x4a/0x3d0
[  919.681805]  kthread+0xfb/0x130
[  919.681807]  ? process_one_work+0x3b0/0x3b0
[  919.681809]  ? kthread_park+0x90/0x90
[  919.681811]  ret_from_fork+0x35/0x40

LorenzoBianconi, 2019-10-11:
> This dmesg output is also triggered through the periodical mac_work.
>
> [ 919.681641] BUG: scheduling while atomic: kworker/u16:19/2668/0x00000202
> [ 919.681645] Modules linked in: usb_storage ccm fuse [...]

looking at the code mt76_rr() runs usb_control_msg() that must not be run
in irq context. I will work on it.

Regards,
Lorenzo

thillux, 2019-10-11:
If I can assist testing, let me know. Furthermore, while reading the code of the driver, I've seen that mt76x02_mac_cc_reset calls a maybe unnecessary mt76_rr(dev, MT_CH_IDLE). As of now, only the MT_CH_BUSY register value is used later. Active time is obtained by tracking time between channel survey invocations if I understand this correctly.

sgruszka, 2019-10-14:
Simply reading register should be taken out of spin_lock_bh(&dev->cc_lock); section as it was before air-fairnesses changes.

LorenzoBianconi, 2019-10-14:
>
>
> Simply reading register should be taken out of
> spin_lock_bh(&dev->cc_lock); section as it was before air-fairnesses
> changes.
Yes, I will move the spin_lock inside .update_survey() and read cc_busy on
a local variable.

Regards,
Lorenzo

LorenzoBianconi, 2019-10-14:
@thillux could you please the patch you can find [here](https://github.com/LorenzoBianconi/mt76/commit/128f77a4d845de5c3b211b92c9e288958ba26655)?

thillux, 2019-10-14:
@LorenzoBianconi I'll test it this evening. Thanks a lot so far!

thillux, 2019-10-14:
I tested the patch and cannot see the above messages anymore.

LorenzoBianconi, 2019-10-14:
>
> I tested the patch and cannot see the above messages anymore.

cool, thx for testing. I will work on an upstream fix. Can I add your tested by?

Regards,
Lorenzo

thillux, 2019-10-14:
Sure, feel free to add my tested by.

1Jeff1, 2019-10-15:
@LorenzoBianconi Trying to test a a6210 on openwrt x86 SBC,I found this bug ,after applied your patch the adapter is working without errors,Thanks for posting this fix

LorenzoBianconi, 2019-10-15:
@thillux @1Jeff1 : I updated the fix, could you please try this [patch](https://github.com/LorenzoBianconi/mt76/commit/69c0ced23111979ca8b57d296413adac532db02d)?

thillux, 2019-10-15:
@LorenzoBianconi I tested your last fix. It also works for me. I put the NIC in station mode, queried the survey data in a endless loop in the shell and ran iperf in order to generate some traffic. No dmesg output like the one above occured.

thillux, 2019-10-15:
I noticed, that channel transmit time is not updated correctly. But that also was the case before your patch.

Bidirectional iperf traffic does not increase the transmit channel time.

iw dev wlan0 survey dump | grep 5180 -A 4
        frequency:                      5180 MHz [in use]
        channel active time:            629121 ms
        channel busy time:              382522 ms
        channel transmit time:          2 ms

LorenzoBianconi, 2019-10-15:
> I noticed, that channel transmit time is not updated correctly. But that
> also was the case before your patch.
>
> Bidirectional iperf traffic does not increase the transmit channel time.
>
> iw dev wlan0 [...]

Uhm, interesting..I double checked on mt76x2e and it works as expected so
maybe it is something usb related. I will look into it.

Regards,
Lorenzo

1Jeff1, 2019-10-15:
Tested your new patch , STA and AP are working,I can't comment how the device perform before since I got it yesterday,iperf test shows a good performance
```
root@OpenWrt:~# iw wlan0 info
Interface wlan0
        ifindex 22
        wdev 0x7
        addr zz:40:uu:57:0d:xx
        ssid NETGEAR-A6210
        type AP
        wiphy 0
        channel 153 (5765 MHz), width: 80 MHz, center1: 5775 MHz
        txpower 18.00 dBm
        multicast TXQ:
                qsz-byt qsz-pkt flows   drops   marks   overlmt hashcol tx-bytes        tx-packets
                0       0       150     0       0       0       0       37234           165
root@OpenWrt:~# iperf3 -c 192.168.1.190
Connecting to host 192.168.1.190, port 5201
[  5] local 192.168.1.1 port 41796 connected to 192.168.1.190 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  48.6 MBytes   407 Mbits/sec    0   1.28 MBytes       
[  5]   1.00-2.00   sec  57.5 MBytes   482 Mbits/sec    0   1.53 MBytes       
[  5]   2.00-3.00   sec  45.0 MBytes   377 Mbits/sec    0   1.53 MBytes       
[  5]   3.00-4.00   sec  52.5 MBytes   440 Mbits/sec    0   1.53 MBytes       
[  5]   4.00-5.00   sec  56.2 MBytes   472 Mbits/sec    0   1.53 MBytes       
[  5]   5.00-6.00   sec  57.5 MBytes   482 Mbits/sec    0   1.53 MBytes       
[  5]   6.00-7.00   sec  56.2 MBytes   472 Mbits/sec    0   1.53 MBytes       
[  5]   7.00-8.00   sec  48.8 MBytes   409 Mbits/sec    0   1.53 MBytes       
[  5]   8.00-9.00   sec  55.0 MBytes   461 Mbits/sec    0   1.53 MBytes       
[  5]   9.00-10.00  sec  57.5 MBytes   482 Mbits/sec    0   1.53 MBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   535 MBytes   449 Mbits/sec    0             sender
[  5]   0.00-10.05  sec   532 MBytes   444 Mbits/sec                  receiver

iperf Done.
root@OpenWrt:~# ubus call system board
{
        "kernel": "4.19.79",
        "hostname": "OpenWrt",
        "system": "Intel(R) Atom(TM) x5-Z8350  CPU @ 1.44GHz",
        "model": "AAEON MF-001",
        "board_name": "aaeon-mf-001",
        "release": {
                "distribution": "Atomic-Pi",
                "version": "SNAPSHOT",
                "revision": "r11225-089b4f16aa",
                "target": "x86/64",
                "description": "Atomic-Pi SNAPSHOT r11225-089b4f16aa"
        }
}
root@OpenWrt:~# 
```

closed 2019-10-15 by thillux

LorenzoBianconi, 2019-10-18:
> I noticed, that channel transmit time is not updated correctly. But that also was the case before your patch.
>
> Bidirectional iperf traffic does not increase the transmit channel time.
>
> iw dev wlan0 [...]

@thillux the issue should be resolved by this [patch](https://patchwork.kernel.org/patch/11197275/)

thillux, 2019-10-18:
@LorenzoBianconi Thanks lot again! I'll test it next week.

thillux, 2019-10-21:
@LorenzoBianconi It seems to work with this patch.

iw dev wlan0 survey dump | grep 5180 -A 3
        frequency:                      5180 MHz [in use]
        channel active time:            173442 ms
        channel busy time:              30502 ms
        channel transmit time:          2776 ms

1Jeff1, 2019-10-22:
@LorenzoBianconi I had also tried the patch and transmit time now is beeing updated.
        #iw dev wlan0 survey dump | grep 5765 -A 4
        frequency:                      5765 MHz [in use]
        channel active time:            35211043 ms
        channel busy time:              1045721 ms
        channel BSS receive time:       5663 ms
        channel transmit time:          74598 ms