openwrt/mt76 #320: local_softirq_pending 08 with usb interface (MT7612U, AVM AC860)

thillux, 2019-10-15:
I'm currently seeing some dmesg lines:

NOHZ: local_softirq_pending 08

These seem to be caused by the mt76 driver for my USB NIC with MT7612U chip. My current intuition says, that mt76u_rx_tasklet() in usb.c should call local_bh_disable() and local_bh_enable() while processing frames. I'm testing tomorrow if using rcu_read_lock_bh()/rcu_read_unlock_bh() instead of rcu_read_lock()/rcu_read_unlock() already do the trick here.

LorenzoBianconi, 2019-10-16:
@thillux are you testing latest version of the previous patch? (now in master branch)
Could you please provide syslog?

thillux, 2019-10-16:
I'll re-test with the current master branch. If I can still see the message, I'll provide some logs.

thillux, 2019-10-16:
I tested the current master 50b1e9bd25cdab113.

[  611.621045] NOHZ: local_softirq_pending 08
[  617.356352] NOHZ: local_softirq_pending 08
[  617.356702] NOHZ: local_softirq_pending 08
[  617.356755] NOHZ: local_softirq_pending 08

The local_softirq_pending 08 message still appears. In order to perform some stress testing, I use two MT7612U NICs. One only in monitor mode, one in station mode with additional monitor vif. iperf and ping with a high rate runs over the station interface. After some time, these local_softirq_pending messages float around in the dmesg. I attached the complete dmesg output of my current boot as reference.

[softirq_08_no_bh_lock.log](https://github.com/openwrt/mt76/files/3733847/softirq_08_no_bh_lock.log)

The problem here seems to be, that ieee80211_rx can be called without bottom halves disabled (local_bh_disable() called before). local_bh_disable also has to be called before handing tx completion notifications to mac80211.

For the case of ieee80211_rx I tracked the calls throughout the driver. You can find a graph in the next attachment, I marked in which function local_bh_disable gets called and in which not.

[nohz.pdf](https://github.com/openwrt/mt76/files/3733873/nohz.pdf)

thillux, 2019-10-16:
@LorenzoBianconi  I made some small changes locally and ensured, that ieee80211_rx_napi,ieee80211_tx_status and ieee80211_tx_status_ext are always called with bh disabled. After these changes I was not able to reproduce the pending message from above.

thillux, 2019-10-17:
I have uploaded my patch for usb devices to my github profile. Find it at https://github.com/thillux/mt76/commit/7f53c0eb3599c6a0426737083ae9b8db5c7778a3.

closed 2019-11-21 by thillux