openwrt/mt76 #323: mt76: Low tx throughput/High tx retries in STA mode

namidairo, 2019-10-22:
Has anyone else run into issues with tx throughput via mt76 when it is in STA (WDS/4addr on) mode? (Although I ran into a similar throughput issue when trying to run it as a 802.11s mesh point as well)

Running a mt7612 on a Xiaomi Miwifi Mini, on an OpenWRT snapshot (therefore up to commit d3a589586d1b1d9e1422e31f642a500ff0195f20)

```
[   12.166396] mt76x2e 0000:01:00.0: card - bus=0x1, slot = 0x0 irq=4
[   12.173039] mt76x2e 0000:01:00.0: ASIC revision: 76120044
[   12.426850] mt76x2e 0000:01:00.0: ROM patch build: 20141115060606a
[   12.439723] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
[   12.445406] mt76x2e 0000:01:00.0: Build: 1
[   12.449587] mt76x2e 0000:01:00.0: Build Time: 201507311614____
[   12.464409] mt76x2e 0000:01:00.0: Firmware running!
```

From the mt76 station to the AP is as follows:

```
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec   173 KBytes  1.41 Mbits/sec   53   4.24 KBytes
[  5]   1.00-2.00   sec   136 KBytes  1.11 Mbits/sec   75   4.24 KBytes
[  5]   2.00-3.00   sec   414 KBytes  3.39 Mbits/sec   64   4.24 KBytes
[  5]   3.00-4.00   sec   872 KBytes  7.15 Mbits/sec   58   4.24 KBytes
[  5]   4.00-5.00   sec   503 KBytes  4.12 Mbits/sec   48   4.24 KBytes
[  5]   5.00-6.00   sec   749 KBytes  6.14 Mbits/sec   51   4.24 KBytes
[  5]   6.00-7.00   sec  1.05 MBytes  8.84 Mbits/sec   59   2.83 KBytes
[  5]   7.00-8.00   sec   573 KBytes  4.69 Mbits/sec   56   2.83 KBytes
[  5]   8.00-9.00   sec  1024 KBytes  8.39 Mbits/sec   56   5.66 KBytes
[  5]   9.00-10.00  sec   707 KBytes  5.79 Mbits/sec   51   4.24 KBytes
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec  6.08 MBytes  5.10 Mbits/sec  571             sender
[  5]   0.00-10.04  sec  6.03 MBytes  5.04 Mbits/sec                  receiver
```

However in the other direction, it appears to be fine (From the AP to the STA)

```
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.03   sec  29.4 MBytes   238 Mbits/sec    0    426 KBytes
[  5]   1.03-2.01   sec  30.2 MBytes   259 Mbits/sec    0    475 KBytes
[  5]   2.01-3.02   sec  24.7 MBytes   205 Mbits/sec    0    475 KBytes
[  5]   3.02-4.01   sec  27.5 MBytes   233 Mbits/sec    0    475 KBytes
[  5]   4.01-5.02   sec  26.2 MBytes   218 Mbits/sec    0    475 KBytes
[  5]   5.02-6.04   sec  30.2 MBytes   250 Mbits/sec    0    475 KBytes
[  5]   6.04-7.02   sec  28.8 MBytes   245 Mbits/sec    0    475 KBytes
[  5]   7.02-8.03   sec  21.2 MBytes   176 Mbits/sec    0    475 KBytes
[  5]   8.03-9.04   sec  20.0 MBytes   165 Mbits/sec    0    475 KBytes
[  5]   9.04-10.01  sec  29.6 MBytes   257 Mbits/sec    0    475 KBytes
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.01  sec   268 MBytes   224 Mbits/sec    0             sender
[  5]   0.00-10.06  sec   267 MBytes   223 Mbits/sec                  receiver
```

And here's the output of `iw dev wlan0 station dump`:

```
inactive time:  10 ms
        rx bytes:       50401221
        rx packets:     78919
        tx bytes:       31363544
        tx packets:     25110
        tx retries:     1123
        tx failed:      1
        beacon loss:    0
        beacon rx:      10873
        rx drop misc:   4125
        signal:         -54 [-62, -54] dBm
        signal avg:     -51 [-58, -51] dBm
        beacon signal avg:      -49 dBm
        tx bitrate:     780.0 MBit/s VHT-MCS 8 80MHz short GI VHT-NSS 2
        tx duration:    19562061 us
        rx bitrate:     780.0 MBit/s VHT-MCS 8 80MHz short GI VHT-NSS 2
        rx duration:    15335478 us
        airtime weight: 256
        expected throughput:    67.16Mbps
        authorized:     yes
        authenticated:  yes
        associated:     yes
        preamble:       long
        WMM/WME:        yes
        MFP:            no
        TDLS peer:      no
        DTIM period:    2
        beacon interval:100
        short slot time:yes
        connected time: 1114 seconds
```

Gingernut1978, 2019-12-15:
I've run into this problem this weekend.

[https://forum.openwrt.org/t/wds-client-ap-tx-rate-problem/50503](url)

I can add any info needed to debug this.

Gingernut1978, 2019-12-15:
@namidairo 

If you can, test latest snapshot and see if the problem is resolved. 

The few tests I've done are positive.

namidairo, 2019-12-17:
Tested against the snapshot before remembering that it's still on an older commit of mt76 from October, no change.

Pointed the Makefile at the current master and rolled my own build, no change.

namidairo, 2019-12-31:
Looks like this particular issue is related to https://github.com/openwrt/mt76/commit/abcb544579191d30177ff0cd1b2671ade0fb9098, setting a RTS threshold brings the throughput up to expected values.

Legogris, 2020-01-23:
@namidairo What value did you put for RTS threshold? I am having severe packet loss issues on MT7621E.

namidairo, 2020-01-23:
RTS: 40
Fragmentation: 3000

I'd probably experiment with the latter.

Sadly I haven't had time to bisect where it breaks (or if it ever worked)

nbd168, 2020-01-23:
Please try the latest revision of OpenWrt master or 19.07. I've fixed a critical issue in hostapd.

Legogris, 2020-01-24:
iperf3 tests look good so far on `master`! :tada: 
For the record, I don't know if it's the same issue, but I was also consistently having clients would be associated with AP but unable to reach network after some time (5min - 2h or so) until manually reassociated. Disabling rekeying made it slightly less frequent, I think, but it still happened dozens of times per day. Will report back if that occurs again.