openwrt/mt76 #325: After 10/10/2019 builds from openwrt my devices can't connect to the 5ghz wifi from router

leonardcoutinho, 2019-10-31:
After 10/10 builds my devices can't connect to the 5ghz wifi from router. My router is an D-link Dir 810l A1 (mt7620A+7610en). I have few knowledge about source code and linux, but I can help if needed to find a solution.

leonardcoutinho, 2019-10-31:
in newer builds my phone see the 5g network but refuse to connect, my laptop can't conect too.

xavier83, 2019-10-31:
Welcome to my world https://github.com/openwrt/mt76/issues/199 :stuck_out_tongue_closed_eyes:

leonardcoutinho, 2019-10-31:
I saved 10/10/19 snapshot build, but 2.4g is not full stable, in latest snapshot(after 10/10, or 4.14.146 kernel), 2.4g is stable but 5g don't work. I hope to it will be fixed in next days, because I saw dev guys working in code.

leonardcoutinho, 2019-10-31:
> Welcome to my world #199 😝

I think is same problem >_<

xavier83, 2019-10-31:
> I saved 10/10/19 snapshot build, but 2.4g is not full stable, in latest snapshot(after 10/10, or 4.14.146 kernel), 2.4g is stable but 5g don't work. I hope to it will be fixed in next days, because I saw dev guys working in code.

Hope so. it has been a year now :smile:

kamnxt, 2019-11-04:
On my router (a I-O DATA WN-AX1167GR2, community supported), I'm getting a similar issue. Devices start connecting but cannot associate completely:

```
iwlwifi 0000:02:00.0: No beacon heard and the time event is over already...
wlp2s0: Connection to AP [...] lost
wlp2s0: authenticate with [...]
wlp2s0: send auth to [...] (try 1/3)
iwlwifi 0000:02:00.0: Unhandled alg: 0x707
wlp2s0: authenticated
wlp2s0: associate with [...] (try 1/3)
wlp2s0: RX AssocResp from [...] (capab=0x111 status=0 aid=2)
wlp2s0: associated
iwlwifi 0000:02:00.0: No beacon heard and the time event is over already...
wlp2s0: Connection to AP [...] lost
```

Oddly enough LuCi is showing the devices in the connected list for a short while before they disappear.

leonardcoutinho, 2019-11-13:
I will check tomorrow. I installed 19.07RC1 and both of radios works, 5ghz give 100mbit speed, but 2.4ghz won't past 35mbit, even with 40mhz forced option. My laptop card is an Intel AX200, an 2x2 wifi 6 card.

LorenzoBianconi, 2019-11-23:
@leonardcoutinho could you please try this [patch](https://github.com/LorenzoBianconi/wireless-drivers-next/commit/81f0cfa6f34d4f657c2ed75ebed61ecae24d133e)? Thanks

Compunctus, 2019-11-28:
> @leonardcoutinho could you please try this [patch](https://github.com/LorenzoBianconi/wireless-drivers-next/commit/81f0cfa6f34d4f657c2ed75ebed61ecae24d133e)? Thanks

This fixed the same issue (was unable to associate to ap) for my tplink archer c1 v1! Thanks.

leonardcoutinho, 2019-12-02:
I can't test these patchs now because router is in my work now, and is running stable with 19.07 rc1. When these patchs will merge with snapshot builds or stable builds?

leonardcoutinho, 2019-12-02:
thanks for your work Lorenzo

leonardcoutinho, 2019-12-30:
I try snapshot from 29/12/19 and Wifi 5ghz have same problem, mt76 file in these build is from 27/12/19. These builds have Lorenzo patch?

kamnxt, 2020-01-05:
@leonardcoutinho seems like they should have it: https://github.com/openwrt/mt76/commit/38f4c579898d799657b227831a7b4bab156f785d

I'm also still having the problem with the 2019-12-27 version of mt76.

kamnxt, 2020-01-08:
After doing some late-night wiresharking, it seems like there's no issues with authentication and association - the device (in this case my phone) seems to connect fine. The router never sends any data back (but does respond to management frames like block ack requests).

So: it seems like the router receives everything just fine, and is able to send management frames but not data frames?

leonardcoutinho, 2020-01-09:
In 09/01/20 build from snapshot My 5ghz is working(MT7610e). 2.4ghz on otherhand only works with one stream(150mbps max in 40mhz)(mt7620a chipset), and not 2 stream which my router has(dir 810l A). I don't use 2.4ghz for speed, if works is fine to me. I will test more and give my feedback. Thanks everyone.

eklitzke, 2020-01-13:
I found a similar issue on the RedHat bugzilla here: https://bugzilla.redhat.com/show_bug.cgi?id=1599036 . It seems like there might be some issues with Intel wifi firmware and PMF (more info about PMF [here](https://fedoramagazine.org/troubleshoot-pmf-f28/)). Can you try disabling PMF and see if that resolves the issue?

leonardcoutinho, 2020-01-14:
@eklitzke is not Intel card problem, because my phone can't connect too in earlier versions, and it uses an snapdragon 660.

leonardcoutinho, 2020-01-14:
My problem now with latest snapshot and openwrt stable build 19.07, is with the router, seens not stable at all like version stable 19.07 rc2. I will downgrade router to rc2 until this problem is fixed, and investigate it later.

boecko, 2020-02-09:
The current mt76/master (721673759d8229a42ab39a9a7cba15261f1dfcf9) in openwrt-master fails on 5ghz with my tp-link650re1 (MediaTek MT7621 ) - says 'password wrong' all the time.

I've changed my openwrt-master package/kernel/mt76/Makefile to mt76/openwrt-19.07 (8a78567983a16869b77a0254b4917027df4a7ad9). This on works on 5ghz, but is fairly old and has no noise survey.

boecko, 2020-02-09:
I did a git bisect and placed the revision to package/kernel/mt76/Makefile
good: 5ghz worked
bad  : 5ghz auth faild

Here is the log https://gist.github.com/boecko/659890826012e9f56f8fa42df5967bc7

first bad commit: [caa47af1d6f2d7bb0f1a288e46d544f9b99222d6] mt76: mt7615: add multiple wiphy support to the rx path

nbd168, 2020-02-13:
@boecko, please try the latest master revision (not yet in OpenWrt). If it still doesn't work, you should open a separate ticket, since your issue has nothing to do with the one previously reported here.

leonardcoutinho, 2020-02-14:
I will close this thread because with newer master builds both wifi(2.4 and 5ghz) works stable

closed 2020-02-14 by leonardcoutinho

boecko, 2020-02-14:
@nbd168 
i can confirm, that the issue is fixed with the current mt76/master

Thanks!