openwrt/mt76 #327: mt7615: slow tx in non-ideal conditions — problematic rate selection?

intelfx, 2019-11-06:
I'm having here a pair of Xiaomi routers, [Mi WiFi R3G (mir3g)](https://openwrt.org/toh/xiaomi/mir3g) and [Mi Router 3 Pro (mir3p)](https://openwrt.org/toh/xiaomi/xiaomi_r3p_pro) together with an Intel 9260 client in my laptop.

For the purposes of the test, I've been using 23 dBm transmit power and a 80MHz-wide channel 161 on both APs.

The immediate problem is that under near-ideal conditions (i. e. me standing a few meters away from the router) both routers show effective tx throughput in range of 50-60 MB/s (measured with `iperf3 -R` to a standalone server), but under non-ideal conditions (i. e. moving into another room that's separated from the router by a wall under a steep angle) the mir3g shows 30-40 MB/s (stable) while mir3p degrades to 15-20 MB/s (unstable and fluctuating).

Thing is, forcing a fixed mid-high rate (e. g. `echo 386 > /sys/kernel/debug/ieee80211/phy1/rc/fixed_rate_idx` for the described scenario) makes mir3p perform at least as good as mir3g.

Looking at the statistics (`/sys/kernel/debug/ieee80211/phy1/netdev:wlan1/stations/<mac>/rc_stats`) without forcing the rate, I'm not even sure if the driver over- or underestimates the rate as it jumps around quite wildly between VHT80/SGI/1/MCS[7-8] and VHT80/SGI/2/MCS[4-5] and sometimes VHT80/SGI/2/MCS9 (with none of these rates being the experimentally discovered optimal 386).

Could it be more of the same as in 0b6520bd1c98b57e5f0835a0fa8d6c8ca16d2191?

I can test any changes and post detailed statistics/anything else if you need.

Mushoz, 2019-12-07:
There's been quite a few commits in the mean time. Has the situation been improved already?

intelfx, 2019-12-08:
@Mushoz As far as I can tell, there was no relevant commits since the issue was created. In either case, I'm tracking the openwrt trunk and the problem is still there.

Mushoz, 2019-12-25:
More people reporting identical issues: https://forum.openwrt.org/t/odds-of-timely-d-link-dir-882-or-878-support-in-2018/10725/37?u=mushoz

nbd168, 2020-01-23:
Please try the latest revision of OpenWrt master or 19.07. I've fixed a critical issue in hostapd.

intelfx, 2020-01-23:
Great news. I'll try the next nightly.

Mushoz, 2020-01-23:
Can I update hostapd in 19.07.0 via opkg? The makefile didn't receive a version bump, though. Or is reflashing mandatory?

intelfx, 2020-01-23:
@Mushoz I think Felix refers to this commit (https://github.com/openwrt/openwrt/commit/4a58a871c46a589263f565ee7b755d5f3d695580 for `openwrt-19.07` branch, https://github.com/openwrt/openwrt/commit/c07f6e8659ea1348c75c04dac2924616f0042293 for `master` branch). It's not a part of any releases yet — if you are staying on 19.07, then you will need to build it yourself.

I personally am waiting for the next nightlies and will test.

intelfx, 2020-01-24:
@nbd168 Nope — `r12106-c6c4701def` (latest nightly), nothing changed.

WiredLife, 2020-10-17:
Problem still exists, no solution?

garyttirn, 2020-10-18:
TX has gotten much better in latest drivers and even RX seem to be as good as it was around July 2020.

Results with lastest snapshot (17.10.2020) with DIR-882 and Intel 9260
(Signal 866.7 Mbit/s, 80 MHz, VHT-MCS 9, VHT-NSS 2, Short GI. Channel 100):

Station RX:
```
root@zenbook:~# iperf3 -R -c 192.168.254.7 
Connecting to host 192.168.254.7, port 5201
Reverse mode, remote host 192.168.254.7 is sending
[  5] local 192.168.254.42 port 58588 connected to 192.168.254.7 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec  41.2 MBytes   345 Mbits/sec                  
[  5]   1.00-2.00   sec  45.2 MBytes   379 Mbits/sec                  
[  5]   2.00-3.00   sec  49.8 MBytes   418 Mbits/sec                  
[  5]   3.00-4.00   sec  45.6 MBytes   383 Mbits/sec                  
[  5]   4.00-5.00   sec  50.9 MBytes   427 Mbits/sec                  
[  5]   5.00-6.00   sec  49.0 MBytes   411 Mbits/sec                  
[  5]   6.00-7.00   sec  51.4 MBytes   431 Mbits/sec                  
[  5]   7.00-8.00   sec  52.1 MBytes   437 Mbits/sec                  
[  5]   8.00-9.00   sec  52.6 MBytes   442 Mbits/sec                  
[  5]   9.00-10.00  sec  48.9 MBytes   411 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   497 MBytes   417 Mbits/sec   66             sender
[  5]   0.00-10.00  sec   487 MBytes   408 Mbits/sec                  receiver

iperf Done.
```

Station TX:
```
root@zenbook:~# iperf3 -c 192.168.254.7 
Connecting to host 192.168.254.7, port 5201
[  5] local 192.168.254.42 port 58592 connected to 192.168.254.7 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  63.6 MBytes   533 Mbits/sec   76   1.81 MBytes       
[  5]   1.00-2.00   sec  71.2 MBytes   598 Mbits/sec    0   1.83 MBytes       
[  5]   2.00-3.00   sec  66.2 MBytes   556 Mbits/sec   46   1.60 MBytes       
[  5]   3.00-4.00   sec  66.2 MBytes   556 Mbits/sec  140   1.82 MBytes       
[  5]   4.00-5.00   sec  66.2 MBytes   556 Mbits/sec    0   1.47 MBytes       
[  5]   5.00-6.00   sec  63.8 MBytes   535 Mbits/sec    0   1.23 MBytes       
[  5]   6.00-7.00   sec  61.2 MBytes   514 Mbits/sec    0   1.47 MBytes       
[  5]   7.00-8.00   sec  57.5 MBytes   482 Mbits/sec    0   1.26 MBytes       
[  5]   8.00-9.00   sec  63.8 MBytes   535 Mbits/sec  134   1.31 MBytes       
[  5]   9.00-10.00  sec  63.8 MBytes   535 Mbits/sec    0   1.99 MBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   644 MBytes   540 Mbits/sec  396             sender
[  5]   0.00-10.00  sec   642 MBytes   539 Mbits/sec                  receiver
```

garyttirn, 2020-12-21:
Nice work, both TX and RX seem to be all but on par now.
Thanks for your efforts.

Results with lastest snapshot (21.12.2020) with DIR-882 and Intel 9260
(Signal 866.7 Mbit/s, 80 MHz, VHT-MCS 9, VHT-NSS 2, Short GI. Channel 100):

Station RX:

```
zenbook:~$ iperf3 -R -c 192.168.254.7
Connecting to host 192.168.254.7, port 5201
Reverse mode, remote host 192.168.254.7 is sending
[  5] local 192.168.254.42 port 59668 connected to 192.168.254.7 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec  67.5 MBytes   566 Mbits/sec                  
[  5]   1.00-2.00   sec  67.8 MBytes   569 Mbits/sec                  
[  5]   2.00-3.00   sec  63.6 MBytes   534 Mbits/sec                  
[  5]   3.00-4.00   sec  71.7 MBytes   602 Mbits/sec                  
[  5]   4.00-5.00   sec  73.2 MBytes   614 Mbits/sec                  
[  5]   5.00-6.00   sec  71.4 MBytes   599 Mbits/sec                  
[  5]   6.00-7.00   sec  58.0 MBytes   487 Mbits/sec                  
[  5]   7.00-8.00   sec  64.5 MBytes   541 Mbits/sec                  
[  5]   8.00-9.00   sec  52.1 MBytes   437 Mbits/sec                  
[  5]   9.00-10.00  sec  50.3 MBytes   422 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   652 MBytes   547 Mbits/sec  139             sender
[  5]   0.00-10.00  sec   640 MBytes   537 Mbits/sec                  receiver

iperf Done.
```

Station TX:

```
zenbook:~$ iperf3 -c 192.168.254.7
Connecting to host 192.168.254.7, port 5201
[  5] local 192.168.254.42 port 59676 connected to 192.168.254.7 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  68.6 MBytes   576 Mbits/sec    0   1.44 MBytes       
[  5]   1.00-2.00   sec  63.8 MBytes   535 Mbits/sec    0   2.31 MBytes       
[  5]   2.00-3.00   sec  75.0 MBytes   629 Mbits/sec    0   2.60 MBytes       
[  5]   3.00-4.00   sec  65.0 MBytes   545 Mbits/sec    0   1.96 MBytes       
[  5]   4.00-5.00   sec  67.5 MBytes   566 Mbits/sec    0   1.76 MBytes       
[  5]   5.00-6.00   sec  67.5 MBytes   566 Mbits/sec    0   1.51 MBytes       
[  5]   6.00-7.00   sec  68.8 MBytes   577 Mbits/sec    0   2.02 MBytes       
[  5]   7.00-8.00   sec  67.5 MBytes   566 Mbits/sec    0   1.51 MBytes       
[  5]   8.00-9.00   sec  71.2 MBytes   598 Mbits/sec    0   2.68 MBytes       
[  5]   9.00-10.00  sec  65.0 MBytes   545 Mbits/sec    0   5.66 KBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   680 MBytes   570 Mbits/sec    0             sender
[  5]   0.00-10.00  sec   678 MBytes   568 Mbits/sec                  receiver
```