openwrt/mt76 #328: MT7610EN - No Radio

AndyX90, 2019-11-10:
I am using a TP-Link Archer VR200v (MT7610EN for 5GHz) and after installing kmod-mt76x0e no wifi is coming up.

On current OpenWRT Snapshot(Kernel 4.19.81).

Lines in Kernel Log:

> [   15.155349] mt76x0e 0000:02:00.0: enabling device (0000 -> 0002)
> [   15.161540] mt76x0e 0000:02:00.0: ASIC revision: 76100002
> [   15.198773] mt76x0e 0000:02:00.0: Firmware Version: 0.1.00
> [   15.453278] mt76x0e 0000:02:00.0: driver does not support default EEPROM
> [   15.458757] mt76x0e: probe of 0000:02:00.0 failed with error -22

LorenzoBianconi, 2019-11-10:
It seems to me an eeprom value issue. Could you please provide device
eeprom?

AndyX90, 2019-11-10:
Sure, i will provide anything you need!
According to [OpenWRT-TOH](https://openwrt.org/toh/tp-link/vr200v) the flash layout on stock firmware is the following: 

```
partition | address space | name | filesystem | function
mtd0 | 0x00000000-0x00020000 | boot | n/a | U-Boot v2.2.67
mtd1 | 0x00020000-0x00160000 | kernel | RAM executable | kernel 2.6.32.42
mtd2 | 0x00160000-0x00fb0000 | rootfs | squashfs | root
mtd3 | 0x00fb0000-0x00fc0000 | radioDECT | n/a | ???
mtd4 | 0x00fc0000-0x00fd0000 | config | XML string | user configuration
mtd5 | 0x00fd0000-0x00fe0000 | romfile | n/a | ???
mtd6 | 0x00fe0000-0x00ff0000 | rom | n/a | ???
mtd7 | 0x00ff0000-0x01000000 | radio | n/a | radio settings
```
I was able to backup all mtd-partitions except mtd7, which seem to be not mounted on stock firmware.
Can you tell me exactly what you need?

Thanks!

LorenzoBianconi, 2019-11-15:
@AndyX90 I think we need the radio partition

h-s-c, 2019-11-18:
> mt76x0e 0000:02:00.0: driver does not support default EEPROM
I think this is a byte-order issue.

Here is the radio partition from my VR200v: 
[OpenWrt.mtd9.zip](https://github.com/openwrt/mt76/files/3860875/OpenWrt.mtd9.zip)

And here is the EEPROM file that is also in the original firmware and is maybe used instead of the device partition:
[MT7610E-V10-FEM-1ANT.zip](https://github.com/openwrt/mt76/files/3860913/MT7610E-V10-FEM-1ANT.zip)

LorenzoBianconi, 2019-11-18:
@h-s-c could you please double check mt76_eeprom_init (run by mt76x0_load_eeprom) is able to read mtd data partition? It seems to me mt76_eeprom_init fails (found = 0) and the driver falls back trying to access efuse data. BTW the files you provided seems very similar (just few differences)

h-s-c, 2019-11-19:
Sorry for the confusion. You are right. I misremembered the error message.
The error indicates a missing mediatek,mtd-eeprom property in the device tree.

If you wire that all up you get another error:

> mt76x0e 0000:02:00.0: EEPROM data check failed 1076

mt76_eeprom_init reads the mtd partition fine just in the wrong byte-order.
mt76x0_check_eeprom checks for 0x7610 but gets 0x1076.

If i byte swap dev->eeprom.data after mtd_read inside mt76_get_of_eeprom the driver loads fine.
I'm just not sure how a proper upstreamable solution would look like.
Here is my patch for reference.

[0001-mt76-add-mtd-eeprom-swap-function.patch.txt](https://github.com/openwrt/mt76/files/3865306/0001-mt76-add-mtd-eeprom-swap-function.patch.txt)

LorenzoBianconi, 2019-11-19:
@h-s-c re-looking at file you provided it seems the eeprom data is written in big endian while mt76 assumes it is written in little endian

h-s-c, 2019-11-19:
@LorenzoBianconi That's the gist of it.

LorenzoBianconi, 2019-11-19:
@h-s-c could you please try this [patch](https://github.com/LorenzoBianconi/wireless-drivers-next/commit/6835b43caccfd47d335afe65b72b55780afe74e1)

h-s-c, 2019-11-20:
@LorenzoBianconi mt76x0_check_eeprom does not use mt76x02_eeprom_get it uses it own get_unaligned_le16(dev->mt76.eeprom.data) call to check the EEPROM header.

> [   29.748049] mt76x0e 0000:02:00.0: enabling device (0000 -> 0002)
[   29.754277] mt76x0e 0000:02:00.0: ASIC revision: 76100002
[   29.766702] mt76x0e 0000:02:00.0: Firmware Version: 0.1.00
[   29.981292] mt76x0e 0000:02:00.0: EEPROM data check failed: 1076
[   30.024384] mt76x0e 0000:02:00.0: driver does not support default EEPROM
[   30.029869] mt76x0e: probe of 0000:02:00.0 failed with error -22

LorenzoBianconi, 2019-11-20:
@h-s-c ack, I will review the code

LorenzoBianconi, 2019-11-20:
@h-s-c I guess it is better to follow your approach and swap eeprom data. Could you please try [patch](https://github.com/LorenzoBianconi/wireless-drivers-next/commit/72010702b6993ec3fb2616d0b458bbf999b33747)? Thanks

h-s-c, 2019-11-21:
@LorenzoBianconi dev->eeprom.data is a void*

>error: dereferencing 'void *' pointer [-Werror]
error: taking address of expression of type 'void' [-Werror]

> struct mt76_dev {
    ...
    struct debugfs_blob_wrapper eeprom;
    ...
};

> struct debugfs_blob_wrapper {
	void *data;
	unsigned long size;
};

LorenzoBianconi, 2019-11-21:
@h-s-c ack, updated [here](https://github.com/LorenzoBianconi/wireless-drivers-next/commit/f578ab8d3e0206776c6725b49e9cd7fb04d5ee20)

h-s-c, 2019-11-22:
@LorenzoBianconi of_property_read_bool(np, "big-endian") does not work because np points to the wrong node at that point.

> list = of_get_property(np, "mediatek,mtd-eeprom", &size);
...
np = of_find_node_by_phandle(phandle);
...
if (of_property_read_bool(np, "big-endian")) {

Device tree:
>mediatek,mtd-eeprom = <&radio 0x0000>;
big-endian;
ieee80211-freq-limit = <5000000 6000000>;

LorenzoBianconi, 2019-11-22:
> @LorenzoBianconi of_property_read_bool(np, "big-endian") does not work because np points to the wrong node at that point.
>
>
> Device tree:

@h-s-c my idea is to add the 'big-endian' properly into the radio partition node (e.g 'factory' node) instead of into the wireless device one (e.g. mt76). Could you please try to add the property into the radio partition node in the board dts?

nbd168, 2019-11-22:
I don't think it should be added to the partition node, that probably won't sit well with upstream either, since it's overloading a different namespace.
Please add the property to the wireless device node.

LorenzoBianconi, 2019-11-22:
> I don't think it should be added to the partition node, that probably won't sit well with upstream either, since it's overloading a different namespace.
> Please add the property to the wireless device node.

ack, but do we have the same issue for 'label' in this case?

LorenzoBianconi, 2019-11-22:
@h-s-c [this](https://github.com/LorenzoBianconi/wireless-drivers-next/commit/7938c5e6645da25a2445ccb1987ee9983b8d8a68) is updated version with big-ending property added in wireless device node instead of in partition one. Could you please give it a whirl?

h-s-c, 2019-11-22:
@LorenzoBianconi Driver loads fine. Thank you very much!

> [   29.845130] mt76x0e 0000:02:00.0: enabling device (0000 -> 0002)
[   29.851346] mt76x0e 0000:02:00.0: ASIC revision: 76100002
[   29.863235] mt76x0e 0000:02:00.0: Firmware Version: 0.1.00
[   30.081321] mt76x0e 0000:02:00.0: EEPROM ver:01 fae:00

LorenzoBianconi, 2019-11-22:
> @LorenzoBianconi Driver loads fine. Thank you very much!

Just to be on the same page, have you used last [patch](https://github.com/LorenzoBianconi/wireless-drivers-next/commit/7938c5e6645da25a2445ccb1987ee9983b8d8a68)?
Is the driver working fine now? (passing traffic and so on). If so I will post the fix with your tested-by tag

Thanks for testing

h-s-c, 2019-11-22:
@LorenzoBianconi Yes i used the last patch.
SSID shows up but i cant connect with any of my devices.
No errors/warnings/messages in either system or kernel log while trying to connect.
Cant see anything obvious on my Android looking at adb logcat.
I tried multiple channel, 20/40/80 MHz and N/AC combinations.

> cat /etc/config/wireless
>
>config wifi-device 'radio0'
        option type 'mac80211'
        option path 'pci0000:01/0000:01:00.0/0000:02:00.0'
        option hwmode '11a'
        option htmode 'VHT80'
        option legacy_rates '0'
        option country 'DE'
        option channel '36'
>
>config wifi-iface 'wifinet0'
        option device 'radio0'
        option mode 'ap'
        option encryption 'none'
        option ssid 'REDACTED'
        option network 'lan'

> iwinfo
ESSID: "REDACTED"
          Access Point: REDACTED
          Mode: Master  Channel: 36 (5.180 GHz)
          Tx-Power: 12 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: unknown
          Bit Rate: unknown
          Encryption: none
          Type: nl80211  HW Mode(s): 802.11nac
          Hardware: 14C3:7650 14C3:7650 [MediaTek MT7610E]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy0

LorenzoBianconi, 2019-11-22:
@h-s-c could you please provide logread() output?

h-s-c, 2019-11-22:
@LorenzoBianconi 

 logread of wlan down/wlan up after that a connection attempt (wireless.radio0.log_level=1)
>Fri Nov 22 20:08:28 2019 daemon.notice hostapd: Remove interface 'wlan0'
Fri Nov 22 20:08:28 2019 daemon.notice hostapd: wlan0: interface state ENABLED->DISABLED
Fri Nov 22 20:08:28 2019 daemon.notice hostapd: wlan0: AP-DISABLED
Fri Nov 22 20:08:28 2019 daemon.notice hostapd: wlan0: CTRL-EVENT-TERMINATING
Fri Nov 22 20:08:28 2019 daemon.notice hostapd: nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Fri Nov 22 20:08:28 2019 kern.info kernel: [  311.704096] device wlan0 left promiscuous mode
Fri Nov 22 20:08:28 2019 kern.info kernel: [  311.707369] br-lan: port 2(wlan0) entered disabled state
Fri Nov 22 20:08:28 2019 daemon.notice netifd: Network device 'wlan0' link is down
Fri Nov 22 20:08:33 2019 daemon.notice netifd: radio0 (4051): sh: out of range
Fri Nov 22 20:08:34 2019 daemon.notice hostapd: Configuration file: /var/run/hostapd-phy0.conf (phy wlan0) --> new PHY
Fri Nov 22 20:08:35 2019 kern.info kernel: [  318.672671] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
Fri Nov 22 20:08:35 2019 kern.info kernel: [  318.689727] br-lan: port 2(wlan0) entered blocking state
Fri Nov 22 20:08:35 2019 kern.info kernel: [  318.693773] br-lan: port 2(wlan0) entered disabled state
Fri Nov 22 20:08:35 2019 kern.info kernel: [  318.699620] device wlan0 entered promiscuous mode
Fri Nov 22 20:08:35 2019 daemon.notice hostapd: wlan0: interface state UNINITIALIZED->COUNTRY_UPDATE
Fri Nov 22 20:08:35 2019 daemon.notice hostapd: wlan0: interface state COUNTRY_UPDATE->HT_SCAN
Fri Nov 22 20:08:35 2019 daemon.err hostapd: Using interface wlan0 with hwaddr ec:08:6b:22:6f:2b and ssid "REDACTED"
Fri Nov 22 20:08:36 2019 kern.info kernel: [  319.760463] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
Fri Nov 22 20:08:36 2019 kern.info kernel: [  319.765999] br-lan: port 2(wlan0) entered blocking state
Fri Nov 22 20:08:36 2019 kern.info kernel: [  319.770904] br-lan: port 2(wlan0) entered forwarding state
Fri Nov 22 20:08:36 2019 daemon.notice hostapd: wlan0: interface state HT_SCAN->ENABLED
Fri Nov 22 20:08:36 2019 daemon.notice hostapd: wlan0: AP-ENABLED
Fri Nov 22 20:08:36 2019 user.notice root: ip link set dev wlan0 up
Fri Nov 22 20:08:37 2019 daemon.notice netifd: Network device 'wlan0' link is up

Managment traffic of some connection attempts:
[wifi_pcap.zip](https://github.com/openwrt/mt76/files/3881087/wifi_pcap.zip)

LorenzoBianconi, 2019-11-22:
@h-s-c looking at the patch we are going to swap even the mac address, what is the mac address showed by ifconfig or ip a? Could you please try to overwrite it using 'mac-address' property in wireless device dts node?

h-s-c, 2019-11-22:
@LorenzoBianconi I already used the mac-adress property to try to follow stock firmware behaviour...
After removing it everything works fine (Traffic etc.).
Thanks for the help.

[Working Patch](https://github.com/LorenzoBianconi/wireless-drivers-next/commit/7938c5e6645da25a2445ccb1987ee9983b8d8a68)

LorenzoBianconi, 2019-11-22:
@h-s-c so, to be on the same page..adding the latest patch and removing the Mac address entry in the dts, it works fine. Correct?

h-s-c, 2019-11-22:
@LorenzoBianconi  Correct.

LorenzoBianconi, 2019-11-23:
Ack, thx for testing..I will post the patch adding your tested by

Regards

LorenzoBianconi, 2019-11-23:
@h-s-c Maybe I found why 'mtd-mac-address' property breaks the connection on mt76x0e. Could you please try to apply this [patch](https://github.com/LorenzoBianconi/wireless-drivers-next/commit/81f0cfa6f34d4f657c2ed75ebed61ecae24d133e) ontop of the big endian fix and reintroduce the mtd-mac-address property in the board device tree? Thanks

h-s-c, 2019-11-23:
@LorenzoBianconi I can confirm that the patch fixes the mtd-mac-address property on mt76x0e.

rozhuk-im, 2019-11-25:
Hi!
I have DIR-510L, looks like it have same issue with 5GHz WiFi.
I try patch but got:
```
Mon Nov 25 01:24:48 2019 kern.err kernel: [   13.943103] mt76x0e 0000:01:00.0: card - bus=0x1, slot = 0x0 irq=4
Mon Nov 25 01:24:48 2019 kern.info kernel: [   13.955712] mt76x0e 0000:01:00.0: ASIC revision: 76100002
Mon Nov 25 01:24:48 2019 kern.info kernel: [   13.969097] mt76x0e 0000:01:00.0: Firmware Version: 0.1.00
Mon Nov 25 01:24:48 2019 kern.err kernel: [   14.063096] mt76x0e 0000:01:00.0: EEPROM data check failed: 1076
Mon Nov 25 01:24:48 2019 kern.err kernel: [   14.087557] mt76x0e 0000:01:00.0: driver does not support default EEPROM
Mon Nov 25 01:24:48 2019 kern.warn kernel: [   14.101059] mt76x0e: probe of 0000:01:00.0 failed with error -22
```
my dst have ```big-endian;```
```
&pcie0 {
	mt76x0e@0,0 {
		reg = <0x0000 0 0 0 0>;
		mtd-mac-address = <&config 0xe490>;
		mtd-mac-address-increment = <(2)>;
		mediatek,mtd-eeprom = <&config 0xe05d>;
		big-endian;
		ieee80211-freq-limit = <5000000 6000000>;
	};
};
```
and I add changes from https://github.com/LorenzoBianconi/wireless-drivers-next/commit/7938c5e6645da25a2445ccb1987ee9983b8d8a68 before build to
/mnt/owrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/mt76-2019-10-10-d3a58958/eeprom.c
/mnt/owrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/backports-5.4-rc8-1/drivers/net/wireless/mediatek/mt76/eeprom.c

LorenzoBianconi, 2019-11-25:
@rozhuk-im could you please provide the config partition?

rozhuk-im, 2019-11-25:
Full Flash: http://www.netlab.linkpc.net/download/tmp/DIR-510L/dir510l_new.bin

LorenzoBianconi, 2019-11-25:
@rozhuk-im looking at the config partition it seems to me already in little endian, please remove the big-endian property in the wireless device dts node (but please keep both patches applied)

rozhuk-im, 2019-11-25:
I revert dts to remove ```big-endian;```, but still can not connect:
```
Mon Nov 25 01:36:42 2019 daemon.info hostapd: wlan0: STA cc:c6:36:2a:8f:95 IEEE 802.11: authenticated
Mon Nov 25 01:36:42 2019 daemon.info hostapd: wlan0: STA cc:c6:36:2a:8f:95 IEEE 802.11: authenticated
Mon Nov 25 01:36:46 2019 daemon.info hostapd: wlan0: STA cc:c6:36:2a:8f:95 IEEE 802.11: authenticated
Mon Nov 25 01:36:46 2019 daemon.info hostapd: wlan0: STA cc:c6:36:2a:8f:95 IEEE 802.11: authenticated
Mon Nov 25 01:36:46 2019 daemon.info hostapd: wlan0: STA cc:c6:36:2a:8f:95 IEEE 802.11: authenticated
Mon Nov 25 01:36:46 2019 daemon.info hostapd: wlan0: STA cc:c6:36:2a:8f:95 IEEE 802.11: associated (aid 1)
Mon Nov 25 01:36:46 2019 daemon.info hostapd: wlan0: STA cc:c6:36:2a:8f:95 IEEE 802.11: associated (aid 1)
Mon Nov 25 01:36:47 2019 daemon.info hostapd: wlan0: STA cc:c6:36:2a:8f:95 IEEE 802.11: authenticated
Mon Nov 25 01:36:47 2019 daemon.info hostapd: wlan0: STA cc:c6:36:2a:8f:95 IEEE 802.11: authenticated
Mon Nov 25 01:36:55 2019 daemon.info hostapd: wlan0: STA cc:c6:36:2a:8f:95 IEEE 802.11: deauthenticated due to local deauth request
```

LorenzoBianconi, 2019-11-26:
@rozhuk-im Is the mtd patch applied?

rozhuk-im, 2019-12-01:
@LorenzoBianconi only https://github.com/LorenzoBianconi/wireless-drivers-next/commit/7938c5e6645da25a2445ccb1987ee9983b8d8a68

LorenzoBianconi, 2019-12-01:
> @LorenzoBianconi only [LorenzoBianconi/wireless-drivers-next@7938c5e](https://github.com/LorenzoBianconi/wireless-drivers-next/commit/7938c5e6645da25a2445ccb1987ee9983b8d8a68)

You need to apply even this [patch](https://github.com/LorenzoBianconi/wireless-drivers-next/commit/81f0cfa6f34d4f657c2ed75ebed61ecae24d133e)

rozhuk-im, 2019-12-01:
What is a proper way to add/apply these patches?

LorenzoBianconi, 2019-12-01:
> What is a proper way to add/apply these patches?

you can cherry-pick or am the patch from [here](https://github.com/LorenzoBianconi/mt76/commit/7e6c821881180ac934d41b6b86f49956572082e3)

rozhuk-im, 2019-12-01:
But I'm building openwrt "image" not mt76.

AndyX90, 2019-12-10:
Will this patch be merged to openwrt/mt76?

closed 2019-12-19 by AndyX90