openwrt/mt76 #329: mt7615e @ R6260 antenna detect 4x4 instead 3x3

nicefile, 2019-11-12:
It seems to be detection issue with antenna count . Netgear R6260 have 3 antennas . 2 external and one internal .  My mt76 compilation is from 10.10.2019
On mt76 iw phy info shows :
Available Antennas: TX 0xf RX 0xf
1111b 
Also MCS  numbers is counted up to 4 streams
I'm unable to set antenna number to 1110
root@OpenWrt:~# iw phy1 set antenna 0xe 0xe
command failed: Not supported (-122)

for reference source driver there is no issues with detection .
mt7615_antenna_default_reset(): TxPath = 3, RxPath = 3

LorenzoBianconi, 2019-11-14:
@nicefile could you please provide your eeprom data?

LorenzoBianconi, 2019-11-14:
@nicefile could you please to test this [patch](https://github.com/LorenzoBianconi/wireless-drivers-next/commit/513773605844a283806e579c51fc58489f9a42a6)?

nicefile, 2019-11-14:
@LorenzoBianconi 
[mt7615.eeprom1.bin.zip](https://github.com/openwrt/mt76/files/3847848/mt7615.eeprom1.bin.zip)
Yes I can build and test above patch

LorenzoBianconi, 2019-11-14:
@nicefile I updated the patch, the new patch is this [one](https://github.com/LorenzoBianconi/wireless-drivers-next/commit/2ad8ff3a8e7a8f2cd752cdb0f2e6699675d33f7d).

nicefile, 2019-11-14:
@LorenzoBianconi It work now 3x3  .Also signal meter shows 70% instead 27% previously
iw phy3 info
Wiphy phy3
        max # scan SSIDs: 4
        max scan IEs length: 2247 bytes
        max # sched scan SSIDs: 0
        max # match sets: 0
        max # scan plans: 1
        max scan plan interval: -1
        max scan plan iterations: 0
        Retry short limit: 7
        Retry long limit: 4
        Coverage class: 0 (up to 0m)
        Available Antennas: TX 0x7 RX 0x7
        Supported interface modes:
                 * managed
                 * AP
                 * AP/VLAN
                 * monitor
                 * mesh point
        Band 2:
                Capabilities: 0x1ff
                        RX LDPC
                        HT20/HT40
                        SM Power Save disabled
                        RX Greenfield
                        RX HT20 SGI
                        RX HT40 SGI
                        TX STBC
                        RX STBC 1-stream
                        Max AMSDU length: 3839 bytes
                        No DSSS/CCK HT40
                Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
                Minimum RX AMPDU time spacing: 4 usec (0x05)
                HT TX/RX MCS rate indexes supported: 0-23
                VHT Capabilities (0x338001fa):
                        Max MPDU length: 11454
                        Supported Channel Width: 160 MHz, 80+80 MHz
                        RX LDPC
                        short GI (80 MHz)
                        short GI (160/80+80 MHz)
                        TX STBC
                        RX antenna pattern consistency
                        TX antenna pattern consistency
                VHT RX MCS set:
                        1 streams: MCS 0-9
                        2 streams: MCS 0-9
                        3 streams: MCS 0-9
                        4 streams: not supported
                        5 streams: not supported
                        6 streams: not supported
                        7 streams: not supported
                        8 streams: not supported
                VHT RX highest supported: 0 Mbps
                VHT TX MCS set:
                        1 streams: MCS 0-9
                        2 streams: MCS 0-9
                        3 streams: MCS 0-9
                        4 streams: not supported
                        5 streams: not supported
                        6 streams: not supported
                        7 streams: not supported
                        8 streams: not supported
                VHT TX highest supported: 0 Mbps
                Frequencies:
                        * 5180 MHz [36] (20.0 dBm)
                        * 5200 MHz [40] (20.0 dBm)
                        * 5220 MHz [44] (20.0 dBm)
                        * 5240 MHz [48] (20.0 dBm)
                        * 5260 MHz [52] (20.0 dBm) (radar detection)
                        * 5280 MHz [56] (20.0 dBm) (radar detection)
                        * 5300 MHz [60] (20.0 dBm) (radar detection)
                        * 5320 MHz [64] (20.0 dBm) (radar detection)
                        * 5500 MHz [100] (22.0 dBm)
                        * 5520 MHz [104] (22.0 dBm)
                        * 5540 MHz [108] (22.0 dBm)
                        * 5560 MHz [112] (22.0 dBm)
                        * 5580 MHz [116] (22.0 dBm)
                        * 5600 MHz [120] (22.0 dBm)
                        * 5620 MHz [124] (22.0 dBm)
                        * 5640 MHz [128] (22.0 dBm)
                        * 5660 MHz [132] (22.0 dBm)
                        * 5680 MHz [136] (22.0 dBm)
                        * 5700 MHz [140] (22.0 dBm)
                        * 5745 MHz [149] (disabled)
                        * 5765 MHz [153] (disabled)
                        * 5785 MHz [157] (disabled)
                        * 5805 MHz [161] (disabled)
                        * 5825 MHz [165] (disabled)
        valid interface combinations:
                 * #{ managed, AP, mesh point } <= 4,
                   total <= 4, #channels <= 1, STA/AP BI must match, radar detect widths: { 20 MHz (no HT), 20 MHz, 40 MHz, 80 MHz, 80+80 MHz, 160 MHz }

        HT Capability overrides:
                 * MCS: ff ff ff ff ff ff ff ff ff ff
                 * maximum A-MSDU length
                 * supported channel width
                 * short GI for 40 MHz
                 * max A-MPDU length exponent
                 * min MPDU start spacing
        Supported extended features:
                * [ RRM ]: RRM
                * [ CQM_RSSI_LIST ]: multiple CQM_RSSI_THOLD records
                * [ CONTROL_PORT_OVER_NL80211 ]: control port over nl80211
                * [ TXQS ]: FQ-CoDel-enabled intermediate TXQs
                * [ AIRTIME_FAIRNESS ]: airtime fairness scheduling

LorenzoBianconi, 2019-11-14:
>
>
> @LorenzoBianconi <https://github.com/LorenzoBianconi> It work now 3x3
> .Also signal meter shows 70% instead 27% previously

Cool :) Have you tested the latest version? If so I will post it upstream
adding your Tested-by

Regards,
Lorenzo

> iw phy3 info
> Wiphy phy3
> max # scan SSIDs: 4
> max scan IEs length: 2247 bytes
> max # sched scan SSIDs: 0
> max # match sets: 0
> max # scan plans: 1
> max scan plan interval: -1
> max scan plan iterations: 0 [...]

nicefile, 2019-11-14:
@LorenzoBianconi 
lastest openwrt master commit

LorenzoBianconi, 2019-11-14:
> @LorenzoBianconi
> lastest openwrt master commit

I mean my last [patch](https://github.com/LorenzoBianconi/wireless-drivers-next/commit/2ad8ff3a8e7a8f2cd752cdb0f2e6699675d33f7d)

nicefile, 2019-11-14:
@LorenzoBianconi 
Yes lastest patch

LorenzoBianconi, 2019-11-14:
> @LorenzoBianconi <https://github.com/LorenzoBianconi>
> Yes lastest patch
Ok, cool. I will post it soon.

Regards,
Lorenzo

nicefile, 2019-11-15:
@LorenzoBianconi 
Just one question . Will be possible to use AC@160Mhz wide channel  on 3x3 antenna ? I Think not basing my current tests . It's constantly stays @80Mhz no matter what I do

LorenzoBianconi, 2019-11-15:
@nicefile I do not think bandwidth depends on # of streams

LorenzoBianconi, 2019-11-20:
@nicefile I guess it can be closed now

closed 2019-11-22 by nicefile

patricius972, 2020-06-07:
Hi, It seems i have the same problem, how do i get the eeprom data?

nicefile, 2020-06-07:
@patricius972  and your output for command shows what ? (phy0 or phy1) 
iw phy0 info
Eeprom data is relevant to this bug.

patricius972, 2020-06-07:
Here is what i have, also my 2.4G on phy1 is not showing up

Wiphy phy0
	max # scan SSIDs: 4
	max scan IEs length: 2257 bytes
	max # sched scan SSIDs: 0
	max # match sets: 0
	max # scan plans: 1
	max scan plan interval: -1
	max scan plan iterations: 0
	Retry short limit: 7
	Retry long limit: 4
	Coverage class: 0 (up to 0m)
	Device supports T-DLS.
	Available Antennas: TX 0x3 RX 0x3
	Configured Antennas: TX 0x3 RX 0x3
	Supported interface modes:
		 * IBSS
		 * managed
		 * AP
		 * AP/VLAN
		 * monitor
		 * mesh point
		 * P2P-client
		 * P2P-GO
	Band 1:
		Capabilities: 0x1fe
			HT20/HT40
			SM Power Save disabled
			RX Greenfield
			RX HT20 SGI
			RX HT40 SGI
			TX STBC
			RX STBC 1-stream
			Max AMSDU length: 3839 bytes
			No DSSS/CCK HT40
		Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
		Minimum RX AMPDU time spacing: No restriction (0x00)
		HT TX/RX MCS rate indexes supported: 0-15
		Frequencies:
			* 2412 MHz [1] (30.0 dBm)
			* 2417 MHz [2] (30.0 dBm)
			* 2422 MHz [3] (30.0 dBm)
			* 2427 MHz [4] (30.0 dBm)
			* 2432 MHz [5] (30.0 dBm)
			* 2437 MHz [6] (30.0 dBm)
			* 2442 MHz [7] (30.0 dBm)
			* 2447 MHz [8] (30.0 dBm)
			* 2452 MHz [9] (30.0 dBm)
			* 2457 MHz [10] (30.0 dBm)
			* 2462 MHz [11] (30.0 dBm)
			* 2467 MHz [12] (30.0 dBm)
			* 2472 MHz [13] (30.0 dBm)
			* 2484 MHz [14] (disabled)
	valid interface combinations:
		 * #{ IBSS } <= 1, #{ managed, AP, mesh point, P2P-client, P2P-GO } <= 4,
		   total <= 4, #channels <= 1, STA/AP BI must match
	HT Capability overrides:
		 * MCS: ff ff ff ff ff ff ff ff ff ff
		 * maximum A-MSDU length
		 * supported channel width
		 * short GI for 40 MHz
		 * max A-MPDU length exponent
		 * min MPDU start spacing
	Supported extended features:
		* [ RRM ]: RRM
		* [ CQM_RSSI_LIST ]: multiple CQM_RSSI_THOLD records
		* [ CONTROL_PORT_OVER_NL80211 ]: control port over nl80211
		* [ TXQS ]: FQ-CoDel-enabled intermediate TXQs
		* [ AIRTIME_FAIRNESS ]: airtime fairness scheduling

phy1:
Wiphy phy1
	max # scan SSIDs: 4
	max scan IEs length: 2304 bytes
	max # sched scan SSIDs: 0
	max # match sets: 0
	max # scan plans: 1
	max scan plan interval: 0
	max scan plan iterations: 0
	Retry short limit: 7
	Retry long limit: 4
	Coverage class: 1 (up to 450m)
	Device supports T-DLS.
	Available Antennas: TX 0xf RX 0xf
	Configured Antennas: TX 0xf RX 0xf
	Supported interface modes:
		 * IBSS
		 * managed
		 * AP
		 * AP/VLAN
		 * monitor
		 * mesh point
		 * P2P-client
		 * P2P-GO
	Band 2:
		Capabilities: 0x1ff
			RX LDPC
			HT20/HT40
			SM Power Save disabled
			RX Greenfield
			RX HT20 SGI
			RX HT40 SGI
			TX STBC
			RX STBC 1-stream
			Max AMSDU length: 3839 bytes
			No DSSS/CCK HT40
		Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
		Minimum RX AMPDU time spacing: No restriction (0x00)
		HT TX/RX MCS rate indexes supported: 0-31
		VHT Capabilities (0x338001fa):
			Max MPDU length: 11454
			Supported Channel Width: 160 MHz, 80+80 MHz
			RX LDPC
			short GI (80 MHz)
			short GI (160/80+80 MHz)
			TX STBC
			RX antenna pattern consistency
			TX antenna pattern consistency
		VHT RX MCS set:
			1 streams: MCS 0-9
			2 streams: MCS 0-9
			3 streams: MCS 0-9
			4 streams: MCS 0-9
			5 streams: not supported
			6 streams: not supported
			7 streams: not supported
			8 streams: not supported
		VHT RX highest supported: 0 Mbps
		VHT TX MCS set:
			1 streams: MCS 0-9
			2 streams: MCS 0-9
			3 streams: MCS 0-9
			4 streams: MCS 0-9
			5 streams: not supported
			6 streams: not supported
			7 streams: not supported
			8 streams: not supported
		VHT TX highest supported: 0 Mbps
		Frequencies:
			* 5180 MHz [36] (6.0 dBm)
			* 5200 MHz [40] (6.0 dBm)
			* 5220 MHz [44] (6.0 dBm)
			* 5240 MHz [48] (6.0 dBm)
			* 5260 MHz [52] (6.0 dBm) (radar detection)
			* 5280 MHz [56] (6.0 dBm) (radar detection)
			* 5300 MHz [60] (6.0 dBm) (radar detection)
			* 5320 MHz [64] (6.0 dBm) (radar detection)
			* 5500 MHz [100] (6.0 dBm) (radar detection)
			* 5520 MHz [104] (6.0 dBm) (radar detection)
			* 5540 MHz [108] (6.0 dBm) (radar detection)
			* 5560 MHz [112] (6.0 dBm) (radar detection)
			* 5580 MHz [116] (6.0 dBm) (radar detection)
			* 5600 MHz [120] (6.0 dBm) (radar detection)
			* 5620 MHz [124] (6.0 dBm) (radar detection)
			* 5640 MHz [128] (6.0 dBm) (radar detection)
			* 5660 MHz [132] (6.0 dBm) (radar detection)
			* 5680 MHz [136] (6.0 dBm) (radar detection)
			* 5700 MHz [140] (6.0 dBm) (radar detection)
			* 5745 MHz [149] (6.0 dBm)
			* 5765 MHz [153] (6.0 dBm)
			* 5785 MHz [157] (6.0 dBm)
			* 5805 MHz [161] (6.0 dBm)
			* 5825 MHz [165] (6.0 dBm)
	valid interface combinations:
		 * #{ IBSS } <= 1, #{ managed, AP, mesh point, P2P-client, P2P-GO } <= 4,
		   total <= 4, #channels <= 1, STA/AP BI must match, radar detect widths: { 20 MHz (no HT), 20 MHz, 40 MHz, 80 MHz, 80+80 MHz, 160 MHz }

	HT Capability overrides:
		 * MCS: ff ff ff ff ff ff ff ff ff ff
		 * maximum A-MSDU length
		 * supported channel width
		 * short GI for 40 MHz
		 * max A-MPDU length exponent
		 * min MPDU start spacing
	Supported extended features:
		* [ VHT_IBSS ]: VHT-IBSS
		* [ RRM ]: RRM
		* [ SET_SCAN_DWELL ]: scan dwell setting
		* [ CQM_RSSI_LIST ]: multiple CQM_RSSI_THOLD records
		* [ CONTROL_PORT_OVER_NL80211 ]: control port over nl80211
		* [ TXQS ]: FQ-CoDel-enabled intermediate TXQs
		* [ AIRTIME_FAIRNESS ]: airtime fairness scheduling

patricius972, 2020-06-08:
fixed my issue, was an eeprom reading issue.