openwrt/mt76 #330: MT7615 LED

dengqf6, 2019-12-03:
I got a router whose wireless LED is connected to MT7615 GPIO#14 directly, but the LED is not working with mt76 driver.

Is `led-sources` required in the DTS?

LorenzoBianconi, 2019-12-04:
@dengqf6 mt7615 has some pwm led pins, are you sure the led is connected to a generic gpio? (I have no 7615 pin layout)

LorenzoBianconi, 2019-12-04:
@dengqf6 now I think I got the point, GPIO14 should be a led pin. I will send you a patch to test.

dengqf6, 2019-12-05:
@LorenzoBianconi Yes. AKA pin 35.

LorenzoBianconi, 2019-12-05:
@dengqf6 could you please try this [patch](https://github.com/LorenzoBianconi/wireless-drivers-next/commit/60b1c72208c634d48d89c23cbcf382e2c4f74651)? I do not think you need to add any node in the device tree since the default LED is 0

dengqf6, 2019-12-08:
Tried. Did not work.

LorenzoBianconi, 2019-12-08:
Hi,

thx for testing..could you please double check the led entry is properly
created in /sys/class/leds/mt76-phyX?
Moreover, could you please double check the trigger?
You can try to change led pin and polarity in debug filesystem led_pin and
led_active

Regards,
Lorenzo

dengqf6, 2019-12-09:
@LorenzoBianconi The mt76-phyX were not registered.
The mt76 version is openwrt/mt76@d3a58958 (in OpenWrt master). Is that the cause?

LorenzoBianconi, 2019-12-09:
> The mt76 version is d3a5895

Have you applied the patch I pointed out in the previous message?

dengqf6, 2019-12-09:
Yes

LorenzoBianconi, 2019-12-09:
@dengqf6 : got it, we are missing this [patch](https://github.com/LorenzoBianconi/mt76/commit/e1cb717eb6c5f0f5f616edfd77670e00dcad1391) for mt76 github repo.
Could you please try to use this [mt76](https://github.com/LorenzoBianconi/mt76/tree/mt76-devel) branch?

dengqf6, 2019-12-10:
It's working now. Thanks!

closed 2019-12-29 by dengqf6

i3roly, 2020-02-11:
hi,

i am having the same issue with no WIFI LEDS. 

> obj-$(CPTCFG_MT7615E) += mt7615e.o
EXTRA_CFLAGS += -Werror -DCONFIG_MT76_LEDS

mt7615e-y := pci.o init.o dma.o eeprom.o main.o mmio.o mcu.o mac.o \
             trace.o debugfs.o

is present in the makefile. could you please give some suggestions as to why the LEDs are not working? i am surprised they work for dengqf6 but not myself...