openwrt/mt76 #334: mt76x2u: BEACON-RESP-RX always empty

PolynomialDivision, 2019-12-06:
I tried to get beacon response messages, but the answer is always

    wlan0: BEACON-RESP-RX 00:xx:xx:xx:xx:xx 62 04

Actually it should be something like

    wlan0: BEACON-RESP-RX 00:xx:xx:xx:xx:xx 62 04 [here some binary data...]

I checked everything with a Nexus 5X and it works fine.

LorenzoBianconi, 2019-12-06:
Could you please provide more details? Could you please provide a
reproducer?

Regards,
Lorenzo

PolynomialDivision, 2019-12-09:
There are different beacon request modes.
* BEACON_REPORT_MODE_PASSIVE
* BEACON_REPORT_MODE_ACTIVE
* BEACON_REPORT_MODE_TABLE

The station says it supports all the modes. Otherwise, `hostapd` would say, that the client does not support those. I'm working with `OpenWrt`, and there you can easily reproduce the issue.

First enable rrm

    ubus call hostapd.wlan0 bss_mgmt_enable '{"neighbor_report":True, "beacon_report":True}'

Then request a beacon report

     ubus call hostapd.wlan0 rrm_beacon_req '{"addr":"00:xx:xx:xx:xx:xx", "op_class":0, "channel":1,"duration":1,"mode":X,"bssid":"ff:ff:ff:ff:ff:ff", "ssid":""}'

Where X is 0 or 1.

You further, can use `hostapd_cli`-tool. I can search for the commands, if you need them.

If you follow this, you will notice, that the clients won't respond with a measurement.
If you set X to `BEACON_REPORT_MODE_TABLE`, it works.