openwrt/mt76 #336: mt7603e seems to always pick the lowest rate possible in legacy mode

Lahvuun, 2019-12-20:
LuCI shows RX Rate / TX Rate as:
54.0 Mbit/s, 20MHz
6.0 Mbit/s, 20MHz

If I set `legacy_rates` to `1`, TX Rate drops further down to 1.0 Mbit/s.

Speedtest on my phone seems to verify this, I get ~4Mbps download without legacy rates, <1Mbps with. Upload is a little better at around 10-15Mbps.

Happens with both WMM on and off.

With `htmode 'HT20'`  or `htmode 'HT40'`, the speeds are:
72.2 Mbit/s, 20MHz, MCS 7, Short GI
72.2 Mbit/s, 20MHz, MCS 7, Short GI

But only if the device connects in N mode. If it has trouble doing so and drops down to G (such as with my raspberry pi zero wireless), I still get legacy speeds.

---

Config:
```
config wifi-device 'radio0'
        option type 'mac80211'
        option hwmode '11g'
        option path 'pci0000:00/0000:00:01.0/0000:02:00.0'
        option country 'UA'
        option channel 'auto'
        option legacy_rates '0'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option key '########'
        option encryption 'psk2'
        option ssid 'OpenWrtLA2G'
        option dtim_period '2'
```

Scan from the raspberry pi zero wireless:
```
pi@raspberrypi:~ $ sudo iw dev wlan0 scan ssid OpenWrtLA2G
BSS 04:d9:f5:ec:7a:f0(on wlan0) -- associated
        last seen: 1334.225s [boottime]
        TSF: 0 usec (0d, 00:00:00)
        freq: 2472
        beacon interval: 100 TUs
        capability: ESS Privacy ShortPreamble ShortSlotTime (0x0431)
        signal: -25.00 dBm
        last seen: 0 ms ago
        SSID: OpenWrtLA2G
        Supported rates: 6.0* 9.0 12.0* 18.0 24.0* 36.0 48.0 54.0
        DS Parameter set: channel 13
        TIM: DTIM Count 0 DTIM Period 2 Bitmap Control 0x0 Bitmap[0] 0x0
        Country: UA     Environment: Indoor/Outdoor
                Channels [1 - 13] @ 20 dBm
        ERP: <no flags>
        RSN:     * Version: 1
                 * Group cipher: CCMP
                 * Pairwise ciphers: CCMP
                 * Authentication suites: PSK
                 * Capabilities: 1-PTKSA-RC 1-GTKSA-RC (0x0000)
        BSS Load:
                 * station count: 2
                 * channel utilisation: 0/255
                 * available admission capacity: 0 [*32us]
        Extended capabilities:
                 * Extended Channel Switching
                 * UTF-8 SSID
                 * Operating Mode Notification
```
With legacy rates:

```
pi@raspberrypi:~ $ sudo iw dev wlan0 scan
BSS 04:d9:f5:ec:7a:f0(on wlan0) -- associated
        last seen: 1958.006s [boottime]
        TSF: 0 usec (0d, 00:00:00)
        freq: 2472
        beacon interval: 100 TUs
        capability: ESS Privacy ShortPreamble ShortSlotTime (0x0431)
        signal: -24.00 dBm
        last seen: 0 ms ago
        SSID: OpenWrtLA2G
        Supported rates: 1.0* 2.0* 5.5* 11.0* 6.0 9.0 12.0 18.0
        DS Parameter set: channel 13
        TIM: DTIM Count 0 DTIM Period 2 Bitmap Control 0x0 Bitmap[0] 0x0
        Country: UA     Environment: Indoor/Outdoor
                Channels [1 - 13] @ 20 dBm
        ERP: <no flags>
        Extended supported rates: 24.0 36.0 48.0 54.0
        RSN:     * Version: 1
                 * Group cipher: CCMP
                 * Pairwise ciphers: CCMP
                 * Authentication suites: PSK
                 * Capabilities: 16-PTKSA-RC 1-GTKSA-RC (0x000c)
        BSS Load:
                 * station count: 2
                 * channel utilisation: 7/255
                 * available admission capacity: 0 [*32us]
        Extended capabilities:
                 * Extended Channel Switching
                 * UTF-8 SSID
                 * Operating Mode Notification
```

---

I can still set different speeds through `iw dev wlan0 set bitrates legacy-2.4 24` and it works, LuCI lists the connections at 24Mbit/s, speedtest also does better. However, I can only set the rates marked by the asterisks (6, 12, 24 and 1, 2, 5.5, 11 with legacy rates), doing something like `iw dev wlan0 set bitrates legacy-2.4 54` doesn't seem to work and I get `command failed: Invalid argument (-22)`

---

Tested this with a snapshot:

```
root@OpenWrtLA:~# cat /etc/openwrt_release
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_REVISION='r11678-9e1e432469'
DISTRIB_TARGET='ramips/mt7621'
DISTRIB_ARCH='mipsel_24kc'
DISTRIB_DESCRIPTION='OpenWrt SNAPSHOT r11678-9e1e432469'
DISTRIB_TAINTS=''
```

As well as snapshot built from source:
```
root@OpenWrtLA:~# cat /etc/openwrt_release
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_REVISION='r11697-b1e8a390ea'
DISTRIB_TARGET='ramips/mt7621'
DISTRIB_ARCH='mipsel_24kc'
DISTRIB_DESCRIPTION='OpenWrt SNAPSHOT r11697-b1e8a390ea'
DISTRIB_TAINTS='no-all'
```

With the patch that (I hope) gives me the latest driver commit:

```diff
diff --git a/package/kernel/mt76/Makefile b/package/kernel/mt76/Makefile
index 04c1761d88..1dd58e0d77 100644
--- a/package/kernel/mt76/Makefile
+++ b/package/kernel/mt76/Makefile
@@ -8,8 +8,8 @@ PKG_LICENSE_FILES:=

 PKG_SOURCE_URL:=https://github.com/openwrt/mt76
 PKG_SOURCE_PROTO:=git
-PKG_SOURCE_DATE:=2019-10-10
-PKG_SOURCE_VERSION:=d3a589586d1b1d9e1422e31f642a500ff0195f20
+PKG_SOURCE_DATE:=2019-12-14
+PKG_SOURCE_VERSION:=3ef5f3444a94e450f16260a85c2ade3237e813bc
 PKG_MIRROR_HASH:=46df12037db0d3438cb0ae97cded24a6562502d5cbd2d6d6d4e72ca807b0407e

 PKG_MAINTAINER:=Felix Fietkau <nbd@nbd.name>
```

nbd168, 2020-01-23:
Please try the latest revision of OpenWrt master or 19.07. I've fixed a critical issue in hostapd.

Lahvuun, 2020-01-23:
Still happening with OpenWrt SNAPSHOT r12104-071a637276, unfortunately.

robho, 2020-03-19:
I'm seeing this bug with Netgear AC1200 (R6220) and OpenWrt 19.07.2 and also with a master snapshot from March 18 (132ff90f1d).

The wlan client is a Squeezebox radio (802.11g device). Luci reports that the TX rate for the client is stuck at 1 Mbps (this causes audio stuttering for some high bitrate audio streams (flac)).

Manually setting transfer rates, as suggested above, helps bump the client to a higher transfer rate. Specifically, running `iw dev wlan0 set bitrates legacy-2.4 24 36 48 54` bumps the TX rate of the client to the new lowest bit rate, 24 Mbps.

robho, 2020-08-29:
FYI, I just tested a snapshot build from today (2020-08-29, 42d14ace3e) and the problem is still there.

hamster65, 2020-10-23:
this explains why I am having connection problems with all of my esp8266 and pi devices. I too get the 1 Mbit... never noticed before. I am on  OpenWrt 19.07.4 r11208-ce6496d796 on a Netgear R6220 with MediaTek MT7603E 802.11bgn. I have added support for WPA3.

robho, 2021-08-06:
I've tested 21.02-rc3 and it looks like this has been fixed. My Squeezebox radio and an ESP32 device both have more dynamic data rates which increase if needed.

Lahvuun, 2021-08-15:
Indeed, I tried 21.02.0-rc4 and this seems to have been fixed. The pi zero is now getting ~16Mbit/s on speedtest, which I guess is reasonable considering the wifi congestion in my apartment building and how small the antennae is.

Would've been nice to know which commit did it, but oh well.

Thank you!

closed 2021-08-15 by Lahvuun