openwrt/mt76 #338: Kernel panic wile connecting to WiFi

mat-alfa, 2019-12-27:
I got the following kernel panic while connecting to WiFi AP:

```
[  622.937603] CPU 0 Unable to handle kernel paging request at virtual address 00000000, epc == 803f93ac, ra == 875bcda0
[  622.948966] Oops[#1]:
[  622.951336] CPU: 0 PID: 8 Comm: kworker/u2:1 Not tainted 4.14.160 #0
[  622.957932] Workqueue: phy0 mt76u_init [mt76_usb]
[  622.962789] task: 87c26760 task.stack: 87c60000
[  622.967460] $ 0   : 00000000 80560000 00000000 00000000
[  622.972864] $ 4   : 875de144 00000002 87c26760 fffffff8
[  622.978266] $ 8   : fffffffd 0000fc00 00000400 000000c0
[  622.983669] $12   : 000d927a 09a8999a ffffffff 00000200
[  622.989072] $16   : 875de144 875dd500 00001798 87c61da8
[  622.994475] $20   : 00000000 00000000 80500000 875de148
[  622.999877] $24   : 804e5040 dd979c02                  
[  623.005281] $28   : 87c60000 87c61d88 804f0000 875bcda0
[  623.010685] Hi    : 000d927a
[  623.013659] Lo    : 09a8999a
[  623.016679] epc   : 803f93ac __mutex_lock.isra.2+0x12c/0x358
[  623.022528] ra    : 875bcda0 mt76u_queues_deinit+0x514/0x758 [mt76_usb]
[  623.029349] Status: 1000fc03	KERNEL EXL IE 
[  623.033677] Cause : 0080000c (ExcCode 03)
[  623.037815] BadVA : 00000000
[  623.040791] PrId  : 00019374 (MIPS 24Kc)
[  623.044840] Modules linked in: ath9k ath9k_common rt2800usb rt2800lib pppoe ppp_async mt76x0u mt76x0_common ath9k_hw ath rtl8187 rt2x00usb rt2x00lib pppox ppp_generic nf_conntrack_ipv6 mt76x2u mt76x2_common mt76x02_usb mt76x02_lib mt76_usb mt76 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_FLOWOFFLOAD slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_mangle iptable_filter ip_tables crc_ccitt compat ledtrig_usbport nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6
[  623.118983]  eeprom_93cx6 ehci_platform ehci_hcd gpio_button_hotplug usbcore nls_base usb_common
[  623.128074] Process kworker/u2:1 (pid: 8, threadinfo=87c60000, task=87c26760, tls=00000000)
[  623.136688] Stack : 80000280 87f2b800 000000c0 875bc0c0 0000012c 0000fc00 00000013 00000004
[  623.145325]         875de148 00000000 875de150 00000004 875de144 875dd500 00001798 87f31000
[  623.153962]         00000000 00000000 80500000 fffffffe 804f0000 875bcda0 875dd518 fffffffe
[  623.162600]         804f0000 875bcc88 87c61e30 875dd500 875dd500 875525e4 87252c00 87579024
[  623.171236]         871448a8 804f5578 875dd500 87c61e5f 875dd500 8755583c 00000000 804f5578
[  623.179874]         ...
[  623.182403] Call Trace:
[  623.184937] [<803f93ac>] __mutex_lock.isra.2+0x12c/0x358
[  623.190431] [<875bcda0>] mt76u_queues_deinit+0x514/0x758 [mt76_usb]
[  623.196905] Code: afa20024  ac930008  afb70020 <ac530000> 8f830000  8c820004  14530006  afa30028  c0820000 
[  623.206988] 
[  623.209418] ---[ end trace 2001b30435e52931 ]---
[  623.216405] Kernel panic - not syncing: Fatal exception
```

The testing hardware is AWUS036ACM(MT7612U) with R36A(ar71xx, mips 24kc), and the firmware I tested is openwrt-19.07 snapshot with following mt76 driver packages:
```
kmod-mt76-core - 4.14.160+2019-08-22-2a0edbb4-1
kmod-mt76-usb - 4.14.160+2019-08-22-2a0edbb4-1
kmod-mt76x0-common - 4.14.160+2019-08-22-2a0edbb4-1
kmod-mt76x02-common - 4.14.160+2019-08-22-2a0edbb4-1
kmod-mt76x02-usb - 4.14.160+2019-08-22-2a0edbb4-1
kmod-mt76x0u - 4.14.160+2019-08-22-2a0edbb4-1
kmod-mt76x2-common - 4.14.160+2019-08-22-2a0edbb4-1
kmod-mt76x2u - 4.14.160+2019-08-22-2a0edbb4-1
```

The problem occurs after wpa_supplicant authentication, at the point between `wpa_cli status` showing COMPLETE and starting to run DHCP.

LorenzoBianconi, 2019-12-28:
> I got the following kernel panic while connecting to WiFi AP:
>
> [  622.937603] CPU 0 Unable to handle kernel paging request at virtual address 00000000, epc == 803f93ac, ra == 875bcda0
> [  622.948966] [...]

You are running a quite old driver, could you please try the latest one?

Regards,
Lorenzo

> The problem occurs after wpa_supplicant authentication, at the point between wpa_cli status showing COMPLETE and starting to run DHCP.

mat-alfa, 2019-12-28:
Yes, thanks!
I just tried the latest version(rev=4cb1195), but failed to build driver. I got the following compile error logs: 

```
make  -C "/tmp/build/28bfe605/openwrt/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/linux-4.14.160" HOSTCFLAGS="-O2 -I/tmp/build/28bfe605/openwrt/staging_dir/host/include -I/tmp/build/28bfe605/openwrt/staging_dir/hostpkg/include -I/tmp/build/28bfe605/openwrt/staging_dir/target-mips_24kc_musl/host/include -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="mips-openwrt-linux-musl-" ARCH="mips" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Sat Dec 28 14:24:51 2019" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/tmp/build/28bfe605/openwrt/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls= KERNELRELEASE=4.14.160 CONFIG_MT76_USB=m CONFIG_MT76x02_LIB=m CONFIG_MT76x02_USB=m CONFIG_MT76x0_COMMON=m CONFIG_MT76x0U=m CONFIG_MT76x2_COMMON=m CONFIG_MT76x2U=m SUBDIRS="/tmp/build/28bfe605/openwrt/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/mt76-2019-08-22-4cb1195a" NOSTDINC_FLAGS="-I/tmp/build/28bfe605/openwrt/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/mt76-2019-08-22-4cb1195a -I/tmp/build/28bfe605/openwrt/staging_dir/target-mips_24kc_musl/usr/include/mac80211-backport/uapi -I/tmp/build/28bfe605/openwrt/staging_dir/target-mips_24kc_musl/usr/include/mac80211-backport -I/tmp/build/28bfe605/openwrt/staging_dir/target-mips_24kc_musl/usr/include/mac80211/uapi -I/tmp/build/28bfe605/openwrt/staging_dir/target-mips_24kc_musl/usr/include/mac80211 -include backport/autoconf.h -include backport/backport.h -DCONFIG_MAC80211_MESH" modules
make[4]: Entering directory '/tmp/build/28bfe605/openwrt/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/linux-4.14.160'
  CC [M]  /tmp/build/28bfe605/openwrt/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/mt76-2019-08-22-4cb1195a/mmio.o
  CC [M]  /tmp/build/28bfe605/openwrt/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/mt76-2019-08-22-4cb1195a/util.o
  CC [M]  /tmp/build/28bfe605/openwrt/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/mt76-2019-08-22-4cb1195a/trace.o
  CC [M]  /tmp/build/28bfe605/openwrt/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/mt76-2019-08-22-4cb1195a/dma.o
  CC [M]  /tmp/build/28bfe605/openwrt/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/mt76-2019-08-22-4cb1195a/mac80211.o
/tmp/build/28bfe605/openwrt/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/mt76-2019-08-22-4cb1195a/mac80211.c: In function 'mt76_get_survey':
/tmp/build/28bfe605/openwrt/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/mt76-2019-08-22-4cb1195a/mac80211.c:627:22: error: 'SURVEY_INFO_TIME_BSS_RX' undeclared (first use in this function); did you mean 'SURVEY_INFO_TIME_RX'?
    survey->filled |= SURVEY_INFO_TIME_BSS_RX;
                      ^~~~~~~~~~~~~~~~~~~~~~~
                      SURVEY_INFO_TIME_RX
/tmp/build/28bfe605/openwrt/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/mt76-2019-08-22-4cb1195a/mac80211.c:627:22: note: each undeclared identifier is reported only once for each function it appears in
/tmp/build/28bfe605/openwrt/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/mt76-2019-08-22-4cb1195a/mac80211.c:636:10: error: 'struct survey_info' has no member named 'time_bss_rx'; did you mean 'time_busy'?
  survey->time_bss_rx = div_u64(state->cc_bss_rx, 1000);
          ^~~~~~~~~~~
          time_busy
make[5]: *** [scripts/Makefile.build:327: /tmp/build/28bfe605/openwrt/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/mt76-2019-08-22-4cb1195a/mac80211.o] Error 1
make[4]: *** [Makefile:1542: _module_/tmp/build/28bfe605/openwrt/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/mt76-2019-08-22-4cb1195a] Error 2
make[4]: Leaving directory '/tmp/build/28bfe605/openwrt/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/linux-4.14.160'
make[3]: *** [Makefile:261: /tmp/build/28bfe605/openwrt/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/mt76-2019-08-22-4cb1195a/.built] Error 2
make[3]: Leaving directory '/tmp/build/28bfe605/openwrt/package/kernel/mt76'
time: package/kernel/mt76/compile#2.05#0.42#4.12
make[2]: *** [package/Makefile:113: package/kernel/mt76/compile] Error 2
make[2]: Leaving directory '/tmp/build/28bfe605/openwrt'
make[1]: *** [package/Makefile:107: /tmp/build/28bfe605/openwrt/staging_dir/target-mips_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/tmp/build/28bfe605/openwrt'
make: *** [/tmp/build/28bfe605/openwrt/include/toplevel.mk:227: world] Error 2
>>> ret: 1
```

LorenzoBianconi, 2019-12-28:
> Yes, thanks!
> I just tried the latest version(rev=4cb1195), but failed to build driver.
> I got the following compile error logs:
>
> make  -C [...]

You need to update even the mac80211 pkg

mat-alfa, 2019-12-28:
I changed to openwrt's master branch and build mt76 driver with latest version(rev=4cb1195a)
```
kmod-mt76-core - 4.14.156+2019-12-29-4cb1195a-1
kmod-mt76-usb - 4.14.156+2019-12-29-4cb1195a-1
kmod-mt76x0-common - 4.14.156+2019-12-29-4cb1195a-1
kmod-mt76x02-common - 4.14.156+2019-12-29-4cb1195a-1
kmod-mt76x02-usb - 4.14.156+2019-12-29-4cb1195a-1
kmod-mt76x0u - 4.14.156+2019-12-29-4cb1195a-1
kmod-mt76x2-common - 4.14.156+2019-12-29-4cb1195a-1
kmod-mt76x2u - 4.14.156+2019-12-29-4cb1195a-1
```

The latest error log I got is:
```
[  521.637682] wlan2: authenticate with c8:d3:a3:59:cc:6e
[  523.175618] wlan2: send auth to c8:d3:a3:59:cc:6e (try 1/3)
[  523.189030] wlan2: authenticated
[  523.212581] wlan2: associate with c8:d3:a3:59:cc:6e (try 1/3)
[  523.244119] wlan2: RX AssocResp from c8:d3:a3:59:cc:6e (capab=0x431 status=0 aid=6)
[  523.287017] wlan2: associated
[  523.326811] IPv6: ADDRCONF(NETDEV_CHANGE): wlan2: link becomes ready
[  524.824472] CPU 0 Unable to handle kernel paging request at virtual address 00000000, epc == 8042f3d0, ra == 87578d50
[  524.835754] Oops[#1]:
[  524.838120] CPU: 0 PID: 3327 Comm: kworker/u2:3 Not tainted 4.14.156 #0
[  524.844984] Workqueue: mt76u mt76u_deinit [mt76_usb]
[  524.850118] task: 87dbd240 task.stack: 86f74000
[  524.854792] $ 0   : 00000000 805a0000 00000000 00000000
[  524.860194] $ 4   : 8764aad0 00000002 87dbd240 fffffff8
[  524.865596] $ 8   : fffffffd 0000fc00 80533a00 00000000
[  524.871000] $12   : 0014d9a9 de6169be 001c9d4a 9d8cb525
[  524.876402] $16   : 8764aad0 87649d20 00001798 87fb5c00
[  524.881805] $20   : 00000000 86f75da8 80540000 fffffffe
[  524.887208] $24   : 80523040 4024db72                  
[  524.892610] $28   : 86f74000 86f75d88 8764aad4 87578d50
[  524.898015] Hi    : 000000f7
[  524.900989] Lo    : 33333371
[  524.903981] epc   : 8042f3d0 __mutex_lock.isra.2+0x134/0x378
[  524.909822] ra    : 87578d50 mt76u_deinit+0x418/0xa6c [mt76_usb]
[  524.916015] Status: 1000fc03	KERNEL EXL IE 
[  524.920344] Cause : 0080000c (ExcCode 03)
[  524.924482] BadVA : 00000000
[  524.927450] PrId  : 00019374 (MIPS 24Kc)
[  524.931498] Modules linked in: ath9k ath9k_common rt2800usb rt2800lib pppoe ppp_async mt76x0u mt76x0_common ath9k_hw ath rtl8812au rtl8187 rt2x00usb rt2x00lib pppox ppp_generic nf_conntrack_ipv6 mt76x2u mt76x2_common mt76x02_usb mt76x02_lib mt76_usb mt76 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_FLOWOFFLOAD slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_mangle iptable_filter ip_tables crc_ccitt compat ledtrig_usbport nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT
[  525.004369]  x_tables nf_reject_ipv6 eeprom_93cx6 ehci_platform ehci_hcd gpio_button_hotplug usbcore nls_base usb_common
[  525.015610] Process kworker/u2:3 (pid: 3327, threadinfo=86f74000, task=87dbd240, tls=00000000)
[  525.024492] Stack : 87649d20 00000004 00001718 87fb5c00 00000000 00000000 80540000 fffffffe
[  525.033131]         8764aad4 00000000 805a0000 87c29780 8764aad0 87649d20 00001798 87fb5c00
[  525.041767]         00000000 00000000 80540000 fffffffe 80530000 87578d50 00000001 87c2a370
[  525.050404]         805339d8 87c2a370 86f75e30 87649d20 86f75e5f 875127bc 8764903c 805339d8
[  525.059042]         87dbd270 87dbd270 8764ab00 87649d20 86f75e5f 87515c30 00000000 805339a8
[  525.067679]         ...
[  525.070208] Call Trace:
[  525.072740] [<8042f3d0>] __mutex_lock.isra.2+0x134/0x378
[  525.078235] [<87578d50>] mt76u_deinit+0x418/0xa6c [mt76_usb]
[  525.084104] Code: afa20024  ac950008  afbe0020 <ac550000> 8f830000  8c820004  1055000b  afa30028  8f820000 
[  525.094183] 
[  525.097555] ---[ end trace 73ae89ab0ca1b3c9 ]---
[  525.104713] Kernel panic - not syncing: Fatal exception
[  525.111111] Rebooting in 3 seconds..
```

LorenzoBianconi, 2019-12-29:
> I changaster branch and build mt76 driver with latest version(rev=4cb1195a)
>
> kmod-mt76-core - 4.14.156+2019-12-29-4cb1195a-1
> kmod-mt76-usb - 4.14.156+2019-12-29-4cb1195a-1
> kmod-mt76x0-common - 4.14.156+2019-12-29-4cb1195a-1
> kmod-mt76x02-common - 4.14.156+2019-12-29-4cb1195a-1
> kmod-mt76x02-usb - 4.14.156+2019-12-29-4cb1195a-1
> kmod-mt76x0u - 4.14.156+2019-12-29-4cb1195a-1
> kmod-mt76x2-common - 4.14.156+2019-12-29-4cb1195a-1
> kmod-mt76x2u - 4.14.156+2019-12-29-4cb1195a-1
>
> The latest error log I got is:
>
> [  521.637682] wlan2: authenticate with c8:d3:a3:59:cc:6e
> [  523.175618] wlan2: send auth to c8:d3:a3:59:cc:6e (try 1/3)
> [  523.189030] wlan2: authenticated
> [  523.212581] wlan2: associate with c8:d3:a3:59:cc:6e (try 1/3)
> [  523.244119] wlan2: RX AssocResp from c8:d3:a3:59:cc:6e (capab=0x431 status=0 aid=6)
> [  523.287017] wlan2: associated
> [  523.326811] IPv6: ADDRCONF(NETDEV_CHANGE): wlan2: link becomes ready
> [  524.824472] CPU 0 Unable to handle kernel paging request at virtual address 00000000, epc == 8042f3d0, ra == 87578d50
> [  524.835754] Oops[#1]:
> [  524.838120] CPU: 0 PID: 3327 Comm: kworker/u2:3 Not tainted 4.14.156 #0
> [  524.844984] Workqueue: mt76u mt76u_deinit [mt76_usb]
> [  524.850118] task: 87dbd240 task.stack: 86f74000
> [  524.854792] $ 0   : 00000000 805a0000 00000000 00000000
> [  524.860194] $ 4   : 8764aad0 00000002 87dbd240 fffffff8
> [  524.865596] $ 8   : fffffffd 0000fc00 80533a00 00000000
> [  524.871000] $12   : 0014d9a9 de6169be 001c9d4a 9d8cb525
> [  524.876402] $16   : 8764aad0 87649d20 00001798 87fb5c00
> [  524.881805] $20   : 00000000 86f75da8 80540000 fffffffe
> [  524.887208] $24   : 80523040 4024db72
> [  524.892610] $28   : 86f74000 86f75d88 8764aad4 87578d50
> [  524.898015] Hi    : 000000f7
> [  524.900989] Lo    : 33333371
> [  524.903981] epc   : 8042f3d0 __mutex_lock.isra.2+0x134/0x378
> [  524.909822] ra    : 87578d50 mt76u_deinit+0x418/0xa6c [mt76_usb]
> [  524.916015] Status: 1000fc03	KERNEL EXL IE
> [  524.920344] Cause : 0080000c (ExcCode 03)
> [  524.924482] BadVA : 00000000
> [  524.927450] PrId  : 00019374 (MIPS 24Kc)
> [  524.931498] Modules linked in: ath9k ath9k_common rt2800usb rt2800lib pppoe ppp_async mt76x0u mt76x0_common ath9k_hw ath rtl8812au rtl8187 rt2x00usb rt2x00lib pppox ppp_generic nf_conntrack_ipv6 mt76x2u mt76x2_common mt76x02_usb mt76x02_lib mt76_usb mt76 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_FLOWOFFLOAD slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_mangle iptable_filter ip_tables crc_ccitt compat ledtrig_usbport nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT
> [  525.004369]  x_tables nf_reject_ipv6 eeprom_93cx6 ehci_platform ehci_hcd gpio_button_hotplug usbcore nls_base usb_common
> [  525.015610] Process kworker/u2:3 (pid: 3327, threadinfo=86f74000, task=87dbd240, tls=00000000)
> [  525.024492] Stack : 87649d20 00000004 00001718 87fb5c00 00000000 00000000 80540000 fffffffe
> [  525.033131]         8764aad4 00000000 805a0000 87c29780 8764aad0 87649d20 00001798 87fb5c00
> [  525.041767]         00000000 00000000 80540000 fffffffe 80530000 87578d50 00000001 87c2a370
> [  525.050404]         805339d8 87c2a370 86f75e30 87649d20 86f75e5f 875127bc 8764903c 805339d8
> [  525.059042]         87dbd270 87dbd270 8764ab00 87649d20 86f75e5f 87515c30 00000000 805339a8
> [  525.067679]         ...
> [  525.070208] Call Trace:
> [  525.072740] [<8042f3d0>] __mutex_lock.isra.2+0x134/0x378
> [  525.078235] [<87578d50>] mt76u_deinit+0x418/0xa6c [mt76_usb]
> [  525.084104] Code: afa20024  ac950008  afbe0020 <ac550000> 8f830000  8c820004  1055000b  afa30028  8f820000
> [  525.094183]
> [  525.097555] ---[ end trace 73ae89ab0ca1b3c9 ]---
> [  525.104713] Kernel panic - not syncing: Fatal exception
> [  525.111111] Rebooting in 3 seconds..

Are you able to bisect the tree to identify the offending commit? I am afk
now

pepe2k, 2019-12-31:
@LorenzoBianconi I can confirm the problem on OpenWrt 19.07 snapshot (mt76 @ 2a0edbb447). Updating mt76 to latest version from openwrt-19.07 branch (f97c33eeb2) solves it.

Cheers,
Piotr

mat-alfa, 2020-01-01:
Yes.

git bisect tells 6493e23 is the first bad commit.

My test results are:
 * faf5e6f , good , the merge base of `master` and `openwrt-19.07` branches
 * efcbee1 , bad
 * 1c8d69e , bad
 * d246beb , bad
 * ea46624 , bad
 * af83ee8 , good
 * 6493e23 , bad

Beside, the f97c33e (current head commit or openwrt-19.07) has build error as below: 
```
  CC [M]  /tmp/build/28bfe605/openwrt/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/mt76-2019-12-29-f97c33e/eeprom.o
/tmp/build/28bfe605/openwrt/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/mt76-2019-12-29-f97c33e/eeprom.c: In function 'mt76_eeprom_override':
/tmp/build/28bfe605/openwrt/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/mt76-2019-12-29-f97c33e/eeprom.c:100:1: error: label 'out' defined but not used [-Werror=unused-label]
 out:
 ^~~
cc1: all warnings being treated as errors
make[4]: *** [scripts/Makefile.build:327: /tmp/build/28bfe605/openwrt/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/mt76-2019-12-29-f97c33e/eeprom.o] Error 1
make[3]: *** [Makefile:1545: _module_/tmp/build/28bfe605/openwrt/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/mt76-2019-12-29-f97c33e] Error 2
make[3]: Leaving directory '/tmp/build/28bfe605/openwrt/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/linux-4.14.156'
make[2]: *** [Makefile:261: /tmp/build/28bfe605/openwrt/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/mt76-2019-12-29-f97c33e/.built] Error 2
make[2]: Leaving directory '/tmp/build/28bfe605/openwrt/package/kernel/mt76'
time: package/kernel/mt76/compile#0.79#0.24#2.35
make[1]: *** [package/Makefile:113: package/kernel/mt76/compile] Error 2
make[1]: Leaving directory '/tmp/build/28bfe605/openwrt'
make: *** [/tmp/build/28bfe605/openwrt/include/toplevel.mk:227: package/mt76/compile] Error 2
```

LorenzoBianconi, 2020-01-02:
> Yes.
> 
> Based on information above, I tried `git bisect` and found [6493e23](https://github.com/openwrt/mt76/commit/6493e234b40d36c7e6c1f5cdd120899c8973d3ef) is the first bad commit.
> 
> My test results are:
> 
> * [faf5e6f](https://github.com/openwrt/mt76/commit/faf5e6fedd7e8831ad811f3074f3bb6ef8cbcbee) , good , the merge base of `master` and `openwrt-19.07` branches
> * [efcbee1](https://github.com/openwrt/mt76/commit/efcbee1d8bfcde9df0cc79aa3558166b582a208b) , bad
> * [1c8d69e](https://github.com/openwrt/mt76/commit/1c8d69e7d666fe09b6651dfcc7fb5dbe6ad1ed8f) , bad
> * [d246beb](https://github.com/openwrt/mt76/commit/d246beb2335088a1e3969070593db09a14ed3973) , bad
> * [ea46624](https://github.com/openwrt/mt76/commit/ea46624355ec77667cf7b291a0e9cc1ce49846cb) , bad
> * [af83ee8](https://github.com/openwrt/mt76/commit/af83ee8de92875fdad3861e188482b8ca706c205) , good
> * [6493e23](https://github.com/openwrt/mt76/commit/6493e234b40d36c7e6c1f5cdd120899c8973d3ef) , bad
> 
> Beside, the [f97c33e](https://github.com/openwrt/mt76/commit/f97c33eeb2e376a5da8bc9f3d1fabe5153ce6274) (current head commit or openwrt-19.07) has build error as below:
> 

This 'not monotonous' behavior seems weird to me. Moreover I tested last nbd [wireless tree](https://github.com/nbd168/wireless) and it works fine.
@pepe2k : is mt76 master branch working for you? 

> ```
>   CC [M]  /tmp/build/28bfe605/openwrt/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/mt76-2019-12-29-f97c33e/eeprom.o [...]

This should be fixed in latest mt76 tree by this [commit](https://github.com/openwrt/mt76/commit/1c8e76f4de3e7c41ca54732ddb4a0f6bdce53111)

pepe2k, 2020-01-03:
@LorenzoBianconi 

> This 'not monotonous' behavior seems weird to me. Moreover I tested last nbd [wireless tree](https://github.com/nbd168/wireless) and it works fine.
> @pepe2k : is mt76 master branch working for you?

Unfortunately, no. Just tried current OpenWrt master with mt76 @ 38f4c57 on R36A (ar71xx, QCA9531, 4.14.160) with AWUS036ACM (MT7612U):

```
[  371.544901] CPU 0 Unable to handle kernel paging request at virtual address 00000000, epc == 8042fbb0, ra == 830d4d50
[  371.556157] Oops[#1]:
[  371.558521] CPU: 0 PID: 11 Comm: kworker/u2:2 Not tainted 4.14.160 #0
[  371.565204] Workqueue: mt76u mt76u_deinit [mt76_usb]
[  371.570331] task: 83823ac0 task.stack: 8386c000
[  371.575004] $ 0   : 00000000 80590000 00000000 00000000
[  371.580407] $ 4   : 82edaad0 00000002 83823ac0 fffffff8
[  371.585810] $ 8   : fffffffd 0000fc00 8052da00 00000000
[  371.591212] $12   : 000b2285 ae53a1a9 00108845 89da44c4
[  371.596615] $16   : 82edaad0 82ed9d20 00001798 832edf00
[  371.602019] $20   : 00000000 8386dda8 80530000 fffffffe
[  371.607421] $24   : 8051d040 76274d1b                  
[  371.612824] $28   : 8386c000 8386dd88 82edaad4 830d4d50
[  371.618228] Hi    : 000000f7
[  371.621203] Lo    : 33333371
[  371.624196] epc   : 8042fbb0 __mutex_lock.isra.2+0x134/0x378
[  371.630043] ra    : 830d4d50 mt76u_deinit+0x418/0xa6c [mt76_usb]
[  371.636237] Status: 1000fc03KERNEL EXL IE 
[  371.640557] Cause : 0080000c (ExcCode 03)
[  371.644696] BadVA : 00000000
[  371.647671] PrId  : 00019374 (MIPS 24Kc)
[  371.651720] Modules linked in: ath9k ath9k_common pppoe ppp_async mt76x0u mt76x0_common ath9k_hw ath pppox ppp_generic nf_conntrack_ipv6 mt76x2u mt76x2_common mt76x02_usb mt76x02_lib mt7601u mt76_usb mt76 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_FLOWOFFLOAD slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_mangle iptable_filter ip_tables crc_ccitt compat ledtrig_usbport nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 ehci_platform ehci_hcd gpio_button_hotplug
[  371.726123]  usbcore nls_base usb_common
[  371.730180] Process kworker/u2:2 (pid: 11, threadinfo=8386c000, task=83823ac0, tls=00000000)
[  371.738884] Stack : 833d009c 83210b4c 82ed8bc0 8386ddac 000001ff 8008ac50 8386ddac 83b74b00
[  371.747519]         82edaad4 00000000 83b74b48 83210c38 82edaad0 82ed9d20 00001798 832edf00
[  371.756157]         00000000 00000000 80530000 fffffffe 80530000 830d4d50 00000040 8389d850
[  371.764794]         8052d9d8 8389d850 8386de30 82ed9d20 8386de5f 831c27bc 833d48ec 8052d9d8
[  371.773431]         83823ac0 83823af0 82edab00 82ed9d20 8386de5f 831c5c30 00000000 8052d9a8
[  371.782069]         ...
[  371.784598] Call Trace:
[  371.787130] [<8042fbb0>] __mutex_lock.isra.2+0x134/0x378
[  371.792622] [<830d4d50>] mt76u_deinit+0x418/0xa6c [mt76_usb]
[  371.798463] Code: afa20024  ac950008  afbe0020 <ac550000> 8f830000  8c820004  1055000b  afa30028  8f820000 
[  371.808546] 
[  371.810920] ---[ end trace c62f0601f6730eb0 ]---
[  371.818101] Kernel panic - not syncing: Fatal exception
[  371.824420] Rebooting in 3 seconds..
```

I will double check that with different card/on different target (with 4.19 kernel/DTS) and will let you know.

> This should be fixed in latest mt76 tree by this [commit](https://github.com/openwrt/mt76/commit/1c8e76f4de3e7c41ca54732ddb4a0f6bdce53111)

This has to be also backported to `openwrt-19.07` branch.

Cheers,
Piotr

pepe2k, 2020-01-03:
@LorenzoBianconi 

Same problem on ath79 with 4.19.91 (just after association with AP):

```
[  155.450084] wlan0: authenticate with 44:xx:xx:xx:xx:04
[  157.170252] wlan0: send auth to 44:xx:xx:xx:xx:04 (try 1/3)
[  157.179343] wlan0: authenticated
[  157.185276] wlan0: associate with 44:xx:xx:xx:xx:04 (try 1/3)
[  157.193415] wlan0: RX AssocResp from 44:xx:xx:xx:xx:04 (capab=0x11 status=0 aid=1)
[  157.207612] wlan0: associated
[  157.231811] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[  158.847038] CPU 0 Unable to handle kernel paging request at virtual address 00000000, epc == 8048a388, ra == 809c8d50
[  158.858178] Oops[#1]:
[  158.860538] CPU: 0 PID: 66 Comm: kworker/u2:2 Not tainted 4.19.91 #0
[  158.867143] Workqueue: mt76u mt76u_deinit [mt76_usb]
[  158.872264] $ 0   : 00000000 80d20000 00000000 00000000
[  158.877665] $ 4   : 808f2c10 00000002 839a0bc0 fffffff8
[  158.883060] $ 8   : fffffffd 838d043c 00000000 80808080
[  158.888462] $12   : f47f7f7f fefefeff 00000000 00010001
[  158.893855] $16   : 808f2c10 808f1d60 00001798 82e91300
[  158.899249] $20   : 839abda8 00000000 805a0000 fffffffe
[  158.904643] $24   : 80591040 8cd0b4c1
[  158.910038] $28   : 839aa000 839abd88 808f2c14 809c8d50
[  158.915442] Hi    : 000000f7
[  158.918416] Lo    : 33333371
[  158.921403] epc   : 8048a388 __mutex_lock.isra.6+0x134/0x350
[  158.927258] ra    : 809c8d50 mt76u_deinit+0x418/0xa6c [mt76_usb]
[  158.933451] Status: 1000dc03 KERNEL EXL IE
[  158.937771] Cause : 0080000c (ExcCode 03)
[  158.941909] BadVA : 00000000
[  158.944876] PrId  : 00019374 (MIPS 24Kc)
[  158.948924] Modules linked in: ath9k ath9k_common pppoe ppp_async mt76x0u mt76x0_common iptable_nat ipt_MASQUERADE ath9k_hw ath xt_state xt_nat xt_conntrack xt_REDIRECT xt_FLOWOFFLOAD pppox ppp_generic nf_nat_ipv4 nf_nat nf_flow_table_hw nf_flow_table nf_conntrack_rtcache nf_conntrack mt76x2u mt76x2_common mt76x02_usb mt76x02_lib mt7601u mt76_usb mt76 mac80211 ipt_REJECT cfg80211 xt_time xt_tcpudp xt_multiport xt_mark xt_mac xt_limit xt_comment xt_TCPMSS xt_LOG slhc nf_reject_ipv4 nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 iptable_mangle iptable_filter ip_tables crc_ccitt compat nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 ehci_platform ehci_hcd gpio_button_hotplug usbcore nls_base usb_common
[  159.017644] Process kworker/u2:2 (pid: 66, threadinfo=f458c730, task=73e4c2bf, tls=00000000)
[  159.026347] Stack : 8298f09c 80a10ab4 808f0be0 839abdac 000001ff 80089184 839abdac 82b00380
[  159.034975]         808f2c14 00000000 82b003c0 80a10ba0 808f2c10 808f1d60 00001798 82e91300
[  159.043604]         00000000 00000000 805a0000 fffffffe 80590000 809c8d50 00000040 8382b510
[  159.052232]         805987a8 8382b4e0 839abe30 808f1d60 839abe5f 809227ac 80977024 805987a8
[  159.060861]         839a0bc0 839a0bf0 808f2c40 808f1d60 839abe5f 80925c10 00000000 80598778
[  159.069489]         ...
[  159.072019] Call Trace:
[  159.074551] [<8048a388>] __mutex_lock.isra.6+0x134/0x350
[  159.080058] [<809c8d50>] mt76u_deinit+0x418/0xa6c [mt76_usb]
[  159.085937] Code: afa20024  ac940008  afbe0020 <ac540000> 8c820004  00a08825  14540003  00808025  0c12261a
[  159.096011]
[  159.098365] ---[ end trace 50bdadf18f1ee9bc ]---
[  159.105204] Kernel panic - not syncing: Fatal exception
[  159.111454] Rebooting in 3 seconds..
```

MT7610U on the same platform also crashes:

```
[  187.685274] CPU 0 Unable to handle kernel paging request at virtual address 00000000, epc == 8048a388, ra == 809c8d50
[  187.696339] Oops[#1]:
[  187.698696] CPU: 0 PID: 66 Comm: kworker/u2:2 Not tainted 4.19.91 #0
[  187.705327] Workqueue: phy1 mt76x02_mac_work [mt76x02_lib]
[  187.710988] $ 0   : 00000000 00000001 00000000 00000000
[  187.716390] $ 4   : 829a2c10 00000002 839a0bc0 fffffff8
[  187.721784] $ 8   : fffffffd 838d043c 00000080 80808080
[  187.727178] $12   : 7f7f7f7f fefefeff 00000000 d8a390b7
[  187.732573] $16   : 829a2c10 829a1d60 00001134 829a1d60
[  187.737975] $20   : 839abdb0 829a1e7c 805a0000 fffffffe
[  187.743369] $24   : 80591040 c6e84b89
[  187.748763] $28   : 839aa000 839abd90 829a2c14 809c8d50
[  187.754158] Hi    : 00137e32
[  187.757132] Lo    : d96e1bd8
[  187.760124] epc   : 8048a388 __mutex_lock.isra.6+0x134/0x350
[  187.765973] ra    : 809c8d50 mt76u_deinit+0x418/0xa6c [mt76_usb]
[  187.772167] Status: 1000dc03 KERNEL EXL IE
[  187.776487] Cause : 0080000c (ExcCode 03)
[  187.780625] BadVA : 00000000
[  187.783592] PrId  : 00019374 (MIPS 24Kc)
[  187.787640] Modules linked in: ath9k ath9k_common pppoe ppp_async mt76x0u mt76x0_common iptable_nat ipt_MASQUERADE ath9k_hw ath xt_state xt_nat xt_conntrack xt_REDIRECT xt_FLOWOFFLOAD pppox ppp_generic nf_nat_ipv4 nf_nat nf_flow_table_hw nf_flow_table nf_conntrack_rtcache nf_conntrack mt76x2u mt76x2_common mt76x02_usb mt76x02_lib mt7601u mt76_usb mt76 mac80211 ipt_REJECT cfg80211 xt_time xt_tcpudp xt_multiport xt_mark xt_mac xt_limit xt_comment xt_TCPMSS xt_LOG slhc nf_reject_ipv4 nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 iptable_mangle iptable_filter ip_tables crc_ccitt compat nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 ehci_platform ehci_hcd gpio_button_hotplug usbcore nls_base usb_common
[  187.856360] Process kworker/u2:2 (pid: 66, threadinfo=6c485b03, task=3f34a5d3, tls=00000000)
[  187.865064] Stack : 80591040 c6e84b89 805987a8 8382b4e0 839aa000 839abdc8 828f9540 80488584
[... cut ...]
[  187.910743] Call Trace:
[  187.913277] [<8048a388>] __mutex_lock.isra.6+0x134/0x350
[  187.918785] [<809c8d50>] mt76u_deinit+0x418/0xa6c [mt76_usb]
[  187.924665] Code: afa20024  ac940008  afbe0020 <ac540000> 8c820004  00a08825  14540003  00808025  0c12261a
[  187.934737]
[  187.936379] ---[ end trace a773edb0e5203c9f ]---
[  187.943006] Kernel panic - not syncing: Fatal exception
[  187.949239] Rebooting in 3 seconds..
```

I will try to find out offending commit during the weekend.

Cheers,
Piotr

pepe2k, 2020-01-04:
More news on this one, it seems the problem is platform (MIPS?) and/or device (card, host?) related. On RPi 3B+ with current OpenWrt master it works.

LorenzoBianconi, 2020-01-04:
>
>
> More news on this one, it seems the problem is platform (MIPS?) and/or
> device (card, host?) related. On RPi 3B+ with current OpenWrt master it
> works.
I tried with last OpenWrt trunk on mt7622 device (ARM) + mt76x0u and it
works fine for me

Regards,
Lorenzo

mat-alfa, 2020-01-04:
@LorenzoBianconi Could we donate hardware if you need MIPS related devices to test?

Sincerely,
Mat

wynn1212, 2020-01-08:
I have the same problem but I don't know if it's off-topic here.
I was intending to use my USB Wifi dongle (ASUS USB-AC51) as a client to connect my android hotspot to share my network.
But When I join my wifi and click save & apply. The configuration saved and a few seconds later. My router crashed and rebooted.
And every time it rebooted. It crashes again when my USB wifi dongle is plugged in and trying to connect my hotspot. The solution to stop crashing is unplugging my USB wifi dongle.
But if I eventually wait long enough without unplugging my wifi dongle. It'll successfully boot and connected to my hotspot. But the problem is it is unstable and will crash randomly(one scenario is another device is connected to this router via router's internal wifi)

My devices: D-Team Newifi D2(mt7621).
OpenWRT Version: 19.07-rc2 and 19.07.0
Installed module: kmod-mt76x0u_4.14.162+2020-01-04-8a785679-1_mipsel_24kc
USB Wifi dongle: ASUS USB-AC51

output via `/sys/kernel/debug/crashlog` from OpenWRT 19.07.0
```
Time: 1578494335.996120
Modules:        pppoe@8ea30000+23e0     ppp_async@8f70e000+1b90 mt76x0u@8ea2a000+1180   mt76x0_common@8ea18000+70a0     rndis_host@8ea06000+1320   pppox@8ec79000+4b0       ppp_generic@8ed90000+5970       nf_conntrack_ipv6@8f780000+1590 mt76x2e@8ee60000+26c0   mt76x2_common@8ee4c000+2a40     mt76x02_usb@8ee76000+16a0   mt76x02_lib@8ece0000+9640       mt7603e@8f7a0000+90a0   mt76_usb@8eebc000+22b0  mt76@8f798000+7920      mac80211@8e980000+6c980     iptable_nat@8ee35000+2c0        ipt_REJECT@8ee31000+3c0 ipt_MASQUERADE@8ee93000+2a0     cfg80211@8f7c0000+369a0 cdc_ether@8ee7a000+12e0 xt_time@8f717000+6e0        xt_tcpudp@8f700000+700  xt_state@8e925000+2c0   xt_nat@8e8ed000+600     xt_multiport@8e933000+500       xt_mark@8e8ef000+2a0xt_mac@8e8ee000+260     xt_limit@8f729000+4a0   xt_conntrack@8f6be000+900       xt_comment@8f6d3000+1e0 xt_TCPMSS@8f5fc000+ae0  xt_REDIRECT@8f709000+2c0    xt_LOG@8ee8f000+300     xt_FLOWOFFLOAD@8e95a000+bb0     xt_CT@8f795000+a60      usbnet@8f790000+4490    ums_usbat@8e918000+2080 ums_sddr55@8e8d4000+1740    ums_sddr09@8ee94000+23c0        ums_karma@8f779000+6a0  ums_jumpshot@8f77d000+fc0       ums_isd200@8ee7e000+1710        ums_freecom@8f77f000+880    ums_datafab@8ee74000+1320       ums_cypress@8e97f000+a20        ums_alauda@8ee68000+22a0        slhc@8ee64000+13c0      nf_reject_ipv4@8e8fb000+840 nf_nat_redirect@8e974000+580    nf_nat_masquerade_ipv4@8e97d000+650     nf_conntrack_ipv4@8ee5a000+1420 nf_nat_ipv4@8e91f000+f30   nf_nat@8ee48000+37d0     nf_log_ipv4@8f756000+d60        nf_flow_table_hw@8ee32000+8d0   nf_flow_table@8ee40000+3350     nf_defrag_ipv6@8eeba000+12e0nf_defrag_ipv4@8ee36000+440     nf_conntrack_rtcache@8ee39000+aa0       nf_conntrack@8ee20000+10550     iptable_mangle@8ee14000+380     iptable_filter@8ee16000+280 ip_tables@8eeac000+2b30 crc_ccitt@8ee19000+3e0  compat@8ee98000+1740    fuse@8ee00000+12740     ledtrig_usbport@8ee9c000+ab0    nf_log_ipv6@8eea8000+1020   nf_log_common@8f757000+aa0      ip6table_mangle@8f755000+4a0    ip6table_filter@8f752000+280    ip6_tables@8ee90000+2a20   ip6t_REJECT@8e91e000+400 x_tables@8f75c000+3290  nf_reject_ipv6@8efff000+9e0     msdos@8f70c000+1980     vfat@8e8f8000+21c0      fat@8ee80000+ccc0  udf@8e960000+137b0       crc_itu_t@8e95b000+3e0  ntfs@8e940000+17430     isofs@8e910000+5770     hfsplus@8f760000+156a0  autofs4@8f710000+5510   nls_utf8@8f6c1000+340       nls_iso8859_1@8e85e000+b20      nls_cp437@8f72c000+1120 uas@8ef90000+2dc0       usb_storage@8e860000+9ee0       sd_mod@8ef88000+7430        scsi_mod@8efe0000+15750 f2fs@8e880000+41eb0     ext4@8e800000+5db20     mbcache@8efbf000+e10    jbd2@8efd0000+d450      exfat@8efa0000+18610        mii@8ef9a000+1480       crc32c_generic@8ef7b000+560     crc32_generic@8ef79000+560      leds_gpio@8f63c000+b60  xhci_plat_hcd@8f66a000+15c0 xhci_pci@8f637000+bc0   xhci_mtk@8f62a000+1460  xhci_hcd@8f640000+160f0 gpio_button_hotplug@8f5f6000+1ac0       usbcore@8f600000+1fbd0  nls_base@8f5e8000+1430      usb_common@8f5dc000+8c0
<6>[    4.113516] 8021q: 802.1Q VLAN Support v1.8
<6>[    4.120383] hctosys: unable to open rtc device (rtc0)
<6>[    4.133653] VFS: Mounted root (squashfs filesystem) readonly on device 31:5.
<6>[    4.145618] Freeing unused kernel memory: 1260K
<4>[    4.150149] This architecture does not have kernel memory protection.
<14>[    4.961198] init: Console is alive
<14>[    4.964925] init: - watchdog -
<6>[    5.149209] mtk_soc_eth 1e100000.ethernet eth0: port 0 link up
<14>[    5.991289] kmodloader: loading kernel modules from /etc/modules-boot.d/*
<6>[    6.172015] usbcore: registered new interface driver usbfs
<6>[    6.177644] usbcore: registered new interface driver hub
<6>[    6.183107] usbcore: registered new device driver usb
<4>[    6.195826] xhci-mtk 1e1c0000.xhci: 1e1c0000.xhci supply vbus not found, using dummy regulator
<4>[    6.204541] xhci-mtk 1e1c0000.xhci: 1e1c0000.xhci supply vusb33 not found, using dummy regulator
<6>[    6.213535] xhci-mtk 1e1c0000.xhci: xHCI Host Controller
<6>[    6.218862] xhci-mtk 1e1c0000.xhci: new USB bus registered, assigned bus number 1
<6>[    6.233363] xhci-mtk 1e1c0000.xhci: hcc params 0x01401198 hci version 0x96 quirks 0x0000000000210010
<6>[    6.242537] xhci-mtk 1e1c0000.xhci: irq 20, io mem 0x1e1c0000
<6>[    6.249295] hub 1-0:1.0: USB hub found
<6>[    6.253123] hub 1-0:1.0: 2 ports detected
<6>[    6.257719] xhci-mtk 1e1c0000.xhci: xHCI Host Controller
[... cut ...]
<4>[ 7851.543106] [<80045f50>] 0x80045f50
<4>[ 7851.546574] [<8004be98>] 0x8004be98
<4>[ 7851.550043] [<8004bd68>] 0x8004bd68
<4>[ 7851.553512] [<8004bd68>] 0x8004bd68
<4>[ 7851.556981] [<8004bd68>] 0x8004bd68
<4>[ 7851.560449] [<80006f78>] 0x80006f78
<4>[ 7851.563925]
<4>[ 7851.565403] Code: afa2002c  ae16000c  afbe0028 <ac560000> 8f830000  8e020008  14560006  afa30030  c2020000
<4>[ 7851.575126]
<4>[ 7851.578656] ---[ end trace 2391d9ffa7b36dfd ]---

===================================
Time: 1578494336.3602
```

LorenzoBianconi, 2020-02-17:
@wynn1212 @mat-alfa : could you please try the latest mt76 version available on openwrt? Does it have the same issue?

LorenzoBianconi, 2020-02-18:
@mat-alfa @wynn1212 : does it happen when you plug the card for the first time or for example when reboot the device with the dongle already plugged?

mat-alfa, 2020-02-18:
I tried openwrt 19.07.1, it works well. Didn't appear panic when plug and boot for the first time, also reboot with dongle already plugged.
GJ! and thank you very much for your help!

NOTE: 
 * The driver I tested is: `kmod-mt76x2u_4.14.167+2020-01-04-8a785679-1`
 * The testing hardware is AWUS036ACM(MT7612U) with R36A(ar71xx, mips 24kc)

pepe2k, 2020-02-18:
@LorenzoBianconi 

On current OpenWrt master with mt76 @ 721673759d82 it still crashes.

Same platform as before:
- R36A (QCA9531) on target ar71xx
- AWUS036ACM (MT7612U)
- kernel 4.14.169

1. Dongle plugged before system bootup:

```
[   35.178302] wlan0: authenticate with 44:d1:fa:XX:XX:XX
[   36.842256] wlan0: send auth to 44:d1:fa:XX:XX:XX (try 1/3)
[   36.848906] CPU 0 Unable to handle kernel paging request at virtual address 00000000, epc == 8042fd30, ra == 83138f6c
[   36.860021] Oops[#1]:
[   36.862381] CPU: 0 PID: 11 Comm: kworker/u2:2 Not tainted 4.14.169 #0
[   36.869096] Workqueue: phy0 mt76x02_mac_work [mt76x02_lib]
[   36.874766] task: 83823ac0 task.stack: 8386c000
[   36.879439] $ 0   : 00000000 00000001 00000000 00000000
[   36.884842] $ 4   : 83142b10 00000002 83823ac0 fffffff8
[   36.890245] $ 8   : fffffffd 83805014 83beda00 8c129824
[   36.895647] $12   : 000000e7 00000004 00000100 00000000
[   36.901050] $16   : 83142b10 83141d60 00001134 83141d60
[   36.906454] $20   : 00000000 8386ddb0 80530000 fffffffe
[   36.911856] $24   : 00000010 44bcee7a                  
[   36.917259] $28   : 8386c000 8386dd90 83142b14 83138f6c
[   36.922662] Hi    : 001c38ec
[   36.925628] Lo    : 61ee75f8
[   36.928625] epc   : 8042fd30 __mutex_lock.isra.2+0x134/0x378
[   36.934475] ra    : 83138f6c mt76u_deinit+0x5e8/0x9e4 [mt76_usb]
[   36.940672] Status: 1000fc03KERNEL EXL IE 
[   36.945001] Cause : 0080000c (ExcCode 03)
[   36.949140] BadVA : 00000000
[   36.952115] PrId  : 00019374 (MIPS 24Kc)
[   36.956164] Modules linked in: ath9k ath9k_common pppoe ppp_async mt76x0u mt76x0_common ath9k_hw ath pppox ppp_generic nf_conntrack_ipv6 mt76x2u mt76x2e mt76x2_common mt76x02_usb mt76x02_lib mt7603e mt7601u mt76_usb mt76 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_FLOWOFFLOAD slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_mangle iptable_filter ip_tables crc_ccitt compat ledtrig_usbport nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 ehci_platform
[   37.029403]  ehci_hcd gpio_button_hotplug usbcore nls_base usb_common
[   37.036064] Process kworker/u2:2 (pid: 11, threadinfo=8386c000, task=83823ac0, tls=00000000)
[   37.044769] Stack : 80530000 fffffffe 80530000 8033e654 81092100 83804000 00000001 8009c238
[   37.053406]         83142b14 00000000 00000000 83823ac0 83142b10 83141d60 00001134 83141d60
[   37.062043]         00000000 83141e7c 80530000 fffffffe 80530000 83138f6c 833a0f00 83168024
[   37.070681]         833a0840 802f755c 83b32080 83141d60 83142628 831218bc 82fbc8fc 802f7600
[   37.079317]         8310b024 82fbc8ec 83140bc0 83141d60 83855a80 83212bf8 83823ac0 00000000
[   37.087955]         ...
[   37.090485] Call Trace:
[   37.093014] [<8042fd30>] __mutex_lock.isra.2+0x134/0x378
[   37.098508] [<83138f6c>] mt76u_deinit+0x5e8/0x9e4 [mt76_usb]
[   37.104380] Code: afa20024  ac950008  afbe0020 <ac550000> 8f830000  8c820004  1055000b  afa30028  8f820000 
[   37.114460] 
[   37.116327] ---[ end trace ad60764e39bc276a ]---
```

2. Dongle plugged after system start and wifi brought up manually:

```
[   71.402377] usb 1-1: new high-speed USB device number 2 using ehci-platform
[   71.772371] usb 1-1: reset high-speed USB device number 2 using ehci-platform
[   71.973817] mt76x2u 1-1:1.0: ASIC revision: 76120044
[   72.105623] mt76x2u 1-1:1.0: ROM patch build: 20141115060606a
[   72.275724] mt76x2u 1-1:1.0: Firmware Version: 0.0.00
[   72.281016] mt76x2u 1-1:1.0: Build: 1
[   72.284879] mt76x2u 1-1:1.0: Build Time: 201507311614____
[   73.199788] ieee80211 phy1: Selected rate control algorithm 'minstrel_ht'
root@OpenWrt:/# wifi
'radio0' is disabled
'radio0' is disabled
root@OpenWrt:/# [  101.228399] IPv6: ADDRCONF(NETDEV_UP): wlan1: link is not ready
[  107.137953] wlan1: authenticate with 44:d1:fa:XX:XX:XX
[  108.802626] wlan1: send auth to 44:d1:fa:XX:XX:XX (try 1/3)
[  108.811832] wlan1: authenticated
[... 16 lines cut ...]
[  110.493174] $28   : 83868000 83869d88 82e2ab14 83138f6c
[  110.498578] Hi    : 000000f7
[  110.501553] Lo    : 33333371
[  110.504548] epc   : 8042fd30 __mutex_lock.isra.2+0x134/0x378
[  110.510393] ra    : 83138f6c mt76u_deinit+0x5e8/0x9e4 [mt76_usb]
[  110.516587] Status: 1000fc03KERNEL EXL IE 
[  110.520916] Cause : 0080000c (ExcCode 03)
[  110.525054] BadVA : 00000000
[  110.528021] PrId  : 00019374 (MIPS 24Kc)
[  110.532070] Modules linked in: ath9k ath9k_common pppoe ppp_async mt76x0u mt76x0_common ath9k_hw ath pppox ppp_generic nf_conntrack_ipv6 mt76x2u mt76x2e mt76x2_common mt76x02_usb mt76x02_lib mt7603e mt7601u mt76_usb mt76 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_FLOWOFFLOAD slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_mangle iptable_filter ip_tables crc_ccitt compat ledtrig_usbport nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 ehci_platform
[... cut ...]
[  110.646496]         8052d9f8 83823af0 83869e30 82e29d60 83869e5f 831527ac 82c158ec 8052d9f8
[  110.655134]         83822f00 83822f30 82e2ab40 82e29d60 83869e5f 83155c20 00000000 8052d9c8
[  110.663772]         ...
[  110.666299] Call Trace:
[  110.668832] [<8042fd30>] __mutex_lock.isra.2+0x134/0x378
[  110.674325] [<83138f6c>] mt76u_deinit+0x5e8/0x9e4 [mt76_usb]
[  110.680166] Code: afa20024  ac950008  afbe0020 <ac550000> 8f830000  8c820004  1055000b  afa30028  8f820000 
[  110.690249] 
[  110.692626] ---[ end trace 8cb8cfb12a221fe5 ]---
```

wynn1212, 2020-02-19:
@LorenzoBianconi 
Well.I tried OpenWRT 19.07.1 It still crash when I plug in the USB wifi dongle and attempt to connect to my hotspot. And after that It'll crash-looping my router until I unplug the USB wifi dongle .

- My devices: D-Team Newifi D2(mt7621).
- OpenWRT Version cause the crash: 19.07.1
- Installed module: kmod-mt76x0u - 4.14.167+2020-01-04-8a785679-1
- USB Wifi dongle: ASUS USB-AC51

Crashlog via `/sys/kernel/debug/crashlog` from OpenWRT 19.07.1
```
Time: 1581129094.897038
Modules:        pppoe@8ea4c000+23e0     ppp_async@8f728000+1b90 mt76x0u@8ea3a000+1180   mt76x0_common@8ea30000+70a0     rndis_host@8ea26000+1320     pppox@8edbe000+4b0      ppp_generic@8edb8000+5970       mt76x2e@8f76c000+26c0   mt76x2_common@8f72c000+2a40  mt76x02_usb@8e92a000+16a0       mt76x02_lib@8ece0000+9640       mt7603e@8e930000+90a0   mt76_usb@8e92c000+22b0  mt76@8e920000+7920   mac80211@8e980000+6c840 iptable_nat@8f797000+2c0        ipt_REJECT@8ee57000+3c0 ipt_MASQUERADE@8f6dd000+2a0 cfg80211@8e940000+36b80  cdc_ether@8e908000+12e0 xt_time@8f7f6000+6e0    xt_tcpudp@8f7f4000+700  xt_state@8f7f2000+2c0   xt_nat@8f7df000+600  xt_multiport@8f7dc000+500       xt_mark@8f7e2000+2a0    xt_mac@8f769000+260     xt_limit@8f7bf000+4a0   xt_conntrack@8eff1000+900    xt_comment@8f7d9000+1e0 xt_TCPMSS@8f7d7000+ae0  xt_REDIRECT@8f7d4000+2c0        xt_LOG@8f714000+300 xt_FLOWOFFLOAD@8f7a4000+c10      xt_CT@8f7ee000+a60      usbnet@8f7e8000+4490    ums_usbat@8f738000+2080 ums_sddr55@8f76a000+1740     ums_sddr09@8f7e4000+23c0        ums_karma@8f77c000+6a0  ums_jumpshot@8f7a5000+fc0       ums_isd200@8f7c8000+1710    ums_freecom@8f7a0000+880 ums_datafab@8f7bc000+1320       ums_cypress@8ef29000+a20        ums_alauda@8f7b4000+22a0        slhc@8f7ae000+13c0   nf_reject_ipv4@8f778000+840     nf_nat_redirect@8f77d000+580    nf_nat_masquerade_ipv4@8eef4000+650     nf_conntrack_ipv4@8ef0c000+1420      nf_nat_ipv4@8f6f8000+f30        nf_log_ipv4@8f6dc000+d60        nf_flow_table_hw@8f75a000+8d0nf_flow_table@8effc000+3350     nf_defrag_ipv4@8f751000+440     nf_conntrack_rtcache@8f6d9000+aa0       iptable_mangle@8f795000+380  iptable_filter@8f798000+280     ip_tables@8ee50000+2b30 crc_ccitt@8f79b000+3e0  compat@8f772000+1740    fuse@8f780000+12740  ledtrig_usbport@8f6da000+ab0    ip6table_nat@8f71c000+2c0       nf_conntrack_ipv6@8f776000+1590 nf_defrag_ipv6@8ef0a000+12e0 nf_nat_ipv6@8f770000+11d0       ip6t_NPT@8f754000+600   ip6t_MASQUERADE@8f756000+220    nf_nat_masquerade_ipv6@8f758000+730  nf_nat@8f73c000+37d0    nf_conntrack@8f740000+10550     nf_log_ipv6@8f72a000+1020       nf_log_common@8ee5f000+aa0  ip6table_mangle@8f6f9000+4a0     ip6table_filter@8f715000+280    ip6_tables@8f720000+2a20        ip6t_REJECT@8f717000+400    x_tables@8eff8000+3290   nf_reject_ipv6@8f71b000+9e0     msdos@8ef08000+1980     vfat@8ee54000+21c0      fat@8ee30000+ccc0   udf@8f700000+137b0       crc_itu_t@8f6fb000+3e0  ntfs@8f6e0000+17410     isofs@8efe0000+5770     hfsplus@8f6c0000+156c0  autofs4@8ee58000+5510        nls_utf8@8ef49000+340   nls_iso8859_1@8ff1f000+b20      nls_cp437@8ee66000+1120 uas@8e868000+2dc0   usb_storage@8efb0000+9ee0        sd_mod@8e860000+7430    scsi_mod@8efc0000+15750 f2fs@8e880000+41ed0     ext4@8e800000+5db20 mbcache@8ef99000+e10     jbd2@8efa0000+d450      exfat@8ef80000+18610    mii@8ef6e000+1480       crc32c_generic@8eed4000+560 crc32_generic@8f651000+560       leds_gpio@8f5ca000+b60  xhci_plat_hcd@8f618000+15c0     xhci_pci@8f605000+bc0   xhci_mtk@8f608000+1460       xhci_hcd@8f620000+160f0 gpio_button_hotplug@8f5cc000+1ac0       usbcore@8f5e0000+1fd50  nls_base@8f5be000+1430       usb_common@8f5b4000+8c0
<5>[    2.523680] Creating 4 MTD partitions on "spi0.0":
<5>[    2.528460] 0x000000000000-0x000000030000 : "u-boot"
<5>[    2.534559] 0x000000030000-0x000000040000 : "u-boot-env"
<5>[    2.540789] 0x000000040000-0x000000050000 : "factory"
<5>[    2.546768] 0x000000050000-0x000002000000 : "firmware"
<5>[    2.553079] 2 uimage-fw partitions found on MTD device firmware
<5>[    2.558984] Creating 2 MTD partitions on "firmware":
<5>[    2.563971] 0x000000000000-0x0000001c963c : "kernel"
<5>[    2.569906] 0x0000001c963c-0x000001fb0000 : "rootfs"
<5>[    2.575796] mtd: device 5 (rootfs) set to be root filesystem
<5>[    2.581567] 1 squashfs-split partitions found on MTD device rootfs
<5>[    2.587762] 0x000000450000-0x000001fb0000 : "rootfs_data"
<6>[    2.594844] libphy: Fixed MDIO Bus: probed
<6>[    2.664546] libphy: mdio: probed
<6>[    4.067425] mtk_soc_eth 1e100000.ethernet: loaded mt7530 driver
<6>[    4.074077] mtk_soc_eth 1e100000.ethernet eth0: mediatek frame engine at 0xbe100000, irq 21
<6>[    4.085294] NET: Registered protocol family 10
<6>[    4.091448] Segment Routing with IPv6
<6>[    4.095263] NET: Registered protocol family 17
<6>[    4.099768] bridge: filtering via arp/ip/ip6tables is no longer available by default. Update your scripts to load br_netfilter if you need this.
<6>[    4.112683] 8021q: 802.1Q VLAN Support v1.8
<6>[    4.119506] hctosys: unable to open rtc device (rtc0)
<6>[    4.132683] VFS: Mounted root (squashfs filesystem) readonly on device 31:5.
<6>[    4.144631] Freeing unused kernel memory: 1260K
<4>[    4.149162] This architecture does not have kernel memory protection.
<14>[    4.969406] init: Console is alive
<14>[    4.973122] init: - watchdog -
<6>[    5.023727] mtk_soc_eth 1e100000.ethernet eth0: port 3 link up
<6>[    5.928426] mtk_soc_eth 1e100000.ethernet eth0: port 2 link up
[... cut ...]
<4>[   32.117109] [<80045f40>] 0x80045f40
<4>[   32.120577] [<8004be88>] 0x8004be88
<4>[   32.124046] [<8004bd58>] 0x8004bd58
<4>[   32.127525] [<8004bd58>] 0x8004bd58
<4>[   32.131004] [<8004bd58>] 0x8004bd58
<4>[   32.134473] [<80006f78>] 0x80006f78
<4>[   32.137949] 
<4>[   32.139426] Code: afa2002c  ae16000c  afbe0028 <ac560000> 8f830000  8e020008  14560006  afa30030  c2020000 
<4>[   32.149151] 
<4>[   32.151459] ---[ end trace 8318868de7f2b062 ]---

===================================
Time: 158
```

mat-alfa, 2020-02-19:
@LorenzoBianconi 
I did some more tests and found my previous test is a special case. The WiFi AP I tested with is an old version LEDE 17.01 device, DIR-505.(I attach the screenshot below)
[image]

After changing the testing AP to newer devices ( ex: Android's hotspot ), I got the same crash-loop as wynn1212 described. I attach the crashlog and uci's wireless configs below
[crashlog.txt](https://github.com/openwrt/mt76/files/4226107/crashlog.txt)
[wireless.uci.txt](https://github.com/openwrt/mt76/files/4226178/wireless.uci.txt)

LorenzoBianconi, 2020-02-19:
@mat-alfa: could you please enable symbol table information and debug info?
Global build settings --> Kernel build options --> Compile the kernel with symbol table information/Compile the kernel with debug information

then can you provide the following info?

let's assume the system crash reporting:
- [<83138f6c>] mt76u_deinit+0x5e8/0x9e4 [mt76_usb]

can you run gdb on usb.o

$<openwrt-dir>/staging_dir/toolchain-mipsel_24kc_gcc-8.3.0_musl/bin/gdb  <openwrt-dir>/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-<version>/usb.o

inside gdb:
(gdb) info address mt76u_deinit
(gdb) l *(<result>+0x5e8)

this should returns where the system crashes

Thanks

LorenzoBianconi, 2020-02-20:
@mat-alfa @pepe2k @wynn1212 as suggested by @nbd168 this is probably a cache corruption issue introduced by commit 09fcbc214b102a489f360a081fba156ae460c142 ("mt76: usb: use max packet length for m76u_copy"). I partially reverted the commit and I make the data buffer aligned to the cacheline. Could you please try the following this [patch](https://github.com/LorenzoBianconi/mt76/commit/3af3a32bc3278846d048f963fb299904df3f94e1)?

sgruszka, 2020-02-20:
>  I partially reverted the commit and I make the data buffer aligned to the cacheline. 

kmalloc data is  aligned to cache line, also it's size is multiple of cache line size (even if we request less data). What is the problem IIUC is that after 09fcbc2 reg_val share cache line with other fields of mt76_usb struct , what  cause corruption of those fields  when dma operation on reg_val is performed. So IMHO right solution would be remove reg_val and use only usb->data for usb requests, like it was before commit 307be50cc5f4f06fd7acb561f03b1de587339810 (except we do not need put/get_unaligned any longer)

LorenzoBianconi, 2020-02-20:
@sgruszka yes, I was thinking to the same approach right now. I think we can remove reg_val and use usb->data directly. I will post a patch to try this solution

LorenzoBianconi, 2020-02-20:
@mat-alfa @pepe2k @wynn1212 : could you please test this [patch](https://github.com/LorenzoBianconi/mt76/commit/55b04c29ed3842ed07e5cfaf4b05d9561aaa6070)?

pepe2k, 2020-02-20:
@LorenzoBianconi 

> could you please test this [patch](https://github.com/LorenzoBianconi/mt76/commit/55b04c29ed3842ed07e5cfaf4b05d9561aaa6070)?

Tested on current OpenWrt master and mt76 @ 8f8e9161 with the patch (same configuration as mentioned in https://github.com/openwrt/mt76/issues/338#issuecomment-587926080). Problem seems to be solved. Thanks!

```
44:D1:FA:XX:XX:XX  -41 dBm / unknown (SNR -41)  360 ms ago
        RX: 866.7 MBit/s, VHT-MCS 9, 80MHz, VHT-NSS 2      9901 Pkts.
        TX: 866.7 MBit/s, VHT-MCS 9, 80MHz, VHT-NSS 2     23338 Pkts.
        expected throughput: 313.5 MBit/s
```

nbd168, 2020-02-20:
Fixed in fd892bc033fbfad7b1a5bcba1432709a74327a1e, thanks for testing

closed 2020-02-20 by nbd168

wynn1212, 2020-02-21:
@LorenzoBianconi Sorry for reply late!

> @mat-alfa @pepe2k @wynn1212 : could you please test this [patch](https://github.com/LorenzoBianconi/mt76/commit/55b04c29ed3842ed07e5cfaf4b05d9561aaa6070)?

Tested on current OpenWrt master and mt76 with this patch

> Fixed in [fd892bc](https://github.com/openwrt/mt76/commit/fd892bc033fbfad7b1a5bcba1432709a74327a1e), thanks for testing

Everything seems to be normal now. No more crashing.

- My devices: D-Team Newifi D2(mt7621).
- OpenWRT Version: OpenWrt SNAPSHOT r12257-c8f8e59816 
- Installed module: kmod-mt76x0u - 4.14.169+2020-02-20-fd892bc0-1
- USB Wifi dongle: ASUS USB-AC51

[image]

Note: I can only build the patched mt76 on OpenWrt master branch not on openwrt-19.07 branch