openwrt/mt76 #339: mt7615e: Unusable 2GHz radio

vgmz95, 2019-12-29:
I have a problem with the mt7615e driver used for the 2GHz band in the Netgear r6700v2 (There's a current [PR](https://github.com/openwrt/openwrt/pull/2614) adding support for openwrt for that device).

When a device connects to the 2GHz radio, the tx/rx speeds are incredibly slow (around 0.6~1 Mbps in perfect conditions) and after a minute or so the device disconnects, and the message: 
`STA 00:00:00:00:00:00 IEEE 802.11: disconnected due to excessive missing ACKs`
 appears in the syslog.

I tried adding the `option disassoc_low_ack '0'` to the 2GHz wireless config and that prevents the device from disconnecting, but it doesn't solve the slow speed issue.

When I connect the same device to the 5GHz band, everything runs fine. I'm using this driver version: 

```
> [   16.391597] mt7615e 0000:01:00.0: HW/SW Version: 0x8a108a10, Build Time: 20180518100604a
> [   16.616003] mt7615e 0000:01:00.0: N9 Firmware Version: _reserved_, Build Time: 20190103180756
> [   16.640111] mt7615e 0000:01:00.0: CR4 Firmware Version: _reserved_, Build Time: 20181207140436
> [   24.315383] mt7615e 0000:02:00.0: HW/SW Version: 0x8a108a10, Build Time: 20180518100604a
> [   24.350005] mt7615e 0000:02:00.0: N9 Firmware Version: _reserved_, Build Time: 20190103180756
> [   24.374465] mt7615e 0000:02:00.0: CR4 Firmware Version: _reserved_, Build Time: 20181207140436

```

vgmz95, 2020-01-06:
I build image for my [R6700v2 ](http://en.techinfodepot.shoutwiki.com/wiki/Netgear_R6700_v2) with the latest [mt76  commit](https://github.com/openwrt/openwrt/commit/1de8fc93ca449942a0fc28234aac8d372542a7b5) and the issue continues, now devices struggle to even connect to the device. When the connection is established, tx/rx speeds continue to be low.

This is the log when a devices tries to connect: 
```
Mon Jan  6 01:09:26 2020 daemon.notice hostapd: wlan0: AP-ENABLED
Mon Jan  6 01:09:27 2020 user.notice root: ip link set dev wlan0 up
Mon Jan  6 01:09:27 2020 daemon.notice netifd: Network device 'wlan0' link is up
Mon Jan  6 01:11:30 2020 daemon.debug hostapd: wlan0: STA 00:00:00:00:00:00 IEEE 802.11: authentication OK (open system)
Mon Jan  6 01:11:30 2020 daemon.debug hostapd: wlan0: STA 00:00:00:00:00:00 MLME: MLME-AUTHENTICATE.indication(00:00:00:00:00:00, OPEN_SYSTEM)
Mon Jan  6 01:11:30 2020 daemon.debug hostapd: wlan0: STA 00:00:00:00:00:00 MLME: MLME-DELETEKEYS.request(00:00:00:00:00:00)
Mon Jan  6 01:11:30 2020 daemon.notice hostapd: wlan0: STA 00:00:00:00:00:00 IEEE 802.11: did not acknowledge authentication response
Mon Jan  6 01:11:32 2020 daemon.debug hostapd: wlan0: STA 00:00:00:00:00:00 IEEE 802.11: deauthenticated
Mon Jan  6 01:11:32 2020 daemon.debug hostapd: wlan0: STA 00:00:00:00:00:00 MLME: MLME-DEAUTHENTICATE.indication(00:00:00:00:00:00, 1)
Mon Jan  6 01:11:32 2020 daemon.debug hostapd: wlan0: STA 00:00:00:00:00:00 MLME: MLME-DELETEKEYS.request(00:00:00:00:00:00)
Mon Jan  6 01:11:34 2020 daemon.debug hostapd: wlan0: STA 00:00:00:00:00:00 IEEE 802.11: authentication OK (open system)
Mon Jan  6 01:11:34 2020 daemon.debug hostapd: wlan0: STA 00:00:00:00:00:00 MLME: MLME-AUTHENTICATE.indication(00:00:00:00:00:00, OPEN_SYSTEM)
Mon Jan  6 01:11:34 2020 daemon.debug hostapd: wlan0: STA 00:00:00:00:00:00 MLME: MLME-DELETEKEYS.request(00:00:00:00:00:00)
Mon Jan  6 01:11:34 2020 daemon.notice hostapd: wlan0: STA 00:00:00:00:00:00 IEEE 802.11: did not acknowledge authentication response
Mon Jan  6 01:11:48 2020 daemon.debug hostapd: wlan0: STA 00:00:00:00:00:00 IEEE 802.11: authentication OK (open system)
Mon Jan  6 01:11:48 2020 daemon.debug hostapd: wlan0: STA 00:00:00:00:00:00 MLME: MLME-AUTHENTICATE.indication(00:00:00:00:00:00, OPEN_SYSTEM)
Mon Jan  6 01:11:48 2020 daemon.debug hostapd: wlan0: STA 00:00:00:00:00:00 MLME: MLME-DELETEKEYS.request(00:00:00:00:00:00)
Mon Jan  6 01:11:48 2020 daemon.notice hostapd: wlan0: STA 00:00:00:00:00:00 IEEE 802.11: did not acknowledge authentication response
Mon Jan  6 01:12:16 2020 daemon.debug hostapd: wlan0: STA 00:00:00:00:00:00 IEEE 802.11: authentication OK (open system)
Mon Jan  6 01:12:16 2020 daemon.debug hostapd: wlan0: STA 00:00:00:00:00:00 MLME: MLME-AUTHENTICATE.indication(00:00:00:00:00:00, OPEN_SYSTEM)
Mon Jan  6 01:12:16 2020 daemon.debug hostapd: wlan0: STA 00:00:00:00:00:00 MLME: MLME-DELETEKEYS.request(00:00:00:00:00:00)
Mon Jan  6 01:12:16 2020 daemon.info hostapd: wlan0: STA 00:00:00:00:00:00 IEEE 802.11: authenticated
Mon Jan  6 01:12:16 2020 daemon.debug hostapd: wlan0: STA 00:00:00:00:00:00 IEEE 802.11: association OK (aid 1)
Mon Jan  6 01:12:16 2020 daemon.info hostapd: wlan0: STA 00:00:00:00:00:00 IEEE 802.11: associated (aid 1)
Mon Jan  6 01:12:16 2020 daemon.debug hostapd: wlan0: STA 00:00:00:00:00:00 MLME: MLME-ASSOCIATE.indication(00:00:00:00:00:00)
Mon Jan  6 01:12:16 2020 daemon.debug hostapd: wlan0: STA 00:00:00:00:00:00 MLME: MLME-DELETEKEYS.request(00:00:00:00:00:00)
Mon Jan  6 01:12:16 2020 daemon.debug hostapd: wlan0: STA 00:00:00:00:00:00 IEEE 802.11: binding station to interface 'wlan0'
Mon Jan  6 01:12:16 2020 daemon.debug hostapd: wlan0: STA 00:00:00:00:00:00 WPA: event 1 notification
Mon Jan  6 01:12:16 2020 daemon.debug hostapd: wlan0: STA 00:00:00:00:00:00 WPA: start authentication
Mon Jan  6 01:12:16 2020 daemon.debug hostapd: wlan0: STA 00:00:00:00:00:00 IEEE 802.1X: unauthorizing port
Mon Jan  6 01:12:16 2020 daemon.debug hostapd: wlan0: STA 00:00:00:00:00:00 WPA: sending 1/4 msg of 4-Way Handshake
Mon Jan  6 01:12:17 2020 daemon.debug hostapd: wlan0: STA 00:00:00:00:00:00 WPA: received EAPOL-Key frame (2/4 Pairwise)
Mon Jan  6 01:12:17 2020 daemon.debug hostapd: wlan0: STA 00:00:00:00:00:00 WPA: sending 3/4 msg of 4-Way Handshake
Mon Jan  6 01:12:17 2020 daemon.debug hostapd: wlan0: STA 00:00:00:00:00:00 WPA: received EAPOL-Key frame (4/4 Pairwise)
Mon Jan  6 01:12:17 2020 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 00:00:00:00:00:00
Mon Jan  6 01:12:17 2020 daemon.debug hostapd: wlan0: STA 00:00:00:00:00:00 IEEE 802.1X: authorizing port
Mon Jan  6 01:12:17 2020 daemon.info hostapd: wlan0: STA 00:00:00:00:00:00 WPA: pairwise key handshake completed (RSN)
Mon Jan  6 01:12:49 2020 daemon.info hostapd: wlan0: STA 00:00:00:00:00:00 IEEE 802.11: disconnected due to excessive missing ACKs
Mon Jan  6 01:12:49 2020 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED 00:00:00:00:00:00
Mon Jan  6 01:12:49 2020 daemon.debug hostapd: wlan0: STA 00:00:00:00:00:00 MLME: MLME-DISASSOCIATE.indication(00:00:00:00:00:00, 34)
Mon Jan  6 01:12:49 2020 daemon.debug hostapd: wlan0: STA 00:00:00:00:00:00 MLME: MLME-DELETEKEYS.request(00:00:00:00:00:00)
Mon Jan  6 01:13:19 2020 daemon.info hostapd: wlan0: STA 00:00:00:00:00:00 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
[... cut ...]
                * [ CONTROL_PORT_OVER_NL80211 ]: control port over nl80211
                * [ TXQS ]: FQ-CoDel-enabled intermediate TXQs
                * [ AIRTIME_FAIRNESS ]: airtime fairness scheduling
```

And these are loaded drivers: 
```
cfg80211              259321  4 mt7615e,mt7603e,mt76,mac80211
mac80211              490633  3 mt7615e,mt7603e,mt76
mt76                   35081  2 mt7615e,mt7603e
mt7603e                38700  0
mt7615e                43248  0
```

If further information is needed, please let me know.

nbd168, 2020-01-23:
Please try the latest revision of OpenWrt master or 19.07. I've fixed a critical issue in hostapd.

vgmz95, 2020-01-25:
I built an image with the latest mt76 driver and I still getting low tx/rx rates and the message: 
`daemon.info hostapd: wlan1: STA 00:00:00:00:00:00 IEEE 802.11: disconnected due to excessive missing ACKs`

CHKDSK88, 2020-02-07:
On actual OpenWrt master is posible to connect only once. After couple seconds connection freezes. After client disconnect is impossible to connect anymore:

````
Fri Feb  7 13:44:55 2020 daemon.info hostapd: wlan0: STA 94:87:e0:09:e9:d0 IEEE 802.11: authenticated
Fri Feb  7 13:44:55 2020 daemon.info hostapd: wlan0: STA 94:87:e0:09:e9:d0 IEEE 802.11: associated (aid 1)
Fri Feb  7 13:44:55 2020 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 94:87:e0:09:e9:d0
Fri Feb  7 13:44:55 2020 daemon.info hostapd: wlan0: STA 94:87:e0:09:e9:d0 WPA: pairwise key handshake completed (RSN)

Fri Feb  7 13:47:52 2020 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED 94:87:e0:09:e9:d0
Fri Feb  7 13:48:03 2020 daemon.notice hostapd: wlan0: STA 94:87:e0:09:e9:d0 IEEE 802.11: did not acknowledge authentication response
Fri Feb  7 13:48:16 2020 daemon.notice hostapd: wlan0: STA 94:87:e0:09:e9:d0 IEEE 802.11: did not acknowledge authentication response
Fri Feb  7 13:48:23 2020 daemon.notice hostapd: wlan0: STA 94:87:e0:09:e9:d0 IEEE 802.11: did not acknowledge authentication response
````

nbd168, 2020-02-20:
Please try the latest version, I've made some more fixes in the mean time

CHKDSK88, 2020-02-21:
@nbd168 

That changes didn't  affect this issue:

````
Fri Feb 21 20:49:05 2020 daemon.info hostapd: wlan0: STA 94:87:e0:09:e9:d0 IEEE 802.11: authenticated
Fri Feb 21 20:49:05 2020 daemon.info hostapd: wlan0: STA 94:87:e0:09:e9:d0 IEEE 802.11: associated (aid 1)
Fri Feb 21 20:49:05 2020 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 94:87:e0:09:e9:d0
Fri Feb 21 20:49:05 2020 daemon.info hostapd: wlan0: STA 94:87:e0:09:e9:d0 WPA: pairwise key handshake completed (RSN)
Fri Feb 21 20:54:08 2020 daemon.info hostapd: wlan0: STA 94:87:e0:09:e9:d0 IEEE 802.11: disconnected due to excessive missing ACKs
Fri Feb 21 20:54:08 2020 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED 94:87:e0:09:e9:d0
Fri Feb 21 20:54:10 2020 daemon.notice hostapd: wlan0: STA 94:87:e0:09:e9:d0 IEEE 802.11: did not acknowledge authentication response
Fri Feb 21 20:54:32 2020 daemon.notice hostapd: wlan0: STA 94:87:e0:09:e9:d0 IEEE 802.11: did not acknowledge authentication response
Fri Feb 21 20:54:38 2020 daemon.info hostapd: wlan0: STA 94:87:e0:09:e9:d0 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Fri Feb 21 20:54:43 2020 daemon.notice hostapd: wlan0: STA 94:87:e0:09:e9:d0 IEEE 802.11: did not acknowledge authentication response
````

Flukethoughts, 2020-03-05:
This also effects the edimax rg21s