openwrt/mt76 #341: How to max tx power from 19dbm to 30dbm?

usama7628674, 2020-01-04:
I've edup-ac1605 mt76x2u.Running a quick iwconfig command show tx-power=19dbm.I've tried increasing it but it does not go beyond 19dbm.Its stuck there.

usama7628674, 2020-01-04:
This is dmesg output when i connect device to my laptop
[image]

Is it old version looking at ROM patch build and Build Time which says 2014 and 2015 respectively.Should I update my driver?
Right now its plug and play on kali linux.

rufengsuixing, 2020-01-06:
try to set country to Australia

usama7628674, 2020-01-07:
`iwlist wlan1 txpower` shows

`Txpower=19dbm    (79dbm)`  which is very low.

usama7628674, 2020-01-07:
@rufengsuixing It does not work this way

usama7628674, 2020-01-07:
This is tied to eeprom and I think devs havent implemented tx-power control.

usama7628674, 2020-01-07:
RTL8812au is the best chipset mediatek is shit full of bugs

closed 2020-01-11 by usama7628674