openwrt/mt76 #344: mt7615: MCU message timeout - 5 GHz, VHT80

porentak, 2020-01-06:
Hello,

I'm working on MT7621 based platform using MT7615N PCIe card. Device is configured to use channel 36, AC, 80 MHz channel bandwidth.
 
In case of using iperf or transferring some bigger files (test with 500MB) over ssh, access to LuCI over WiFi, I get errors like:
[ 3712.079933] mt7615e 0000:02:00.0: Message 73 (seq 7) timeout
[ 3738.319832] mt7615e 0000:02:00.0: Message 73 (seq 8) timeout
[ 3764.559685] mt7615e 0000:02:00.0: Message 73 (seq 9) timeout

After this errors, WiFi is not working anymore.

I can not reproduce the issue if:
 - using 20 or 40MHz channel bandwidth
 - using only ICMP as a traffic (interval 0.01s), transferring smaller files

I can reproduce the issue even if:
 - client does not support AC
 - country code is changed

OpenWRT: master branch, commit  a3010a7f8dbe04972efef16ed7d81b5fd72e9a94
mt76 driver: master branch, commit 4cb1195a153087d84995f410285c8ecb587b1456 (2019-12-27)

My WiFi config:

config wifi-device 'radio0'
        option type 'mac80211'
        option channel '36'
        option hwmode '11a'
        option path 'pci0000:00/0000:00:01.0/0000:02:00.0'
        option htmode 'VHT80'
        option country 'US'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option encryption 'none'
        option ssid 'OpenWrtMT76'

Can anyone else reproduce this?

Is 80MHz band supported in MT76 driver?

Any comments, ideas what to look for are more than appreciated.

Thank you.

porentak, 2020-01-10:
After replacing PCIe WiFi card with new one issue is gone.

For anyone else having similar issues:
 - WiFi card: http://wiki.banana-pi.org/BPI-MT7615_802.11_ac_wifi_4x4_dual-band_module
 - the only difference I have noticed between working and non working card was the WiFi chip temperature (non working one had 10C higher temperature - cat /sys/kernel/debug/ieee80211/phy0/mt76/temperature)

closed 2020-01-10 by porentak

ptpt52, 2021-03-24:
@porentak @nbd168 
hello, does it support to get temperature  on mt7613 chip?
I try `cat /sys/kernel/debug/ieee80211/phy1/mt76/temperature` always get 0