openwrt/mt76 #346: wr840n v6.20 / mt7603e

Npeca75, 2020-01-10:
Hi 
i like to ask for help with tplink wr840n v6.20 device
i was successfully compiled latest OpenWrt SNAPSHOT r11954-7936cb94a9 but the WiFi is almost blind
There is no scan data most of time. Or, if AP occasionally scan something, it is heavy reduced
for example, only 1 AP from 10 was shown in scan list

Same driver, same snapshot work flawlessly on 840n v5 which is basically same, only lower amount of RAM

is there any LOG or other debug info which will help to resolve this problem?

sanitariu, 2020-01-11:
This is related to your dts file being wrong. I did it working for 6.0,6.1,6.2 and 14.
I am attaching all the files for 6.0. Just modify them adding 6.1,6.2.
Next do make menuconfig and choose your model.

[mt7628an_tplink_tl-wr840n-v6.dts.zip](https://github.com/openwrt/mt76/files/4049225/mt7628an_tplink_tl-wr840n-v6.dts.zip)
[02_network.zip](https://github.com/openwrt/mt76/files/4049226/02_network.zip)
[mt76x8.mk.zip](https://github.com/openwrt/mt76/files/4049227/mt76x8.mk.zip)

Note you must have a little free space 3 blocks at least so you settings remain after restarts.
If someone have the will he can create patches for openwrt.
Leds must be fixed too for versions 6.1,6.2.
I will do it in the next days.....

Npeca75, 2020-01-11:
Hi @sanitariu 
tnx for your help
i tried with your DTS and still the same behaviour. RX is blind. TX seems to work OK, other AP's are "see" my test620AP with proper power, but RX ... nope

about LED's
`
    /* LED used is dual-color,dual lead LED */
    leds {
        compatible = "gpio-leds";

        lan {
            label = "tl-wr840n-v620:green:lan";
            gpios = <&gpio1 7 GPIO_ACTIVE_LOW>;
        };

        wlan {
            label = "tl-wr840n-v620:green:wlan";
            gpios = <&gpio1 12 GPIO_ACTIVE_LOW>;
        };

        led_power: orange {
            label = "tl-wr840n-v620:orange:wan";
            gpios = <&gpio1 10 GPIO_ACTIVE_HIGH>;
        };

        green {
            label = "tl-wr840n-v620:green:wan";
            gpios = <&gpio1 8 GPIO_ACTIVE_LOW>;
        };
    };
`
this way, LED's working fine for me

and about space:
i don't need "full" release. all my 840n V5 are working like "dumb" AP, they has only VLAN's and SSID's for each VLAN, with my custom, minimal web, so the space is not issue
overlayfs:/overlay      896.0K    240.0K    656.0K  27% /

unfortunately, i ran out of V5, and only similar AP is V6.20, that is the reason why i tried so hard to make OpenWrt on it

anyway, thank you for your help

Npeca75, 2020-01-11:
hi @sanitariu 
please ignore my prev post
you was right
with your DTS wifi works as expected
90 mbps up&down, scan is fine

it looks like i made mistake when i tried variations with DTS file, so, my bad
thank you again, you are a king ! :)

i will close this issue, because it is not related to wifi driver
sorry for noise

closed 2020-01-11 by Npeca75

sanitariu, 2020-01-11:
Thanks for the leds it will save me time next week :)