openwrt/mt76 #347: mt76x2u IBSS unable to communicate with other stations

eSoares, 2020-01-13:
I have a Netgear AC6210, using mt76x2u setting up an IBSS network: `iw dev <> ibss join test 2417`
A second interface using the ath9k driver is setup on the same IBSS network but no communication is possible between the two. (A simple ping fails.)

Using a third interface in monitor mode (also ath9k) is visible the traffic of both stations, Beacons and ARPs being made, but is like they are in different IBSS networks. As far as I saw, the output of dmesg confirms this `No active IBSS STAs - trying to scan for other IBSS networks with same SSID (merge)`.

Communication between two ath9k interfaces in IBSS has no problem at all.

Is there a solution for this?

LorenzoBianconi, 2020-01-14:
> I have a Netgear AC6210, using mt76x2u setting up an IBSS network: iw dev
> <> ibss join test 2417
> A second interface using the ath9k driver is setup on the same IBSS
> network but no communication is [...]
Are you running an encrypted connection? If so I guess you need to
use wpad-mesh-openssl.

Regards,
Lorenzo

eSoares, 2020-01-14:
No encryption was used, I simply setup the network with `iw dev <iface> ibss join test 2427`.
I'm using IBSS not mesh.

Thanks for the reply,
Eduardo

LorenzoBianconi, 2020-01-14:
>
> No encryption was used, I simply setup the network with iw dev <iface> ibss join test 2427.
> I'm using IBSS not mesh.
>
> Thanks for the reply,
> Eduardo

Ah ok, sorry..I misread..I will test it..are you using latest mt76 trunk?

Regards,
Lorenzo

eSoares, 2020-01-14:
I'm using the version from the kernel that I'm running in my workstation, 5.3.0-26.
But will try a more recent version later today.

eSoares, 2020-01-14:
Strange thing, testing in a older kernel (5.3.0-23) everything works.
What is the best version to test the latest version of the driver?

Thanks for the help,
Eduardo

LorenzoBianconi, 2020-01-14:
> Strange thing, testing in a older kernel (5.3.0-23) everything works.
> What is the best version to test the latest version of the driver?
> 
> Thanks for the help,
> Eduardo

I tried using nbd mt76 [tree](https://github.com/nbd168/wireless) and mt76x0u is connecting properly in ibss mode to an ath9k peer

eSoares, 2020-01-15:
> I tried using nbd mt76 [tree](https://github.com/nbd168/wireless) and mt76x0u is connecting properly in ibss mode to an ath9k peer

Using the tree from nbd is working for me also. Not working 100% of the times, but is hard to actually find what makes it not working.
I notice some messages being printed out on dmesg:
```
[ 1239.728695] mt76x2u 2-1:1.0: rx urb failed: -32
[ 1239.825026] mt76x2u 2-1:1.0: vendor request req:07 off:1134 failed:-32
[ 1239.930330] mt76x2u 2-1:1.0: vendor request req:07 off:1720 failed:-32
[ 1240.060768] mt76x2u 2-1:1.0: vendor request req:07 off:108c failed:-32
[ 1240.229666] mt76x2u 2-1:1.0: vendor request req:06 off:108c failed:-32 
``` 
Thanks for the help, this version of the driver looks to be working better. 
Will close the issue. 

Thanks,
Eduardo

closed 2020-01-15 by eSoares