openwrt/mt76 #348: ZBT WE1326. High latency when on 5GHz network.

4k3or3et, 2020-01-15:
Hi Guys

i just updated my ZBT WE1326 to stable 19.07.0. When my clients are connected to 5GHz i have very high pings to local IP of the router. Pings are between 50ms to timeout. It looks very unstable i would say. When connected to 2.4GHz interface i get around 1ms all the time. Tried on few devices. Tried different channels for that interface. No luck.

EDIT:

I have notice that when i change channel width from 80MHz do 20MHz situation gets a little better. In that case ping are lower up to 250ms. Changing channel width back to 80MHz gives usually around 500ms to timeout. In both casses pings are lower than 100ms very rare.

Any clues how to resolve the issue?

Thank you.

stefanomaxenti, 2020-01-22:
It appears to be related to the country code. Changing it to 00 (World) seems to fix the problem, even though it's not the correct long-term solution. This problem wasn't present in the RC1.

nbd168, 2020-01-22:
What country code was it using before, and what channel did you test on?

stefanomaxenti, 2020-01-22:
I use channel 36, 80Mhz width, automatic power. I've never tried DFS channels.
IT country code is the one having problems (high spikes). 00 is ok.

tkso1997, 2020-01-22:
I can confirm this behavior with current master and 19.07 on a Xiaomi R3P using mt7615.
With country code 00 (World) it looks pretty consistent:

`PING 192.168.39.1 (192.168.39.1): 56 data bytes
64 bytes from 192.168.39.1: icmp_seq=0 ttl=64 time=1.849 ms
64 bytes from 192.168.39.1: icmp_seq=1 ttl=64 time=3.751 ms
64 bytes from 192.168.39.1: icmp_seq=2 ttl=64 time=3.488 ms
64 bytes from 192.168.39.1: icmp_seq=3 ttl=64 time=3.634 ms
64 bytes from 192.168.39.1: icmp_seq=4 ttl=64 time=1.939 ms
64 bytes from 192.168.39.1: icmp_seq=5 ttl=64 time=3.484 ms
64 bytes from 192.168.39.1: icmp_seq=6 ttl=64 time=2.009 ms
64 bytes from 192.168.39.1: icmp_seq=7 ttl=64 time=3.626 ms
64 bytes from 192.168.39.1: icmp_seq=8 ttl=64 time=3.475 ms
64 bytes from 192.168.39.1: icmp_seq=9 ttl=64 time=1.939 ms
64 bytes from 192.168.39.1: icmp_seq=10 ttl=64 time=3.491 ms
64 bytes from 192.168.39.1: icmp_seq=11 ttl=64 time=3.736 ms
64 bytes from 192.168.39.1: icmp_seq=12 ttl=64 time=3.477 ms
64 bytes from 192.168.39.1: icmp_seq=13 ttl=64 time=3.477 ms
64 bytes from 192.168.39.1: icmp_seq=14 ttl=64 time=3.508 ms
64 bytes from 192.168.39.1: icmp_seq=15 ttl=64 time=3.561 ms
64 bytes from 192.168.39.1: icmp_seq=16 ttl=64 time=3.640 ms
64 bytes from 192.168.39.1: icmp_seq=17 ttl=64 time=3.530 ms
64 bytes from 192.168.39.1: icmp_seq=18 ttl=64 time=3.569 ms
64 bytes from 192.168.39.1: icmp_seq=19 ttl=64 time=3.458 ms
64 bytes from 192.168.39.1: icmp_seq=20 ttl=64 time=3.799 ms
64 bytes from 192.168.39.1: icmp_seq=21 ttl=64 time=3.663 ms
64 bytes from 192.168.39.1: icmp_seq=22 ttl=64 time=3.442 ms
64 bytes from 192.168.39.1: icmp_seq=23 ttl=64 time=3.420 ms
64 bytes from 192.168.39.1: icmp_seq=24 ttl=64 time=3.423 ms
64 bytes from 192.168.39.1: icmp_seq=25 ttl=64 time=3.478 ms
64 bytes from 192.168.39.1: icmp_seq=26 ttl=64 time=3.344 ms
64 bytes from 192.168.39.1: icmp_seq=27 ttl=64 time=3.435 ms
64 bytes from 192.168.39.1: icmp_seq=28 ttl=64 time=3.489 ms
64 bytes from 192.168.39.1: icmp_seq=29 ttl=64 time=3.491 ms
64 bytes from 192.168.39.1: icmp_seq=30 ttl=64 time=3.520 ms
64 bytes from 192.168.39.1: icmp_seq=31 ttl=64 time=2.959 ms
64 bytes from 192.168.39.1: icmp_seq=32 ttl=64 time=3.503 ms
64 bytes from 192.168.39.1: icmp_seq=33 ttl=64 time=3.522 ms
64 bytes from 192.168.39.1: icmp_seq=34 ttl=64 time=3.556 ms`

while with country code DE (Germany) ist bouncing up and down:

`PING 192.168.39.1 (192.168.39.1): 56 data bytes
64 bytes from 192.168.39.1: icmp_seq=0 ttl=64 time=92.240 ms
64 bytes from 192.168.39.1: icmp_seq=1 ttl=64 time=203.219 ms
64 bytes from 192.168.39.1: icmp_seq=2 ttl=64 time=267.411 ms
64 bytes from 192.168.39.1: icmp_seq=3 ttl=64 time=138.087 ms
64 bytes from 192.168.39.1: icmp_seq=4 ttl=64 time=292.063 ms
64 bytes from 192.168.39.1: icmp_seq=5 ttl=64 time=3.871 ms
64 bytes from 192.168.39.1: icmp_seq=6 ttl=64 time=131.518 ms
64 bytes from 192.168.39.1: icmp_seq=7 ttl=64 time=244.223 ms
64 bytes from 192.168.39.1: icmp_seq=8 ttl=64 time=140.520 ms
64 bytes from 192.168.39.1: icmp_seq=9 ttl=64 time=81.165 ms
64 bytes from 192.168.39.1: icmp_seq=10 ttl=64 time=3.475 ms
64 bytes from 192.168.39.1: icmp_seq=11 ttl=64 time=82.041 ms
64 bytes from 192.168.39.1: icmp_seq=12 ttl=64 time=249.711 ms
64 bytes from 192.168.39.1: icmp_seq=13 ttl=64 time=217.108 ms
64 bytes from 192.168.39.1: icmp_seq=14 ttl=64 time=1.867 ms
64 bytes from 192.168.39.1: icmp_seq=15 ttl=64 time=147.073 ms
64 bytes from 192.168.39.1: icmp_seq=16 ttl=64 time=191.727 ms
64 bytes from 192.168.39.1: icmp_seq=17 ttl=64 time=48.921 ms
64 bytes from 192.168.39.1: icmp_seq=18 ttl=64 time=66.245 ms
64 bytes from 192.168.39.1: icmp_seq=19 ttl=64 time=292.740 ms
64 bytes from 192.168.39.1: icmp_seq=20 ttl=64 time=262.893 ms
64 bytes from 192.168.39.1: icmp_seq=21 ttl=64 time=164.762 ms
64 bytes from 192.168.39.1: icmp_seq=22 ttl=64 time=2.047 ms
64 bytes from 192.168.39.1: icmp_seq=23 ttl=64 time=2.036 ms
64 bytes from 192.168.39.1: icmp_seq=24 ttl=64 time=357.254 ms
64 bytes from 192.168.39.1: icmp_seq=25 ttl=64 time=7.466 ms
64 bytes from 192.168.39.1: icmp_seq=26 ttl=64 time=154.660 ms
64 bytes from 192.168.39.1: icmp_seq=27 ttl=64 time=46.383 ms
64 bytes from 192.168.39.1: icmp_seq=28 ttl=64 time=59.513 ms
64 bytes from 192.168.39.1: icmp_seq=29 ttl=64 time=324.081 ms
64 bytes from 192.168.39.1: icmp_seq=30 ttl=64 time=112.726 ms
64 bytes from 192.168.39.1: icmp_seq=31 ttl=64 time=27.144 ms
64 bytes from 192.168.39.1: icmp_seq=32 ttl=64 time=295.050 ms
64 bytes from 192.168.39.1: icmp_seq=33 ttl=64 time=36.802 ms
64 bytes from 192.168.39.1: icmp_seq=34 ttl=64 time=177.042 ms
64 bytes from 192.168.39.1: icmp_seq=35 ttl=64 time=11.292 ms
64 bytes from 192.168.39.1: icmp_seq=36 ttl=64 time=186.320 ms
64 bytes from 192.168.39.1: icmp_seq=37 ttl=64 time=26.493 ms
64 bytes from 192.168.39.1: icmp_seq=38 ttl=64 time=116.887 ms
64 bytes from 192.168.39.1: icmp_seq=39 ttl=64 time=262.082 ms
64 bytes from 192.168.39.1: icmp_seq=40 ttl=64 time=214.138 ms
64 bytes from 192.168.39.1: icmp_seq=41 ttl=64 time=116.194 ms
64 bytes from 192.168.39.1: icmp_seq=42 ttl=64 time=193.344 ms`

4k3or3et, 2020-01-22:
Changing country code to '00 (World)' resolves the issue. Thanks a lot guys.

tkso1997, 2020-01-22:
leave it open, as its probably a bug, right?

nbd168, 2020-01-23:
What kind of clients are you using? Does country code US trigger the same issues?

tkso1997, 2020-01-23:
Just tried US and it seems to be as good as WORLD! 
I am using a MacBook Pro (Broadcom BCM43xx 1.0).

by the way and just for completeness, the 2.4ghz network is also a bit laggy, I played with the 
country code but could not see any difference. 

Let me know if you want me to test other things / settings!

nbd168, 2020-01-23:
I've spent some time debugging this, and I found out that it's not an mt76 bug at all. The bug was in hostapd: The code applying regulatory limits (only present on ETSI regdomains) to WMM settings had a few issues that were corrupting the WMM IE.
I've fixed this in master and 19.07, and it should work fine for all regdomains now.

closed 2020-01-23 by nbd168

tkso1997, 2020-01-23:
thanks its working! :)

stefanomaxenti, 2020-01-24:
I can confirm it's working perfectly. Thanks!!

xavier83, 2020-07-27:
@4k3or3et I have the same device ZBT WE1326 and haven't been able to get 5Ghz working. Can you please point me to the last working commit and some sample settings corresponding to it. Thanks.