openwrt/mt76 #349: mt7615: slow AC wireless speed on Linksys EA8100-AH

jackcolentern, 2020-01-22:
Hi, I am trying to add support for Linksys EA8100-AH (MT7621 soc with dual MT7615N)
The AC speed is about 50~150Mbps when connected to Oneplus 6, but the original firmware can do around 350Mbps,
Thanks!

I noticed that in LUCI, Network -> Wireless -> Associated Stations, it is only showing

RX Rate / TX Rate
6.0 Mbit/s, 20MHz
585.1 Mbit/s, 80MHz, VHT-MCS 6, VHT-NSS 2, Short GI

it this normal?

jackcolentern, 2020-01-22:
I also noticed that the TX speed (from Router to device) is very good while rx speed (from device to Router) is slow

nbd168, 2020-01-23:
Please try the latest revision of OpenWrt master or 19.07. I've fixed a critical issue in hostapd.

jackcolentern, 2020-01-23:
> Please try the latest revision of OpenWrt master or 19.07. I've fixed a critical issue in hostapd.

Hi, I rebuilt and tested the latest master, unfortunately the problem is still here

jackcolentern, 2020-01-23:
After several rounds of testing, i can confirm that some time i see speed above 300Mbps or some brief amount of time, but most of the time the speed is still at 90~150Mbps

adiastra, 2020-06-30:
I would like to confirm the low transmission speeds on the latest release 

- Model	MTC Wireless Router WR1201
- Architecture	MediaTek MT7621 ver:1 eco:3
- Firmware Version	OpenWrt 19.07.2 r10947-65030d81f3 / LuCI openwrt-19.07 branch git-20.155.55664-f35803e

I have a Strong brand AC1200 router (mine is actually hootoo but its identical) and even have the second CPU enabled (it comes populated but not enabled) 

I still get average speeds below 250.

prabuselva, 2020-07-24:
I would like to confirm low 5GHz Speed on my Xiaomi R3P pro on 19.07.3 

* Model	Xiaomi Mi Router 3 Pro
* Architecture	MediaTek MT7621 ver:1 eco:3
* Firmware Version	OpenWrt 19.07.3 r11063-85e04e9f46 / LuCI openwrt-19.07 branch git-20.136.49537-fb2f363
* Kernel Version	4.14.180

My Speed differences,

* Laptop -> R3P Stock Firmware : averags around **480~510 Mpbs** (maxes at 550 Mbps) 
* Laptop -> R3P OpenWRT 19.07.3 : averages around 240-260 Mbps (maxes at 300 Mbps)

Now  its  very consistent