openwrt/mt76 #350: AirPrint and Soundtouch Speaker connection loss

tkso1997, 2020-02-04:
I am Using a Xiaomi 3Pro with Openwrt. The Wifi Chip is a mt7615.

After about one day of usage my AirPrint Printer and Bose soundtouch Speaker are no longer visible my Macbook and my Ios devices (for the soundtouch the Bose app). Both the printer and the speaker are still connected the wifi (at least the luci wireless page states that). After I rebooted the router both things work again for the next couple of hours. I think it has something to do with the wifi as using the closed source drivers (MTK with the same version of OpenWrt) does not lead to that kind of weird disconnection issues after some time. 

the issue happens at least with the last master builds and 19.07.X 

Any ideas what can cause the problems or is there anything I could do to help debugging it?

nbd168, 2020-02-04:
What's the revision number of the latest master build that you tried?

tkso1997, 2020-02-04:
OpenWrt SNAPSHOT, r12173-01c8f2e97c

tkso1997, 2020-04-15:
I did some more research and testing. 
As soon as my printer and Soundtuch speaker becoming unavailable I tried to ping their IP address from my MacBook. The result is that there is no ping going through. BUT I can still ping them from the router! I If I restart the wifi or reboot the router the connection is working again for a couple of hours before it starts agin...   
The problem exists till the current master and I suspect its caused by hosted-2.9. I made an 19.07.2 image with the old hostapd-2.8 (from  4 May 2019, commit 8af79550e6c280717660f66032d89d21007b15d2) and it works perfectly, no disconnects after a couple of days. Where should I report that, who is responsible for the hostapd implementation in openwrt.

tkso1997, 2020-05-05:
Problem still exists on newest master, while the same snapshot works on r7800 and c2600 (both use athk10) without this issue. Therefore I guess it has something to do with the mt76 wifi drivers. Its a really annoying bug! Let me know If I can do something to help narrowing the problem down ...

tkso1997, 2020-05-12:
apparently changing the time interval for the rekeying GTK to 0 sort of solves the issue. Maybe this information helps to find the underlying problem? Again: athk10 in the same network works fine.