openwrt/mt76 #351: mt7615 on netgear r6260

bricco1981, 2020-02-08:
`root@OpenWrt:~# iw dev
phy#1
	Interface wlan1
		ifindex 11
		wdev 0x100000005
		addr 8c:3b:ad:xx:xx:xx
		ssid OpenWrt-5ghz
		type managed
		channel 100 (5500 MHz), width: 80 MHz, center1: 5530 MHz
		txpower 4.00 dBm
		multicast TXQ:
			qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcoltx-bytes	tx-packets
			0	0	0	0	0	0	0	00
phy#0
	Interface wlan0
		ifindex 3
		wdev 0x1
		addr 8c:3b:ad:xx:xx:xx
		type managed
		txpower 23.00 dBm
		multicast TXQ:
			qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcoltx-bytes	tx-packets
			0	0	0	0	0	0	0	00
root@OpenWrt:~# 
`hi, txpower on mt 7615 only 4dbm, also the rx rate still low and only at 20/40 Mhz, tx rate looks good.

nicefile, 2020-02-12:
Got no problem using recent official 19.07.01 @R6260
 I have region IN (India) under OpenWrt and 22db output for 5Ghz no problem

bricco1981, 2020-02-12:
still same problem on last snapshot, i don't see any 19.07.01 for this device, mabie you used image for r 6350

bricco1981, 2020-02-15:
i'm trying to compile and install snapshot every day but this problem still here, very bad range, the radio mt7603 is over the top, working really good.

closed 2020-04-01 by bricco1981