openwrt/mt76 #354: missing band 1 for phy1  (dual MT7615 radios [DIR-882])

i3roly, 2020-02-10:
hello,

first, i do not have the IEEE freq limit encoded in my DTS, so i find it weird that the second radio only shows the 5ghz band. why would one radio show both but the other only show one? i 

> root@DD-WRT:~# iw phy
Wiphy phy1
        max # scan SSIDs: 4
        max scan IEs length: 2247 bytes
        max # sched scan SSIDs: 0
        max # match sets: 0
        max # scan plans: 1
        max scan plan interval: -1
        max scan plan iterations: 0
        Retry short limit: 7
        Retry long limit: 4
        Coverage class: 5 (up to 2250m)
        Device supports RSN-IBSS.
        Supported Ciphers:
                * WEP40 (00-0f-ac:1)
                * WEP104 (00-0f-ac:5)
                * TKIP (00-0f-ac:2)
                * CCMP-128 (00-0f-ac:4)
                * CCMP-256 (00-0f-ac:10)
                * GCMP-128 (00-0f-ac:8)
                * GCMP-256 (00-0f-ac:9)
                * CMAC (00-0f-ac:6)
                * CMAC-256 (00-0f-ac:13)
                * GMAC-128 (00-0f-ac:11)
                * GMAC-256 (00-0f-ac:12)
        Available Antennas: TX 0xf RX 0xf
        Configured Antennas: TX 0xf RX 0xf
        Supported interface modes:
                 * IBSS
                 * managed
                 * AP
                 * AP/VLAN
                 * monitor
        Band 2:
                Capabilities: 0x1ff
                        RX LDPC
                        HT20/HT40
                        SM Power Save disabled
                        RX Greenfield
                        RX HT20 SGI
                        RX HT40 SGI
                        TX STBC
                        RX STBC 1-stream
                        Max AMSDU length: 3839 bytes
                        No DSSS/CCK HT40
                Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
                Minimum RX AMPDU time spacing: 4 usec (0x05)
                HT TX/RX MCS rate indexes supported: 0-31
                VHT Capabilities (0x338001fa):
                        Max MPDU length: 11454
                        Supported Channel Width: 160 MHz, 80+80 MHz
                        RX LDPC
                        short GI (80 MHz)
                        short GI (160/80+80 MHz)
                        TX STBC
                        RX antenna pattern consistency
                        TX antenna pattern consistency
                VHT RX MCS set:
                        1 streams: MCS 0-9
                        2 streams: MCS 0-9
                        3 streams: MCS 0-9
                        4 streams: MCS 0-9
                        5 streams: not supported
                        6 streams: not supported
                        7 streams: not supported
                        8 streams: not supported
[... 335 lines cut ...]
        HT Capability overrides:
                 * MCS: ff ff ff ff ff ff ff ff ff ff
                 * maximum A-MSDU length
                 * supported channel width
                 * short GI for 40 MHz
                 * max A-MPDU length exponent
                 * min MPDU start spacing
        Device supports TX status socket option.
        Device supports HT-IBSS.
        Device supports SAE with AUTHENTICATE command
        Device supports low priority scan.
        Device supports scan flush.
        Device supports AP scan.
        Device supports per-vif TX power setting
        Driver supports full state transitions for AP/GO clients
        Driver supports a userspace MPM
        Device supports active monitor (which will ACK incoming frames)
        Device supports configuring vdev MAC-addr on create.
        Supported extended features:
                * [ VHT_IBSS ]: VHT-IBSS
                * [ RRM ]: RRM
                * [ FILS_STA ]: STA FILS (Fast Initial Link Setup)
                * [ CQM_RSSI_LIST ]: multiple CQM_RSSI_THOLD records
                * [ CONTROL_PORT_OVER_NL80211 ]: control port over nl80211
                * [ TXQS ]: FQ-CoDel-enabled intermediate TXQs

nbd168, 2020-02-14:
Supported bands are enabled based on EEPROM data. This does not look like a bug to me

closed 2020-02-14 by nbd168

tarori, 2020-02-19:
I have a same problem.
Band 1 worked on previous openwrt master branch, but not work on current master.
I found that this commit https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=ea5078014d3f8a6f662e41cf9dec23028ca303e3 broke band 1.
So, it should be fixed.

i3roly, 2020-02-19:
tarori: do you find that this commit possibly broke the mt76 driver in terms of performance?

it is strange it performs so poorly

nbd168, 2020-02-19:
Please try latest master branch.

tarori, 2020-02-21:
Both band 0 and 1 worked with the latest commit. Thanks.
Band 1 (11ac) worked well though band 0 (11n) is very slow (~ 2Mbps). But it seemed to be the different problem.