openwrt/mt76 #355: MT7615 LEDs not working (even after adding -DCONFIG_MT76_LEDS)

i3roly, 2020-02-11:
hi,

i noticed lorenzo helped dengqf6 fix his issue with the absent wifi LEDS. it is a similar situation for me, except i added the macro and the lights are still not functioning.

@dengqf6 : got it, we are missing this [patch](https://github.com/LorenzoBianconi/mt76/commit/e1cb717eb6c5f0f5f616edfd77670e00dcad1391) for mt76 github repo.
Could you please try to use this [mt76](https://github.com/LorenzoBianconi/mt76/tree/mt76-devel) branch?

_Originally posted by @LorenzoBianconi in https://github.com/openwrt/mt76/issues/330#issuecomment-563343237_

any suggestinos?

dengqf6, 2020-02-11:
Are those LEDs connected to the MT7615 chip directly?

i3roly, 2020-02-11:
hi dengqf6

i am not sure. i can poke around the proprietary driver directory to see, but it seems these LEDs are not able to be manipulated through the gpio export interface in /sys/class

i can screw with all of the frontl ights on a DIR882 except for the two wifi LEDs.

if four of the six LEDS on the front of the router are directly manipulable  (on/off etc) using /sys/class/gpio/export, then shouldn't the WIFI leds be the same?

i noticed you commented on the DIR-878 ticket, and that would be a router very similar to mine. i note that this router's current configuration does not have anything for the wifi LEDs.

i3roly, 2020-02-13:
> Are those LEDs connected to the MT7615 chip directly?

i also noticed that toggling the WAN LED is very difficult, if not "impossible" (not really, just haven't screwed with stuff enough yet) without RALINK_GPIO.

with RALINK_GPIO i can at least get all LEDS save the WIFI working, and with  GPIO_BUTTON_HOTPLUG i can use the buttons.

from my perspective it seems i have done enough to prepare this router for the full-fledged mt76 support.

the TODO list is as follows:

1. see if we can add RFKILL support to the mt76 driver. that will at least render one button useful
2. see if we can add LED Support within the MT76 driver. i will use my familiarity with mt_wifi to see if i can help here.
3. i dunno wtf to use the WPS button for, but let's find a good purpose for that. i would rather use the protruding buttons (SES/WPS) to disable each radio instead of one doing both, but i suspect end users won't mind holding a button to toggle modes and such.

Djfe, 2023-01-08:
Do we need an entry here? (like ucidef_set_led_wlan)
https://github.com/openwrt/openwrt/blob/fc33c41c21362b7186aa051a2140623943fa3143/target/linux/ramips/mt7621/base-files/etc/board.d/01_leds#L86

or is it missing a device package?
`kmod-mt7615e` (or something similar)
https://github.com/Znevna/openwrt/blob/c4b806d5c4ccc653968620e6e9aec93bc4e370e5/target/linux/ramips/image/mt7621.mk#L604

Or did this get fixed over time?

A router where it works and similar lines were added:
https://github.com/openwrt/openwrt/pull/11601/files