openwrt/mt76 #358: MT7621: Clients disconnecting.

Openwrtfunboy, 2020-02-15:
Hi! I have MT7621 and when to router connected more than 10 peoples router start disconnecting some peoples(have two networks guest and main on 2,4).

There is my settings:
config wifi-iface 'default_radio0'
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option wpa_disable_eapol_key_retries '1'
        option key 'WIFIPASS'
        option ssid 'WIFINAME'
        option encryption 'psk2+ccmp'
        option disassoc_low_ack '0'
(guest network have same settings)

I also noticed that before the client is disconnected from the network, it will lose the Internet.

Other info:
Newifi-D2
MediaTek MT7621 ver:1 eco:3
OpenWrt 19.07.1 r10911-c155900f66 / LuCI openwrt-19.07 branch git-20.045.27998-49999e9

I see it problem in other issue topic.
1: https://github.com/openwrt/mt76/issues/169#issuecomment-477673633
2: https://github.com/openwrt/mt76/issues/169#issuecomment-480727386
I also noticed that before the client is disconnected from the network, it will lose the Internet.

Openwrtfunboy, 2020-02-16:
I also suggest that this is because I have two wifi networks. When I had one and 8 active clients were connected to it, everything was fine, but as soon as the clients began to connect also to the guest, then there were problems.

I will try to unite the networks, but this is not a solution and causes a lot of problems, because many clients will have to rewrite the password.

Openwrtfunboy, 2020-02-16:
I found peoples with same problem:

Openwrtfunboy, 2020-02-16:
I found peoples with same problem:
1: https://github.com/openwrt/mt76/issues/146#issuecomment-458372609
2: https://github.com/openwrt/mt76/issues/146#issuecomment-484582241

Its seriously problem, OpenWRT team need to fix it!

Openwrtfunboy, 2020-02-16:
I'll try to clarify again:
When many clients were connected to the main network, everything was fine, but when they connected to the guest as well, big problems started.

Openwrtfunboy, 2020-02-16:
Empirically, I found out that the number of networks does not affect the problem. Even if you transfer clients to the same network, disconnections still occur.

Openwrtfunboy, 2020-02-17:
Empirically, it was found out that not even the number of clients, but the distance to the router affects the outages. If there are ~ 3 clients next to the router, then everyone else will experience problems. If these clients are not near the routers, then everything will be fine

JFtico, 2020-02-18:
Also seeing this on another MT7621/MT7603e platform, a ZBT WE3526 (512MB RAM) running OpenWRT 19.07.0

In a rural, uncongested WiFi setting, It handles six or so very active clients fine, even if very close (within 3') to the unit. I have not tested a mix of near + far like the OP.
But what is occurring is that if the wifi context is extremely congested (8+ APs on every channel) in an urban setting (my mixed-use office building), even three clients do not remain connected for more than 30 minutes. Even locking in signal strength and channels, it still happens.

Openwrtfunboy, 2020-02-19:
> Also seeing this on another MT7621/MT7603e platform, a ZBT WE3526 (512MB RAM) running OpenWRT 19.07.0
>
> In a rural, uncongested WiFi setting, It handles six or so very active clients fine, even if [...]

And before everything was good? If so, on which version was it?

JFtico, 2020-02-19:
> And before everything was good? If so, on which version was it?

It was 18.06.4 and the Wifi was worse, with regular crashes caught in the system log.

The new behavior is puzzling, as there is no crash (or anything) in the system log. The clients just lose connectivity, and eventually (minutes) are able to reconnect.

on 19.07, it's OK in an isolated Wifi environment (only two or three competing APs, low client count), It is much, much worse the more local WiFi contention there is. 

Does anyone know if MT76 Wifi has an ANI (Adaptive Noise Isolation) function & associated settings?
If so, maybe turning that off in congested environments might address the disconnects.

Openwrtfunboy, 2020-02-19:
> It was 18.06.4 and the Wifi was worse, with regular crashes caught in the system log.
>
> The new behavior is puzzling, as there is no crash (or anything) in the system log. The clients just lose [...]

In general, I studied this topic and talked with a lot of people in the local Russian forum, as well as with the seller of my router. Apparently, the only solution will be to migrate to Padavan firmware.

This is a popular and very old issue. It is unlikely that OpenWRT will ever fix that.

Openwrtfunboy, 2020-02-19:
**IF HERE IS A DEVELOPER OPENWRT, PLEASE REPLACE DRIVER FOR MT7603 TAKEN FROM FIRMWARE PADAVAN OR PANDORABOX, WHERE 2,4 GHz. CHANNEL WORKS GOOD.

ENOUGH TO IGNORE THIS FOREIGN PROBLEM!!!**

_I am more and more disappointed in OpenWRT._

Rising-Sun, 2020-02-20:
If you know the solution please ignore openwrt. 

Big thanks to nbd and the team. The driver  may not be perfect but I can use my xiaomi mir3g as my daily driver which was unthinkable 1-2 years ago! I appreciate the big efforts made to the driver! 

Using latest snapshot in an environment with >20 APs visible from my router on the 2.4 GHz network

JFtico, 2020-02-20:
I also appreciate the hard work of the OpenWRT team to bring one of the few Open Source WiFi platforms forward.

> Using latest snapshot in an environment with >20 APs visible from my router on the 2.4 GHz network

That is great to hear and hopefully a sign the code is handling corner cases better.

I do want to inquire why 19.07 (released in February 2020) has four and six-year-old firmware blobs, are these the most recent available?

` kern.warn kernel: [   14.471114] PCI: Enabling device 0000:00:01.0 (0004 -> 0006)
 kern.info kernel: [   14.476961] mt7603e 0000:02:00.0: ASIC revision: 76030010
 kern.info kernel: [   15.551185] mt7603e 0000:02:00.0: Firmware Version: ap_pcie
 kern.info kernel: [   15.556763] mt7603e 0000:02:00.0: Build Time: 20160107100755
 kern.info kernel: [   15.600261] mt7603e 0000:02:00.0: firmware init done
 kern.debug kernel: [   15.772599] ieee80211 phy0: Selected rate control algorithm 'minstrel_ht'
 kern.warn kernel: [   15.788556] bus=0x1, slot = 0x0, irq=0xff
 kern.warn kernel: [   15.792689] PCI: Enabling device 0000:00:00.0 (0004 -> 0006)
 kern.info kernel: [   15.798531] mt76x2e 0000:01:00.0: ASIC revision: 76120044
 kern.info kernel: [   16.511248] mt76x2e 0000:01:00.0: ROM patch build: 20141115060606a
 kern.info kernel: [   16.521248] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
 kern.info kernel: [   16.526733] mt76x2e 0000:01:00.0: Build: 1
 kern.info kernel: [   16.530869] mt76x2e 0000:01:00.0: Build Time: 201507311614____
 kern.info kernel: [   16.550267] mt76x2e 0000:01:00.0: Firmware running!
`
There is an open pull request from July 2019 https://github.com/openwrt/mt76/pull/295 indicating that firmware from 2017 is available for the 7603/7621 pair.
Are there problems with that 2017 version, if not, other than doing a custom build, any builds include that?

nbd168, 2020-02-20:
As far as I can tell, there are no relevant changes in the newer firmware blop that make it worth upgrading to.

JFtico, 2020-02-20:
@nbd168 Thanks for the update on the firmware blobs, so the fix for the client's disconnecting identified in this thread is in an existing build or still an open issue?

Thanks in advance

Openwrtfunboy, 2020-02-20:
> As far as I can tell, there are no relevant changes in the newer firmware blop that make it worth upgrading to.

Is work in progress or planned to correct this error on the mt7621/mt7603e platform in the future?

Openwrtfunboy, 2020-02-20:
> If you know the solution please ignore openwrt.
>
> Big thanks to nbd and the team. The driver may not be perfect but I can use my xiaomi mir3g as my daily driver which was unthinkable 1-2 years ago! I [...]

However, this does not solve the existing problem. I switched to another firmware only because I cannot use OpenWRT. Look at my nickname again. Thanks.

nbd168, 2020-02-20:
You could try this: `echo 1 > /sys/kernel/debug/ieee80211/phy0/mt76/dynamic_sensitivity`

Openwrtfunboy, 2020-02-21:
> You could try this: `echo 1 > /sys/kernel/debug/ieee80211/phy0/mt76/dynamic_sensitivity`

**It really works and is a fix.** But before the developers include this in the main branch, I would like to see feedback from other participants who also applied this fix. What about you, @JFtico?

Thanks, @nbd168 I love you!

JFtico, 2020-02-21:
Thank you, will give that a shot this weekend.

Openwrtfunboy, 2020-02-23:
> Thank you, will give that a shot this weekend.

Hi! Did you test it, @JFtico?

9000000, 2020-02-24:
bản cập nhật gần đây nhất mà tôi sử dụng "OpenWrt SNAPSHOT r12288-1173719817" không biết đã sửa lỗi này hay chưa nhưng các bản trước tôi cũng gặp trường hợp tương tự. Còn phiên bản hiện tại tôi đang sử dụng thì không gặp lỗi này

Openwrtfunboy, 2020-02-24:
> The latest update that I use "OpenWrt SNAPSHOT r12288-1173719817" does not know if I've fixed this error yet but the previous versions I have encountered the same case. And the current version I am using does not encounter this error

Yes, because it fixed now by default in new builds.

JFtico, 2020-02-24:
@Openwrtfunboy @nbd168 In the testing this weekend, the disconnects were indeed fixed.

However, there is still a problem, even though the stations remain connected to wifi, the devices lose Internet connectivity.  The device must cycle the wifi connection before it can re-establish Internet access. So while there is progress, I await to test with a 19.0.7.2 build before I'd call this fixed, unless there is a commit for the loss of internet as well.

Openwrtfunboy, 2020-02-24:
> @Openwrtfunboy @nbd168 In the testing this weekend, the disconnects were indeed fixed.
>
> However, there is still a problem, even though the stations remain connected to wifi, the devices lose [...]

Can you provide as much information as possible? Your observations? Assumptions?

JFtico, 2020-02-24:
> Can you provide as much information as possible? Your observations? Assumptions?

I was able to see that the devices were still connected to the Wifi, but they were showing "No internet".

For some reason, the devices wouldn't disconnect and connect back. I had to manually disconnect and connect back.

Openwrtfunboy, 2020-02-24:
> I was able to see that the devices were still connected to the Wifi, but they were showing "No internet".
>
> For some reason, the devices wouldn't disconnect and connect back. I had to manually [...]

**Please provide even more information.** How long does this happen, how many clients, do the same outages occur with LAN users?

Rising-Sun, 2020-02-24:
> Yes, because it fixed now by default in new builds.

No it shouldn’t be fixed because the new Mt76 driver hasn’t be pushed to the openwrt master branch, you have to manually change the makefile of the Mt76 driver.

JFtico, 2020-02-24:
> Please provide even more information. 

The WiFi devices were losing the internet after a couple of hours.
Not all devices were losing the internet, the TV never lost Internet.

This was with around 23-26 devices connected at all times.

Wired devices remained connected, so it was something in the wifi stack.

Openwrtfunboy, 2020-02-25:
> > Yes, because it fixed now by default in new builds.
> 
> No it shouldn’t be fixed because the new Mt76 driver hasn’t be pushed to the openwrt master branch, you have to manually change the makefile of the Mt76 driver.

Take a look: https://github.com/openwrt/mt76/commit/master
https://github.com/openwrt/mt76/commit/0a53dcda520346d45c4b280a19ff73cd50b0414b

Openwrtfunboy, 2020-02-25:
> The WiFi devices were losing the internet after a couple of hours.
> Not all devices were losing the internet, the TV never lost Internet.
>
> This was with around 23-26 devices connected at all times. [...]

So, we can wait only @nbd168

9000000, 2020-02-25:
> I was able to see that the devices were still connected to the Wifi, but they were showing "No internet".
>
> For some reason, the devices wouldn't disconnect and connect back. I had to manually [...]

I have the same situation, I use xiaomi r3g v1 router, my house only uses a few devices

nbd168, 2020-02-25:
I've pushed the mt76 update to OpenWrt master

Openwrtfunboy, 2020-02-25:
> I've pushed the mt76 update to OpenWrt master

@9000000  and @JFtico  have bug with wifi.

Openwrtfunboy, 2020-02-26:
@JFtico if you have another problem(not disconnecting) - open another issue. Now there my problem is solved.

closed 2020-02-26 by Openwrtfunboy

ergg, 2020-04-22:
Hi , I did have exactly the same problem after the "echo 1 > /sys/kernel/debug/ieee80211/phy0/mt76/dynamic_sensitivity" my 5GHz Network works now and I can stop/start the Interfaces without any issues.