openwrt/mt76 #360: MT7615 poor performance (or lack thereof) on DIR882 (in general)

i3roly, 2020-02-17:
hello everyone,

i am having many issues with the MT76 driver on the MT7615.

i was willing to look past the lack of WLAN leds, but another disappointing thing is that on higher 5GHz widths (such as VHT160), older devices seem to be able to detect, but not connect to the AP.

right now i'm running hostapd in auto mode on VHT160, and my 2012 Mac Pro (as an example) can see the network but not connect. i would like to get some feedback on what the problems may be, because it is frustrating. i want this driver to work the way you guys do, but it is still not even able to touch the proprietary variant.

given how much work BS has invested in the mac80211 side of things (i'd call him the king of ath5/9/10k), i would like to leverage his hard work for the mt7615.

the 2Ghz is not much better. i can connect without issue but the speeds sometime will be OK (like 70Mbps) and then it'll drop like a rock. it is super weird.

please help. below is my 5Ghz config to give you an idea

> root@DD-WRT:~# cat /tmp/ath1_hostap.conf 
driver=nl80211
ctrl_interface=/var/run/hostapd
wmm_ac_bk_cwmin=4
wmm_ac_bk_cwmax=10
wmm_ac_bk_aifs=7
wmm_ac_bk_txop_limit=0
wmm_ac_bk_acm=0
wmm_ac_be_aifs=3
wmm_ac_be_cwmin=4
wmm_ac_be_cwmax=10
wmm_ac_be_acm=0
wmm_ac_vi_aifs=2
wmm_ac_vi_cwmin=3
wmm_ac_vi_cwmax=4
wmm_ac_vi_txop_limit=94
wmm_ac_vi_acm=0
wmm_ac_vo_aifs=2
wmm_ac_vo_cwmin=2
wmm_ac_vo_cwmax=3
wmm_ac_vo_txop_limit=47
wmm_ac_vo_acm=0
tx_queue_data3_aifs=7
tx_queue_data3_cwmin=15
tx_queue_data3_cwmax=1023
tx_queue_data3_burst=0
tx_queue_data2_aifs=3
tx_queue_data2_cwmin=15
tx_queue_data2_cwmax=63
tx_queue_data1_aifs=1
tx_queue_data1_cwmin=7
tx_queue_data1_cwmax=15
tx_queue_data1_burst=3.0
tx_queue_data0_aifs=1
tx_queue_data0_cwmin=3
tx_queue_data0_cwmax=7
tx_queue_data0_burst=1.5
country_code=IE
tx_queue_data2_burst=2.0
wmm_ac_be_txop_limit=64
ieee80211n=1
dynamic_ht40=0
ht_capab=[HT40+][LDPC][SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC1]
vht_capab=[RXLDPC][SHORT-GI-80][TX-STBC-2BY1][RX-STBC-1][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN][MAX-MPDU-11454][MAX-A-MPDU-LEN-EXP7]
ieee80211ac=1
ieee80211d=1
ieee80211h=1
vht_oper_chwidth=1
vht_oper_centr_freq_seg0_idx=42
hw_mode=a
basic_rates=60 120 240
channel=36
frequency=5180
beacon_int=100
airtime_mode=0

interface=ath1
preamble=0
disassoc_low_ack=1
wmm_enabled=1
uapsd_advertisement_enabled=0
bssid=F6:68:5E:AF:C7:13
ignore_broadcast_ssid=0
max_num_sta=256
dtim_period=2
ssid=AGT1337-5.0G
bridge=br0
logger_syslog=-1
logger_stdout=-1
logger_stdout_level=2
eapol_version=1
eapol_key_index_workaround=0
wpa=2
ieee80211w=1
sae_require_mfp=1
wpa_passphrase=xxxxxx
wpa_key_mgmt=WPA-PSK SAE WPA-PSK-SHA256 
sae_groups=19 20 21
wpa_pairwise=CCMP CCMP-256
group_mgmt_cipher=AES-128-CMAC
okc=1
wpa_group_rekey=3600
signal_connect=-128
signal_stay=-128
signal_poll_time=10
signal_strikes=3

bss=ath1.1
disassoc_low_ack=1
wmm_enabled=1
uapsd_advertisement_enabled=0
bssid=F4:68:5E:AF:C7:13
ignore_broadcast_ssid=0
max_num_sta=256
dtim_period=2
ssid=x1
bridge=br0
logger_syslog=-1
logger_stdout=-1
logger_stdout_level=2
eapol_version=1
eapol_key_index_workaround=0
wpa=2
ieee80211w=1
sae_require_mfp=1
wpa_passphrase=highband!
wpa_key_mgmt=WPA-PSK SAE WPA-PSK-SHA256 
sae_groups=19 20 21
wpa_pairwise=CCMP CCMP-256
group_mgmt_cipher=AES-128-CMAC
okc=1
wpa_group_rekey=3600
signal_connect=-128
signal_stay=-128
signal_poll_time=10
signal_strikes=3

bss=ath1.2
disassoc_low_ack=1
wmm_enabled=1
uapsd_advertisement_enabled=0
bssid=F0:68:5E:AF:C7:13
ignore_broadcast_ssid=0
max_num_sta=256
dtim_period=2
ssid=----to the fellas out there
bridge=br0
logger_syslog=-1
logger_stdout=-1
logger_stdout_level=2
eapol_version=1
eapol_key_index_workaround=0
wpa=2
ieee80211w=1
sae_require_mfp=1
wpa_passphrase=xxxx
wpa_key_mgmt=WPA-PSK SAE WPA-PSK-SHA256 
sae_groups=19 20 21
wpa_pairwise=CCMP CCMP-256
group_mgmt_cipher=AES-128-CMAC
okc=1
wpa_group_rekey=3600
signal_connect=-128
signal_stay=-128
signal_poll_time=10
signal_strikes=3

bss=ath1.3
disassoc_low_ack=1
wmm_enabled=1
uapsd_advertisement_enabled=0
bssid=FC:68:5E:AF:C7:13
ignore_broadcast_ssid=0
max_num_sta=256
dtim_period=2
ssid=---with ladies to impress
bridge=br0
logger_syslog=-1
logger_stdout=-1
logger_stdout_level=2
eapol_version=1
eapol_key_index_workaround=0
wpa=2
ieee80211w=1
sae_require_mfp=1
wpa_passphrase=xxxx
wpa_key_mgmt=WPA-PSK SAE WPA-PSK-SHA256 
sae_groups=19 20 21
wpa_pairwise=CCMP CCMP-256
group_mgmt_cipher=AES-128-CMAC
okc=1
wpa_group_rekey=3600
signal_connect=-128
signal_stay=-128
signal_poll_time=10
signal_strikes=3

bss=ath1.4
disassoc_low_ack=1
wmm_enabled=1
uapsd_advertisement_enabled=0
bssid=F8:68:5E:AF:C7:13
ignore_broadcast_ssid=0
max_num_sta=256
dtim_period=2
ssid=--it's easy to do
bridge=br0
logger_syslog=-1
logger_stdout=-1
logger_stdout_level=2
eapol_version=1
eapol_key_index_workaround=0
wpa=2
ieee80211w=1
sae_require_mfp=1
wpa_passphrase=xxxx
wpa_key_mgmt=WPA-PSK SAE WPA-PSK-SHA256 
sae_groups=19 20 21
wpa_pairwise=CCMP CCMP-256
group_mgmt_cipher=AES-128-CMAC
okc=1
wpa_group_rekey=3600
signal_connect=-128
signal_stay=-128
signal_poll_time=10
signal_strikes=3

bss=ath1.5
disassoc_low_ack=1
wmm_enabled=1
uapsd_advertisement_enabled=0
bssid=E4:68:5E:AF:C7:13
ignore_broadcast_ssid=0
max_num_sta=256
dtim_period=2
ssid=--just follow these steps:
bridge=br0
logger_syslog=-1
logger_stdout=-1
logger_stdout_level=2
eapol_version=1
eapol_key_index_workaround=0
wpa=2
ieee80211w=1
sae_require_mfp=1
wpa_passphrase=xxxx
wpa_key_mgmt=WPA-PSK SAE WPA-PSK-SHA256 
sae_groups=19 20 21
wpa_pairwise=CCMP CCMP-256
group_mgmt_cipher=AES-128-CMAC
okc=1
wpa_group_rekey=3600
signal_connect=-128
signal_stay=-128
signal_poll_time=10
signal_strikes=3

i3roly, 2020-02-17:
looks like those "damned chinese" really have done a number on mt76 in terms of the mt7615 guys.

i will be eagerly, probably periodically, checking on your progress here.

i have fans to please, i am sorry

hisirbr, 2020-02-19:
Hi all!
I also have the same issue with very poor performance of DIR-882 (MT7621+2xMT7615), especially for 2.4GHz.
All tests were done with the latest commit of 'master' branch of this repository (using iperf3 for WiFi<-->LAN eth port and speedtest for WiFi<-->WAN port). 

For 5GHz speed is about 30-250Mbps (I use different 1x1 and 2x2 clients (e.g. Intel 8265), OS's, etc.)
For 2.4GHz clients can connect to the AP, but speed is very low (1-5Mbps for download and below 1Mbps for upload) and often client is disconnected during the test. Encryption doesn't matter.
WiFi config for 2.4GHz:
```
config wifi-device 'radio0'
        option type 'mac80211'
        option hwmode '11g'
        option path 'pci0000:00/0000:00:00.0/0000:01:00.0'
        option legacy_rates '0'
        option htmode 'HT40'
        option channel '11'
        option country 'RU'
```
I found that `option noscan '1'` can help to improve 2.4GHz performance (I get at least 20-30Mbps in this case). But anyway WiFi is still unstable and random disconnects often occurs:
`IEEE 802.11: disconnected due to excessive missing ACKs`.

Temperature of the chip is stable:
```
root@OpenWrt:/# cat /sys/kernel/debug/ieee80211/phy0/mt76/temperature 
Temperature: 48
```

@nbd168
If I you need any information to fix this bugs, please let me know and I'll try to provide it.
I mean some specific info from debugfs during the test or wireshark logs or some specific bits from 'factory' partition.

hisirbr, 2020-02-19:
Additional info:
Results of speedtest with Intel 8265AC and Windows10 (ISP limit is 100Mbps upload and download).
2.4GHz with `option noscan '1'`: download: 70Mbps, upload: 15Mbps
5Ghz without any additional options: download: >90Mbps, upload: 60Mbps.