openwrt/mt76 #365: MT7621|MT7612EN: 5GHz low bitrait and poor speed.

Openwrtfunboy, 2020-02-24:
Only one device (TV box) is connected to my 5 GHz network, the first time after rebooting the router (or chip) the bitrate is from 130 to 450 mbs, which gives the entire speed of my connection, but after a while (about a day) the bitrate drops to 30- 60, which gives an extremely low speed even for viewing in 720. Is there any way to fix the bitrate on the 5GHz channel?

Openwrtfunboy, 2020-02-24:
I also noticed that the bitrait stopped at 65 megabits.

Openwrtfunboy, 2020-02-24:
I noticed that if several clients are connected to the network, then there is no problem. Perhaps i can somehow manually fixate the minimum bitrate?

Openwrtfunboy, 2020-02-25:
@JFtico Can you test 5Ghz?

closed 2020-03-24 by Openwrtfunboy

carras, 2021-04-01:
any luck with this? I have issues too.