openwrt/mt76 #366: Kernel panic

taeasy, 2020-02-25:
Device: Newifi-D2
OpenWrt version: 19.07.1 r10911-c155900f66

[27226.008011] mtk_soc_eth 1e100000.ethernet eth0: port 3 link down
[27228.980608] mtk_soc_eth 1e100000.ethernet eth0: port 3 link up
[60114.949063] mtk_soc_eth 1e100000.ethernet eth0: port 1 link up
[60146.647874] mtk_soc_eth 1e100000.ethernet eth0: port 2 link down
[60155.584307] mtk_soc_eth 1e100000.ethernet eth0: port 2 link up
[60223.819350] ------------[ cut here ]------------
[60223.824060] WARNING: CPU: 2 PID: 19 at backports-4.19.98-1/net/mac80211/tx.c:4270 0x8f7ac914 [mac80211@8f780000+0x6c840]
[60223.834925] Modules linked in: pppoe ppp_async pppox ppp_generic nf_conntrack_ipv6 mt76x2e mt76x2_common mt76x02_lib mt7603e mt76 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_FLOWOFFLOAD xt_CT slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_mangle iptable_filter ip_tables crc_ccitt compat ledtrig_usbport nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 leds_gpio xhci_plat_hcd xhci_pci xhci_mtk xhci_hcd gpio_button_hotplug usbcore nls_base usb_common
[60223.905881] CPU: 2 PID: 19 Comm: ksoftirqd/2 Not tainted 4.14.167 #0
[60223.912213] Stack : 00000000 00000000 00000000 8f1309b0 00000000 00000000 00000000 00000000
[60223.920557]         00000000 00000000 00000000 00000000 00000000 00000001 8fc83b70 ac07f5be
[60223.928898]         8fc83c08 00000000 00000000 00006798 00000038 8049b6d8 00000008 00000000
[60223.937235]         00000000 80550000 000dd299 00000000 8fc83b50 00000000 00000000 8f7e7ba0
[60223.945575]         8f7ac914 000010ae 8f596480 8f1309b0 00000003 802ac758 00000008 806b0008
[60223.953916]         ...
[60223.956355] Call Trace:
[60223.956410] [<8049b6d8>] 0x8049b6d8
[60223.962314] [<8f7ac914>] 0x8f7ac914 [mac80211@8f780000+0x6c840]
[60223.968221] [<802ac758>] 0x802ac758
[60223.971700] [<8000c1a0>] 0x8000c1a0
[60223.975174] [<8000c1a8>] 0x8000c1a8
[60223.978650] [<80484544>] 0x80484544
[60223.982140] [<8f7ac914>] 0x8f7ac914 [mac80211@8f780000+0x6c840]
[60223.988030] [<800729c4>] 0x800729c4
[60223.991503] [<8002e5f8>] 0x8002e5f8
[60223.994984] [<8f7ac914>] 0x8f7ac914 [mac80211@8f780000+0x6c840]
[60224.000905] [<8002e6e0>] 0x8002e6e0
[60224.004381] [<80053adc>] 0x80053adc
[60224.007858] [<8f7ac914>] 0x8f7ac914 [mac80211@8f780000+0x6c840]
[60224.013757] [<8f7af14c>] 0x8f7af14c [mac80211@8f780000+0x6c840]
[60224.019657] [<80067be8>] 0x80067be8
[60224.023138] [<80017f6c>] 0x80017f6c
[60224.026644] [<8ec57b10>] 0x8ec57b10 [mt76x02_lib@8ec50000+0x9640]
[60224.032723] [<8f7af5b0>] 0x8f7af5b0 [mac80211@8f780000+0x6c840]
[60224.038639] [<8ec57b50>] 0x8ec57b50 [mt76x02_lib@8ec50000+0x9640]
[60224.044720] [<8f7b6270>] 0x8f7b6270 [mac80211@8f780000+0x6c840]
[60224.050629] [<8f7b634c>] 0x8f7b634c [mac80211@8f780000+0x6c840]
[60224.056529] [<8ec54a48>] 0x8ec54a48 [mt76x02_lib@8ec50000+0x9640]
[60224.062604] [<800329c4>] 0x800329c4
[60224.066083] [<804a24d8>] 0x804a24d8
[60224.069567] [<8004fc60>] 0x8004fc60
[60224.073047] [<8004fc60>] 0x8004fc60
[60224.076525] [<80032c98>] 0x80032c98
[60224.080014] [<8004fe08>] 0x8004fe08
[60224.083492] [<8049da58>] 0x8049da58
[60224.086990] [<8004be88>] 0x8004be88
[60224.090461] [<8004bd58>] 0x8004bd58
[60224.093942] [<8004bd58>] 0x8004bd58
[60224.097419] [<8004bd58>] 0x8004bd58
[60224.100892] [<80006f78>] 0x80006f78
[60224.104376] 
[60224.106046] ---[ end trace 9451b22d11d2504d ]---
[60248.509536] mtk_soc_eth 1e100000.ethernet eth0: port 1 link down
[60260.376655] mtk_soc_eth 1e100000.ethernet eth0: port 1 link up
[60287.745437] mtk_soc_eth 1e100000.ethernet eth0: port 1 link down
[72968.373293] mtk_soc_eth 1e100000.ethernet eth0: port 2 link down
[72970.371221] mtk_soc_eth 1e100000.ethernet eth0: port 2 link up
[72988.002931] mtk_soc_eth 1e100000.ethernet eth0: port 2 link down

taeasy, 2020-02-26:
[    0.000000] Linux version 4.14.167 (builder@buildhost) (gcc version 7.5.0 (OpenWrt GCC 7.5.0 r10911-c155900f66)) #0 SMP Wed Jan 29 16:05:35 2020
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
[    0.000000] bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
[    0.000000] MIPS: machine is Newifi-D2
[    0.000000] Determined physical RAM map:
[    0.000000]  memory: 1c000000 @ 00000000 (usable)
[    0.000000]  memory: 04000000 @ 20000000 (usable)
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] VPE topology {2,2} total 4
[    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000]   HighMem  [mem 0x0000000010000000-0x0000000023ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000001bffffff]
[    0.000000]   node   0: [mem 0x0000000020000000-0x0000000023ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000023ffffff]
[    0.000000] On node 0 totalpages: 131072
[    0.000000] free_area_init_node: node 0, pgdat 80572720, node_mem_map 81003000
[    0.000000]   Normal zone: 512 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 65536 pages, LIFO batch:15
[    0.000000]   HighMem zone: 65536 pages, LIFO batch:15
[    0.000000] random: get_random_bytes called from 0x80575744 with crng_init=0
[    0.000000] percpu: Embedded 14 pages/cpu s26224 r8192 d22928 u57344
[    0.000000] pcpu-alloc: s26224 r8192 d22928 u57344 alloc=14*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 130560
[    0.000000] Kernel command line: console=ttyS0,115200 rootfstype=squashfs,jffs2
[    0.000000] PID hash table entries: 1024 (order: 0, 4096 bytes)
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes)
[    0.000000] Writing ErrCtl register=00000002
[    0.000000] Readback ErrCtl register=00000002
[    0.000000] Memory: 512044K/524288K available (4745K kernel code, 242K rwdata, 588K rodata, 1260K init, 255K bss, 12244K reserved, 0K cma-reserved, 262144K highmem)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000] NR_IRQS: 256
[    0.000000] CPU Clock: 880MHz
[    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4343773742 ns
[    0.000010] sched_clock: 32 bits at 440MHz, resolution 2ns, wraps every 4880645118ns
[    0.007826] Calibrating delay loop... 586.13 BogoMIPS (lpj=2930688)
[    0.073985] pid_max: default: 32768 minimum: 301
[    0.078762] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.085271] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.094150] Hierarchical SRCU implementation.
[    0.099307] smp: Bringing up secondary CPUs ...
[    9.236754] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    9.236762] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    9.236774] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    9.236917] CPU1 revision is: 0001992f (MIPS 1004Kc)
[    0.164056] Synchronize counters for CPU 1: done.
[    4.348591] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    4.348600] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    4.348607] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    4.348679] CPU2 revision is: 0001992f (MIPS 1004Kc)
[    0.255082] Synchronize counters for CPU 2: done.
[    7.153354] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    7.153361] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    7.153368] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    7.153448] CPU3 revision is: 0001992f (MIPS 1004Kc)
[    0.340264] Synchronize counters for CPU 3: done.
[    0.370125] smp: Brought up 1 node, 4 CPUs
[    0.378027] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.387813] futex hash table entries: 1024 (order: 3, 32768 bytes)
[... 383 lines cut ...]
[278181.255290] mtk_soc_eth 1e100000.ethernet eth0: port 1 link down
[278942.322610] mtk_soc_eth 1e100000.ethernet eth0: port 1 link up
[278962.582736] mtk_soc_eth 1e100000.ethernet eth0: port 2 link down
[278967.386419] mtk_soc_eth 1e100000.ethernet eth0: port 2 link up
[279168.151993] mtk_soc_eth 1e100000.ethernet eth0: port 3 link down
[279171.273452] mtk_soc_eth 1e100000.ethernet eth0: port 3 link up
[279301.763273] mtk_soc_eth 1e100000.ethernet eth0: port 2 link down
[279342.858362] mtk_soc_eth 1e100000.ethernet eth0: port 2 link up
[279344.075927] mtk_soc_eth 1e100000.ethernet eth0: port 2 link down
[279346.408348] mtk_soc_eth 1e100000.ethernet eth0: port 2 link up
[279347.970827] mtk_soc_eth 1e100000.ethernet eth0: port 2 link down
[... cut ...]
[279829.761149] mtk_soc_eth 1e100000.ethernet eth0: port 2 link up
[279866.033841] mtk_soc_eth 1e100000.ethernet eth0: port 2 link down
[279962.570842] mtk_soc_eth 1e100000.ethernet eth0: port 2 link up
[280117.857310] mtk_soc_eth 1e100000.ethernet eth0: port 1 link down
[280141.357513] mtk_soc_eth 1e100000.ethernet eth0: port 1 link up
[280291.837095] mtk_soc_eth 1e100000.ethernet eth0: port 1 link down
[280807.843929] mtk_soc_eth 1e100000.ethernet eth0: port 3 link down
[280810.864138] mtk_soc_eth 1e100000.ethernet eth0: port 3 link up

Openwrtfunboy, 2020-04-03:
Too sad, have same :(