openwrt/mt76 #367: MT7621/7603 contant hardware restarts logged

JFtico, 2020-02-28:
Using Master from 2/27/2020 OpenWrt SNAPSHOT r12324-35890514bb / LuCI Master git-20.057.42551-7c6c043

Hardware: ZBT-WE3526 w/512MB 

Unit ran fine for about 24hrs, then became very sluggish, looking at logs showed it was in a non-stop loop of restarts with this being logged:

`Fri Feb 28 15:15:23 2020 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED 00:51:ed:f2:cc:96
Fri Feb 28 15:15:24 2020 kern.err kernel: [101017.595239] mt76x2e 0000:01:00.0: MCU message 2 (seq 8) timed out
Fri Feb 28 15:15:24 2020 daemon.info hostapd: wlan0: STA 00:51:ed:f2:cc:96 IEEE 802.11: authenticated
Fri Feb 28 15:15:24 2020 daemon.info hostapd: wlan0: STA 00:51:ed:f2:cc:96 IEEE 802.11: authenticated
Fri Feb 28 15:15:24 2020 daemon.info hostapd: wlan0: STA 00:51:ed:f2:cc:96 IEEE 802.11: authenticated
Fri Feb 28 15:15:24 2020 daemon.info hostapd: wlan0: STA 00:51:ed:f2:cc:96 IEEE 802.11: authenticated
Fri Feb 28 15:15:24 2020 daemon.info hostapd: wlan0: STA 00:51:ed:f2:cc:96 IEEE 802.11: associated (aid 6)
Fri Feb 28 15:15:24 2020 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 00:51:ed:f2:cc:96
Fri Feb 28 15:15:24 2020 daemon.info hostapd: wlan0: STA 00:51:ed:f2:cc:96 WPA: pairwise key handshake completed (RSN)
Fri Feb 28 15:15:24 2020 kern.info kernel: [101017.716526] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
Fri Feb 28 15:15:24 2020 kern.info kernel: [101017.722297] mt76x2e 0000:01:00.0: Build: 1
Fri Feb 28 15:15:24 2020 kern.info kernel: [101017.726754] mt76x2e 0000:01:00.0: Build Time: 201507311614____
Fri Feb 28 15:15:24 2020 kern.info kernel: [101017.755236] mt76x2e 0000:01:00.0: Firmware running!
Fri Feb 28 15:15:24 2020 kern.info kernel: [101017.765269] ieee80211 phy1: Hardware restart was requested
Fri Feb 28 15:15:25 2020 kern.err kernel: [101018.795192] mt76x2e 0000:01:00.0: MCU message 2 (seq 10) timed out
Fri Feb 28 15:15:25 2020 kern.info kernel: [101018.916468] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
Fri Feb 28 15:15:25 2020 kern.info kernel: [101018.922041] mt76x2e 0000:01:00.0: Build: 1
Fri Feb 28 15:15:25 2020 kern.info kernel: [101018.926287] mt76x2e 0000:01:00.0: Build Time: 201507311614____
Fri Feb 28 15:15:25 2020 kern.info kernel: [101018.955173] mt76x2e 0000:01:00.0: Firmware running!
Fri Feb 28 15:15:25 2020 kern.info kernel: [101018.965256] ieee80211 phy1: Hardware restart was requested
Fri Feb 28 15:15:26 2020 kern.err kernel: [101019.995148] mt76x2e 0000:01:00.0: MCU message 2 (seq 12) timed out
Fri Feb 28 15:15:26 2020 kern.info kernel: [101020.116404] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
Fri Feb 28 15:15:26 2020 kern.info kernel: [101020.121974] mt76x2e 0000:01:00.0: Build: 1
Fri Feb 28 15:15:26 2020 kern.info kernel: [101020.126237] mt76x2e 0000:01:00.0: Build Time: 201507311614____
Fri Feb 28 15:15:26 2020 kern.info kernel: [101020.155186] mt76x2e 0000:01:00.0: Firmware running!
Fri Feb 28 15:15:26 2020 kern.info kernel: [101020.165149] ieee80211 phy1: Hardware restart was requested
Fri Feb 28 15:15:27 2020 kern.err kernel: [101021.195105] mt76x2e 0000:01:00.0: MCU message 2 (seq 14) timed out
Fri Feb 28 15:15:28 2020 kern.info kernel: [101021.316377] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
Fri Feb 28 15:15:28 2020 kern.info kernel: [101021.321948] mt76x2e 0000:01:00.0: Build: 1
Fri Feb 28 15:15:28 2020 kern.info kernel: [101021.326201] mt76x2e 0000:01:00.0: Build Time: 201507311614____
Fri Feb 28 15:15:28 2020 kern.info kernel: [101021.355085] mt76x2e 0000:01:00.0: Firmware running!
Fri Feb 28 15:15:28 2020 kern.info kernel: [101021.365135] ieee80211 phy1: Hardware restart was requested
Fri Feb 28 15:15:29 2020 kern.err kernel: [101022.395066] mt76x2e 0000:01:00.0: MCU message 2 (seq 1) timed out
Fri Feb 28 15:15:29 2020 kern.info kernel: [101022.516321] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
Fri Feb 28 15:15:29 2020 kern.info kernel: [101022.522384] mt76x2e 0000:01:00.0: Build: 1
Fri Feb 28 15:15:29 2020 kern.info kernel: [101022.526830] mt76x2e 0000:01:00.0: Build Time: 201507311614____
Fri Feb 28 15:15:29 2020 kern.info kernel: [101022.555062] mt76x2e 0000:01:00.0: Firmware running!
Fri Feb 28 15:15:29 2020 kern.info kernel: [101022.565399] ieee80211 phy1: Hardware restart was requested
Fri Feb 28 15:15:30 2020 kern.err kernel: [101023.595025] mt76x2e 0000:01:00.0: MCU message 2 (seq 3) timed out
Fri Feb 28 15:15:30 2020 kern.info kernel: [101023.716256] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
Fri Feb 28 15:15:30 2020 kern.info kernel: [101023.721833] mt76x2e 0000:01:00.0: Build: 1
Fri Feb 28 15:15:30 2020 kern.info kernel: [101023.726089] mt76x2e 0000:01:00.0: Build Time: 201507311614____
Fri Feb 28 15:15:30 2020 kern.info kernel: [101023.754998] mt76x2e 0000:01:00.0: Firmware running!
... ad-infinitum`

Both 2.4 (6 or 7 devices) and 5Ghz clients (3 or so) connected.

I was validating that the fix for #358  was working, and while I did not see disconnects, this new problem is worse.

Please let me know if there is further info I can provide.

sombree, 2020-02-29:
I have same issue on latest (OpenWRT master/4d87963) build. 5GHz was already a bit slower on this build vs build from ~1 month ago. During some heavy traffic 5GHz network died and kernel log was spammed with this:
```
[283722.493880] mt76x2e 0000:02:00.0: MCU message 2 (seq 14) timed out
[283722.615134] mt76x2e 0000:02:00.0: Firmware Version: 0.0.00
[283722.621156] mt76x2e 0000:02:00.0: Build: 1
[283722.625540] mt76x2e 0000:02:00.0: Build Time: 201507311614____
[283722.653890] mt76x2e 0000:02:00.0: Firmware running!
[283722.663887] ieee80211 phy1: Hardware restart was requested
[283723.693880] mt76x2e 0000:02:00.0: MCU message 2 (seq 1) timed out
[283723.815141] mt76x2e 0000:02:00.0: Firmware Version: 0.0.00
[283723.820714] mt76x2e 0000:02:00.0: Build: 1
[283723.824966] mt76x2e 0000:02:00.0: Build Time: 201507311614____
[283723.853935] mt76x2e 0000:02:00.0: Firmware running!
[283723.863924] ieee80211 phy1: Hardware restart was requested
[283724.893891] mt76x2e 0000:02:00.0: MCU message 2 (seq 3) timed out
[283725.015158] mt76x2e 0000:02:00.0: Firmware Version: 0.0.00
[283725.020729] mt76x2e 0000:02:00.0: Build: 1
[283725.025128] mt76x2e 0000:02:00.0: Build Time: 201507311614____
[283725.053875] mt76x2e 0000:02:00.0: Firmware running!
[283725.063961] ieee80211 phy1: Hardware restart was requested
[283726.093874] mt76x2e 0000:02:00.0: MCU message 2 (seq 5) timed out
[283726.215138] mt76x2e 0000:02:00.0: Firmware Version: 0.0.00
[283726.220708] mt76x2e 0000:02:00.0: Build: 1
[283726.224949] mt76x2e 0000:02:00.0: Build Time: 201507311614____
[283726.253865] mt76x2e 0000:02:00.0: Firmware running!
[283726.263886] ieee80211 phy1: Hardware restart was requested
[283727.293873] mt76x2e 0000:02:00.0: MCU message 2 (seq 7) timed out
[283727.415144] mt76x2e 0000:02:00.0: Firmware Version: 0.0.00
[283727.420716] mt76x2e 0000:02:00.0: Build: 1
[283727.424970] mt76x2e 0000:02:00.0: Build Time: 201507311614____
[283727.453916] mt76x2e 0000:02:00.0: Firmware running!
[283727.463916] ieee80211 phy1: Hardware restart was requested
[283728.493875] mt76x2e 0000:02:00.0: MCU message 2 (seq 9) timed out
[283728.615114] mt76x2e 0000:02:00.0: Firmware Version: 0.0.00
[283728.620686] mt76x2e 0000:02:00.0: Build: 1
[283728.624930] mt76x2e 0000:02:00.0: Build Time: 201507311614____
[283728.653866] mt76x2e 0000:02:00.0: Firmware running!
```
There was no problem on older build.

kyoto44, 2020-03-06:
> I have same issue on latest (OpenWRT master/4d87963) build. 5GHz was already a bit slower on this build vs build from ~1 month ago. During some heavy traffic 5GHz network died and kernel log was spammed with this:
> 
> ```
> [283722.493880] mt76x2e 0000:02:00.0: MCU message 2 (seq 14) timed out
> [283722.615134] mt76x2e 0000:02:00.0: Firmware Version: 0.0.00
> [283722.621156] mt76x2e 0000:02:00.0: Build: 1
> [283722.625540] mt76x2e 0000:02:00.0: Build Time: 201507311614____
> [283722.653890] mt76x2e 0000:02:00.0: Firmware running!
> [283722.663887] ieee80211 phy1: Hardware restart was requested
> [283723.693880] mt76x2e 0000:02:00.0: MCU message 2 (seq 1) timed out
> [283723.815141] mt76x2e 0000:02:00.0: Firmware Version: 0.0.00
> [283723.820714] mt76x2e 0000:02:00.0: Build: 1
> [283723.824966] mt76x2e 0000:02:00.0: Build Time: 201507311614____
> [283723.853935] mt76x2e 0000:02:00.0: Firmware running!
> [283723.863924] ieee80211 phy1: Hardware restart was requested
> [283724.893891] mt76x2e 0000:02:00.0: MCU message 2 (seq 3) timed out
> [283725.015158] mt76x2e 0000:02:00.0: Firmware Version: 0.0.00
> [283725.020729] mt76x2e 0000:02:00.0: Build: 1
> [283725.025128] mt76x2e 0000:02:00.0: Build Time: 201507311614____
> [283725.053875] mt76x2e 0000:02:00.0: Firmware running!
> [283725.063961] ieee80211 phy1: Hardware restart was requested
> [283726.093874] mt76x2e 0000:02:00.0: MCU message 2 (seq 5) timed out
> [283726.215138] mt76x2e 0000:02:00.0: Firmware Version: 0.0.00
> [283726.220708] mt76x2e 0000:02:00.0: Build: 1
> [283726.224949] mt76x2e 0000:02:00.0: Build Time: 201507311614____
> [283726.253865] mt76x2e 0000:02:00.0: Firmware running!
> [283726.263886] ieee80211 phy1: Hardware restart was requested
> [283727.293873] mt76x2e 0000:02:00.0: MCU message 2 (seq 7) timed out
> [283727.415144] mt76x2e 0000:02:00.0: Firmware Version: 0.0.00
> [283727.420716] mt76x2e 0000:02:00.0: Build: 1
> [283727.424970] mt76x2e 0000:02:00.0: Build Time: 201507311614____
> [283727.453916] mt76x2e 0000:02:00.0: Firmware running!
> [283727.463916] ieee80211 phy1: Hardware restart was requested
> [283728.493875] mt76x2e 0000:02:00.0: MCU message 2 (seq 9) timed out
> [283728.615114] mt76x2e 0000:02:00.0: Firmware Version: 0.0.00
> [283728.620686] mt76x2e 0000:02:00.0: Build: 1
> [283728.624930] mt76x2e 0000:02:00.0: Build Time: 201507311614____
> [283728.653866] mt76x2e 0000:02:00.0: Firmware running!
> ```
> 
> There was no problem on older build.
> 
> Edit:
> Hardware: YouHua WR1200JS - MT7621 with MT7603+MT7612.

I have same device and when 5G is enabled after some time devices just freezes and all connections lost

Openwrtfunboy, 2020-03-06:
Have same platform, but dont have this problem.

Newifi3 D2.

I think this is the problem only of this router.

Openwrtfunboy, 2020-03-08:
Have same problem :((

fape, 2020-03-08:
I have something similar. After I flashed 19.07.2 to my wr1200js, once a day I need to restart. (I usually unplug-plug the power) because there is no wifi. I'm using only 2.4ghz (mt7603e). I'm trying to get more info eg log. I'm going to connect via cable too.

nbd168, 2020-03-11:
Fixed in the latest version

closed 2020-03-11 by nbd168

JFtico, 2020-03-23:
Confirm fixed in snapshot  r12529-228bb84744 .

I have been running for 10 days with zero issues.

LZeugirdor, 2024-12-06:
I get this occasionally after an extended period of runtime on the router.

marianomd, 2025-01-09:
> Fixed in the latest version

This is happenning on 24.10.0-rc2 (MT7621/7603).

```
[1107468.070115] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
[1107468.075802] mt76x2e 0000:01:00.0: Build: 1
[1107468.080284] mt76x2e 0000:01:00.0: Build Time: 201607111443____
[1107468.106328] mt76x2e 0000:01:00.0: Firmware running!
[1107468.115093] ieee80211 phy1: Hardware restart was requested
```