openwrt/mt76 #368: mt7615e   In XiaoMi R3P :disconnected due to excessive missing ACKs

mingxiaoyu, 2020-03-07:
Hi  Openwrt Team
2.4G is disconnected at least twice a day.
 5G is fine.

Sat Mar  7 20:29:46 2020 daemon.info hostapd: wlan0: STA ac:c1:ee:30:07:d4 IEEE 802.11: disconnected due to excessive missing ACKs
Sat Mar  7 20:29:46 2020 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED ac:c1:ee:30:07:d4
Sat Mar  7 20:29:46 2020 daemon.info hostapd: wlan0: STA ac:c1:ee:30:07:d4 IEEE 802.11: authenticated
Sat Mar  7 20:29:46 2020 daemon.info hostapd: wlan0: STA ac:c1:ee:30:07:d4 IEEE 802.11: associated (aid 1)
Sat Mar  7 20:29:46 2020 daemon.notice hostapd: wlan0: AP-STA-CONNECTED ac:c1:ee:30:07:d4

[   18.288301] [EXFAT] mounted successfully
[   18.495814] mt7615e 0000:01:00.0: N9 Firmware Version: _reserved_, Build Time: 20190103180756
[   18.513159] mt7615e 0000:01:00.0: CR4 Firmware Version: _reserved_, Build Time: 20181207140436
[   26.172749] ieee80211 phy0: Selected rate control algorithm 'minstrel_ht'
[   26.175452] bus=0x2, slot = 0x1, irq=0x0
[   26.198259] mt7615e 0000:02:00.0: HW/SW Version: 0x8a108a10, Build Time: 20180518100604a
[   26.198259] 
[   26.222652] mt7615e 0000:02:00.0: N9 Firmware Version: _reserved_, Build Time: 20190103180756
[   26.238023] mt7615e 0000:02:00.0: CR4 Firmware Version: _reserved_, Build Time: 20181207140436
[   33.894443] ieee80211 phy1: Selected rate control algorithm 'minstrel_ht'
[   33.905013] PPP generic driver version 2.4.2

mingxiaoyu, 2020-03-07:
OpenWrt version  19.07

mingxiaoyu, 2020-03-08:
23070.569668] ------------[ cut here ]------------
[23070.574311] WARNING: CPU: 3 PID: 0 at net/sched/sch_generic.c:320 dev_watchdog+0x1ac/0x324
[23070.582596] NETDEV WATCHDOG: eth0 (mtk_soc_eth): transmit queue 0 timed out
[23070.589528] Modules linked in: xt_FULLCONENAT pppoe ppp_async pppox ppp_mppe ppp_generic nf_nat_pptp nf_conntrack_pptp mt7615e mt76 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_socket xt_recent xt_policy xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_esp xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TPROXY xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_FLOWOFFLOAD xt_DSCP xt_CT xt_CLASSIFY usblp ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda ts_fsm ts_bm slhc nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_redirect nf_nat_proto_gre nf_nat_masquerade_ipv4 nf_nat_irc nf_conntrack_ipv4
[23070.660722]  nf_nat_ipv4 nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_defrag_ipv4 nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_rtcache nf_conntrack_proto_gre nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast ts_kmp nf_conntrack_amanda iptable_raw iptable_mangle iptable_filter ipt_ah ipt_ECN ip_tables crc_ccitt compat fuse sch_cake tcp_bbr sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred ledtrig_usbport cryptodev xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport
[23070.731198]  ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_nat nf_conntrack_ipv6 nf_defrag_ipv6 nf_nat_ipv6 ip6t_NPT ip6t_MASQUERADE nf_nat_masquerade_ipv6 nf_nat nf_conntrack nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 ifb sit xfrm6_mode_tunnel xfrm6_mode_transport xfrm6_mode_beet ipcomp6 xfrm6_tunnel esp6 ah6 xfrm4_tunnel xfrm4_mode_tunnel xfrm4_mode_transport xfrm4_mode_beet ipcomp esp4 ah4 tunnel6 tunnel4 ip_tunnel tun xfrm_user xfrm_ipcomp af_key xfrm_algo vfat fat nls_utf8 nls_iso8859_1 nls_cp936 nls_cp437 sha256_generic sha1_generic jitterentropy_rng drbg md5 hmac echainiv ecb des_generic cbc authenc usb_storage leds_gpio xhci_plat_hcd xhci_pci xhci_mtk xhci_hcd
[23070.801892]  ohci_pci ohci_platform ohci_hcd ehci_pci ehci_platform sd_mod scsi_mod ehci_hcd gpio_button_hotplug ext4 mbcache jbd2 exfat usbcore nls_base usb_common crc32c_generic
[23070.817901] CPU: 3 PID: 0 Comm: swapper/3 Not tainted 4.14.171 #0
[23070.823972] Stack : 00000000 8fde8c40 805e0000 80071e48 80600000 805a37f8 00000000 00000000
[23070.832317]         8056d8d4 8fc0fdc4 8fc3d18c 805dea07 805686d0 00000001 8fc0fd68 1cc2824b
[23070.840657]         00000000 00000000 80760000 000075e8 00000000 0000016c 00000008 00000000
[23070.848988]         00000000 805e0000 000c7aed 00000000 00000000 80600000 00000000 803aa6b0
[23070.857322]         00000009 00000140 00000003 8fde8c40 00000000 802d1688 0000000c 8075000c
[23070.865661]         ...
[23070.868097] Call Trace:
[23070.870562] [<8000c700>] show_stack+0x58/0x100
[23070.875006] [<804a15a4>] dump_stack+0xa4/0xe0
[23070.879347] [<8002b1e8>] __warn+0xe0/0x114
[23070.883426] [<8002b24c>] warn_slowpath_fmt+0x30/0x3c
[23070.888370] [<803aa6b0>] dev_watchdog+0x1ac/0x324
[23070.893084] [<80089344>] call_timer_fn.isra.3+0x24/0x84
[23070.898288] [<80089560>] run_timer_softirq+0x1bc/0x248
[23070.903426] [<804c0568>] __do_softirq+0x128/0x2ec
[23070.908117] [<8002fc54>] irq_exit+0xac/0xc8
[23070.912292] [<80284a9c>] plat_irq_dispatch+0xfc/0x138
[23070.917321] [<80007608>] except_vec_vi_end+0xb8/0xc4
[23070.922265] [<80008fd0>] r4k_wait_irqoff+0x1c/0x24
[23070.927128] ---[ end trace 399d83f30e733166 ]---
[23070.931756] mtk_soc_eth 1e100000.ethernet eth0: transmit timed out
[23070.937911] mtk_soc_eth 1e100000.ethernet eth0: dma_cfg:80000065
[23070.943954] mtk_soc_eth 1e100000.ethernet eth0: tx_ring=0, base=0e720000, max=0, ctx=180, dtx=180, fdx=179, next=180
[23070.954487] mtk_soc_eth 1e100000.ethernet eth0: rx_ring=0, base=0dec0000, max=0, calc=3040, drx=3041
[23070.967413] mtk_soc_eth 1e100000.ethernet: 0x100 = 0x6060000c, 0x10c = 0x80818
[23070.980140] mtk_soc_eth 1e100000.ethernet: PPE started

Openwrtfunboy, 2020-03-08:
Type to SSH:
echo 1 > /sys/kernel/debug/ieee80211/phy0/mt76/dynamic_sensitivity

lukasz1992, 2020-03-08:
@Openwrtfunboy It won't work on mt7615 driver, only mt7603 has it.
@mingxiaoyu this is another problem, described here https://forum.openwrt.org/t/mtk-soc-eth-watchdog-timeout-after-r11573

mingxiaoyu, 2020-03-08:
@lukasz1992  Thank you for your reply. But 2.4g disconnected only the first log. 

The third log is the sudden slowdown of the network

lukasz1992, 2020-03-12:
@mingxiaoyu Maybe last commit bef8f8a5966dbea0d68ba70ee7f526c65198ebd1  will help?

mingxiaoyu, 2020-03-14:
unfortunately...it's not work. 
when got  "802.11: disconnected due to excessive missing ACKs" ,  my  laptop  network will  slowly... 

log:Sat Mar 14 16:01:22 2020 daemon.info hostapd: wlan0: STA 2c:d0:5a:2a:16:b5 IEEE 802.11: disconnected due to excessive missing ACKs
Sat Mar 14 16:01:22 2020 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED 2c:d0:5a:2a:16:b5
Sat Mar 14 16:01:23 2020 daemon.info hostapd: wlan0: STA 2c:d0:5a:2a:16:b5 IEEE 802.11: authenticated
Sat Mar 14 16:01:23 2020 daemon.info hostapd: wlan0: STA 2c:d0:5a:2a:16:b5 IEEE 802.11: associated (aid 6)
Sat Mar 14 16:01:23 2020 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 2c:d0:5a:2a:16:b5
Sat Mar 14 16:01:23 2020 daemon.info hostapd: wlan0: STA 2c:d0:5a:2a:16:b5 RADIUS: starting accounting session F4D5B77B4CA680B7
Sat Mar 14 16:01:23 2020 daemon.info hostapd: wlan0: STA 2c:d0:5a:2a:16:b5 WPA: pairwise key handshake completed (RSN)
Sat Mar 14 16:01:23 2020 daemon.info dnsmasq-dhcp[19508]: DHCPREQUEST(br-lan) 192.168.31.155 2c:d0:5a:2a:16:b5
Sat Mar 14 16:01:23 2020 daemon.info dnsmasq-dhcp[19508]: DHCPACK(br-lan) 192.168.31.155 2c:d0:5a:2a:16:b5 DESKTOP-EAT7R47
Sat Mar 14 16:04:52 2020 daemon.warn dnsmasq[19508]: possible DNS-rebind attack detected: appservice.qq.com
Sat Mar 14 16:04:52 2020 daemon.warn dnsmasq[19508]: possible DNS-rebind attack detected: appservice.qq.com
Sat Mar 14 16:06:59 2020 daemon.info hostapd: wlan0: STA 2c:d0:5a:2a:16:b5 IEEE 802.11: disconnected due to excessive missing ACKs
Sat Mar 14 16:06:59 2020 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED 2c:d0:5a:2a:16:b5
Sat Mar 14 16:07:00 2020 daemon.info hostapd: wlan0: STA 2c:d0:5a:2a:16:b5 IEEE 802.11: authenticated
Sat Mar 14 16:07:00 2020 daemon.info hostapd: wlan0: STA 2c:d0:5a:2a:16:b5 IEEE 802.11: associated (aid 6)
Sat Mar 14 16:07:00 2020 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 2c:d0:5a:2a:16:b5
Sat Mar 14 16:07:00 2020 daemon.info hostapd: wlan0: STA 2c:d0:5a:2a:16:b5 RADIUS: starting accounting session F4D5B77B4CA680B7
Sat Mar 14 16:07:00 2020 daemon.info hostapd: wlan0: STA 2c:d0:5a:2a:16:b5 WPA: pairwise key handshake completed (RSN)
Sat Mar 14 16:07:00 2020 daemon.info dnsmasq-dhcp[19508]: DHCPREQUEST(br-lan) 192.168.31.155 2c:d0:5a:2a:16:b5
Sat Mar 14 16:07:00 2020 daemon.info dnsmasq-dhcp[19508]: DHCPACK(br-lan) 192.168.31.155 2c:d0:5a:2a:16:b5 DESKTOP-EAT7R47
Sat Mar 14 16:07:01 2020 daemon.info hostapd: wlan0: STA 2c:d0:5a:2a:16:b5 IEEE 802.11: disconnected due to excessive missing ACKs
Sat Mar 14 16:07:01 2020 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED 2c:d0:5a:2a:16:b5
Sat Mar 14 16:07:03 2020 daemon.info hostapd: wlan0: STA 2c:d0:5a:2a:16:b5 IEEE 802.11: authenticated
Sat Mar 14 16:07:03 2020 daemon.info hostapd: wlan0: STA 2c:d0:5a:2a:16:b5 IEEE 802.11: associated (aid 6)
Sat Mar 14 16:07:03 2020 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 2c:d0:5a:2a:16:b5
Sat Mar 14 16:07:03 2020 daemon.info hostapd: wlan0: STA 2c:d0:5a:2a:16:b5 RADIUS: starting accounting session F4D5B77B4CA680B7
Sat Mar 14 16:07:03 2020 daemon.info hostapd: wlan0: STA 2c:d0:5a:2a:16:b5 WPA: pairwise key handshake completed (RSN)
Sat Mar 14 16:07:03 2020 daemon.info dnsmasq-dhcp[19508]: DHCPREQUEST(br-lan) 192.168.31.155 2c:d0:5a:2a:16:b5
Sat Mar 14 16:07:03 2020 daemon.info dnsmasq-dhcp[19508]: DHCPACK(br-lan) 192.168.31.155 2c:d0:5a:2a:16:b5 DESKTOP-EAT7R47
Sat Mar 14 16:07:04 2020 daemon.info dnsmasq-dhcp[19508]: DHCPREQUEST(br-lan) 192.168.31.155 2c:d0:5a:2a:16:b5
Sat Mar 14 16:07:04 2020 daemon.info dnsmasq-dhcp[19508]: DHCPACK(br-lan) 192.168.31.155 2c:d0:5a:2a:16:b5 DESKTOP-EAT7R47
Sat Mar 14 16:07:05 2020 daemon.info hostapd: wlan0: STA 2c:d0:5a:2a:16:b5 IEEE 802.11: disconnected due to excessive missing ACKs
Sat Mar 14 16:07:05 2020 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED 2c:d0:5a:2a:16:b5
Sat Mar 14 16:07:07 2020 daemon.info hostapd: wlan0: STA 2c:d0:5a:2a:16:b5 IEEE 802.11: authenticated
Sat Mar 14 16:07:07 2020 daemon.info hostapd: wlan0: STA 2c:d0:5a:2a:16:b5 IEEE 802.11: associated (aid 6)
Sat Mar 14 16:07:07 2020 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 2c:d0:5a:2a:16:b5
Sat Mar 14 16:07:07 2020 daemon.info hostapd: wlan0: STA 2c:d0:5a:2a:16:b5 RADIUS: starting accounting session F4D5B77B4CA680B7
Sat Mar 14 16:07:07 2020 daemon.info hostapd: wlan0: STA 2c:d0:5a:2a:16:b5 WPA: pairwise key handshake completed (RSN)
Sat Mar 14 16:07:07 2020 daemon.info dnsmasq-dhcp[19508]: DHCPREQUEST(br-lan) 192.168.31.155 2c:d0:5a:2a:16:b5
Sat Mar 14 16:07:07 2020 daemon.info dnsmasq-dhcp[19508]: DHCPACK(br-lan) 192.168.31.155 2c:d0:5a:2a:16:b5 DESKTOP-EAT7R47
Sat Mar 14 16:07:11 2020 daemon.info hostapd: wlan0: STA 2c:d0:5a:2a:16:b5 IEEE 802.11: disconnected due to excessive missing ACKs
Sat Mar 14 16:07:11 2020 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED 2c:d0:5a:2a:16:b5
Sat Mar 14 16:07:12 2020 daemon.info hostapd: wlan0: STA 2c:d0:5a:2a:16:b5 IEEE 802.11: authenticated
Sat Mar 14 16:07:12 2020 daemon.info hostapd: wlan0: STA 2c:d0:5a:2a:16:b5 IEEE 802.11: associated (aid 6)
Sat Mar 14 16:07:12 2020 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 2c:d0:5a:2a:16:b5
Sat Mar 14 16:07:12 2020 daemon.info hostapd: wlan0: STA 2c:d0:5a:2a:16:b5 RADIUS: starting accounting session F4D5B77B4CA680B7
Sat Mar 14 16:07:12 2020 daemon.info hostapd: wlan0: STA 2c:d0:5a:2a:16:b5 WPA: pairwise key handshake completed (RSN)
Sat Mar 14 16:07:12 2020 daemon.info dnsmasq-dhcp[19508]: DHCPREQUEST(br-lan) 192.168.31.155 2c:d0:5a:2a:16:b5
[... cut ...]
Sat Mar 14 16:07:51 2020 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 2c:d0:5a:2a:16:b5
Sat Mar 14 16:07:51 2020 daemon.info hostapd: wlan0: STA 2c:d0:5a:2a:16:b5 RADIUS: starting accounting session F4D5B77B4CA680B7
Sat Mar 14 16:07:51 2020 daemon.info hostapd: wlan0: STA 2c:d0:5a:2a:16:b5 WPA: pairwise key handshake completed (RSN)
Sat Mar 14 16:07:51 2020 daemon.info dnsmasq-dhcp[19508]: DHCPREQUEST(br-lan) 192.168.31.155 2c:d0:5a:2a:16:b5
Sat Mar 14 16:07:51 2020 daemon.info dnsmasq-dhcp[19508]: DHCPACK(br-lan) 192.168.31.155 2c:d0:5a:2a:16:b5 DESKTOP-EAT7R47

closed 2023-03-24 by mingxiaoyu