openwrt/mt76 #370: Poor signal in both wifi modules ZBT WG108 router - MT7621

christiansasig, 2020-03-11:
Hello, we have tested the latest stable version of openwrt firmware 19.07.2 on the wg108 router from the official openwrt page at the following link:
[https://downloads.openwrt.org/releases/19.07.2/targets/ramips/mt7621/openwrt-19.07.2-ramips-mt7621-zbt-we1326-squashfs-sysupgrade.bin](url)

It works correctly on the Wg108 router we only found a problem with the wifi modules the wifi driver does not increase the output power.

There is poor performance in both wifi modules. This problem turns on the mt7603 2.4ghz and the mt7612 5ghz.

 I installed the compatible firmware for the zbt-wg108 router "openwrt_ZBT-WE1326_V7" and the 2.4Ghz and 5Ghz wifi modules work correctly. This firmware runs on Linux kernel 3.10.
[https://drive.google.com/drive/folders/1ZFfQQ-7KYNgKc53z_G3_8BOW8Lcl0oEp](url)

Extract the drivers (.ko) of the wifi modules to test in the openwrt 4.14 kernel "openwrt-19.07.2-ramips-mt7621-zbt-we1326-squashfs-sysupgrade.bin" and the problem of tx power remains.

I am attaching the firmware files for possible help to solve this problem.
regards

JFtico, 2020-03-23:
Give the latest snapshot a try, I had all kinds of wifi issues in 19.07 that were cleared up in builds since March 13.

dobrevdobromir, 2025-01-16:
I upgrade my ZBT WG108 device from WG1326 to WG3526 openwrt software. Wifi works very good.  
[Sysupgrade image] 
(https://downloads.openwrt.org/releases/23.05.5/targets/ramips/mt7621/openwrt-23.05.5-ramips-mt7621-zbtlink_zbt-wg3526-16m-squashfs-sysupgrade.bin) 
ZBT WG108  and ZBT-WG3526 are compatible devices. Speed test under 5Ghz radio shows 270/270Mbit/s