openwrt/mt76 #371: mt76x0u: aukey ac600 usb wifi adapter WF-R5: kernel panic

everloop2, 2020-03-12:
Openwrt-19.07.2, Linux 4.14.171, SpeedportW504V_TypeA
-> installed kmod-mt76x0u + auto dependencies -> activated AP -> kernel panic and reboot loop

-> connect stick to USB port

```
[  249.746781] dwc2 1e101000.usb: Mode Mismatch Interrupt: currently in Host mode
[  249.752624] dwc2 1e101000.usb: Mode Mismatch Interrupt: currently in Host mode
[  249.946878] usb 1-1: new high-speed USB device number 3 using dwc2
[  249.952330] dwc2 1e101000.usb: Mode Mismatch Interrupt: currently in Host mode
[  249.958705] dwc2 1e101000.usb: Mode Mismatch Interrupt: currently in Host mode
[  250.188173] dwc2 1e101000.usb: Mode Mismatch Interrupt: currently in Host mode
[  250.193969] dwc2 1e101000.usb: Mode Mismatch Interrupt: currently in Host mode
[  250.386191] usb 1-1: reset high-speed USB device number 3 using dwc2
[  250.394441] dwc2 1e101000.usb: Mode Mismatch Interrupt: currently in Host mode
[  250.400224] dwc2 1e101000.usb: Mode Mismatch Interrupt: currently in Host mode
[  250.620064] mt76x0u 1-1:1.0: ASIC revision: 76100002 MAC revision: 76502000
[  251.979341] mt76x0u 1-1:1.0: EEPROM ver:02 fae:04
[  252.868746] ieee80211 phy1: Selected rate control algorithm 'minstrel_ht'
```
-> lsusb -t
```
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=dwc2/1p, 480M
    |__ Port 1: Dev 3, If 0, Class=Vendor Specific Class, Driver=mt76x0u, 480M
```

-> iw list -> dBm ?false? values
```
Wiphy phy1
        max # scan SSIDs: 4
        max scan IEs length: 2243 bytes
        max # sched scan SSIDs: 0
        max # match sets: 0
        max # scan plans: 1
        max scan plan interval: -1
        max scan plan iterations: 0
        Retry short limit: 7
        Retry long limit: 4
        Coverage class: 0 (up to 0m)
        Available Antennas: TX 0x1 RX 0x1
        Configured Antennas: TX 0x1 RX 0x1
        Supported interface modes:
                 * IBSS
                 * managed
                 * AP
                 * AP/VLAN
                 * monitor
                 * mesh point
        Band 1:
                Capabilities: 0x17e
                        HT20/HT40
                        SM Power Save disabled
                        RX Greenfield
                        RX HT20 SGI
                        RX HT40 SGI
                        RX STBC 1-stream
                        Max AMSDU length: 3839 bytes
                        No DSSS/CCK HT40
                Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
                Minimum RX AMPDU time spacing: 4 usec (0x05)
                HT TX/RX MCS rate indexes supported: 0-7
                Frequencies:
                        * 2412 MHz [1] (14.0 dBm)
                        * 2417 MHz [2] (14.0 dBm)
                        * 2422 MHz [3] (14.0 dBm)
                        * 2427 MHz [4] (14.0 dBm)
                        * 2432 MHz [5] (14.0 dBm)
                        * 2437 MHz [6] (14.0 dBm)
                        * 2442 MHz [7] (14.0 dBm)
                        * 2447 MHz [8] (14.0 dBm)
[... 117 lines cut ...]
[  306.265682]         ...
[  306.268048] Call Trace:
[  306.268153] [<832dda4c>] 0x832dda4c [mt76_usb@832dc000+0x2390]
[  306.276083] [<800540ec>] 0x800540ec
[  306.279501] [<832c18f8>] 0x832c18f8 [mt76x02_lib@832c0000+0x9a40]
[  306.285433] [<832d256c>] 0x832d256c [mt76@832d0000+0x7940]
[  306.290698] [<8004590c>] 0x8004590c
[  306.294098] [<832c358c>] 0x832c358c [mt76x02_lib@832c0000+0x9a40]
[  306.299990] [<80496af8>] 0x80496af8
[  306.303392] [<80043e0c>] 0x80043e0c
[  306.306754] [<80044708>] 0x80044708
[  306.310132] [<80044390>] 0x80044390
[  306.313508] [<8004a8a0>] 0x8004a8a0
[  306.316887] [<8004a738>] 0x8004a738
[  306.320266] [<8004a738>] 0x8004a738
[  306.323647] [<8000a198>] 0x8000a198
[  306.327035] 
[  306.328451] Code: afa20024  ac930008  afb50020 <ac530000> 8f830000  8c820004  14530017  afa30028  3c028054 
[  306.337913] 
[  306.342990] ---[ end trace 8bad6e2b20cc19a1 ]---
[  306.350644] Kernel panic - not syncing: Fatal exception
[  306.356630] Rebooting in 3 seconds..
```
-> bootlog without stick:
[SpeedportW504V_TypeA_bootlog.txt](https://github.com/openwrt/mt76/files/4324604/SpeedportW504V_TypeA_bootlog.txt)

LorenzoBianconi, 2020-04-10:
@everloop2 could you please test latest version?

everloop2, 2020-04-24:
No more kernel panic - thanks, working :D

tested:

AccessPoint:

2.4GHz (channel 5 and 13) -> mode: legacy, n

5.180GHz (channel 36) -> mode: n, ac

STA (WWAN client):

2.4GHz -> mode: n

-> iw list -> still shows ?false? dBm values (in DE should be 20dBm max.) -> no major

so, some still good boxes can be upgraded to 5GHz wifi now for a few euros ;)

f.e.: https://openwrt.org/toh/arcadyan/vgv7510kw22 - Arcadyan VGV7510KW22 (o2 Box 6431) 2x500MHz and vDSL

closed 2020-04-24 by everloop2