openwrt/mt76 #373: Please add Mercury UD13 (2c4e:0103) to the support list

CWZ7605, 2020-03-20:
Mercury UD13 wireless card use MT7612UN.
After I use the command `echo '2c4e 0103' > /sys/bus/usb/drivers/mt76x2u/new_id ` it can work perfectly.
Please add this pid and vid to the mt76x2u database.

Thanks.

LorenzoBianconi, 2020-03-20:
Could you please test the following patch:

diff --git a/drivers/net/wireless/mediatek/mt76/mt76x2/usb.c b/drivers/net/wireless/mediatek/mt76/mt76x2/usb.c
index eafa283ca699..7be7fe7a068b 100644
--- a/drivers/net/wireless/mediatek/mt76/mt76x2/usb.c
+++ b/drivers/net/wireless/mediatek/mt76/mt76x2/usb.c
@@ -16,6 +16,7 @@ static const struct usb_device_id mt76x2u_device_table[] = {
 	{ USB_DEVICE(0x0e8d, 0x7612) },	/* Aukey USBAC1200 - Alfa AWUS036ACM */
 	{ USB_DEVICE(0x057c, 0x8503) },	/* Avm FRITZ!WLAN AC860 */
 	{ USB_DEVICE(0x7392, 0xb711) },	/* Edimax EW 7722 UAC */
+	{ USB_DEVICE(0x2c4e, 0x0103) },	/* Mercury UD13 */
 	{ USB_DEVICE(0x0846, 0x9053) },	/* Netgear A6210 */
 	{ USB_DEVICE(0x045e, 0x02e6) },	/* XBox One Wireless Adapter */
 	{ },

LorenzoBianconi, 2020-03-20:
Could you please try [this](https://github.com/LorenzoBianconi/wireless-drivers-next/commit/3c026636718314c03817620304f58bd741ad7d45) patch? Thanks.

CWZ7605, 2020-03-21:
> 
> 
> Could you please try [this](https://github.com/LorenzoBianconi/wireless-drivers-next/commit/3c026636718314c03817620304f58bd741ad7d45) patch? Thanks.

I will try today

CWZ7605, 2020-03-21:
> 
> 
> Could you please try [this](https://github.com/LorenzoBianconi/wireless-drivers-next/commit/3c026636718314c03817620304f58bd741ad7d45) patch? Thanks.

Great! It works. Thanks a lot.

CMCCAdmin, 2020-04-15:
> Mercury UD13 wireless card use MT7612UN.
> After I use the command `echo '2c4e 0103' > /sys/bus/usb/drivers/mt76x2u/new_id ` it can work perfectly.
> Please add this pid and vid to the mt76x2u database.
> 
> Thanks.

你好，你的水星 UD13 是这款么？（ https://item.jd.com/100003723464.html ）网卡是window免驱款，openwrt下使用应该需要使用usb-modeswitch转换模式吧？这个网卡插在openwrt路由器上是否支持ap模式？能不能通过luci（网页）控制网卡发射wifi信号？谢谢了 ！

Hello, is this your Mercury UD13? (Https://item.jd.com/100003723464.html) The network card is a window-free driver. Should we use the usb-modeswitch conversion mode for openwrt? Does this NIC support ap mode when plugged into an openwrt router? Can you control the network card to transmit wifi signal through luci (webpage)? Thank you !

CWZ7605, 2020-04-16:
> 
> 
> > Mercury UD13 wireless card use MT7612UN.
> > After I use the command `echo '2c4e 0103' > /sys/bus/usb/drivers/mt76x2u/new_id ` it can work perfectly.
> > Please add this pid and vid to the mt76x2u database.
> > Thanks.
> 
> 你好，你的水星 UD13 是这款么？（ https://item.jd.com/100003723464.html ）网卡是window免驱款，openwrt下使用应该需要使用usb-modeswitch转换模式吧？这个网卡插在openwrt路由器上是否支持ap模式？能不能通过luci（网页）控制网卡发射wifi信号？谢谢了 ！
> 
> Hello, is this your Mercury UD13? ([Https://item.jd.com/100003723464.html](Https://item.jd.com/100003723464.html)) The network card is a window-free driver. Should we use the usb-modeswitch conversion mode for openwrt? Does this NIC support ap mode when plugged into an openwrt router? Can you control the network card to transmit wifi signal through luci (webpage)? Thank you !

是。水星就这一款UD13

CMCCAdmin, 2020-04-16:
> 是。水星就这一款UD13

那openwrt下能不能作为ap发射信号呢？

CWZ7605, 2020-04-16:
> 那openwrt下能不能作为ap发射信号呢？

能。可以正常使用。
受USB限制，以下点请注意：

1. 2.4G和5G无法同时发射
2. 无法使用中继模式

在openwrt x86版上面测试，传输速度可以达到200Mbps以上（就是迅雷下载能达到20MB/s以上）

CMCCAdmin, 2020-04-16:
> 在openwrt x86版上面测试，传输速度可以达到200Mbps以上（就是迅雷下载能达到20MB/s以上）

好的，谢谢 ！

CMCCAdmin, 2020-04-16:
> 能。可以正常使用。
> 受USB限制，以下点请注意：
>
> 1. 2.4G和5G无法同时发射
> 2. 无法使用中继模式
>
> 在openwrt x86版上面测试，传输速度可以达到200Mbps以上（就是迅雷下载能达到20MB/s以上）

我又来麻烦您了，这个网卡因为windows下是免驱动的，所以这个网卡会有一个存储器用来储存windows下的驱动，如果插在openwrt下首先会识别成cd-rom或者储存吧，那该用什么具体命令来切换到网卡模式呢？

CWZ7605, 2020-04-17:
> 我又来麻烦您了，这个网卡因为windows下是免驱动的，所以这个网卡会有一个存储器用来储存windows下的驱动，如果插在openwrt下首先会识别成cd-rom或者储存吧，那该用什么具体命令来切换到网卡模式呢？

安装usb-modeswitch即可。最新版usb-modeswitch已支持自动切换。

tedaz, 2020-06-23:
With the latest snapshot (kernel 5.4.48), the Mercury UD13 can be recognized and mode switched automatically, tested on Raspberry Pi 4B and VMware Workstation, AP/master mode works and iperf3 gets about 270Mbps.

CWZ7605, 2020-06-23:
> 
> 
> With the latest snapshot (kernel 5.4.48), the Mercury UD13 can be recognized and mode switched automatically, tested on Raspberry Pi 4B and VMware Workstation, AP/master mode works and iperf3 gets about 270Mbps.

Yes, you got it

closed 2020-06-23 by CWZ7605

raynorkenfox, 2020-11-12:
冒昧打扰，我也是想用UD13在虚拟机上做AP，固件已安装kmod-mt76x2、kmod-mt76x2-common（版本均为5.4.69+2020-08-25-b36..0-2），也从openwrt官网下载安装usb-modeswitch（版本2017-12-19-f40f84c2-1）。但现在还是无法识别到，无线选项显示“无线未开启或未关联”。

请问如何解决？

CWZ7605, 2020-11-12:
> 
> 
> 冒昧打扰，我也是想用UD13在虚拟机上做AP，固件已安装kmod-mt76x2、kmod-mt76x2-common（版本均为5.4.69+2020-08-25-b36..0-2），也从openwrt官网下载安装usb-modeswitch（版本2017-12-19-f40f84c2-1）。但现在还是无法识别到，无线选项显示“无线未开启或未关联”。
> 
> 请问如何解决？

建议重新编译固件，编译固件时选择驱动

raynorkenfox, 2020-11-12:
我排查了一下基本可以判断是usb-modeswitch的问题，用lsusb是可以识别到Device ID 2c4e:0103，但输入usb_modeswitch -h却显示not found。请问如果不重新编译的话有什么解决办法呢？

也许我需要一个正常使用的usb-modeswitch ipk包？目前似乎只有官网能下载。。。

CMCCAdmin, 2020-11-12:
> 我排查了一下基本可以判断是usb-modeswitch的问题，用lsusb是可以识别到Device ID 2c4e:0103，但输入usb_modeswitch -h却显示not found。请问如果不重新编译的话有什么解决办法呢？
> 
> 也许我需要一个正常使用的usb-modeswitch ipk包？目前似乎只有官网能下载。。。

编译时要在make menuconfig中选上usb-modeswitch这个包，不需要单独下载，

几个月前我试了个树莓派4b结合完美使用的，但是有一个很严重呃呃问题无法解决就是，每次开机网卡模式不会自动切换，需要手动切换一下，试了装上usbrest，然后执行重置也不管用，最后买个不带驱动模式的，edup ac1602完美使用到现在，并且时usb3.0的接口，近距离能跑但400mbps

CMCCAdmin, 2020-11-12:
> 我排查了一下基本可以判断是usb-modeswitch的问题，用lsusb是可以识别到Device ID 2c4e:0103，但输入usb_modeswitch -h却显示not found。请问如果不重新编译的话有什么解决办法呢？
> 
> 也许我需要一个正常使用的usb-modeswitch ipk包？目前似乎只有官网能下载。。。

我又看了下，既然usb id已经是2c4e:0103，说明切换模式成功了，那就不是modeswitch的问题了。试试通过luci设置固定信道，国家，设置ap模式然后重启

raynorkenfox, 2020-11-13:
> 
> 
> > 我排查了一下基本可以判断是usb-modeswitch的问题，用lsusb是可以识别到Device ID 2c4e:0103，但输入usb_modeswitch -h却显示not found。请问如果不重新编译的话有什么解决办法呢？
> > 也许我需要一个正常使用的usb-modeswitch ipk包？目前似乎只有官网能下载。。。
> 
> 我又看了下，既然usb id已经是2c4e:0103，说明切换模式成功了，那就不是modeswitch的问题了。试试通过luci设置固定信道，国家，设置ap模式然后重启

我用另一块网卡时已经设置了固定信道和国家US了，换上UD13后工作频率那里的模式、频宽、信道、频宽都变成空白了，国家还是US

raynorkenfox, 2020-11-13:
> 
> 
> > 我排查了一下基本可以判断是usb-modeswitch的问题，用lsusb是可以识别到Device ID 2c4e:0103，但输入usb_modeswitch -h却显示not found。请问如果不重新编译的话有什么解决办法呢？
> > 也许我需要一个正常使用的usb-modeswitch ipk包？目前似乎只有官网能下载。。。
> 
> 编译时要在make menuconfig中选上usb-modeswitch这个包，不需要单独下载，
> 
> 几个月前我用这个网卡和树莓派4b结合可以使用，但是有一个很严重的问题无法解决，就是每次树莓派开机网卡模式不会自动切换，需要手动插拔一下网卡，试了装上usbrest，然后执行重置也不管用，最后买个不带驱动模式的edup ac1602完美使用到现在，并且是usb3.0的接口，近距离能跑但400mbps

请教下，我是用GitHub的云编译，make menuconfig中USB SUPPORT没有usb-modeswitch，请问如何添加？

CWZ7605, 2020-11-13:
> 请教下，我是用GitHub的云编译，make menuconfig中USB SUPPORT没有usb-modeswitch，请问如何添加？

这个包是openwrt官方自带的包。
您说没有，我猜您应该是使用了一些“精简”的第三方openwrt吧

raynorkenfox, 2020-11-19:
我跑去买了一块edup ac1602

> 编译时要在make menuconfig中选上usb-modeswitch这个包，不需要单独下载，
>
> 几个月前我用这个网卡和树莓派4b结合可以使用，但是有一个很严重的问题无法解决，就是每次树莓派开机网卡模式不会自动切换，需要手动插拔一下网卡，试了装上usbrest，然后执行重置也不管用，最后买个不带驱动模式的edup ac1602完美使用到现在，并且是usb3.0的接口，近距离能跑但400mbps

你好！我也跑去买了一块edup ac1602，现在能识别出网卡，驱动也装了，无线上的2.4G/5G选项也都齐全，但就是不能工作，提示“无线未开启或未关联”。以下是日志信息：
Thu Nov 19 14:24:10 2020 daemon.notice netifd: radio0 (12930): Device setup failed: INTERFACE_CREATION_FAILED
Thu Nov 19 14:24:10 2020 daemon.notice hostapd: Configuration file: /var/run/hostapd-phy0.conf (phy wlan0) --> new PHY
Thu Nov 19 14:24:10 2020 daemon.err hostapd: Could not read interface wlan0 flags: No such device
Thu Nov 19 14:24:10 2020 daemon.notice hostapd: nl80211: Driver does not support authentication/association or connect commands
Thu Nov 19 14:24:10 2020 daemon.notice hostapd: nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Thu Nov 19 14:24:10 2020 daemon.err hostapd: Could not read interface wlan0 flags: No such device
Thu Nov 19 14:24:10 2020 daemon.err hostapd: nl80211 driver initialization failed.
Thu Nov 19 14:24:10 2020 daemon.notice hostapd: wlan0: CTRL-EVENT-TERMINATING
Thu Nov 19 14:24:10 2020 daemon.err hostapd: hostapd_free_hapd_data: Interface wlan0 wasn't started
Thu Nov 19 14:24:10 2020 daemon.notice netifd: radio0 (12930): Command failed: Invalid argument
Thu Nov 19 14:24:11 2020 daemon.notice netifd: radio0 (12930): Cannot find device "wlan0"
Thu Nov 19 14:24:11 2020 daemon.notice netifd: radio0 (12930): Interface 0 setup failed: IFUP_ERROR
Thu Nov 19 14:24:11 2020 user.info : Loaded config file '/etc/netdata/apps_groups.conf'
Thu Nov 19 14:24:32 2020 user.info : Loaded config file '/etc/netdata/apps_groups.conf'

请问如何解决？谢谢！

Jenfong, 2022-11-03:
Fedora 36 , UD13 ,it's work! 
```[root@localhost ~] # echo '2c4e 0103' > /sys/bus/usb/drivers/mt76x2u/new_id
[root@localhost ~] # lsmod
[root@localhost ~] # modprobe​ mt76x2u
[root@localhost ~] # lsusb 
```