openwrt/mt76 #375: mt7603 lose internet connection

9000000, 2020-03-22:
mt7603 still fails to watch a video file that we fast forward many times it will lose internet connection

All versions of openwrt 18 and 19 I use have this error
My router xiaomi r3g v1

JFtico, 2020-03-23:
Try the latest snapshot, there were some significant WiFi fixes the recent month.

9000000, 2020-03-24:
I have tried many times but still got an error

Openwrtfunboy, 2020-04-10:
Yes, in snapshot from 8 APR i have same. Also have on latest stable Openwrt.

JFtico, 2020-04-10:
I thought it was better, but just looked the kernel logs for a unit with 28 days uptime, running on snapshot  r12529-228bb84744   
This unit is used strictly as an Access Point. So no dnsmasq, or other stuff loading it.
It recovers, as the users have not complained. but obviously something is bad.

Context: primary AP for home with two tech workers, laptops and iPhones, a couple of Android tablets and test phones. IoT stuff (a dozen, but nothing heavy duty, no cameras)
All other gear is hard-wired (via a sperate switch).
Hardware is ZBT-WE3526

Here is the full kernel log. Looks like several 7603 re-starts, but also other proc oops as well.

'[1382323.633626]         00000000 00000000 8f782e90 8ff91880 af7a9e10 8eeb5e40 783b48a3 8020060c
[1382323.642169]         ffffffff 00000001 0fdf5380 8f79002c 8eeb5e40 00000000 8dca2000 8dca3da0
[1382323.650711]         40000000 8f791244 11008403 0000001c 00000000 8ff91880 50800400 8f79119c
[1382323.659267]         ...
[1382323.661887] Call Trace:
[1382323.664517] [<804dff4c>] __schedule+0x7d4/0x878
[1382325.105985] mtk_soc_eth 1e100000.ethernet eth0: transmit timed out
[1382325.112357] mtk_soc_eth 1e100000.ethernet eth0: dma_cfg:80000065
[1382325.118557] mtk_soc_eth 1e100000.ethernet eth0: tx_ring=0, base=0f1e0000, max=0, ctx=3717, dtx=3717, fdx=3704, next=3717
[1382325.129590] mtk_soc_eth 1e100000.ethernet eth0: rx_ring=0, base=0e560000, max=0, calc=2932, drx=2930
[1382330.065787] mtk_soc_eth 1e100000.ethernet eth0: transmit timed out
[1382330.072159] mtk_soc_eth 1e100000.ethernet eth0: dma_cfg:80000065
[1382330.078358] mtk_soc_eth 1e100000.ethernet eth0: tx_ring=0, base=0f1e0000, max=0, ctx=3717, dtx=3717, fdx=3704, next=3717
[1382330.089392] mtk_soc_eth 1e100000.ethernet eth0: rx_ring=0, base=0e560000, max=0, calc=2932, drx=2930
[1382335.105598] mtk_soc_eth 1e100000.ethernet eth0: transmit timed out
[1382335.111974] mtk_soc_eth 1e100000.ethernet eth0: dma_cfg:80000065
[1382335.118172] mtk_soc_eth 1e100000.ethernet eth0: tx_ring=0, base=0f1e0000, max=0, ctx=3717, dtx=3717, fdx=3704, next=3717
[1382335.129210] mtk_soc_eth 1e100000.ethernet eth0: rx_ring=0, base=0e560000, max=0, calc=2932, drx=2930
[1382340.065408] mtk_soc_eth 1e100000.ethernet eth0: transmit timed out
[1382340.071785] mtk_soc_eth 1e100000.ethernet eth0: dma_cfg:80000065
[1382340.077988] mtk_soc_eth 1e100000.ethernet eth0: tx_ring=0, base=0f1e0000, max=0, ctx=3717, dtx=3717, fdx=3704, next=3717
[1382340.089021] mtk_soc_eth 1e100000.ethernet eth0: rx_ring=0, base=0e560000, max=0, calc=2932, drx=2930
[1382345.105210] mtk_soc_eth 1e100000.ethernet eth0: transmit timed out
[1382345.111587] mtk_soc_eth 1e100000.ethernet eth0: dma_cfg:80000065
[1382345.117789] mtk_soc_eth 1e100000.ethernet eth0: tx_ring=0, base=0f1e0000, max=0, ctx=3717, dtx=3717, fdx=3704, next=3717
[1382345.128825] mtk_soc_eth 1e100000.ethernet eth0: rx_ring=0, base=0e560000, max=0, calc=2932, drx=2930
[1382350.065014] mtk_soc_eth 1e100000.ethernet eth0: transmit timed out
[1382350.071391] mtk_soc_eth 1e100000.ethernet eth0: dma_cfg:80000065
[1382350.077593] mtk_soc_eth 1e100000.ethernet eth0: tx_ring=0, base=0f1e0000, max=0, ctx=3717, dtx=3717, fdx=3704, next=3717
[1382350.088630] mtk_soc_eth 1e100000.ethernet eth0: rx_ring=0, base=0e560000, max=0, calc=2932, drx=2930
[1382355.104817] mtk_soc_eth 1e100000.ethernet eth0: transmit timed out
[1382355.111196] mtk_soc_eth 1e100000.ethernet eth0: dma_cfg:80000065
[1382355.117396] mtk_soc_eth 1e100000.ethernet eth0: tx_ring=0, base=0f1e0000, max=0, ctx=3717, dtx=3717, fdx=3704, next=3717
[1382355.128430] mtk_soc_eth 1e100000.ethernet eth0: rx_ring=0, base=0e560000, max=0, calc=2932, drx=2930
[1382360.064626] mtk_soc_eth 1e100000.ethernet eth0: transmit timed out
[1382360.071000] mtk_soc_eth 1e100000.ethernet eth0: dma_cfg:80000065
[1382360.077197] mtk_soc_eth 1e100000.ethernet eth0: tx_ring=0, base=0f1e0000, max=0, ctx=3717, dtx=3717, fdx=3704, next=3717
[1382360.088233] mtk_soc_eth 1e100000.ethernet eth0: rx_ring=0, base=0e560000, max=0, calc=2932, drx=2930
[1382365.104432] mtk_soc_eth 1e100000.ethernet eth0: transmit timed out
[1382365.110809] mtk_soc_eth 1e100000.ethernet eth0: dma_cfg:80000065
[1382365.117007] mtk_soc_eth 1e100000.ethernet eth0: tx_ring=0, base=0f1e0000, max=0, ctx=3717, dtx=3717, fdx=3704, next=3717
[1382365.128064] mtk_soc_eth 1e100000.ethernet eth0: rx_ring=0, base=0e560000, max=0, calc=2932, drx=2930
[1382370.064232] mtk_soc_eth 1e100000.ethernet eth0: transmit timed out
[1382370.070603] mtk_soc_eth 1e100000.ethernet eth0: dma_cfg:80000065
[1382370.076804] mtk_soc_eth 1e100000.ethernet eth0: tx_ring=0, base=0f1e0000, max=0, ctx=3717, dtx=3717, fdx=3704, next=3717
[1382370.087841] mtk_soc_eth 1e100000.ethernet eth0: rx_ring=0, base=0e560000, max=0, calc=2932, drx=2930
[1382375.104051] mtk_soc_eth 1e100000.ethernet eth0: transmit timed out
[1382375.110427] mtk_soc_eth 1e100000.ethernet eth0: dma_cfg:80000065
[1382375.116626] mtk_soc_eth 1e100000.ethernet eth0: tx_ring=0, base=0f1e0000, max=0, ctx=3717, dtx=3717, fdx=3704, next=3717
[1382375.127664] mtk_soc_eth 1e100000.ethernet eth0: rx_ring=0, base=0e560000, max=0, calc=2932, drx=2930
[1382380.063848] mtk_soc_eth 1e100000.ethernet eth0: transmit timed out
[1382380.070227] mtk_soc_eth 1e100000.ethernet eth0: dma_cfg:80000065
[1382380.076432] mtk_soc_eth 1e100000.ethernet eth0: tx_ring=0, base=0f1e0000, max=0, ctx=3717, dtx=3717, fdx=3704, next=3717
[... cut ...]
[1383060.049978] mtk_soc_eth 1e100000.ethernet eth0: tx_ring=0, base=0f1e0000, max=0, ctx=3717, dtx=3717, fdx=3704, next=3717
[1383060.061013] mtk_soc_eth 1e100000.ethernet eth0: rx_ring=0, base=0e560000, max=0, calc=2932, drx=2930
[1383060.837362] INFO: rcu_sched detected expedited stalls on CPUs/tasks: { 0-... } 153481 jiffies s: 51353 root: 0x1/.
[1383060.847893] blocking rcu_node structures:
[1383060.852065] Task dump for CPU 0:
[1383060.855451] kworker/u8:1    R  running task        0 23749      2 0x00100004
[1383060.862708] Workq'

JFtico, 2020-04-13:
And since rebooting 3 days ago, another mt7603 crash.
Unit recovered, as I never noticed.

`[226013.618068] ------------[ cut here ]------------
[226013.622946] WARNING: CPU: 1 PID: 5731 at backports-5.4-rc8-1/net/mac80211/sta_info.c:1937 ieee80211_sta_update_pending_airtime+0x110/0x214 [mac80211]
[226013.636401] STA f0:c3:71:85:10:92 AC 2 txq pending airtime underflow: 4294966936, 360
[226013.636408] Modules linked in: pppoe ppp_async pppox ppp_generic nf_conntrack_ipv6 mt76x2e mt76x2_common mt76x02_lib mt7603e mt76 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_FLOWOFFLOAD xt_CT slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_mangle iptable_filter ip_tables crc_ccitt compat ledtrig_usbport ledtrig_heartbeat nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 mmc_block mtk_sd mmc_core leds_gpio xhci_plat_hcd xhci_pci xhci_mtk xhci_hcd
[226013.714724]  gpio_button_hotplug usbcore nls_base usb_common
[226013.720474] CPU: 1 PID: 5731 Comm: kworker/u8:0 Not tainted 4.14.172 #0
[226013.727162] Workqueue: phy0 mt7603_mac_work [mt7603e]
[226013.732276] Stack : 80591d48 8eacdbdc 80600000 8f0f7300 80591d60 8ec07fc8 8ec6d7fc 00000791
[226013.740700]         ffffffff 800778e0 8fd1838c 00000000 80600000 00000001 8eacdba0 ac07f5b5
[226013.749133]         00000000 00000000 00000000 00000000 807784a0 0000012b 00000007 00000000
[226013.757582]         00000000 00000000 000b187a 6d5f3330 00000000 80630000 00000000 8ec07fc8
[226013.766008]         8ec6d7fc 00000791 ffffffff 00000001 00000008 802e8690 00000004 80770004
[226013.774430]         ...
[226013.776958] Call Trace:
[226013.779509] [<8000c4d4>] show_stack+0x58/0x100
[226013.784030] [<804c7784>] dump_stack+0xa4/0xe0
[226013.788470] [<8002fd60>] __warn+0xe0/0x140
[226013.792636] [<8002f9a4>] warn_slowpath_fmt+0x30/0x3c
[226013.797777] [<8ec07fc8>] ieee80211_sta_update_pending_airtime+0x110/0x214 [mac80211]
[226013.805638] [<8ec01e4c>] ieee80211_report_low_ack+0x160/0x4f4 [mac80211]
[226013.812534] ---[ end trace dc4b731b6a56048d ]---`

JFtico, 2020-04-14:
Seeing stuff like this about once a day:

`[225754.393859] ieee80211 phy1: Hardware restart was requested
[225756.464920] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
[225756.470491] mt76x2e 0000:01:00.0: Build: 1
[225756.474732] mt76x2e 0000:01:00.0: Build Time: 201507311614____
[225756.503643] mt76x2e 0000:01:00.0: Firmware running!
[225756.513901] ieee80211 phy1: Hardware restart was requested
[225763.334935] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
[225763.340579] mt76x2e 0000:01:00.0: Build: 1
[225763.344840] mt76x2e 0000:01:00.0: Build Time: 201507311614____
[225763.373489] mt76x2e 0000:01:00.0: Firmware running!
[225763.383567] ieee80211 phy1: Hardware restart was requested
[225907.238670] ------------[ cut here ]------------
[225907.243580] WARNING: CPU: 2 PID: 10799 at backports-5.4-rc8-1/net/mac80211/sta_info.c:1937 ieee80211_sta_update_pending_airtime+0x110/0x214 [mac80211]
[225907.257144] STA 64:5d:86:7d:0f:b4 AC 2 txq pending airtime underflow: 4294966952, 344
[225907.257168] Modules linked in: pppoe ppp_async pptp pppox ppp_mppe ppp_generic nf_nat_pptp nf_conntrack_pptp nf_conntrack_ipv6 mt76x2e mt76x2_common mt76x02_lib mt7603e mt76 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_FLOWOFFLOAD xt_DSCP xt_CT xt_CLASSIFY ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda ts_fsm ts_bm slhc nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_redirect nf_nat_proto_gre nf_nat_masquerade_ipv4 nf_nat_irc nf_conntrack_ipv4 nf_nat_ipv4 nf_nat_h323
[225907.336143]  nf_nat_amanda nf_nat nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_rtcache nf_conntrack_proto_gre nf_conntrack_irc nf_conntrack_h323 nf_conntrack_broadcast ts_kmp nf_conntrack_amanda iptable_raw iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt compat sch_cake nf_conntrack sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred ledtrig_usbport ledtrig_heartbeat nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 msdos ip_gre gre ifb ip_tunnel tun vfat fat nls_utf8 nls_iso8859_1 nls_cp437 sha1_generic ecb arc4 mmc_block usb_storage mtk_sd mmc_core leds_gpio
[225907.407297]  xhci_plat_hcd xhci_pci xhci_mtk xhci_hcd ohci_platform ohci_hcd sd_mod scsi_mod gpio_button_hotplug ext4 mbcache jbd2 usbcore nls_base usb_common crc32c_generic
[225907.423210] CPU: 2 PID: 10799 Comm: kworker/u8:2 Not tainted 4.14.172 #0
[225907.430036] Workqueue: phy0 mt7603_mac_work [mt7603e]
[225907.435164] Stack : 8055eb08 8e68dbdc 805d0000 8ed43580 8055eb20 8ee87fc8 8eeed7fc 00000791
[225907.443606]         ffffffff 80077720 8e152b4c 00000000 805d0000 00000001 8e68dba0 ac07f5b5
[225907.452046]         00000000 00000000 00000000 00000000 807484a0 0000026c 00000007 00000000
[225907.460483]         00000000 00000000 00068fd4 6d5f3330 00000000 80600000 00000000 8ee87fc8
[225907.468922]         8eeed7fc 00000791 ffffffff 00000001 00000008 802d4f48 00000008 80740008
[225907.477360]         ...
[225907.479891] Call Trace:
[225907.482452] [<8000c4d4>] show_stack+0x58/0x100
[225907.486984] [<80498ce4>] dump_stack+0xa4/0xe0
[225907.491430] [<8002fd60>] __warn+0xe0/0x140
[225907.495604] [<8002f9a4>] warn_slowpath_fmt+0x30/0x3c
[225907.500814] [<8ee87fc8>] ieee80211_sta_update_pending_airtime+0x110/0x214 [mac80211]
[225907.508707] [<8ee81e4c>] ieee80211_report_low_ack+0x160/0x4f4 [mac80211]
[225907.515629] ---[ end trace 1397f52a7f46469a ]---`