openwrt/mt76 #376: mt7615e, 5G low tx-power, missing 2.4G

xuefer, 2020-03-24:
[   17.930530] mt7615e 0000:01:00.0: HW/SW Version: 0x8a108a10, Build Time: 20180518100604a
[   18.531466] mt7615e 0000:01:00.0: N9 Firmware Version: 2.0, Build Time: 20200131181812
[   18.634181] mt7615e 0000:01:00.0: CR4 Firmware Version: _reserved_, Build Time: 20190121161307
iw shows 2 bands but openwrt "wifi config" adds only 1 

rm /etc/config/wireless
wifi config
cat /etc/config/wireless
config wifi-device 'radio0'
        option type 'mac80211'
        option channel '36'
        option hwmode '11a'
        option path 'pci0000:00/0000:00:00.0/0000:01:00.0'
        option htmode 'VHT80'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option ssid 'OpenWrt'
        option encryption 'none'

 iw list | grep -i freq -A10
                Frequencies:
                        * 2412 MHz [1] (6.0 dBm)
                        * 2417 MHz [2] (6.0 dBm)
                        * 2422 MHz [3] (6.0 dBm)
                        * 2427 MHz [4] (6.0 dBm)
                        * 2432 MHz [5] (6.0 dBm)
                        * 2437 MHz [6] (6.0 dBm)
                        * 2442 MHz [7] (6.0 dBm)
                        * 2447 MHz [8] (6.0 dBm)
                        * 2452 MHz [9] (6.0 dBm)
                        * 2457 MHz [10] (6.0 dBm)
--
                Frequencies:
                        * 5180 MHz [36] (6.0 dBm)
                        * 5200 MHz [40] (6.0 dBm)
                        * 5220 MHz [44] (6.0 dBm)
                        * 5240 MHz [48] (6.0 dBm)
                        * 5260 MHz [52] (6.0 dBm) (no IR, radar detection)
                        * 5280 MHz [56] (6.0 dBm) (no IR, radar detection)
                        * 5300 MHz [60] (6.0 dBm) (no IR, radar detection)
                        * 5320 MHz [64] (6.0 dBm) (no IR, radar detection)
                        * 5500 MHz [100] (6.0 dBm) (no IR, radar detection)
                        * 5520 MHz [104] (6.0 dBm) (no IR, radar detection)

connected but can't get ip, (which implies no data can be transfer'ed)

xuefer, 2020-05-09:
after updated to 3e6968593b613006a906822e08a7725e07cfebde:
still low power,  luci says was 6 dBm, now 10 dBm. but actual signal is good in my android pad
was 5G by default, switchable to 2.4G, but now 2.4G only available
"iw list" used to list both 2.4G and 5G. but now 2.4G only

Gingernut1978, 2020-05-23:
Latest snapshot from 22/05/20 also reports very low tx power output on both bands.

`root@R6800:~# iw list
Wiphy phy1
        max # scan SSIDs: 4
        max scan IEs length: 2304 bytes
        max # sched scan SSIDs: 0
        max # match sets: 0
        max # scan plans: 1
        max scan plan interval: 0
        max scan plan iterations: 0
        Retry short limit: 7
        Retry long limit: 4
        Coverage class: 0 (up to 0m)
        Device supports T-DLS.
        Available Antennas: TX 0xf RX 0xf
        Configured Antennas: TX 0xf RX 0xf
        Supported interface modes:
                 * IBSS
                 * managed
                 * AP
                 * AP/VLAN
                 * monitor
                 * mesh point
        Band 2:
                Capabilities: 0x1ff
                        RX LDPC
                        HT20/HT40
                        SM Power Save disabled
                        RX Greenfield
                        RX HT20 SGI
                        RX HT40 SGI
                        TX STBC
                        RX STBC 1-stream
                        Max AMSDU length: 3839 bytes
                        No DSSS/CCK HT40
                Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
                Minimum RX AMPDU time spacing: No restriction (0x00)
                HT TX/RX MCS rate indexes supported: 0-31
                VHT Capabilities (0x338001fa):
                        Max MPDU length: 11454
                        Supported Channel Width: 160 MHz, 80+80 MHz
                        RX LDPC
                        short GI (80 MHz)
                        short GI (160/80+80 MHz)
                        TX STBC
                        RX antenna pattern consistency
                        TX antenna pattern consistency
                VHT RX MCS set:
                        1 streams: MCS 0-9
                        2 streams: MCS 0-9
                        3 streams: MCS 0-9
                        4 streams: MCS 0-9
                        5 streams: not supported
                        6 streams: not supported
                        7 streams: not supported
                        8 streams: not supported
                VHT RX highest supported: 0 Mbps
                VHT TX MCS set:
                        1 streams: MCS 0-9
                        2 streams: MCS 0-9
                        3 streams: MCS 0-9
                        4 streams: MCS 0-9
                        5 streams: not supported
                        6 streams: not supported
                        7 streams: not supported
                        8 streams: not supported
                VHT TX highest supported: 0 Mbps
                Frequencies:
                        * 5180 MHz [36] (6.0 dBm)
[... 88 lines cut ...]
                        * 2457 MHz [10] (6.0 dBm)
                        * 2462 MHz [11] (6.0 dBm)
                        * 2467 MHz [12] (6.0 dBm)
                        * 2472 MHz [13] (6.0 dBm)
                        * 2484 MHz [14] (disabled)
        valid interface combinations:
                 * #{ IBSS } <= 1, #{ managed, AP, mesh point } <= 4,
                   total <= 4, #channels <= 1, STA/AP BI must match, radar detect widths: { 20 MHz (no HT), 20 MHz, 40 MHz, 80 MHz, 80+80 MHz, 160 MHz }

        HT Capability overrides:
                 * MCS: ff ff ff ff ff ff ff ff ff ff
                 * maximum A-MSDU length
                 * supported channel width
                 * short GI for 40 MHz
                 * max A-MPDU length exponent
                 * min MPDU start spacing
        Supported extended features:
                * [ VHT_IBSS ]: VHT-IBSS
                * [ RRM ]: RRM
                * [ SET_SCAN_DWELL ]: scan dwell setting
                * [ CQM_RSSI_LIST ]: multiple CQM_RSSI_THOLD records
                * [ CONTROL_PORT_OVER_NL80211 ]: control port over nl80211
                * [ TXQS ]: FQ-CoDel-enabled intermediate TXQs
                * [ AIRTIME_FAIRNESS ]: airtime fairness scheduling
`

ryderlee1110, 2020-06-04:
This is not a software issue. The eeprom or OTP on your card is empty, so you need to put calibration data into the corresponding region.

Something like MT7615E1_EEPROM.bin
https://github.com/digizer0/openwrt-mt7615/tree/master/mtk/drivers/mt_wifi/src/bin/mt7615

ryderlee1110, 2020-06-05:
This is more like the eerpom on your card is empty, so you need to put RF data into the corresponding region to solve this.

something like MT7615E1_EEPROM.bin https://github.com/digizer0/openwrt-mt7615/tree/master/mtk/drivers/mt_wifi/src/bin/mt7615

patricius972, 2020-06-07:
Hi, 
I have the same problem with a wac124, I could not find any instruction on how to add that eeprom into the right place. Could you point me in the right direction ? I also would like to know how to extract the factory mtd on my device, if there is any guide somwhere i could follow.

patricius972, 2020-06-08:
Just in case that will help anybody, it is a eeprom issue, most likely not read correctly from the mtd.

closed 2020-09-18 by xuefer

RemiNV, 2022-01-09:
I got the same problem on a WRC-1167GST2 router, and replacing the eeprom fixed it as mentioned above.
I had to do some research on how to do that, so to save time for anyone else having this issue:

Good read: [OpenWrt flash layout](https://openwrt.org/docs/techref/flash.layout). See device layout in /proc/mtd.

What I did:

```
dd if=/dev/mtd2 /tmp/mtd2 # Backup the original factory (mtd2 on this device) image, then scp it from device
opkg install kmod-mtd-rw
insmod mtd-rw i_want_a_brick=1 # Use at your own risk
mtd unlock factory
mtd write /tmp/MT7615E1_EEPROM.bin factory # MT7615E1_EEPROM.bin file from link above
```

xuefer, 2022-01-09:
> Just in case that will help anybody, it is a eeprom issue, most likely not read correctly from the mtd.

data is read stably, but the values are incorrect. official driver don't read them but read from /etc/ instead. oem vendor may not have eeprom data written correctly