openwrt/mt76 #378: mt 7628 - 2.4G wifi very low range, unreliable, auth problems

nx518, 2020-03-25:
```
Model	TP-Link Archer C50 v4
Architecture	MediaTek MT7628AN ver:1 eco:2
Firmware Version	OpenWrt 19.07.0 r10860-a3ffeb413b / LuCI openwrt-19.07 branch git-20.081.71863-12f0dac
Kernel Version	4.14.162
```

The 2.4G wifi has several issus on this device: Range is really low,  >4m results in disconnect, auth fails often, speed is low.

I tried disabling Legacy Rates, WMM, Crack, Disass on low ack, but it did not help.

Anyone has this device running with 2.4G successfully?

Vonger, 2020-08-25:
Try this patch https://github.com/openwrt/mt76/pull/426

dtrautmann, 2020-09-03:
Could you please try if this fixes your problem: #442

nx518, 2020-09-03:
Thanks for the note, I will try it within 1 month and report here.

nx518, 2020-09-17:
Tested #442 on my Archer C50 v4.
Fixed the issues for me, thanks!

rotanid, 2020-09-17:
@nx518 please don't close an issue as long as the fix is not merged into the code base, the issue still exists

nx518, 2020-09-17:
ok

adelton, 2022-05-14:
@rotanid Given https://github.com/openwrt/mt76/pull/442 seems to have been merged, should this issues still be open?

rotanid, 2022-05-14:
@adelton ok @nx518

closed 2022-05-19 by nx518